ai-republic/bms-to-inverter #1: Growatt Inverter support via RS 485

sanitariu, 2023-09-08:
Hello,
Do you plan any support for Growatt Inverter support via RS 485 ?

ai-republic, 2023-09-13:
I'm currently implementing a MQTT support with a little webserver to see the states of all battery cells in each battery pack.

To implement the Growatt protocol would be no problem but maybe in a couple of days. If you have the protocol at hand or could organize it, it would be helpful and speed up the process :)

Best regards,
Torsten

sanitariu, 2023-09-13:
I will help with all i can. I would like to make my Daly BMS speak with my Growatt inverter. Currently i am able to read info from the Daly via small python script i found here
https://github.com/dreadnought/python-daly-bms.git

This is the information from my Daly:

daly-bms-cli --device /dev/ttyUSB2 --all

{
  "soc": {
    "total_voltage": 53.9,
    "current": 39.0,
    "soc_percent": 76.4
  },
  "cell_voltage_range": {
    "highest_voltage": 3.379,
    "highest_cell": 16,
    "lowest_voltage": 3.372,
    "lowest_cell": 5
  },
  "temperature_range": {
    "highest_temperature": 28,
    "highest_sensor": 1,
    "lowest_temperature": 28,
    "lowest_sensor": 1
  },
  "mosfet_status": {
    "mode": "charging",
    "charging_mosfet": true,
    "discharging_mosfet": true,
    "capacity_ah": 213.92
  },
  "status": {
    "cells": 16,
    "temperature_sensors": 1,
    "charger_running": false,
    "load_running": false,
    "states": {
      "DI1": false
    },
    "cycles": 63
  },
  "cell_voltages": {
    "1": 3.373,
    "2": 3.373,
    "3": 3.372,
    "4": 3.375,
    "5": 3.372,
    "6": 3.375,
    "7": 3.374,
    "8": 3.377,
    "9": 3.374,
    "10": 3.373,
    "11": 3.372,
    "12": 3.375,
    "13": 3.372,
    "14": 3.375,
    "15": 3.374,
    "16": 3.379
  },
  "temperatures": {
    "1": 28
  },
  "balancing_status": {
    "error": "not implemented"
  },
  "errors": []
}

Also I am able to read data from my Growatt inverter and graph it using grafana via this:
https://github.com/sdsolomo/growatt-x000ES

Daly is using this connection - ch341-uart converter now attached to ttyUSB2 (USB to UART cable connected to Daly device)
Growatt is using this connection - cp210x converter now attached to ttyUSB1 (USB direct cable from the Inverter to Pi device)

If i can help you somehow i would do it even with some remote access if you like.

sanitariu, 2023-09-16:
Here is Growatt protocol
[Growatt PV Inverter Modbus RS485 RTU Protocol v120.pdf](https://github.com/ai-republic/bms-to-inverter/files/12640748/Growatt.PV.Inverter.Modbus.RS485.RTU.Protocol.v120.pdf)

ai-republic, 2023-09-23:
Hi Todor,
sorry for the late reply. I've been away for a few days. I've had a look at the Growatt protocol but have not yet really understood everything as a lot of the knitty gritty stuff for the BMS data is in chinese. I will have to work through it with google translator and try to figure it out. Do you know which mandatory values the Growatt inverter needs from the BMS?

ai-republic, 2023-10-05:
FYI: I've added a growatt module to the project. Just for the structure... there is no functionality yet!

ai-republic, 2023-10-19:
Hi Todor, 
I've implemented the Growatt CAN protocol if you like to test it. 
Note: I do not have a Growatt Inverter so it is purely programmed by the Growatt CAN protocol specifiation for low voltage (12-96V) inverters. 
If you like to test it I would be thankful :)

sanitariu, 2023-10-19:
Hi,
Thanks for the support ! I am going to test it very soon. I must order CAN device to connect the inverter. I can connect now only via RS-485.

sanitariu, 2023-10-28:
I have ordered the same CAN device model as yours. I found also one PYLON_EMU project with dbc file where you can see what you must report via the CAN to the inverter. I am attaching the file if you are interested.
I am going to test tomorrow your version with my CAN device.

[pylon_CAN_210124.txt](https://github.com/ai-republic/bms-to-inverter/files/13194992/pylon_CAN_210124.txt)

sanitariu, 2024-01-31:
Ok trying to simply connect Growatt to can0.
I receive the following errors
[1662498.588369] mcp251xfd spi0.0 can0: CRC read error at address 0x0010 (length=4, data=e0 80 af 70, CRC=0x01f5) retrying.
I am connecting High to High and Low to Low pins.
Here is my /boot/config.txt

# i2c canhat
dtparam=spi=on
dtoverlay=spi1-3cs
dtoverlay=mcp251xfd,spi0-0,interrupt=25
dtoverlay=mcp251xfd,spi1-0,interrupt=24

Also doing this setup ...
# link up
/usr/bin/ip link set can0 up type can bitrate 1000000 dbitrate 8000000 restart-ms 1000 berr-reporting on fd on
/usr/bin/ip link set can1 up type can bitrate 1000000 dbitrate 8000000 restart-ms 1000 berr-reporting on fd on

# buffers
/usr/sbin/ifconfig can0 txqueuelen 65536
/usr/sbin/ifconfig can1 txqueuelen 65536

If i loop can0 to can1 and send something like this

candump can0
cansend can1 000#11.22.33.44

I see the output.
Can i have your settings for the same wave hat and can ?

ai-republic, 2024-02-01:
Hi Todor,
which CAN adaptor are you using? Is it the [Waveshare 2-CH CAN FD Hat](https://www.waveshare.com/2-ch-can-fd-hat.htm)?
Since your `boot.config` setting suggest and you have `can0` and `can1` I guess it might be.

So I believe your Growatt will not have a baudrate of 1000000 but probably 500000. But I will check the documentation.
Also you should not have FD on and also no berr reporting.

My settings for the SMA SI are like this:

```
sudo ip link set can0 up type can bitrate 500000 restart-ms 100
sudo ip link set can0 txqueuelen 65536
```

sanitariu, 2024-02-01:
Yes I have the exact same can model you suggested. I setup to 500kb now. Will I see any output from the inverter even if i did not setup program 36 at the inverter side ? I have the same schema in my Growatt book. Seems like the cables are blue and white-blue. As far as I know i should connect CANH to CANH and CANL to CANL ?

ai-republic, 2024-02-01:
Yes, CAN H to CAN H and CAN L to CAN L.

I'm not sure if you need to set up your inverter to use 2 Lithium and then 36 mode 51 (Growatt) or 52 (Pylontech I think) for the inverter to activate the CAN communication on its BMS port.

You should see some output in the log about the inverter communication. If the Growatt inverter module (with mode 51 on the inverter) doesn't work you can also try to configure the SOLARK inverter since that is also a pure Pylontech CAN protocol implementation with mode 52 on the inverter.

sanitariu, 2024-02-16:
Anyone did a voltage measurements for Growatt CAN H/L ? According to scheme blue/white and blue are pin numbers 4/5. Orange is the ground. I did measure mine and both H/L shows 2.25 voltage. Also resistance between H/L is 160 oms. Is this correct ? I thought CAN H should have more high voltage.
If someone have values for his inverter please share...

ai-republic, 2024-02-17:
As per CAN physical layer specification:
CAN H (≥ 2.5 VDC)  |   2.5 VDC  (idle)   |   2.5 to 3.5 VDC (active)
CAN L (≤ 2.5 VDC)   |   2.5 VDC  (idle)   |   2.5 to 1.5 VDC (active)

The CAN-L and CAN-H waveforms mirror each other about 2.5 V and have a 1 V peak to peak amplitude. The CAN-L waveform switches from 2.5 V down to 1.5 V and the CAN-H waveform switches from 2.5 V up to 3.5 V.

So 2.25V seems a bit low for CAN H even if its idle. Have you connected PIN 6 (GND) also?

sanitariu, 2024-02-17:
No i have not connected Ground because i read that it is dangerous and you do not need it.
Also i do not know how to test exactly if CAN is working at inverter side ... like i would like to send some command and see the result. If you have something in mind like commands i can try connect the ground and test them via cansend/candump ?

ai-republic, 2024-02-18:
You could try to send a frame like:
```
0x13, 0x03, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0xE6, 0x14, 0x0F, 0x00, 0xDC, 0x00, 0x5E, 0x64
```

sanitariu, 2024-02-22:
I tried but nothing happened. Growatt does not return anything. I do not know if this is normal ? Maybe i need to setup the Growatt first ?!
Also i have tried direct in your code /dev/can1 and this is the output:

2024-02-22 10:45:23.680 | ERROR | 2-thread-1 | verter.core.Inverter:70    | Failed to send CAN frame
java.lang.IndexOutOfBoundsException: fromIndex: 1 > toIndex: 0
        at java.util.BitSet.checkRange(BitSet.java:369) ~[?:?]
        at java.util.BitSet.set(BitSet.java:484) ~[?:?]
        at com.airepublic.bmstoinverter.inverter.growatt.can.GrowattInverterCANProcessor.getChargeStates(GrowattInverterCANProcessor.java:224) ~[inverter-growatt-can-0.0.1-SNAPSHOT.jar:?]
        at com.airepublic.bmstoinverter.inverter.growatt.can.GrowattInverterCANProcessor.createAlarms(GrowattInverterCANProcessor.java:135) ~[inverter-growatt-can-0.0.1-SNAPSHOT.jar:?]
        at com.airepublic.bmstoinverter.inverter.growatt.can.GrowattInverterCANProcessor.createSendFrames(GrowattInverterCANProcessor.java:34) ~[inverter-growatt-can-0.0.1-SNAPSHOT.jar:?]
        at com.airepublic.bmstoinverter.core.Inverter.process(Inverter.java:62) ~[core-api-0.0.1-SNAPSHOT.jar:?]
        at com.airepublic.bmstoinverter.inverter.growatt.can.GrowattInverterCANProcessor$Proxy$_$$_WeldClientProxy.process(Unknown Source) ~[inverter-growatt-can-0.0.1-SNAPSHOT.jar:?]
        at com.airepublic.bmstoinverter.BmsToInverter.lambda$start$4(BmsToInverter.java:168) ~[bms-to-inverter-main-0.0.1-SNAPSHOT.jar:?]
        at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:539) [?:?]
        at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:305) [?:?]
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:305) [?:?]
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136) [?:?]
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635) [?:?]
        at java.lang.Thread.run(Thread.java:840) [?:?]

I did not setup anything to growatt yet ! I am still not sure if these voltages 2.25 on blue/white and blue are correct.

ai-republic, 2024-02-23:
Hi Todor,
could you try to see what the inverter sends via `candump`?
That way I can compare and see which protocol the inverter uses.

ai-republic, 2024-02-25:
Hi,
I updated the Configurator to allow setting a BMS Id. Also I changed the poll intervall to be a general BMS configuration (not for each).

You will need to download the new Configurator and re-configure your setup and do a clean install.

jimmy737, 2024-04-12:
> I'm currently implementing a MQTT support with a little webserver to see the states of all battery cells in each battery pack.
>
> To implement the Growatt protocol would be no problem but maybe in a [...]

Hello, Have you succeeded to get cell information? I have the SPF 5000 ES with 6 Growatt ARK 2.5 batteries trying to get the same battery details as you.  Thank you.

ai-republic, 2024-04-13:
Hi Jimmy,
yes, the Growatt binding is working :)

jimmy737, 2024-04-13:
> Hi Jimmy, yes, the Growatt binding is working :)

Thank you. Is there any step by step documentation I can follow to be able to get cell details information from the inverter USB port to an Arduino or raspberry PI? Thank you.

ai-republic, 2024-04-14:
I assume you want to get cell information from the BMS (Battery Management System) - not the inverter.
Your Growatt ARK battery system should also have a CAN communication port. Via this port you can connect to a Raspberry PI with a 2 Chanel CAN hat, e.g. the Waveshare 2CH CAH hat on the first CAN bus. From the PI you connect also on the 2nd CAN bus to the inverter. 
To install this application see the [Wiki](https://github.com/ai-republic/bms-to-inverter/wiki).

If you have any questions feel free to ask :)

jimmy737, 2024-04-14:
Thank you Torsten for your help.  Sorry I am not sure if the ARK battery
has a CAN port. But maybe this is one in the picture below.  I don`t know
it those things are helpful to my project of reading cell data but I
purchased and used a J-LINK flash device to fix a corrupted boot loader and
flash firmware on the BMS board.
So the ARK battery has a USB port. When I put a CONFIG.txt file with
down_fault in it, I get a spreadsheet with cell data. So obviously the BMS
board has it.  So it would be easier to connect the raspberry PI to the USB
port but I don`t know if that is possible.
As far as the CAN port, is it this on the picture?  Thanks again for your
patience in helping me with my project. Based on what I read on forums
other ARK battery owners can benefit from this.

ai-republic, 2024-04-15:
Ah but there is no image ;)
This project reads the cell data and BMS state continuously with realtime data. The Excel file would just be a snapshot of a certain time and already outdated by the time you read it. Also how would you get refreshed data? You would probably need to simulate unplugging and re-connecting the USB.

jimmy737, 2024-04-15:
Sorry about that. Is this the CAN port?
[image]

jimmy737, 2024-04-15:
I connected to that port to upgrade the firmware. Is this tool any good to retreive data in real time?  
[image]

ai-republic, 2024-04-16:
No, this doesn't look like anything I know - not that it means much ;)
The CAN port usually is a RJ45 normal network cable port like you use to connect your computer to your router. And should be freely accessible on your ARK battery case.

jimmy737, 2024-04-16:
Ah got it. Sorry for my ignorance.  Yes. Great! I can test with just one
battery standalone.  I connect to the PCS RJ45 of the battery and the other
end where?
Thank you Torsten. I appreciate your help.

jimmy737, 2024-04-16:
here is a pic of my CAN bus
[image]

jimmy737, 2024-04-16:
Thank you for the quick reply. My Rasperry PI 4 has Solar Assistant but not useful for me. I can reload from scratch mt PI 4 and get where i can run sudo comands. I did install a music server once but I am not a java, json, Linux etc programmer. I am good at following instructions!  So in your post it says to get everything run this below. But I am lost on how/where to run those commands on the PI4. Do I need to install a software called Daly ?  Thank you for your patience.
"Get everything possible:
# daly-bms-cli  -d /dev/ttyUSB0 --all

jimmy737, 2024-04-16:
Sorry me again!   I cannot install daly. I get this error. Any idea how to resolve this?  Thank you.
[image]

jimmy737, 2024-04-16:
Ok. I think I am ready to try connecting to the battery. Does it look ready?  Thank you.
[image]

ai-republic, 2024-04-17:
No, you don't need daly or python.
Once you have connected the RJ45 to your BMS's CAN port you need to connect the other end to your CAN hat on your RPi. What CAN hat do you have? And have you already installed/configured it?
And you need to know which cables of the 8 in the RJ45 are assigned for CAN H and CAN L.
That should be in the manual. Then just follow the [Wiki](https://github.com/ai-republic/bms-to-inverter/wiki).
You will need to install a Java JDK:

```
sudo apt install default-jdk
```

And then download and start the _Configurator_ app as described in the [Wiki](https://github.com/ai-republic/bms-to-inverter/wiki).

jimmy737, 2024-04-17:
Thank you. I install the Configurator on my laptop and configured it. But in the drop [image]
[image]

jimmy737, 2024-04-17:
Thank you. I install the Configurator on my laptop and configured it. But in the drop list
it dows not show an option for PI4 64 bits. Which option to choose?
Also I found the correct pins for the cable to go from the CAN port to the PI4. Thank you.

[image]
[image]

jimmy737, 2024-04-17:
Sorry one more thing. Yesterday we talked about one end of the custom made network cable to the battery and the other end to PI4. But what do I do with the Waveshare RS485 CAN HAT? I dont have one but if need to I`ll one. Thank you.

ai-republic, 2024-04-20:
The PI4 has an AARCH64 OS platform. The RJ 45 goes right into your CAN port of the ARK. Then you will need to either cut the end off and use wire 4 and 5 (as seen above). Make sure you have the plug the right way around when determining the wires.
Another option is to buy something like this:

[image]

You will need to connect the CAN H and CAN L to the [Waveshare 2CH CAN hat](https://www.waveshare.com/2-ch-can-hat.htm) respectively:

[image]

jimmy737, 2024-04-20:
Thank you for the clear explanation.  I will try first option because I live on an island in the caribbean and buying electronic parts is not easy. I'll try that today and report back my progress.  Thank you for all the help. Greatly appreciated.

sanitariu, 2024-04-25:
I also can not detect the inverter via my PI CanHat. Maybe the cabling is not correct. I am using standart RJ-45 cable with connectors and cut the cable. Using blue and white-blue cables for CANL and CANH.
I tried setup the inverter to battery Li and type 51 or 52 and it goes to error 20/4 immediately. The bad thing is that I must connect my PI4 to other power source because when inverter errors it does not output energy. If someone have more success in connecting Growatt ES5000 inverter please share howto.

jimmy737, 2024-04-25:
> I also can not detect the inverter via my PI CanHat. Maybe the cabling is not correct. I am using standart RJ-45 cable with connectors and cut the cable. Using blue and white-blue cables for CANL [...]

I am connecting directly to the Growatt Ark battery the same way as you. But no success.

ai-republic, 2024-04-27:
Hi Jimmy,
the error message that is relevant here is:
_Failed to resolve the interface: can() - errorNumber=19, errorMessage='No such device'_

What CAN ports show up if you use the command:

```
ifconfig -a
```

Usually they are call `can0` and `can1`.

Another thing I noticed on your configuration you configure a Daly BMS. I would expect the Growatt ARK BMS to use the Growatt CAN protocol. If that doesn't work try the Pylon HV or the Pylon CAN protocol and see if you get some data.

Let me know if that helps.

@sanitariu have you checked the documentation which RJ45 wires are CAN H and CAN L for the inverter? I think they were pin 4 and pin 5. Maybe try to switch the cables.

Best regards,
Torsten

funkydeo, 2024-05-07:
Hi there, im just wondering if you managed to get the Growatt SPF (48v) inverter working with the BMS-inverter. I have some Lithium Po4 batteries and it would be nice to input data for BMS on inverter. i have two batteries but they are not charging/discharging equally on bms setting (Li) and so others use (user2) and input own values problem i am facing is that when i set one particular parameter it resets and i have to input it again the next day for the batteries to charge. so far only the bms function works and i will have to swap cables from top battery to bottom as not charging/discharging equally. hope this is the right place i am searchimg for my answer. cheers

funkydeo, 2024-05-07:
Hi Jimmy, thanks very much for your response.  Yes i think i may have to rotate the batteris in this case with my syatem.  At the moment my setting 36 is on 01 which was mentioned by the seller who sold me the battery(OT-LV-5120WH)/inverter(SPF 5000 ES). I havent tried it on 52 yet, is it just a protocol for the battery brand?   Im also waiting for the RS-232 connection from battery to check detailed info. 
thanks

ai-republic, 2024-05-12:
@funkydeo yes, the Growatt SPF should be working with either the Growatt inverter CAN binding or the Pylon inverter CAN binding

KelasHA00, 2024-06-21:
Hi,
i have attempted and just now succeeded in coupling a Daly HV BMS to a Growatt SPH TL3 BH UP thank to this thread.
i was stuck for some time, but changing from Growatt Can to Pylontech HV Can solved the communication to Inverter.
Could you update the wiki to make this clear?

I will definitly send support to this project. Great work!

i have some minor issues with 'Next frame: Buffer (HEX): Null - No bytes recieved' from Daly HV BMS every 8 or so seconds.
But communication works.

ai-republic, 2024-06-22:
Hi @KelasHA00,
great to hear you got it working and thanks for the donation!!!
Yes, the message 'Next frame: Buffer (HEX): Null - No bytes recieved' is normal. Occasionally the read frame buffer has not been received yet or is not fully filled with frame bytes.

Your suggestion is great to add a Wiki page documenting which binding works with which BMS and which Inverter! 
Thanks I will definitely create that and always great to get feedback to make it more complete.

KelasHA00, 2024-06-22:
Well, the description in the Inverter settings leads to think the Growatt Can is the correct one. But i guess it only applies to LV Growatt inverters. I spend quite some hours double checking my wiring and resetting the Pi.

The daly only works on rs485, but should be able to communicate on Can also.
I have some Pi 4bs and both the waveshare 2ch can and rs485 can hat i can test on.
Is this something we can look into?

KelasHA00, 2024-06-22:
One issue i just noticed now is, the inverter is mostly idling the battery when using Lithium setting.
In the log its reporting 'dis-/charge-state: stationary.

Battery at 40% and no change when excess power should trigger a charge.
And no discharge when load is introduced.

Daly is a HV / 48s LiIon series the Inverter a SPH TL3 BH UP running firmware YA1.0.

ai-republic, 2024-06-22:
> Well, the description in the Inverter settings leads to think the Growatt Can is the correct one. But i guess it only applies to LV Growatt inverters. I spend quite some hours double checking my [...]

Yes, the CAN protocol works definetly better with the Daly. I have also 8 Daly's running via the Waveshare 2CH CAN hat on my RPI 4B. Must make sure that the Daly is really a CAN-enabled version (they do advertise CAN but need it specifically enabled on ordering), otherwise you're stuck with RS485.

ai-republic, 2024-06-22:
> ne issue i just noticed now is, the inverter is mostly idling the battery when using Lithium setting. In the log its reporting 'dis-/charge-state: stationary.
> 
> Battery at 40% and no change when excess power should trigger a charge. And no discharge when load is introduced. i will monitor the behavior over the day when more sun is available.
> 
> Daly is a HV / 48s LiIon series the Inverter a SPH TL3 BH UP running firmware YA1.0.

Have you checked the maximum discharge voltage for you Daly? This might cut off at 40% charge.

ai-republic, 2024-08-09:
Closing this issue since I have not heard anything for a long time. Please create a new issue if you encounter any problems.

closed 2024-08-09 by ai-republic

funkydeo, 2024-08-09:
Hi Torsten,

Thanks for the update, I have not had time to tinker with the inverter yet,
but i will let u know if I do encounter any problems once I do get time to
sort it out.

Kind regards,

Nick