ai-republic/bms-to-inverter #2: Building on PI4

sanitariu, 2023-10-04:
Hello,
I am trying to build this on a PI4 device.
Here is the error i got:

[ERROR] Failed to execute goal org.apache.maven.plugins:maven-compiler-plugin:3.10.1:compile (default-compile) on project core-api: Fatal error compiling: error: release version 17 not supported -> [Help 1]

Maybe my mvn is too new ?

root@sol:~/bms-to-inverter # mvn -v
Apache Maven 3.6.3
Maven home: /usr/share/maven
Java version: 11.0.18, vendor: Raspbian, runtime: /usr/lib/jvm/java-11-openjdk-armhf
Default locale: en_US, platform encoding: UTF-8
OS name: "linux", version: "6.1.31-v8+", arch: "arm", family: "unix"

ai-republic, 2023-10-05:
Hi Todor,
it looks like you don't have a Java 17 or newer installed on your PI.
You can download the latest version ([JDK 21](https://jdk.java.net/21/))  and choose your target architecture aarch64.
Then unpack it to e.g. `/usr/lib/jvm` and follow the guide to update the default Java runtime [here](https://linuxhint.com/update-java-raspberry-pi/)

ai-republic, 2023-10-19:
I think I can close this issue

closed 2023-10-19 by ai-republic

sanitariu, 2023-11-20:
dri@dri ~/bms-to-inverter $ mvn -v
Apache Maven 3.9.5 (57804ffe001d7215b5e7bcb531cf83df38f93546)
Maven home: /usr/share/maven-bin-3.9
Java version: 1.8.0_252, vendor: IcedTea, runtime: /opt/icedtea-bin-3.16.0/jre
Default locale: en_US, platform encoding: ANSI_X3.4-1968
OS name: "linux", version: "5.10.128-gentoo", arch: "amd64", family: "unix"

This is my mvn version. Trying to build gave me error flag --release, so i commented out <release> stuff in pom.xml and run again.
Now i have another error:
[INFO] Compiling 19 source files to /home/dri/bms-to-inverter/core-api/target/classes
[INFO] -------------------------------------------------------------
[ERROR] COMPILATION ERROR :
[INFO] -------------------------------------------------------------
[ERROR] /home/dri/bms-to-inverter/core-api/src/main/java/com/airepublic/bmstoinverter/core/EnergyStorageProducer.java:[13,33] cannot access jakarta.enterprise.inject.Produces
  bad class file: /home/dri/.m2/repository/jakarta/enterprise/jakarta.enterprise.cdi-api/4.0.1/jakarta.enterprise.cdi-api-4.0.1.jar(jakarta/enterprise/inject/Produces.class)
    class file has wrong version 55.0, should be 52.0
    Please remove or make sure it appears in the correct subdirectory of the classpath.
[INFO] 1 error
[INFO] -------------------------------------------------------------
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary for bms-to-inverter-parent 0.0.1-SNAPSHOT:
[INFO]
[INFO] bms-to-inverter-parent ............................. SUCCESS [  0.373 s]
[INFO] core-api-0.0.1-SNAPSHOT ............................ FAILURE [  2.370 s]
[INFO] protocol-rs485-0.0.1-SNAPSHOT ...................... SKIPPED
[INFO] protocol-can-0.0.1-SNAPSHOT ........................ SKIPPED
[INFO] bms-daly-common-0.0.1-SNAPSHOT ..................... SKIPPED
[INFO] bms-daly-rs485-0.0.1-SNAPSHOT ...................... SKIPPED
[INFO] bms-daly-can-0.0.1-SNAPSHOT ........................ SKIPPED
[INFO] inverter-sma-can-0.0.1-SNAPSHOT .................... SKIPPED
[INFO] service-mqtt-broker-0.0.1-SNAPSHOT ................. SKIPPED
[INFO] service-mqtt-client-0.0.1-SNAPSHOT ................. SKIPPED
[INFO] bms-to-inverter-main-0.0.1-SNAPSHOT ................ SKIPPED
[INFO] webserver-0.0.1-SNAPSHOT ........................... SKIPPED
[INFO] protocol-modbus-0.0.1-SNAPSHOT ..................... SKIPPED
[INFO] inverter-growatt-rs485-0.0.1-SNAPSHOT .............. SKIPPED
[INFO] inverter-growatt-can-0.0.1-SNAPSHOT ................ SKIPPED
[INFO] bms-jk-can-0.0.1-SNAPSHOT .......................... SKIPPED
[INFO] bms-seplos-can-0.0.1-SNAPSHOT ...................... SKIPPED
[INFO] inverter-deye-can-0.0.1-SNAPSHOT ................... SKIPPED
[INFO] inverter-solark-can-0.0.1-SNAPSHOT ................. SKIPPED
[INFO] bms-pylon-can-0.0.1-SNAPSHOT ....................... SKIPPED
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  3.648 s
[INFO] Finished at: 2023-11-20T15:46:34+02:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-compiler-plugin:3.10.1:compile (default-compile) on project core-api: Compilation failure
[ERROR] /home/dri/bms-to-inverter/core-api/src/main/java/com/airepublic/bmstoinverter/core/EnergyStorageProducer.java:[13,33] cannot access jakarta.enterprise.inject.Produces
[ERROR]   bad class file: /home/dri/.m2/repository/jakarta/enterprise/jakarta.enterprise.cdi-api/4.0.1/jakarta.enterprise.cdi-api-4.0.1.jar(jakarta/enterprise/inject/Produces.class)
[ERROR]     class file has wrong version 55.0, should be 52.0
[ERROR]     Please remove or make sure it appears in the correct subdirectory of the classpath.
[ERROR]
[ERROR] -> [Help 1]
[ERROR]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException
[ERROR]
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <args> -rf :core-api

Can you tell me the OS you build the zip file and i will install the same ?

sanitariu, 2023-11-20:
Ok i build it finally :) Now setting up MQQT