ai-republic/bms-to-inverter #3: Hardware connection to the Inverter

sanitariu, 2023-10-22:
Hello,
I have little technical question about connecting Growatt to PI.
I have 3 ways to connect it. One is via RS-485 to USB but as i see you started only CAN. Next way is connecting via CAN from the inverter to USB port in the PI device. Growatt has one USB port too which i have already once connected to my PI for reading inverter data. Now i am wondering how exactly to connect it to the inverter. I have received CAN to USB modules so i can use that too. Also i think growatt has some kind of password which is growatt+current_date. Can you give a little help how exactly you are expecting the data from the inverter via CAN ?

ai-republic, 2023-10-24:
Hi Todor,
can you tell me the exact model of your Growatt inverter? Then I can have a look in the manual what would be the right connection for the CAN communication. Usually its via a RJ45 plug (network cable) where some pins act as CAN H (high) and CAN L (low). These pins need to be connected to your CAN to USB module. Try and unplug your module then plug it in again and run `dmesg` to check which port it is running on. Then you should be able to find your module running on port ttyS0 or similar.
Please tell me also the brand and model of your CAN to USB module. I use the [Waveshare 2CH-CAN-FD expansion hat](https://www.waveshare.com/2-ch-can-fd-hat.htm) since I read the BMS on one channel and write to the inverter on the other channel so that there is no collision.

sanitariu, 2023-10-24:
Hello,
Thank you very much for your support !
My inverter is - Growatt SPF 5000 XS.
Yes I have bms port and rs-485 port. I also have usb port where currently i have Wifi dongle which is connected to growatt website and is reporting inverter data. Also i can edit inverter settings from that web page. I can edit from inverter buttons too. I have tried and removed the dongle. At that place i have connected via usb cable directly to my PI device.  I am able to read data using growatt-x000ES programe downloaded from github. So i think i have 3 different types of connections i can use. I have ordered and have CAN and RS-485 adapters (usb or ttl) so i can use that too.
My perfect goal is to read battery data from Daly BMS and send it to the inverter. So inverter knows how much battery ...
Currently i have the following can/rs-485 adapters:

1. https://www.aliexpress.com/item/1005005373061818.html?spm=a2g0o.order_list.order_list_main.11.60971802MVAg51
2. https://www.aliexpress.com/item/1005005461959890.html?spm=a2g0o.order_list.order_list_main.17.60971802MVAg51
3. https://www.aliexpress.com/item/1005003395927937.html?spm=a2g0o.order_list.order_list_main.35.60971802MVAg51
4. https://www.aliexpress.com/item/1005005526945498.html?spm=a2g0o.order_list.order_list_main.54.60971802MVAg51
5. https://www.aliexpress.com/item/1005005632050165.html?spm=a2g0o.order_list.order_list_main.60.60971802MVAg51
6. https://www.aliexpress.com/item/1005001621746811.html?spm=a2g0o.order_list.order_list_main.72.60971802MVAg51

I ordered many so i can test all :)
I can order yours but we will wait some days to arrive.

ai-republic, 2023-10-24:
Hi Todor,
thanks for the quick reply. Is https://www.aliexpress.com/item/1005005461959890.html  a CAN or RS485 TTL module?
I also bought lots of modules but some didn't work as expected and the best suitable I think is the [Waveshare 2-Channel CAN FD HAT](https://www.waveshare.com/2-ch-can-fd-hat.htm) module because it has 2 isolated CAN channels. It is a little more expensive but does the job perfectly.
I hope I have time tomorrow, then I will check the inverter specifications?

sanitariu, 2023-10-24:
Hello, this link is CAN TTL i have ordered that. I have all the devices above.

ai-republic, 2023-10-25:
Hi Todor,
I've checked the manual of your inverter and on page 6 they give you all the info how to connect the RJ45 to the inverter on pins 4 (CANH) & 5 (CANL).  

[image]

On your Daly BMS its exactly the other way around pin 4 (CANL) & 5 (CANH).

[image]

I've had a look at the CAN to TTL module but I think that it will only be usable to _read_ CAN frames. But for a bi-directional communication I fear its to slow, since I don't see a quartz timer on it, I think it will only be working at 9600 baud.

sanitariu, 2023-10-25:
I am going to try anyway. Do you think that https://www.aliexpress.com/item/1005005373061818.html is better ? If none of the both adapters are working i am going to order one like yours.

sanitariu, 2023-11-17:
Hi, Finally i got Waveshare 2CH-CAN-FD. Going to install it today and prepare for the tests.
How is your Daly BMS connected to PI ? Mine is via ttl-to-usb adapter and the usb adapter is in PI device.
Next part is installing your software :)

ai-republic, 2023-11-18:
Mode A with resistor jumper ON

ai-republic, 2024-01-15:
Closed. All issues are resolved I guess.

closed 2024-01-15 by ai-republic