ai-republic/bms-to-inverter #5: Daly BMS - ttl

sanitariu, 2023-11-17:
Hello,
Currently you are supporting Daly BMS (RS485 or CAN).
I have bad luck with my daly bms and i am missing the hardware parts in my Daly for CAN/RS485. 
I have only TTL usb.

[image]

Maybe i should use USB to RS-485 but how i am gonna connect it to my PI ? Now i can connect directly via USB
and read data like cells , soc and all of this.

ai-republic, 2023-11-18:
Did you confirm this with the Daly support that CAN is not enabled for your BMS?

The cable in the picture is USB to UART so you should find a new device on your PI. 
Have you tried the `dmesg` command before and after you connect the USB device? Then you should see which device it adds which you need to configure in the `config.properties` file.
This should work the same as RS485 protocol.

sanitariu, 2023-11-18:
Yes i confirm this. There are no CAN/RS-485 hardware parts inside my BMS. Only TTL. Using the converter i have this in PI 
[   59.569599] usb 1-1.3.3: ch341-uart converter now attached to ttyUSB0
I can read data like SOC/cells etc ... via some python tools.
Do i need to attach another converter USB to CAN so i plug direct in my waveshare or i can use direct your software /dev/ttyUSB0 ?

ai-republic, 2023-11-19:
Yes, you only need to configure it in the `config.properties` and adjust it like this:
```
bms.port = /dev/ttyUSB0
```

I guest the port will run with 9600 baud, so you won't have to change anything else. 

First let's test if we get proper data from the Daly BMS over the ttyUSB0 port.
For that you must activate the DALY RS485 BMS dependencies.
To do that, just uncomment them and comment out any other bms or inverter dependency in the [`pom.xml`](https://github.com/ai-republic/bms-to-inverter/tree/main/bms-to-inverter-main/pom.xml) file of the [`bms-to-inverter-main`](https://github.com/ai-republic/bms-to-inverter/tree/main/bms-to-inverter-main) folder.

So the BMS section should only have the following uncommented:
```
<!-- ####################  DALY (RS485)  ################### -->
<dependency>
	<groupId>com.ai-republic.bms-to-inverter</groupId>
	<artifactId>bms-daly-rs485</artifactId>
	<version>${project.version}</version>
</dependency>
```

The inverter and optional services should all be uncommented for now.

Next, let's increase the information that's written to the console.
For that change [`log4j.xml`](https://github.com/ai-republic/bms-to-inverter/blob/main/bms-to-inverter-main/src/main/resources/log4j2.xml) the root log level from `"info"` to `"debug"`:

```
<Loggers>
	<Root level="info">
		<AppenderRef ref="Console" />
	</Root>
</Loggers>
```

Then you should be ready to start - follow the instructions [4. Building the project](https://github.com/ai-republic/bms-to-inverter/wiki/How-to-use#4-building-the-project) to build the project and then copy the created `zip` file to your PI and start as described.
Please post the log on the console so we can check if everything works.

After that we will do the CAN connection...

sanitariu, 2023-11-20:
Ok it is starting but crashing on MQTT service. Here is log of the crash.

root@sol:~/test # java -jar webserver-0.0.1-SNAPSHOT.jar
SLF4J: Class path contains multiple SLF4J providers.
SLF4J: Found provider [ch.qos.logback.classic.spi.LogbackServiceProvider@1fa3a9d]
SLF4J: Found provider [org.apache.logging.slf4j.SLF4JServiceProvider@ebc39e]
SLF4J: See https://www.slf4j.org/codes.html#multiple_bindings for an explanation.
SLF4J: Actual provider is of type [ch.qos.logback.classic.spi.LogbackServiceProvider@1fa3a9d]

  .   ____          _            __ _ _
 /\\ / ___'_ __ _ _(_)_ __  __ _ \ \ \ \
( ( )\___ | '_ | '_| | '_ \/ _` | \ \ \ \
 \\/  ___)| |_)| | | | | || (_| |  ) ) ) )
  '  |____| .__|_| |_|_| |_\__, | / / / /
 =========|_|==============|___/=/_/_/_/
 :: Spring Boot ::                (v3.1.3)

2023-11-20T18:13:34.556+02:00  INFO 6898 --- [           main] c.a.b.webserver.WebserverApplication     : Starting WebserverApplication using Java 17.0.9 with PID 6898 (/root/test/webserver-0.0.1-SNAPSHOT.jar started by root in /root/test)
2023-11-20T18:13:34.567+02:00  INFO 6898 --- [           main] c.a.b.webserver.WebserverApplication     : No active profile set, falling back to 1 default profile: "default"
2023-11-20T18:13:38.174+02:00  INFO 6898 --- [           main] o.s.b.w.embedded.tomcat.TomcatWebServer  : Tomcat initialized with port(s): 8080 (http)
2023-11-20T18:13:38.211+02:00  INFO 6898 --- [           main] o.apache.catalina.core.StandardService   : Starting service [Tomcat]
2023-11-20T18:13:38.212+02:00  INFO 6898 --- [           main] o.apache.catalina.core.StandardEngine    : Starting Servlet engine: [Apache Tomcat/10.1.12]
2023-11-20T18:13:38.535+02:00  INFO 6898 --- [           main] o.a.c.c.C.[Tomcat].[localhost].[/]       : Initializing Spring embedded WebApplicationContext
2023-11-20T18:13:38.539+02:00  INFO 6898 --- [           main] w.s.c.ServletWebServerApplicationContext : Root WebApplicationContext: initialization completed in 3728 ms
2023-11-20T18:13:38.948+02:00  INFO 6898 --- [           main] c.a.b.webserver.WebserverApplication     : Connecting to MQTT at tcp://localhost:1883/energystorage
2023-11-20T18:14:10.500+02:00 ERROR 6898 --- [           main] c.a.b.service.mqtt.MQTTConsumerService   : Error starting MQTT service!

org.apache.activemq.artemis.api.core.ActiveMQConnectionTimedOutException: AMQ219013: Timed out waiting to receive cluster topology. Group:null
	at org.apache.activemq.artemis.core.client.impl.ServerLocatorImpl.createSessionFactory(ServerLocatorImpl.java:741) ~[artemis-core-client-2.28.0.jar!/:2.28.0]
	at com.airepublic.bmstoinverter.service.mqtt.MQTTConsumerService.create(MQTTConsumerService.java:33) ~[service-mqtt-client-0.0.1-SNAPSHOT.jar!/:na]
	at com.airepublic.bmstoinverter.service.mqtt.MQTTConsumerService.create(MQTTConsumerService.java:19) ~[service-mqtt-client-0.0.1-SNAPSHOT.jar!/:na]
	at com.airepublic.bmstoinverter.webserver.WebserverApplication.startMQTTClient(WebserverApplication.java:75) ~[classes!/:na]
	at com.airepublic.bmstoinverter.webserver.WebserverApplication.init(WebserverApplication.java:68) ~[classes!/:na]
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method) ~[na:na]
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:77) ~[na:na]
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[na:na]
	at java.base/java.lang.reflect.Method.invoke(Method.java:568) ~[na:na]
	at org.springframework.beans.factory.annotation.InitDestroyAnnotationBeanPostProcessor$LifecycleMethod.invoke(InitDestroyAnnotationBeanPostProcessor.java:457) ~[spring-beans-6.0.11.jar!/:6.0.11]
	at org.springframework.beans.factory.annotation.InitDestroyAnnotationBeanPostProcessor$LifecycleMetadata.invokeInitMethods(InitDestroyAnnotationBeanPostProcessor.java:401) ~[spring-beans-6.0.11.jar!/:6.0.11]
	at org.springframework.beans.factory.annotation.InitDestroyAnnotationBeanPostProcessor.postProcessBeforeInitialization(InitDestroyAnnotationBeanPostProcessor.java:219) ~[spring-beans-6.0.11.jar!/:6.0.11]
	at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.applyBeanPostProcessorsBeforeInitialization(AbstractAutowireCapableBeanFactory.java:419) ~[spring-beans-6.0.11.jar!/:6.0.11]
	at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.initializeBean(AbstractAutowireCapableBeanFactory.java:1762) ~[spring-beans-6.0.11.jar!/:6.0.11]
	at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.doCreateBean(AbstractAutowireCapableBeanFactory.java:598) ~[spring-beans-6.0.11.jar!/:6.0.11]
	at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.createBean(AbstractAutowireCapableBeanFactory.java:520) ~[spring-beans-6.0.11.jar!/:6.0.11]
	at org.springframework.beans.factory.support.AbstractBeanFactory.lambda$doGetBean$0(AbstractBeanFactory.java:326) ~[spring-beans-6.0.11.jar!/:6.0.11]
	at org.springframework.beans.factory.support.DefaultSingletonBeanRegistry.getSingleton(DefaultSingletonBeanRegistry.java:234) ~[spring-beans-6.0.11.jar!/:6.0.11]
	at org.springframework.beans.factory.support.AbstractBeanFactory.doGetBean(AbstractBeanFactory.java:324) ~[spring-beans-6.0.11.jar!/:6.0.11]
	at org.springframework.beans.factory.support.AbstractBeanFactory.getBean(AbstractBeanFactory.java:200) ~[spring-beans-6.0.11.jar!/:6.0.11]
[... cut ...]
	... 26 common frames omitted
Caused by: org.apache.activemq.artemis.api.core.ActiveMQConnectionTimedOutException: AMQ219013: Timed out waiting to receive cluster topology. Group:null
	at org.apache.activemq.artemis.core.client.impl.ServerLocatorImpl.createSessionFactory(ServerLocatorImpl.java:741) ~[artemis-core-client-2.28.0.jar!/:2.28.0]
	at com.airepublic.bmstoinverter.service.mqtt.MQTTConsumerService.create(MQTTConsumerService.java:33) ~[service-mqtt-client-0.0.1-SNAPSHOT.jar!/:na]
	... 36 common frames omitted

ai-republic, 2023-11-21:
The MQTT broker is running in the application `bms-to-invert-main`. This needs to be running for the webserver to connect to the broker to poll data. So first start the application, then the webserver.

You don't need Mosquito broker.

sanitariu, 2023-11-21:
Yes, i figured that out. I was thinking that it is using my own MQTT broker and i set the same port. I will check that later. Main goal is to see data from my Daly BMS via TTL-to-USB adapter.

ai-republic, 2024-01-15:
Closed this issue. All issues resolved in the current version.

closed 2024-01-15 by ai-republic