ai-republic/bms-to-inverter #6: First time running with Daly TTL

sanitariu, 2023-11-20:
Here is the debug info as requested:

root@sol:~/test # java -jar bms-to-inverter-main-0.0.1-SNAPSHOT.jar

[debug.txt](https://github.com/ai-republic/bms-to-inverter/files/13417971/debug.txt)

also I am attaching my pom.xml and config.properties files

[ai.zip](https://github.com/ai-republic/bms-to-inverter/files/13417952/ai.zip)

sanitariu, 2023-11-20:
I think i need problem with config.properties setting things for RS485 port. It must be /dev/ttyUSB0

sanitariu, 2023-11-20:
This is my last changes. I think i am little ahead. Seems like the problem is opening serial port.

root@sol:~/test # java -jar bms-to-inverter-main-0.0.1-SNAPSHOT.jar
20:11:10.622 INFO : WELD-000900: 5.1.1 (Final)
20:11:11.140 WARN : WELD-ENV-002008: Bean class module-info found in multiple bean archives - this may result in incorrect behavior:
  - WeldBeanDeploymentArchive [id=/root/test/lib/email-javamail-1.0.5.jar],
  - WeldBeanDeploymentArchive [id=/root/test/lib/email-api-1.0.5.jar]
20:11:11.377 INFO : WELD-000101: Transactional services not available. Injection of @Inject UserTransaction not available. Transactional observers will be invoked synchronously.
20:11:12.473 INFO : WELD-ENV-002003: Weld SE container 8212513e-97f5-477c-ac0c-93827d7aaee7 initialized
20:11:13.056 INFO : AMQ221000: live Message Broker is starting with configuration Broker Configuration (clustered=false,journalDirectory=data/journal,bindingsDirectory=data/bindings,largeMessagesDirectory=data/largemessages,pagingDirectory=data/paging)
20:11:13.074 INFO : AMQ221045: libaio is not available, switching the configuration into NIO
20:11:13.148 INFO : AMQ221057: Global Max Size is being adjusted to 1/2 of the JVM max size (-Xmx). being defined as 129761280
20:11:13.220 INFO : AMQ221043: Protocol module found: [artemis-server]. Adding protocol support for: CORE
20:11:13.222 INFO : AMQ221043: Protocol module found: [artemis-mqtt-protocol]. Adding protocol support for: MQTT
20:11:13.224 INFO : AMQ221043: Protocol module found: [artemis-stomp-protocol]. Adding protocol support for: STOMP
20:11:13.358 INFO : AMQ601138: User anonymous@unknown is getting notification info on target resource: null
20:11:13.360 INFO : AMQ601019: User anonymous@unknown is getting mbean info on target resource: org.apache.activemq.artemis.core.management.impl.ActiveMQServerControlImpl@1c29daa
20:11:13.361 INFO : AMQ601138: User anonymous@unknown is getting notification info on target resource: ActiveMQServerImpl::name=localhost
20:11:13.443 INFO : AMQ601019: User anonymous@unknown is getting mbean info on target resource: org.apache.activemq.artemis.core.management.impl.AddressControlImpl@4ccef7
20:11:13.613 INFO : AMQ601019: User anonymous@unknown is getting mbean info on target resource: org.apache.activemq.artemis.core.management.impl.AcceptorControlImpl@5cbff2
20:11:13.959 INFO : AMQ221020: Started NIO Acceptor at 127.0.0.1:61616 for protocols [CORE,MQTT,STOMP]
20:11:13.960 INFO : AMQ221007: Server is now live
20:11:13.960 INFO : AMQ221001: Apache ActiveMQ Artemis Message Broker version 2.30.0 [localhost, nodeID=3065f73d-87d0-11ee-8259-22551d751e71]
20:11:14.279 INFO : AMQ601767: CORE connection e29de773 for user unknown@127.0.0.1:48510 created
20:11:14.325 INFO : AMQ601267: User anonymous@127.0.0.1:48510 is creating a core session on target resource ActiveMQServerImpl::name=localhost with parameters: [312cc9b0-87d0-11ee-8259-22551d751e71, null, ****, 102400, RemotingConnectionImpl [ID=e29de773, clientID=null, nodeID=3065f73d-87d0-11ee-8259-22551d751e71, transportConnection=org.apache.activemq.artemis.core.remoting.impl.netty.NettyServerConnection@ae8165[ID=e29de773, local= /127.0.0.1:61616, remote=/127.0.0.1:48510]], true, true, false, false, null, org.apache.activemq.artemis.core.protocol.core.impl.CoreSessionCallback@16ddc9a, true, {}]
20:11:14.506 INFO : AMQ601065: User anonymous@127.0.0.1:48510 is creating a queue on target resource: ServerSessionImpl() with parameters: [QueueConfiguration [id=null, name=energystorage, address=energystorage, routingType=null, filterString=null, durable=true, user=null, maxConsumers=-1, exclusive=null, groupRebalance=null, groupRebalancePauseDispatch=null, groupBuckets=null, groupFirstKey=null, lastValue=null, lastValueKey=null, nonDestructive=null, purgeOnNoConsumers=false, enabled=null, consumersBeforeDispatch=null, delayBeforeDispatch=null, consumerPriority=null, autoDelete=null, autoDeleteDelay=null, autoDeleteMessageCount=null, ringSize=1, configurationManaged=null, temporary=false, autoCreateAddress=null, internal=null, transient=null, autoCreated=false, fqqn=null]]
20:11:14.518 INFO : AMQ601019: User anonymous@127.0.0.1:48510 is getting mbean info on target resource: org.apache.activemq.artemis.core.management.impl.AddressControlImpl@710eda
20:11:14.618 WARN : AMQ222165: No Dead Letter Address configured for queue energystorage in AddressSettings
20:11:14.618 WARN : AMQ222166: No Expiry Address configured for queue energystorage in AddressSettings
20:11:14.660 INFO : AMQ601019: User anonymous@127.0.0.1:48510 is getting mbean info on target resource: org.apache.activemq.artemis.core.management.impl.QueueControlImpl@1d414f0
20:11:14.709 INFO : AMQ601768: CORE connection e29de773 for user anonymous@127.0.0.1:48510 destroyed
20:11:14.735 INFO : AMQ601767: CORE connection 1e190608 for user unknown@127.0.0.1:48520 created
20:11:14.749 INFO : AMQ601267: User anonymous@127.0.0.1:48520 is creating a core session on target resource ActiveMQServerImpl::name=localhost with parameters: [31708974-87d0-11ee-8259-22551d751e71, null, ****, 102400, RemotingConnectionImpl [ID=1e190608, clientID=null, nodeID=3065f73d-87d0-11ee-8259-22551d751e71, transportConnection=org.apache.activemq.artemis.core.remoting.impl.netty.NettyServerConnection@821afb[ID=1e190608, local= /127.0.0.1:61616, remote=/127.0.0.1:48520]], true, true, false, false, null, org.apache.activemq.artemis.core.protocol.core.impl.CoreSessionCallback@2e2dde, true [... cut ...]
	at com.airepublic.bmstoinverter.BmsToInverter$Proxy$_$$_WeldClientProxy.start(Unknown Source) [bms-to-inverter-main-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.BmsToInverter.main(BmsToInverter.java:68) [bms-to-inverter-main-0.0.1-SNAPSHOT.jar:?]
Caused by: java.lang.ClassNotFoundException: android.content.Context
	at jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:641) ~[?:?]
	at jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(ClassLoaders.java:188) ~[?:?]
	at java.lang.ClassLoader.loadClass(ClassLoader.java:525) ~[?:?]
	... 30 more

ai-republic, 2023-11-21:
I have checked your config.properties.
Do you really have 16 battery packs? A battery pack consists (usually) of more lithium cells like 4 for 12V, 8 for 24V and 16 for 48V, etc. I assume you have 1 battery pack consisting of 16 x 3.6V cells (1P16S), so the `numBatteryPacks` parameter should be 1.

If you need to change the MQTT broker locator port (61616) please be aware to change it also in the `application.properties` of the webserver project if you want use the web-app.

Also, please pull the latest version, I just fixed an issue in the JSerialCommPort to ensure the port is open. That was lost in the commit yesterday and should work now.

sanitariu, 2023-11-21:
Thank you very much for the support ! I hope we will make it working ... even if you need remote shell i can arrange it ! Now seems like ttyUSB0 is opened and much better. Still there is something wrong in reading or sending data from the com port.
Attaching the log files with debug enabled.

[config.properties.txt](https://github.com/ai-republic/bms-to-inverter/files/13424527/config.properties.txt)
[debug.txt](https://github.com/ai-republic/bms-to-inverter/files/13424529/debug.txt)
[pom.xml.txt](https://github.com/ai-republic/bms-to-inverter/files/13424531/pom.xml.txt)

sanitariu, 2023-11-21:
I think something should be added like protocol RS232 instead 485 ...

ai-republic, 2023-11-22:
My bad, the port wasn't initialized properly with the frameLength, baudrate and startFlag. I committed the  fix.

I will try and setup another PI today with the RS485/CAN hat so I can retest the RS485. I've been concentrating on CAN lately and only had one PI ready for testing.

sanitariu, 2023-11-22:
Hello, now is littler better ... we have some responses :)
09:28:44.140 WARN : Command 0x50 to BMS 1 successfully sent and received!
09:28:44.219 WARN : Command 0x53 to BMS 1 successfully sent and received!
09:28:44.289 WARN : Command 0x5a to BMS 1 successfully sent and received!
09:28:44.379 WARN : Command 0x5b to BMS 1 successfully sent and received!

But still have errors for id. Attaching the file.

[debug1.txt](https://github.com/ai-republic/bms-to-inverter/files/13435029/debug1.txt)

ai-republic, 2023-11-22:
Could you run this in debug mode please, then I can have a look what the frame looks like that has been received and why the msg.cmd is null. Thanks Todor!
BTW I've got my 2nd PI setup with the RS485 hat now so I can do some effective debugging again tomorrow

sanitariu, 2023-11-22:
Here is full debug. I see some good data !
10:03:09.365 DEBUG: ratedCellmV=3200, ratedCapacitymAh=280000
This is my battery 280AH
10:03:09.444 DEBUG: battery type={}=0 - this one is emtpy ?
10:03:09.612 DEBUG: maxPackChargeCurrent=2400, maxPackDischargeCurrent=-2400 - this i am not sure ...
Anyway i see the light in the tunnel :)

[fulldebug.txt](https://github.com/ai-republic/bms-to-inverter/files/13435408/fulldebug.txt)

sanitariu, 2023-11-22:
I could not find a way to share an access for you so i am giving it here.
I could not find your email.
Please change the password and erase this once you are in.
ssh to 88.87.8.77 port 44444 username: torsten / password: lPmdj19Msfw
Project is here - bms-to-inverter
Final is in folder - final

I think this will make things much more easy for you to develop and test.
I gave you an access to dialout group so you can open/read to ttyUSB0

sanitariu, 2023-11-22:
Thanks. Much much better ! Now i see info like this
10:41:10.911 INFO :
BMS     SOC       V       A      CellMinV        CellMaxV       CellDiff
1        1.7    52.2    7.4     3.259(#2)       3.271(#0.009)   0.012

Seems like USB (RS232) is working too :)

Here is the log file

[fulldebug.txt](https://github.com/ai-republic/bms-to-inverter/files/13435770/fulldebug.txt)

ai-republic, 2023-11-23:
Ok - that looks all good :)

I think I know where the last exception (gson - android/content/Context) is coming from. I fixed it

sanitariu, 2023-11-23:
Here is the latest debug log. Also i have tested webserver and it shows some data. I think data is not correct.

[fulldebug.txt](https://github.com/ai-republic/bms-to-inverter/files/13447574/fulldebug.txt)
[image]

ai-republic, 2023-11-24:
I fixed the gson error now and made the code more resilient against errors.

The data that's received is correct. That's what's reported by the BMS. The 'Cell 17/18' is actually reported in the 5th frame but can be ignored. It's only because Daly sends a 5th frame to send the 16th cells value. I guess they don't clear their buffer properly - that's where cell 17 and 18 result

sanitariu, 2023-11-24:
Hello,
tested again with latest fixes. Now it is working better. Actually it is flooding the console with info each second which i think is too fast sending / getting data. It should be maybe 5-10 seconds interval. Anyway it was working for 2-3 minutes they stops. Do you know how i can log all the output ?

These are the last lines received:

09:39:52.438 DEBUG: Validation of frame failed!
09:39:52.438 DEBUG: Ignoring frame: RX Buffer (HEX): [0x00, 0x01, 0x95, 0x08, 0x05, 0x0C, 0xCD, 0x0C, 0xC7, 0x0C, 0xCC, 0x01, 0x95]

09:40:06.981 DEBUG: Scanning for expires on energystorage
09:40:06.982 DEBUG: Scanning for expires on energystorage done

Also on the webserver i noticed that you have BMS8 but they should be only 1. Maybe config.properties is not read by the webserver.

ai-republic, 2023-11-25:
Hi Todor,
I could put in a configurable poll interval - that's no problem.

I guess that when it stops maybe the buffers are full on the PI. I'll have a look into that.

You can also log everything to file. I added a configuration to log to console and rotational log file.

sanitariu, 2023-11-26:
Here is the lates log file. I have tried 4-5 seconds but it does not help. I think it is stopping at the same time. Maybe buffer ?! You can login to my machine and test if you like. It is okay for me.

[BMS-to-Inverter.log.gz](https://github.com/ai-republic/bms-to-inverter/files/13466434/BMS-to-Inverter.log.gz)

ai-republic, 2023-11-27:
I've checked the log file and I guess its shutting down the BMS and inverter processing threads because of an uncaught exception coming from the MQTT service. I've generalized the exception handling so there shouldn't be any exceptions falling through.

Also I noticed that there are still 8 battery packs in your system. Is that correct? Or is it only 1? In that case you might need to check the `numBatteryPacks` configuration in `config.properties` and set it to 1.

sanitariu, 2023-11-27:
Hello, i have 1 pack and i fixed it now. Still it reads some data 5-6 times then it does not try anymore.

[BMS-to-Inverter.log.gz](https://github.com/ai-republic/bms-to-inverter/files/13471720/BMS-to-Inverter.log.gz)

sanitariu, 2023-11-27:
Ok i setup email and seems like it is working from 5 minutes :))) I put bms/inverter read data to 10 seconds.
On the webserver you must fix the number of packs. Maybe they are hardcoded.

sanitariu, 2023-11-30:
I tested today with latest fixes. Working for some minutes then it stops.
I think it is from "2023-11-29 18:48:00.909 | DEBUG | 2-thread-1 | s485.JSerialCommPort:132   | Validation of frame failed!" Maybe some data is corrupted or lagged but it should ignore data from that time and do not send to inverter anything ... then just retry on the next period ?
Attaching log file if you need it. Maybe you should try different approach like open com port , read data , close com port , then after the period open/read/close. I think com ports are more unstable than can and sometimes stuck for some period maybe ?!

[BMS-to-Inverter.log.gz](https://github.com/ai-republic/bms-to-inverter/files/13503341/BMS-to-Inverter.log.gz)

ai-republic, 2023-12-06:
Hi Todor,
sorry for the late reply - I was very busy the last few days. 
I logged in, tested and updated the application on your PI. Also made a small script in the home  folder to speed things up with pulling a new version, building and starting the app. 

Now to the problem. I've observed this also on my PI when running RS485 that Daly starts sending rubbish messages address to ID FF (255) or messages that are not to short. I've tried to compensate the short messages by reading data from the stream and searching for the start flag (0xA5) and restarting from there but it didn't make any difference because the data was rubbish.

The next approach I want to try would be to read the RS485 stream and buffer it locally and start reading the messages from there (also better for debugging) and when it starts sending rubbish try to send a reset command.

I will try to implement this on the weekend, since I'm still busy the next couple of days.

sanitariu, 2023-12-11:
Thank you very much for the support ! Waiting for stable 485 version so we can continue with the inverter :)

sanitariu, 2023-12-19:
Hello,
Tested latest fixes and data is read only once.

[debug.log](https://github.com/ai-republic/bms-to-inverter/files/13715045/debug.log)

ai-republic, 2024-01-15:
Closed this issue. All is fixed in the current version.

closed 2024-01-15 by ai-republic