ai-republic/bms-to-inverter #7: Running on Windows

jparfianowicz, 2024-01-16:
Hi, 

I am trying to run the solution on Windows 10. I managed to build the project but running bms-to-inverter throws below error:
Exception in thread "main" org.jboss.weld.exceptions.DeploymentException: WELD-001408: Unsatisfied dependencies for type Inverter with qualifiers @Default
  at injection point [BackedAnnotatedField] @Inject private com.airepublic.bmstoinverter.BmsToInverter.inverter
  at com.airepublic.bmstoinverter.BmsToInverter.inverter(BmsToInverter.java:0)

        at org.jboss.weld.bootstrap.Validator.validateInjectionPointForDeploymentProblems(Validator.java:367)
        at org.jboss.weld.bootstrap.Validator.validateInjectionPoint(Validator.java:285)
        at org.jboss.weld.bootstrap.Validator.validateGeneralBean(Validator.java:141)
        at org.jboss.weld.bootstrap.Validator.validateRIBean(Validator.java:162)
        at org.jboss.weld.bootstrap.Validator.validateBean(Validator.java:515)
        at org.jboss.weld.bootstrap.ConcurrentValidator$1.doWork(ConcurrentValidator.java:64)
        at org.jboss.weld.bootstrap.ConcurrentValidator$1.doWork(ConcurrentValidator.java:62)
        at org.jboss.weld.executor.IterativeWorkerTaskFactory$1.call(IterativeWorkerTaskFactory.java:62)
        at org.jboss.weld.executor.IterativeWorkerTaskFactory$1.call(IterativeWorkerTaskFactory.java:55)
        at org.jboss.weld.executor.CommonForkJoinPoolExecutorServices.lambda$wrap$0(CommonForkJoinPoolExecutorServices.java:70)
        at java.base/java.util.concurrent.ForkJoinTask$AdaptedInterruptibleCallable.exec(ForkJoinTask.java:1461)
        at java.base/java.util.concurrent.ForkJoinTask.doExec(ForkJoinTask.java:373)
        at java.base/java.util.concurrent.ForkJoinPool$WorkQueue.topLevelExec(ForkJoinPool.java:1182)
        at java.base/java.util.concurrent.ForkJoinPool.scan(ForkJoinPool.java:1655)
        at java.base/java.util.concurrent.ForkJoinPool.runWorker(ForkJoinPool.java:1622)
        at java.base/java.util.concurrent.ForkJoinWorkerThread.run(ForkJoinWorkerThread.java:165)

Unfortunately I am not a Java expert. Spent 2h googling for solution but could not make it work. Help would be appreciated.

ai-republic, 2024-01-17:
Hi,
could you please attach your `config.properties` (from bms-to-inverter-main/src/main/resources) and `pom.xml`  (from bms-to-inverter-main) please.

Please be aware that if you are running on Windows you will not be able to run any CAN protocol bindings - only RS485/RS232/UART.

jparfianowicz, 2024-01-17:
pom.xml

<project xmlns="http://maven.apache.org/POM/4.0.0"
	xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:schemaLocation="http://maven.apache.org/POM/4.0.0 https://maven.apache.org/xsd/maven-4.0.0.xsd">
	<modelVersion>4.0.0</modelVersion>
	<artifactId>bms-to-inverter-main</artifactId>

	<parent>
		<groupId>com.ai-republic.bms-to-inverter</groupId>
		<artifactId>bms-to-inverter-parent</artifactId>
		<version>0.0.1-SNAPSHOT</version>
	</parent>

	<name>${project.artifactId}-${project.version}</name>
	<description>Application to communicate between a BMS and inverter</description>

	<properties>
		<encoding>UTF-8</encoding>
		<project.build.sourceEncoding>UTF-8</project.build.sourceEncoding>
	</properties>

	<build>
		<plugins>
			<plugin>
				<artifactId>maven-jar-plugin</artifactId>
				<version>3.3.0</version>
				<configuration>
					<archive>
						<manifest>
							<addClasspath>true</addClasspath>
							<classpathPrefix>lib/</classpathPrefix>
							<mainClass>
								com.airepublic.bmstoinverter.BmsToInverter</mainClass>
						</manifest>
					</archive>
				</configuration>
			</plugin>

			<plugin>
				<groupId>org.apache.maven.plugins</groupId>
				<artifactId>maven-dependency-plugin</artifactId>
				<version>3.6.0</version>
				<executions>
					<execution>
						<id>copy-dependencies</id>
						<phase>package</phase>
						<goals>
							<goal>copy-dependencies</goal>
						</goals>
						<configuration>
							<outputDirectory>${project.build.directory}/lib</outputDirectory>
							<overWriteReleases>false</overWriteReleases>
							<overWriteSnapshots>false</overWriteSnapshots>
							<overWriteIfNewer>true</overWriteIfNewer>
						</configuration>
					</execution>
				</executions>
			</plugin>

			<plugin>
				<groupId>org.apache.maven.plugins</groupId>
				<artifactId>maven-assembly-plugin</artifactId>				
				<executions>
					<execution>
						<id>create-distribution</id>
						<phase>package</phase>
						<goals>
							<goal>single</goal>
						</goals>
						<configuration>
							<descriptors>
								<descriptor>assembly/zip.xml</descriptor>
							</descriptors>
						</configuration>
					</execution>
				</executions>
			</plugin>
		</plugins>
	</build>

	<dependencies>
		<dependency>
			<groupId>org.jboss.weld.se</groupId>
			<artifactId>weld-se-shaded</artifactId>
			<version>5.1.1.Final</version>
		</dependency>

		

		
		

		
		<dependency>
			<groupId>com.ai-republic.bms-to-inverter</groupId>
			<artifactId>bms-daly-rs485</artifactId>
			<version>${project.version}</version>
		</dependency>

		
		
		
		
		
		

		
		
		
		
		
		

		
		
		
		
		
		

		

		
		

		
		
		
		
		
		

		
		<dependency>
			<groupId>com.ai-republic.bms-to-inverter</groupId>
			<artifactId>inverter-growatt-can</artifactId>
			<version>${project.version}</version>
		</dependency>

		
		
		
		
		
		

		
		
		
		
		
		

		

		
		<dependency>
			<groupId>com.ai-republic.bms-to-inverter</groupId>
			<artifactId>service-mqtt-broker</artifactId>
			<version>${project.version}</version>
		</dependency>

		<dependency>
			<groupId>com.ai-republic.bms-to-inverter</groupId>
			<artifactId>service-mqtt-client</artifactId>
			<version>${project.version}</version>
		</dependency>

		
		<dependency>
			<groupId>com.ai-republic.email</groupId>
			<artifactId>email-javamail</artifactId>
			<version>1.0.5</version>
		</dependency>
	</dependencies>
</project>

jparfianowicz, 2024-01-17:
config.properties

###################################################################
###                  System specific settings                   ###
###################################################################

# The number of battery packs (not cells) of the system
numBatteryPacks=1

###################################################################
###                 Protocol specific settings                  ###
###################################################################

# RS485 properties
RS485.baudrate=9600
RS485.startFlag=165
RS485.frameLength=13

# ModBus properties
ModBus.baudrate=9600

###################################################################
###                       BMS settings                          ###
###################################################################

####  Simple single port configuration ####
# BMS port protocol (CAN/RS485/ModBus)
bms.portProtocol=RS485
#bms.portProtocol=CAN

# The port name/device to use to communicate to the BMS  
#bms.portLocator=com3
bms.portLocator=COM7
#bms.portLocator=/dev/ttyS0

#### Or for multiple BMSes connected to multiple ports #### 
#bms.0.portProtocol=CAN
#bms.0.portLocator=can0
#bms.1.portProtocol=CAN
#bms.1.portLocator=can1
#bms.2.portProtocol=CAN
#bms.2.portLocator=can2
#etc...

# Interval to request BMS data (in seconds) 
bms.pollInterval=2

###################################################################
###                    Inverter settings                        ###
###################################################################

# The port name/device to use to communicate to the  inverter  
inverter.portLocator=can1
# Interval to send data to the inverter (in seconds) 
inverter.sendInterval=1

###################################################################
###                 Optional services settings                  ###
###################################################################

#### MQTT properties ####
# The URL to of the MQTT broker  
mqtt.locator=tcp://127.0.0.1:61616
# The topic name on the MQTT broker
mqtt.topic=energystorage

#### Email properties ####
mail.out.debug=true
# SMTP or IMAP address of the outgoing server 
mail.out.host=smtp.gmail.com
# The port of the outgoing server
mail.out.port=587
# smtp for TLS, smtps for SSL
mail.out.type=smtp
# User name to authenticate at the outgoing server
mail.out.username=
# Password to authenticate at the outgoing server
mail.out.password= 
# Disable if using TLS
mail.out.sslEnable=false
# Disable if using SSL
mail.out.tlsEnable=true
# The email address to use when sending emails
mail.out.defaultEmail=
# A (comma separated) list of pre-configured email recipients
mail.recipients=

ai-republic, 2024-01-18:
Hi,
ok thanks. I found the problem and fixed it.
Please pull the latest version.

I would advise to start first only with the dummy inverter configured just to test the connection to your BMS (Daly right?) is working correctly.

kommando828, 2024-01-18:
This is my go to when in a mess with RS485 Modbus, runs under windows as long as Java is installed.

https://github.com/SciFiDryer/ModbusMechanic

Use the register scanner , tools/register scanner, select RTU, set comms and then scan for the expected registers and try different options such as Slave ID, baud rates etc until you get a result.

ai-republic, 2024-01-19:
Hi, 
so up to command 0x95 its running fine. 0x95 is where all the cell voltages are read and it is split up in multiple frames. Daly seems to take longer here to prepare the frames and put them on the wire. 
I've had this problem with someone else's Daly via UART also.
The fix there was to wait a little longer between the readings.
I will add another config parameter to set a delay between commands.

What happens after the first reading round? Does this repeat?

closed 2024-02-14 by ai-republic