ai-republic/bms-to-inverter #9: configuration issues

sanitariu, 2024-01-28:
After latest commits seems like webserver and email is broken. Also there are many errors in cleanrepo.sh

ai-republic, 2024-01-29:
Hey,
yes, I've done lots of changes, so the installation and configuration will be a lot easier. 
See [Wiki - How to use](https://github.com/ai-republic/bms-to-inverter/wiki/How-to-use) for a detailed description on how to use the [Configurator](https://github.com/ai-republic/bms-to-inverter/blob/main/configurator/current/configurator.jar).

I've externaized the whole configuration and it will produce and build the whole application. No need to build the whole application everytime there's a config or logging change ;)

sanitariu, 2024-01-29:
Seems reasonable but this left out people without X installation. For example me :) I do not have any kind of graphical installation. I would like just to git clone, edit manually pom.xml, config.properties then compile with mvn clean package and run. I will try latest code from today maybe it will work as expected.

ai-republic, 2024-01-29:
Ah, but you don't need a UI on your PI. Just use your Windows/MacOS/Ubuntu or whatever system to run the Configurator. Then push it to your PI via WinSCP or similar. Then you just edit your `config/config.properties` and `config/log4j2.xml` manually.

sanitariu, 2024-01-29:
I tried on MacOS. There are problems for example i can not edit BMS type only DALY_CAN available.

ai-republic, 2024-01-30:
You mean the combobox doesn't open on MacOS? Did you pull the new sources? 
That would only be the case if you commented the other BMSes out in the  _configurator_ module `pom.xml` (or later on if you run the configurator start script in the _bms-to-invert-main_ module `pom.xml`). Could you check that please? Maybe you copied the old pom.xml.mine and overwrote the original. You should no longer copy/change the `pom.xml` anymore.

Yes, to activate the Growatt inverter just set the `inverter.type=GROWATT_CAN` and the `inverter.portLocator=can0` if connected to the can0 port.

ai-republic, 2024-01-30:
FYI: I've changed the config property `mail.recipients` to  `mail.out.recipients` so you might need to adjust your `config.properties` if you're updating the application.

closed 2024-02-01 by ai-republic