ai-republic/bms-to-inverter #10: Webserver module

sanitariu, 2024-02-01:
I noticed a bug that after sometime 24 hours the webserver is working but do not refresh values like SoC and other correctly. Actually they stay as they were previous day. App seems to work no crash. I stop and start it and everything is displayed normal updated. If you need more info you can email me for live access.

ai-republic, 2024-02-02:
Thanks Todor, I will have a look into this. Do you have a log file of the _bms-to-inverter-main_ application? Has the MQTT server stopped?

sanitariu, 2024-02-02:
I am sending the log files. It happened again. Can you check if it happens when UART crash ?
If i restart webserver only it will start working good.
[logs.tar.gz](https://github.com/ai-republic/bms-to-inverter/files/14138408/logs.tar.gz)

ai-republic, 2024-02-03:
I can't find any errors in the logs. I suspect it could be a session timeout on the webserver. I will check that.
If possible you could send me a console log of the browser. If you press F12 you usually get the developer console (at least on FF or Edge).

sanitariu, 2024-02-06:
After 1 day it stops always. I tested second browser. Web console 

[yoroi/prod] checking if should inject dapp-connector api [inject.js:63:17](moz-extension://271a6ac3-ddc9-7d4f-a037-2a7dc4542117/js/inject.js)
[yoroi/prod] injecting over 'nothing' [inject.js:66:21](moz-extension://271a6ac3-ddc9-7d4f-a037-2a7dc4542117/js/inject.js)
[yoroi/prod] dapp-connector is successfully injected into xx.xx.xx.xx

sanitariu, 2024-02-06:
Webserver even shows wrong data although i have restarted it.

ai-republic, 2024-02-07:
> [yoroi/prod] checking if should inject dapp-connector api inject.js:63:17
> [yoroi/prod] injecting over 'nothing' inject.js:66:21
> [yoroi/prod] dapp-connector is successfully injected into xx.xx.xx.xx

This is nothing from my application.
I suspect its a timeout in the webserver. I haven't had time to look into it yet.
Can you give me more detailed information on what wrong data it displays after restart?

sanitariu, 2024-02-07:
It is displaying old information from yesterday for example ...

ai-republic, 2024-02-07:
Ok, that would mean that the BMS reading has stopped or the MQTT server. I believe I saw that the BMS reading was still running, so maybe its a timeout or validity of the MQTT topic on the MQTT server.
Thanks, that's a possibility to look into.

sanitariu, 2024-02-16:
For last hours seems like webserver is working. Lets see one more day and if it still refreshes it means problem is fixed.

ai-republic, 2024-02-16:
yes. I've also been keeping an eye on it :)

sanitariu, 2024-02-18:
After testing 1 day ... seems like webserver is not stable and stops refreshing the information for some reason. No crashes or anything just stop. After restart it works again. Maybe timeout or something like that. When you have time look at it.

ai-republic, 2024-03-21:
@sanitariu I think I fixed the webserver issue due to a MQTT producer disconnect. I now try to reconncect if the message could not be sent.

sanitariu, 2024-03-21:
Hi, I have downloaded latest release and this is what i have:

root@sol:~/bms # java -jar -DconfigFile=config/config.properties -Dlog4j2.configurationFile=file:config/log4j2.xml lib/bms-to-inverter-main-0.0.1-SNAPSHOT.jar
2024-03-21 10:08:13.145 | INFO  | main       | erter.core.util.Util:25    | Loading config.properties from: config/config.properties
2024-03-21 10:08:13.240 | INFO  | main       | ootstrap.WeldStartup:150   | WELD-000900: 5.1.1 (Final)
2024-03-21 10:08:13.810 | INFO  | main       | ionDiscoveryStrategy:52    | WELD-ENV-000014: Falling back to Java Reflection for bean-discovery-mode="annotated" discovery. Add org.jboss:jandex to the classpath to speed-up startup.
2024-03-21 10:08:13.867 | WARN  | main       | .environment.se.Weld:1015  | WELD-ENV-002008: Bean class module-info found in multiple bean archives - this may result in incorrect behavior:
  - WeldBeanDeploymentArchive [id=/root/bms/lib/email-api-1.0.5.jar],
  - WeldBeanDeploymentArchive [id=/root/bms/lib/email-javamail-1.0.5.jar]
2024-03-21 10:08:14.088 | INFO  | main       | ootstrap.WeldStartup:228   | WELD-000101: Transactional services not available. Injection of @Inject UserTransaction not available. Transactional observers will be invoked synchronously.
2024-03-21 10:08:15.344 | INFO  | main       | ent.se.WeldContainer:239   | WELD-ENV-002003: Weld SE container c38c6bd4-e4ec-4630-bf40-33dfa6aa3924 initialized
2024-03-21 10:08:15.915 | INFO  | main       | eMQServerLogger_impl:45    | AMQ221000: live Message Broker is starting with configuration Broker Configuration (clustered=false,journalDirectory=data/journal,bindingsDirectory=data/bindings,largeMessagesDirectory=data/largemessages,pagingDirectory=data/paging)
2024-03-21 10:08:15.933 | INFO  | main       | eMQServerLogger_impl:397   | AMQ221045: libaio is not available, switching the configuration into NIO
2024-03-21 10:08:16.009 | INFO  | main       | eMQServerLogger_impl:493   | AMQ221057: Global Max Size is being adjusted to 1/2 of the JVM max size (-Xmx). being defined as 129761280
2024-03-21 10:08:16.084 | INFO  | main       | eMQServerLogger_impl:389   | AMQ221043: Protocol module found: [artemis-server]. Adding protocol support for: CORE
2024-03-21 10:08:16.087 | INFO  | main       | eMQServerLogger_impl:389   | AMQ221043: Protocol module found: [artemis-mqtt-protocol]. Adding protocol support for: MQTT
2024-03-21 10:08:16.089 | INFO  | main       | eMQServerLogger_impl:389   | AMQ221043: Protocol module found: [artemis-stomp-protocol]. Adding protocol support for: STOMP
2024-03-21 10:08:16.228 | INFO  | main       | ogs.AuditLogger_impl:1139  | AMQ601138: User anonymous@unknown is getting notification info on target resource: null
2024-03-21 10:08:16.231 | INFO  | main       | ogs.AuditLogger_impl:187   | AMQ601019: User anonymous@unknown is getting mbean info on target resource: org.apache.activemq.artemis.core.management.impl.ActiveMQServerControlImpl@972170
2024-03-21 10:08:16.233 | INFO  | main       | ogs.AuditLogger_impl:1139  | AMQ601138: User anonymous@unknown is getting notification info on target resource: ActiveMQServerImpl::name=localhost
2024-03-21 10:08:16.317 | INFO  | main       | ogs.AuditLogger_impl:187   | AMQ601019: User anonymous@unknown is getting mbean info on target resource: org.apache.activemq.artemis.core.management.impl.AddressControlImpl@1d8899
2024-03-21 10:08:16.484 | INFO  | main       | ogs.AuditLogger_impl:187   | AMQ601019: User anonymous@unknown is getting mbean info on target resource: org.apache.activemq.artemis.core.management.impl.AcceptorControlImpl@191f50e
2024-03-21 10:08:16.857 | INFO  | main       | eMQServerLogger_impl:205   | AMQ221020: Started NIO Acceptor at 127.0.0.1:61616 for protocols [CORE,MQTT,STOMP]
2024-03-21 10:08:16.858 | INFO  | main       | eMQServerLogger_impl:101   | AMQ221007: Server is now live
2024-03-21 10:08:16.859 | INFO  | main       | eMQServerLogger_impl:53    | AMQ221001: Apache ActiveMQ Artemis Message Broker version 2.30.0 [localhost, nodeID=2b887578-e75a-11ee-be27-8eb748af779a]
2024-03-21 10:08:17.193 | INFO  | y-threads) | ogs.AuditLogger_impl:2843  | AMQ601767: CORE connection 66fccd0b for user unknown@127.0.0.1:36474 created
2024-03-21 10:08:17.238 | INFO  | 6@1eb5790) | ogs.AuditLogger_impl:2171  | AMQ601267: User anonymous@127.0.0.1:36474 is creating a core session on target resource ActiveMQServerImpl::name=localhost with parameters: [2c58e4db-e75a-11ee-be27-8eb748af779a, null, ****, 102400, RemotingConnectionImpl [ID=66fccd0b, clientID=null, nodeID=2b887578-e75a-11ee-be27-8eb748af779a, transportConnection=org.apache.activemq.artemis.core.remoting.impl.netty.NettyServerConnection@d18eae[ID=66fccd0b, local= /127.0.0.1:61616, remote=/127.0.0.1:36474]], true, true, false, false, null, org.apache.activemq.artemis.core.protocol.core.impl.CoreSessionCallback@c9ee37, true, {}]
2024-03-21 10:08:17.496 | INFO  | 6@1eb5790) | ogs.AuditLogger_impl:555   | AMQ601065: User anonymous@127.0.0.1:36474 is creating a queue on target resource: ServerSessionImpl() with parameters: [QueueConfiguration [id=null, name=energystorage, address=energystorage, routingType=null, filterString=null, durable=true, user=null, maxConsumers=-1, exclusive=null, groupRebalance=null, groupRebalancePauseDispatch=null, groupBuckets=null, groupFirstKey=null, lastValue=null, lastValueKey=null, nonDestructive=null, purgeOnNoConsumers=false, enabled=null, consumersBeforeDispatch=null, delayBeforeDispatch=null, consume [... cut ...]
2024-03-21 10:08:18.053 | INFO  | Thread-1   | eMQServerLogger_impl:61    | AMQ221002: Apache ActiveMQ Artemis Message Broker version 2.30.0 [2b887578-e75a-11ee-be27-8eb748af779a] stopped, uptime 2.219 seconds
2024-03-21 10:08:18.054 | INFO  | Thread-1   | tt.MQTTBrokerService:98    | Shutting down MQTT broker on 'tcp://127.0.0.1:61616'...OK
2024-03-21 10:08:18.055 | INFO  | Thread-1   | verter.BmsToInverter:476   | Shutting down MQTT broker threads...OK
2024-03-21 10:08:18.057 | INFO  | Thread-1   | verter.BmsToInverter:484   | Shutting down ports...OK

sanitariu, 2024-03-21:
> The important lines are:
> So there is no configuration file found in your installation folder under config/config.properties

Hi,
There is the file
***@***.***:~/bms # pico config/config.properties
***@***.***:~/bms # ls -la config/config.properties
-rw-r--r-- 1 sani sani 3150 Mar 21 10:14 config/config.properties
***@***.***:~/bms #

I even run like this:

***@***.***:~/bms # java -DconfigFile=/root/bms/config/config.properties -Dlog4j2.configurationFile=file:config/log4j2.xml -jar lib/bms-to-inverter-main-0.0.1-SNAPSHOT.jar

***@***.***:~/bms # ls -la /root/bms/config/config.properties
-rw-r--r-- 1 sani sani 3150 Mar 21 10:14 /root/bms/config/config.properties

###################################################################
###                  System specific settings                   ###
###################################################################

###################################################################
###                       BMS settings                          ###
###################################################################

####  Simple single port configuration ####
# bms.x.type - can be (DALY_CAN, DALY_RS485, JK_CAN, PYLON_CAN or SEPLOS_CAN
# bms.x.portLocator - is the locator/device to use to communicate to the BMS, eg. can0, /dev/ttyUSB0, com3, etc.
# bms.x.pollIntervall - is the interval to request BMS data (in seconds)
# bms.x.delayAfterNoBytes - is the delay after receiving no data (in ms)
bms.0.type=DALY_RS485
bms.0.portLocator=/dev/ttyAMA0
bms.0.pollInterval=5
bms.0.delayAfterNoBytes=200

###################################################################
###                    Inverter settings                        ###
###################################################################
# The inverter type can be NONE, DEYE_CAN, GROWATT_CAN, SMA_SI_CAN, SOLARK_CAN
inverter.type=NONE
# The port name/device to use to communicate to the  inverter
inverter.portLocator=/dev/can1
# Interval to send data to the inverter (in seconds)
inverter.sendInterval=5

I tried to stop Webserver but not starting.

Best Regards,
Todor

sanitariu, 2024-03-21:
I think i found the problem. config.properties should be like that:

####  Simple single port configuration ####
# bms.pollIntervall - is the interval to request BMS data (in seconds)
# bms.x.type - can be (DALY_CAN, DALY_RS485, JK_CAN, PYLON_CAN or SEPLOS_CAN
# bms.x.portLocator - is the locator/device to use to communicate to the BMS, eg. can0, /dev/ttyUSB0, com3, etc.
# bms.x.delayAfterNoBytes - is the delay after receiving no data (in ms)
bms.pollInterval=5
bms.1.type=DALY_RS485
bms.1.id=1
bms.1.portLocator=/dev/ttyAMA0
bms.1.delayAfterNoBytes=200

sanitariu, 2024-03-21:
I will run the webserver and we will know in some hours if it is ok.
Also i have noted the following:

2024-03-21 11:28:34.274 | INFO  | Thread-5   | verter.BmsToInverter:190   | Sending to inverter NONE on /dev/can1...
2024-03-21 11:28:34.277 | DEBUG | Thread-5   | verter.core.Inverter:79    | Inverter received: Buffer (HEX): [null]

Maybe we should not send anything to inverter if type is NONE ?

ai-republic, 2024-03-22:
Hi Todor,
yes, the numbering changed from starting from 0 to starting from 1.
Also the poll interval is now general for all BMS because it makes no sense to differ between multiple BMSes.

The DummyInverter (NONE) is not actually sending or receiving anything. It also doesn't open a port. I added that for the automatic discovery, but yes, I could take the NONE and just not have any inverter thread and processor running. Thanks for pointing it out :)

sanitariu, 2024-03-22:
I tested for 1 day and webserver stopped refreshing data. After restarting only the web module everything is okay. Strange that there is no errors or anything. I can not help you for debug

sanitariu, 2024-04-03:
Do you have any clues for fixing this ? Also why not add some graph charts per day/hour so we can see the history ?

ai-republic, 2024-04-04:
I think its related the number of posts for the MQTT Server on the queue. I have not had time to look into that.

Yeah, the history would be a great thing. Good idea!
It'll need a database in the background and some new views. Let's see when I get around doing that :)

sanitariu, 2024-04-04:
Also i noted that Web server is showing Discharge with -18A while the other console is showing +30A Charging which is the correct one and i see the same on my Growatt display.
Maybe this is old data after it stucked ?!

ai-republic, 2024-05-01:
Hi @sanitariu,
I changed the MQTT handling and hope that the webserver will not loose connection to the MQTT broker

ai-republic, 2024-08-24:
I changed the webserver finally and its working without stopping. I will close this issue. If you encounter any issues please open another ticket ;)

closed 2024-08-24 by ai-republic