ai-republic/bms-to-inverter #11: Growatt Hybrid high voltage / Pylontech BMS high voltage (powercube x1) / CAN

solsrvolt, 2024-02-01:
Hi,
great work. Thx.
A quick question, support your converter the Growatt SPH Hybrid (HV) series and the Pylontech HV BMS system (powercube X1) over CAN?
Thanks for your response.
Best regards
Utz

ai-republic, 2024-02-02:
Hi,
yes, I'm sure I can support this. As far as I know, the Growatt inverters use the Pylontech CAN specification.
But if you could get the CAN specifications from the manufacturers then I can tell for sure or - if it differs - implement the module in a couple of days.

solsrvolt, 2024-02-04:
Hi,
thanks for your quick reply. I attached the high voltage protocol and some other files for this setup (Growatt SPH Hybrid (HV) series and the Pylontech HV BMS system (powercube X1)) . It`s very hard to get the Growatt can bus protocol from the manufacturer. Maybe it will help you.
Many thanks for your support.
Best regads Utz
[pylon_CAN_210124.txt](https://github.com/ai-republic/bms-to-inverter/files/14157972/pylon_CAN_210124.txt)
[CAN-Bus-Protocol-Sermatec-high-voltage-V1.1810kW (1).pdf](https://github.com/ai-republic/bms-to-inverter/files/14157973/CAN-Bus-Protocol-Sermatec-high-voltage-V1.1810kW.1.pdf)
[PYLONTECH.zip](https://github.com/ai-republic/bms-to-inverter/files/14157974/PYLONTECH.zip)

ai-republic, 2024-02-05:
Ok, I will have a look at it possibly this week.
The file _[CAN-Bus-Protocol-Sermatec-high-voltage-V1.1810kW (1).pdf](https://github.com/ai-republic/bms-to-inverter/files/14157973/CAN-Bus-Protocol-Sermatec-high-voltage-V1.1810kW.1.pdf)_ is the protocol for your Pylontech HV BMS, correct?

solsrvolt, 2024-02-05:
Hi Torsten,

yes it is. I have attached a document regarding the growatt Can Bus.
Best regards / Utz 
[Growatt-Battery-BMS.docx](https://github.com/ai-republic/bms-to-inverter/files/14166157/Growatt-Battery-BMS.docx)

ai-republic, 2024-02-10:
Thanks for the sponsoring! :)
I will see what I can implement this weekend starting with the HV BMS support first. Then we can test that we get some proper readings.

solsrvolt, 2024-02-11:
Hi Torsten,

thanks for your great support. 
I will testing in the next days. Replay comes.

Have nice (rest)weekend.

Best regards Utz

ai-republic, 2024-02-11:
Ok.  Be sure to connect CAN H and CAN L correctly of your Powercube X1 to the PI hat. 
This is what I found in the Operation Manual for the Powercube X1:
[image]

So you need to cut a RJ45 network cable (min. CAT 5) and connect pin 4 (CANH) to CANH and pin 5 (CANL) to the CANL  on your PI's CAN hat.

Let me know if you have any questions!

solsrvolt, 2024-02-11:
Hi, thx.
Currently I have a problem, I can only select the daly bms, no any other inverter bms and only "none" inverter brand. I installed the configurator on a pi 4 with JDK, do you have an idea what is the problem?

Thanks. best regrds Utz

solsrvolt, 2024-02-11:
64-bit, newest PI firmware (bookworm)

solsrvolt, 2024-02-11:
Hi,
I solved the problem, but I can't chose the bm pylontech hv protocol, see png attached.
Thx. for your support.
BG / Utz
[image]

ai-republic, 2024-02-12:
Let me check....

solsrvolt, 2024-02-12:
merci, I will testing this week and come back with results.
Bg / Utz

solsrvolt, 2024-02-15:
Hi Tosrten,

I tested, see the result in the attached jpeg. Pls. can you help me?
Thx. BG Utz
[image]

ai-republic, 2024-02-16:
Hi Utz,
so far so good, but does it stop after the last log entry?
Is your BMS connected to can1 or can0?
Please use the _Dummy_ inverter for now, so we can test if we get any readings from the BMS.
Also configure the _log level_ to _debug_ in the Configurator please. In the future please just attach the log file found in `<installation directory>/logs` folder. Then it's easier to read ;)

solsrvolt, 2024-02-17:
Hi Torsten,
I tested, pls. see the attached log. The converter display said "bms com failed". The pylon BMS show no errors on the leds.
I think its not a great problem :-)
Thx. for your support.
BG / Utz
[BMS-to-Inverter-2024-02-12-1.log.gz](https://github.com/ai-republic/bms-to-inverter/files/14319192/BMS-to-Inverter-2024-02-12-1.log.gz)

ai-republic, 2024-02-18:
Hi Utz,
good, looks like the BMS is responding properly to the request. But something is not right possible with the configuration.
Could you please attach your `config.properties` file from the config folder.

jenshermans, 2024-02-18:
Hi,
Subscibed to this.
Having the same growatt HV inverter (SPH 5000TL3 BH-UP).
I have the Pylontech Force H1 battery. But I'm guessing I also could use the pylontech_hv_can module.
Attached my config and log file.

[config.txt](https://github.com/ai-republic/bms-to-inverter/files/14322433/config.txt)
[BMS-to-Inverter.log](https://github.com/ai-republic/bms-to-inverter/files/14322434/BMS-to-Inverter.log)

solsrvolt, 2024-02-18:
Hi Torsten,
in the attachment you find my configuration properties.
Thanks for analysing.
BG Utz
[config.txt](https://github.com/ai-republic/bms-to-inverter/files/14322531/config.txt)

ai-republic, 2024-02-19:
Hi Utz & Jens,
please update your installation (_Clean install_ in the _Configurator_). 
I've fixed the problem with the exception so you should get some proper readings now. I've tested it on my PI giving it simulated data for the SOC, current and voltage readings.

@jenshermans Please reduce your configuration to a minimum for now (see Utz's config) until we get decent data from the BMSes

jenshermans, 2024-02-19:
Hi,
Did a clean install, copied the config file from Utz and this is my log file;
[BMS-to-Inverter.log](https://github.com/ai-republic/bms-to-inverter/files/14335251/BMS-to-Inverter.log)

solsrvolt, 2024-02-19:
Hi Torsten,
here is my clean configuration log file. Now I have some values from the hv pylontechs... , but still errors. The inverter does not recognise the bms.
Thx. BG Utz
[BMS-to-Inverter.txt](https://github.com/ai-republic/bms-to-inverter/files/14335948/BMS-to-Inverter.txt)

jenshermans, 2024-02-20:
Hi Torsten,
I'v added my current log on debug level, Doubting it's an other Pylontech HV protocol...
For the growatt inverter; I can't set any protocols on my SPH, even on installer level.
[BMS-to-Inverter.log](https://github.com/ai-republic/bms-to-inverter/files/14339285/BMS-to-Inverter.log)

ai-republic, 2024-02-20:
Hi,
thanks for the log.
Could you check which protocols or batteries your inverter supports (maybe in the manual or website)? 
Maybe you could also ask Growatt to send you a list of supported batteries or even better their CAN protocol ;)
What I've found was that that they support(ed?) the Pylontech protocol, so the SolArk should work:
[image]

Have you connected the right pins from CAN hat to inverter to your `can1` port?
[image]

solsrvolt, 2024-02-20:
Hi Torsten,

that's tricky, Growatt supported the Pylontech Powercube X1 when the SPH bh-up inverter series was launched. Growatt has not officially supported the memory for some time now. But it still works. A friend is running this setup with the Pylontech CAN protocol 1.18 (see above). I think I read somewhere that Growatt uses the CAN bus protocol 1.01 (that's not open) for the high-voltage series.
I'll check the SolArk settings again today.
BG / Utz

Translated with DeepL.com (free version)

solsrvolt, 2024-02-20:
Hi,
I tested three ways: Growatt-Pylon, SolArk-HVPylon, SolArk-Pylon. Logs attached.
Bg / Utz
[BMS-to-Inverter-Growatt-Pylon.log](https://github.com/ai-republic/bms-to-inverter/files/14342249/BMS-to-Inverter-Growatt-Pylon.log)
[BMS-to-Inverter-SolArk-HVPylon.log](https://github.com/ai-republic/bms-to-inverter/files/14342250/BMS-to-Inverter-SolArk-HVPylon.log)
[BMS-to-Inverter-SolArk-Pylon.log](https://github.com/ai-republic/bms-to-inverter/files/14342251/BMS-to-Inverter-SolArk-Pylon.log)

jenshermans, 2024-02-20:
Hi,

I'm receiving packages from my growatt inverter, uploaded my candump. maybe this could help you a little forward.
Alsof tried the (new) configuration with pylonHV-Growatt and PylonHV-SolArk, logfiles are also added as attachment.
Also found this similar project documents regarding 
 the Growatt inverter. (jk-bms); https://drive.google.com/drive/folders/1YP-UwRgFRqus0CKmYnccU79ZlMlZN57e

[candump.log](https://github.com/ai-republic/bms-to-inverter/files/14348929/candump.log)
[PylonHV_Growatt.log](https://github.com/ai-republic/bms-to-inverter/files/14348930/PylonHV_Growatt.log)
[PylonHV_Solark.log](https://github.com/ai-republic/bms-to-inverter/files/14348932/PylonHV_Solark.log)

sanitariu, 2024-02-21:
Here is a link with the protocol information for low an high voltage 
https://www.amosplanet.org/how-to-commission-the-communication-between-lithium-ion-battery-and-spf-3500-5000es/

https://www.amosplanet.org/wp-content/uploads/2022/04/Growatt_BMS_RS485_protocal_1xSxxP_ESS_V2.02-1.pdf

https://www.amosplanet.org/wp-content/uploads/2022/04/Growatt-BMS-CAN-Bus-protocol-low-voltage-V1.04-1.pdf

ai-republic, 2024-02-21:
Hi
I fixed the missing dependency for the Pylontech HV inverter now. Please pulll the latest _Configurator_ and do a _clean install_.

jenshermans, 2024-02-21:
Hey Torsten,
Downloaded the new files, and the configurator wont open anymore.

ai-republic, 2024-02-21:
Sorry, my bad! Shouldn't do stuff at 11pm :(

Fixed the service descriptors - now it working.

jenshermans, 2024-02-22:
Added the current log.
I hear the BMS relay clicking, and the status led turned blue, so he goes out of error mode.
On the inverter i now get 'batt voltage high'.

my config is:
-bms: pylon_hv_can
-inverter: pylon_hv_can
[BMS-to-Inverter.log](https://github.com/ai-republic/bms-to-inverter/files/14368775/BMS-to-Inverter.log)

ai-republic, 2024-02-22:
Hi Jens & Utz,
I just committed a full implementation of the Pylontech HV inverter module (also corrected some value units). 
Could you please test? Thanks!

jenshermans, 2024-02-22:
BMS reading looks right. Inverter now shows 'bms com fault'.

[BMS-to-Inverter.log](https://github.com/ai-republic/bms-to-inverter/files/14377507/BMS-to-Inverter.log)
[candump bms.log](https://github.com/ai-republic/bms-to-inverter/files/14377508/candump.bms.log)
[candump inverter.log](https://github.com/ai-republic/bms-to-inverter/files/14377509/candump.inverter.log)

ai-republic, 2024-02-23:
Hi Jens,
I fixed some wrong units. Hopefully that will solve the 'bat voltage high'.
Can you please test the latest version?

jenshermans, 2024-02-23:
Current logs attached, Inverter still shows bat voltage high, bms com fault
[BMS-to-Inverter.log](https://github.com/ai-republic/bms-to-inverter/files/14381828/BMS-to-Inverter.log)

ai-republic, 2024-02-23:
Hi Jens,
I'm a little bit confused about your battery configuration.
When I send a broadcast to all battery modules I only receive answers from Battery No. 2 (e.g. 0x4211) but not from Battery No. 1 (e.g. 0x4210). Last digit identifies the battery module. 

On Utz's battery I get the correct Battery No. 1 (e.g. 0x4210) when doing the broadcast.
Could it be that you need to change the DIP switch on your BMS or Battery module to correct the identifer?

When the inverter starts a broadcast to request the BMS data, I now send data from all BMS. 

So I updated some code for more info. Maybe you can update and run it again?

solsrvolt, 2024-02-23:
Hi Torsten,
I tested, the inverter display dispatch BMS Abnormal, BMS Com fault. Can you provide me with the version that worked for Jens until the inverter/bms stopped using bms com fault, bat voltage high? The Powercube BMS schows a red error light.
Thx.
BG/Utz
[BMS-to-Inverter.log](https://github.com/ai-republic/bms-to-inverter/files/14387596/BMS-to-Inverter.log)

jenshermans, 2024-02-23:
Hi Torsten,
It's not possible to change my BMS address, this addresses itself. Its only possible to change bitrate and some "special protocols"
For this current version; Battery is okay and the inverter is showing the first 5 seconds "normal", after that it shows 'BMS com fault' and 'Normal' varying.
But i noticed; the first 5 seconds, (while normal) the SOC is displaying like it should be, after that it disappears.

Did a candump on can0 and can1 (starting from timestamp 2024-02-23 17:36:07.064 in the bms-to-inverter log)

[BMS-to-Inverter.log](https://github.com/ai-republic/bms-to-inverter/files/14387765/BMS-to-Inverter.log)
[candump can0 bms.log](https://github.com/ai-republic/bms-to-inverter/files/14387766/candump.can0.bms.log)
[candump can1 inv.log](https://github.com/ai-republic/bms-to-inverter/files/14387767/candump.can1.inv.log)

ai-republic, 2024-02-24:
Hi Utz,
can you please update to the latest version. You seem to not get any responses from your BMS. Please also check that your CAN device is properly connected and that the BMS is not idle.

ai-republic, 2024-02-24:
Hi Jens,
from the candump can1 (inverter) I can see that everything gets sent to the inverter initially. Then 5 seconds between the sending of the invalid data of 0x4210 and then the valid data 0x4211, etc.

I have fixed that temporarily ignoring the battery for which I don't get data (0x4210 in your case). That should hopefully fix the problem that the inverter switches from normal to error mode.

jenshermans, 2024-02-24:
Hi Torsten,

I'm still getting the bms com error on the inverter, in the logs you also will see 'error communicating to inverter'.
[BMS-to-Inverter.log](https://github.com/ai-republic/bms-to-inverter/files/14392014/BMS-to-Inverter.log)

ai-republic, 2024-02-24:
Hi Jens,
I fixed the exception when sending of the manufacturer.

jenshermans, 2024-02-24:
Still receiving the 'normal/bms com error' status
added a candump of the inverter to attachments. 
[BMS-to-Inverter.log](https://github.com/ai-republic/bms-to-inverter/files/14392048/BMS-to-Inverter.log)
[candump-inv.log](https://github.com/ai-republic/bms-to-inverter/files/14392050/candump-inv.log)

ai-republic, 2024-02-24:
hmm, can you access the inverter's log. Maybe there a more detailed error message

jenshermans, 2024-02-24:
Hope I can help you with this? (via oss.growatt.com)
[Schermafbeelding 2024-02-24 082850](https://github.com/ai-republic/bms-to-inverter/assets/60035424/ca36637d-6157-48a5-9625-35e74a22a356)

ai-republic, 2024-02-24:
Ok, can you tell me your inverter settings for the maximum voltage limit?

jenshermans, 2024-02-24:
As the manual describes, battery voltage range is 100-550vdc.

Running your program unlocks my BMS. So I configured the inverter with lead acid battery and its correct charging current/voltage, and load current/voltage. This works fine.

On lithium mode i cant configure anything as all data comes from the BMS.
I noticed something (don't know if this is the problem) Max charging/discharging current is 25A, bms reports via CAN max charge/discharge 37A

[image](https://github.com/ai-republic/bms-to-inverter/assets/60035424/566a2c40-1c4f-4712-861f-89927dbbf2c2)

ai-republic, 2024-02-25:
Hi,
I updated the Configurator to allow setting a BMS Id. Also I changed the poll intervall to be a general BMS configuration (not for each).

You will need to download the new Configurator and re-configure your setup and do a clean install.

ai-republic, 2024-02-25:
I also fixed the reading of the module voltages. It was the wrong data type (s_int_16 instead of u_int_16).
They reported -15.608V. I fixed that so now its 49.950V so ~50V for a module.

jenshermans, 2024-02-25:
Still receiving the BMS com error. Also added my BMS's data this time. maybe this also can help you.
[BMS-to-Inverter.log](https://github.com/ai-republic/bms-to-inverter/files/14395265/BMS-to-Inverter.log)
[image]
[Screenshot_20240225-055800](https://github.com/ai-republic/bms-to-inverter/assets/60035424/20fa29ea-9af7-4bae-aa71-18daf0f10d22)

jenshermans, 2024-02-25:
Hey Thorsten,

Please note previous screenshot was from my BMS, it has a WiFi module in-built to read this data.

Added a screenshot of the inverter data. 

Can't read any data from the installer app because there is "no BMC connected"

Also added a picture of the BMS specs. I noticed that growatt wants to display the SN of the battery. Don't know if you send this already, but looks like it could be needed.

[image]

[image]

[image]

jenshermans, 2024-02-25:
Added error logs from the growatt app,  it refers to specific bytes
[image]
[image]
[image]

ai-republic, 2024-02-25:
@jenshermans I've added sending the 0x42B0 message

What is the baudrate you set for your `can1` to the inverter?

jenshermans, 2024-02-25:
Running my inverter and BMS at 500kbps
My can startup commands;
```
sudo ip link set can0 up type can bitrate 500000
sudo ip link set can1 up type can bitrate 500000
```

ai-republic, 2024-02-25:
That should be ok then.

ai-republic, 2024-02-25:
Could you please update (clean install). I've changed the inverter binding, so that it will send the messages to the inverter always as 0x42x0 (like 0x4210) instead of 0x42xx (like 0x4211). Let's see if Growatt will accept these.

jenshermans, 2024-02-25:
current log, there is an error communicating to the inverter
[BMS-to-Inverter.log](https://github.com/ai-republic/bms-to-inverter/files/14395642/BMS-to-Inverter.log)

ai-republic, 2024-02-25:
You're `can1` is not connected or needs to be reset:

```
sudo ip link set can1 down
sudo ip link set can1 up type can bitrate 500000
```

ai-republic, 2024-02-25:
Wow! Great!!! Awesome!

Still don't know why your BMS sends the 0x421x, etc. messages. If you would have two or more packs that would be understandable. Musst be some internal preconfigured switch....
I'll need to restructure some code to compensate for this in the future.

@solsrvolt Can you confirm that its working for you too?

ai-republic, 2024-02-25:
@jenshermans Please monitor it for a day or so that everything works as expected. And if you are happy with my work and support please feel free to [sponsor](https://github.com/sponsors/ai-republic) me ;)

[38 more replies]

closed 2024-08-09 by ai-republic