ai-republic/bms-to-inverter #14: Not working normally

sanitariu, 2024-02-12:
Hi, after today update the software is not working properly.
It does not crash it just stops.
For example these are the last lines:

BMS     SOC       V       A      CellMinV        CellMaxV       CellDiff
1        25.6   53.2    8.0     3.328(#2)       3.333(#15)      0.005

2024-02-12 13:45:48.367 | INFO  | $6@96aa05) | ogs.AuditLogger_impl:2299  | AMQ601501: User anonymous@127.0.0.1:48790 is consuming a message from energystorage: Reference[158]:RELIABLE:CoreMessage[messageID=158,durable=true,userID=null,priority=4, timestamp=Mon Feb 12 13:45:48 EET 2024,expiration=0, durable=true, address=energystorage,size=11224,properties=TypedProperties[]]@31771282
2024-02-12 13:45:48.372 | INFO  | $6@96aa05) | ogs.AuditLogger_impl:2291  | AMQ601500: User anonymous@127.0.0.1:39724 sent a message CoreMessage[messageID=158,durable=true,userID=null,priority=4, timestamp=Mon Feb 12 13:45:48 EET 2024,expiration=0, durable=true, address=energystorage,size=11224,properties=TypedProperties[]]@31771282, context: RoutingContextImpl(Address=energystorage, routingType=null, PreviousAddress=energystorage previousRoute:null, reusable=true, version=-2147483646)
..................................................
***** durable queues energystorage:
- queueID=4 address:energystorage name:energystorage filter:null
***** non durable for energystorage:
..................................................
, transaction: null
2024-02-12 13:45:48.377 | INFO  | 2-thread-1 | verter.BmsToInverter:225   | BMS alarms:
        NONE
2024-02-12 13:45:53.378 | INFO  | 2-thread-1 | stoinverter.core.BMS:128   | ---------------------------------> Thread 43
2024-02-12 13:45:53.585 | WARN  | 2-thread-1 | alyBmsRS485Processor:127   | Command 0x90 to BMS 1 successfully sent and received!
2024-02-12 13:45:53.793 | WARN  | 2-thread-1 | alyBmsRS485Processor:127   | Command 0x91 to BMS 1 successfully sent and received!
2024-02-12 13:45:54.000 | WARN  | 2-thread-1 | alyBmsRS485Processor:127   | Command 0x92 to BMS 1 successfully sent and received!
2024-02-12 13:45:54.208 | WARN  | 2-thread-1 | alyBmsRS485Processor:127   | Command 0x93 to BMS 1 successfully sent and received!
2024-02-12 13:45:54.416 | WARN  | 2-thread-1 | alyBmsRS485Processor:127   | Command 0x94 to BMS 1 successfully sent and received!

Not updating. If i stop the webserver it prints info so it is not crashed

ai-republic, 2024-02-12:
Thanks Todor, I will look into it tomorrow!

ai-republic, 2024-02-13:
I fixed clearing the buffers at the start of each reading.
Please update.

sanitariu, 2024-02-13:
Hello, i have updated.
Now stays here and nothing happens.
No more data printed after that line...

2024-02-13 12:53:47.475 | INFO  | main       | verter.BmsToInverter:158   | Starting BMS receiver...
2024-02-13 12:53:47.478 | INFO  | main       | verter.BmsToInverter:167   | Starting inverter sender...
2024-02-13 12:53:48.479 | INFO  | 2-thread-1 | stoinverter.core.BMS:128   | ---------------------------------> Thread 42
2024-02-13 12:53:48.481 | INFO  | 2-thread-1 | toinverter.core.Port:101   | Opening /dev/ttyAMA0 ...
2024-02-13 12:53:48.503 | INFO  | 2-thread-1 | toinverter.core.Port:103   | Opening port /dev/ttyAMA0 SUCCESSFUL

ai-republic, 2024-02-14:
Its working now. I've uploaded the application I generated via the Configurator and tested it on your PI. There was a blocking read when clearing the buffers. Btw, the Configurator application should also work on your Mac now.

sanitariu, 2024-02-14:
Hi, thanks for supporting. Now the app is working but for 2-3 minutes. Seems like when it hits empty data or some wrong data it just stays forever. Yes configurator is working on Mac and i am using it :)

ai-republic, 2024-02-15:
Ok, there was another blocking reading. I've changed buffering the readings now to a non-blocking queue. That solved the issue. I'm running it on your PI now

sanitariu, 2024-02-15:
Hi, perfect ! If it is working for 20-30 minutes i think it is okay. Also you can start the web server, same ip address and port 8080 to check if it updates accordingly.

ai-republic, 2024-02-16:
Hi, all working good. I've also started the webserver and its updating the values.
I haven't had time to look into the timeout on the MQTT client/server, so maybe the web app will stop updating after a day.
Just keep an eye on that.

I will close this issue as it is fixed now.

closed 2024-02-16 by ai-republic