ai-republic/bms-to-inverter #16: Latest code not working

sanitariu, 2024-02-21:
Trying the latest version throws the following error:

e.remoting.impl.netty.NettyServerConnection@4cfeae[ID=b311cd28, local= /127.0.0.1:61616, remote=/127.0.0.1:40888]], true, true, false, false, null, org.apache.activemq.artemis.core.protocol.core.impl.CoreSessionCallback@59f6d4, true, {}]
Exception in thread "main" jakarta.enterprise.inject.CreationException
        at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
        at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:77)
        at java.base/jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
        at java.base/java.lang.reflect.Constructor.newInstanceWithCaller(Constructor.java:499)
        at java.base/java.lang.reflect.ReflectAccess.newInstance(ReflectAccess.java:128)
        at java.base/jdk.internal.reflect.ReflectionFactory.newInstance(ReflectionFactory.java:347)
        at java.base/java.lang.Class.newInstance(Class.java:645)
        at org.jboss.weld.security.NewInstanceAction.run(NewInstanceAction.java:33)
        at java.base/java.security.AccessController.doPrivileged(AccessController.java:569)
        at org.jboss.weld.injection.Exceptions.rethrowException(Exceptions.java:40)
        at org.jboss.weld.injection.Exceptions.rethrowException(Exceptions.java:50)
        at org.jboss.weld.injection.Exceptions.rethrowException(Exceptions.java:90)
        at org.jboss.weld.injection.ConstructorInjectionPoint.newInstance(ConstructorInjectionPoint.java:127)
        at org.jboss.weld.injection.ConstructorInjectionPoint.invokeAroundConstructCallbacks(ConstructorInjectionPoint.java:92)
        at org.jboss.weld.injection.ConstructorInjectionPoint.newInstance(ConstructorInjectionPoint.java:78)
        at org.jboss.weld.injection.producer.AbstractInstantiator.newInstance(AbstractInstantiator.java:28)
        at org.jboss.weld.injection.producer.BasicInjectionTarget.produce(BasicInjectionTarget.java:112)
        at org.jboss.weld.injection.producer.BeanInjectionTarget.produce(BeanInjectionTarget.java:186)
        at org.jboss.weld.bean.ManagedBean.create(ManagedBean.java:160)
        at org.jboss.weld.contexts.AbstractContext.get(AbstractContext.java:96)
        at org.jboss.weld.bean.ContextualInstanceStrategy$DefaultContextualInstanceStrategy.get(ContextualInstanceStrategy.java:100)
        at org.jboss.weld.bean.ContextualInstanceStrategy$ApplicationScopedContextualInstanceStrategy.get(ContextualInstanceStrategy.java:140)
        at org.jboss.weld.bean.ContextualInstance.get(ContextualInstance.java:50)
        at org.jboss.weld.manager.BeanManagerImpl.getReference(BeanManagerImpl.java:684)
        at org.jboss.weld.manager.BeanManagerImpl.getReference(BeanManagerImpl.java:666)
        at org.jboss.weld.injection.producer.AbstractMemberProducer.getReceiver(AbstractMemberProducer.java:123)
        at org.jboss.weld.injection.producer.AbstractMemberProducer.produce(AbstractMemberProducer.java:158)
        at org.jboss.weld.bean.AbstractProducerBean.create(AbstractProducerBean.java:180)
        at org.jboss.weld.contexts.unbound.DependentContextImpl.get(DependentContextImpl.java:64)
        at org.jboss.weld.bean.ContextualInstanceStrategy$DefaultContextualInstanceStrategy.get(ContextualInstanceStrategy.java:100)
        at org.jboss.weld.bean.ContextualInstance.get(ContextualInstance.java:50)
        at org.jboss.weld.manager.BeanManagerImpl.getReference(BeanManagerImpl.java:684)
        at org.jboss.weld.manager.BeanManagerImpl.getReference(BeanManagerImpl.java:666)
        at org.jboss.weld.manager.BeanManagerImpl.getInjectableReference(BeanManagerImpl.java:784)
        at org.jboss.weld.injection.FieldInjectionPoint.inject(FieldInjectionPoint.java:92)
        at org.jboss.weld.util.Beans.injectBoundFields(Beans.java:345)
        at org.jboss.weld.util.Beans.injectFieldsAndInitializers(Beans.java:356)
        at org.jboss.weld.injection.producer.ResourceInjector$1.proceed(ResourceInjector.java:69)
        at org.jboss.weld.injection.InjectionContextImpl.run(InjectionContextImpl.java:48)
        at org.jboss.weld.injection.producer.ResourceInjector.inject(ResourceInjector.java:71)
        at org.jboss.weld.injection.producer.BasicInjectionTarget.inject(BasicInjectionTarget.java:117)
        at org.jboss.weld.bean.ManagedBean.create(ManagedBean.java:161)
        at org.jboss.weld.contexts.AbstractContext.get(AbstractContext.java:96)
        at org.jboss.weld.bean.ContextualInstanceStrategy$DefaultContextualInstanceStrategy.get(ContextualInstanceStrategy.java:100)
        at org.jboss.weld.bean.ContextualInstanceStrategy$ApplicationScopedContextualInstanceStrategy.get(ContextualInstanceStrategy.java:140)
        at org.jboss.weld.bean.ContextualInstance.get(ContextualInstance.java:50)
        at org.jboss.weld.bean.proxy.ContextBeanInstance.getInstance(ContextBeanInstance.java:102)
        at org.jboss.weld.bean.proxy.ProxyMethodHandler.getInstance(ProxyMethodHandler.java:131)
        at com.airepublic.bmstoinverter.BmsToInverter$Proxy$_$$_WeldClientProxy.start(Unknown Source)
        at com.airepublic.bmstoinverter.BmsToInverter.main(BmsToInverter.java:73)
Caused by: java.util.ServiceConfigurationError: com.airepublic.bmstoinverter.core.InverterDescriptor: Provider com.airepublic.bmstoinverter.inverter.pylonhv.can.PylonHVInverterCANProcessor not found
        at java.base/java.util.ServiceLoader.fail(ServiceLoader.java:593)
        at java.base/java.util.ServiceLoader$LazyClassPathLookupIterator.nextProviderClass(ServiceLoader.java:1219)
        at java.base/java.util.ServiceLoader$LazyClassPathLookupIterator.hasNextService(ServiceLoader.java:1228)
        at java.base/java.util.ServiceLoader$LazyClassPathLookupIterator.hasNext(ServiceLoader.java:1273)
        at java.base/java.util.ServiceLoader$2.hasNext(ServiceLoader.java:1309)
        at java.base/java.util.ServiceLoader$3.hasNext(ServiceLoader.java:1393)
        at java.base/java.lang.Iterable.forEach(Iterable.java:74)
        at com.airepublic.bmstoinverter.core.InverterProducer.<init>(InverterProducer.java:27)
        at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
        at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:77)
        at java.base/jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
        at java.base/java.lang.reflect.Constructor.newInstanceWithCaller(Constructor.java:499)
        at java.base/java.lang.reflect.Constructor.newInstance(Constructor.java:480)
        at org.jboss.weld.injection.ConstructorInjectionPoint.newInstance(ConstructorInjectionPoint.java:119)
        ... 37 more

and sits there ...

ai-republic, 2024-02-21:
Could you try to update again with the latest Configurator please?

   

  and sits there ...

sanitariu, 2024-02-21:
I tried but also downloaded again configurator.jar and it does not work too. Complains about missing pylon ...

dri@macbook-sani ~/bms $ java -jar configurator.jar
Exception in thread "main" java.util.ServiceConfigurationError: com.airepublic.bmstoinverter.core.InverterDescriptor: Provider com.airepublic.bmstoinverter.inverter.pylonhv.can.PylonHVInverterCANProcessor not found
	at java.base/java.util.ServiceLoader.fail(ServiceLoader.java:593)
	at java.base/java.util.ServiceLoader$LazyClassPathLookupIterator.nextProviderClass(ServiceLoader.java:1219)
	at java.base/java.util.ServiceLoader$LazyClassPathLookupIterator.hasNextService(ServiceLoader.java:1228)
	at java.base/java.util.ServiceLoader$LazyClassPathLookupIterator.hasNext(ServiceLoader.java:1273)
	at java.base/java.util.ServiceLoader$2.hasNext(ServiceLoader.java:1309)
	at java.base/java.util.ServiceLoader$3.hasNext(ServiceLoader.java:1393)
	at java.base/java.lang.Iterable.forEach(Iterable.java:74)
	at com.airepublic.bmstoinverter.configurator.InverterPanel.createInverterItems(InverterPanel.java:95)
	at com.airepublic.bmstoinverter.configurator.InverterPanel.<init>(InverterPanel.java:20)
	at com.airepublic.bmstoinverter.configurator.Configurator.<init>(Configurator.java:90)
	at com.airepublic.bmstoinverter.configurator.Configurator.main(Configurator.java:469)

ai-republic, 2024-02-21:
OK, thanks. That's weird because I tested it before committing. Maybe I missed something. I'll get onto that first thing tomorrow morning.

ai-republic, 2024-02-21:
Hi Todor,
fixed it now. Please pulll the latest _Configurator_ and do a _clean install_.

sanitariu, 2024-02-21:
Still not working configurator

dri@macbook-sani ~/bms $ java -jar configurator.jar
Exception in thread "main" java.util.ServiceConfigurationError: com.airepublic.bmstoinverter.core.InverterDescriptor: com.airepublic.bmstoinverter.inverter.pylonhv.can.PylonHVInverterCANProcessor not a subtype
	at java.base/java.util.ServiceLoader.fail(ServiceLoader.java:593)
	at java.base/java.util.ServiceLoader$LazyClassPathLookupIterator.hasNextService(ServiceLoader.java:1244)
	at java.base/java.util.ServiceLoader$LazyClassPathLookupIterator.hasNext(ServiceLoader.java:1273)
	at java.base/java.util.ServiceLoader$2.hasNext(ServiceLoader.java:1309)
	at java.base/java.util.ServiceLoader$3.hasNext(ServiceLoader.java:1393)
	at java.base/java.lang.Iterable.forEach(Iterable.java:74)
	at com.airepublic.bmstoinverter.configurator.InverterPanel.createInverterItems(InverterPanel.java:95)
	at com.airepublic.bmstoinverter.configurator.InverterPanel.<init>(InverterPanel.java:20)
	at com.airepublic.bmstoinverter.configurator.Configurator.<init>(Configurator.java:90)
	at com.airepublic.bmstoinverter.configurator.Configurator.main(Configurator.java:469)