ai-republic/bms-to-inverter #17: Batteries with no content shown in the overview and webserver

ai-republic, 2024-02-26:
When using Pylon HV BMS with non-incremental identifiers for batteries there are batteries shown
with no content in the overview and webserver.

ai-republic, 2024-02-26:
I changed the incremental battery list to a battery mapping which dynamically will add the batteries depending on their identifier.

@jenshermans could you test this that there are no more batteries listed with SOC -1 and 0 (zero) values in the log?

jenshermans, 2024-02-26:
[BMS-to-Inverter.log](https://github.com/ai-republic/bms-to-inverter/files/14402436/BMS-to-Inverter.log)

ai-republic, 2024-02-26:
Did you update? Because it's still showing 2 battery packs.

jenshermans, 2024-02-26:
I'm pretty sure i did it, downloaded again and this is the current log.
Still showing multiple BMS's.
[BMS-to-Inverter.log](https://github.com/ai-republic/bms-to-inverter/files/14403641/BMS-to-Inverter.log)

ai-republic, 2024-02-26:
Ok, thank you Jens. I will have to check that again in more detail.

ai-republic, 2024-02-27:
@jenshermans please update again. There was a merge conflict so some code was not committed. I think that's why we're seeing the first invalid BMS. Now it should be fine. I also tested it emulating the Pylon HV BMS with receiving 0x42x1 messages and its looks fine.

jenshermans, 2024-02-27:
Only showing one BMS now.

Noticed a little bug in the configurator, the "BMS-ID" keeps showing 0, even if configured something else.

Here i configured my id to 2;
[image]
When pressing edit, it shows ID: 0, in the edit screen.

[BMS-to-Inverter.log](https://github.com/ai-republic/bms-to-inverter/files/14414365/BMS-to-Inverter.log)

ai-republic, 2024-02-27:
Hi Jens, 
super. And I also fixed the _Configurator_ bug you found.

closed 2024-02-27 by ai-republic