ai-republic/bms-to-inverter #20: Outback Power Skybox Inverter BMS Canbus

carsonfee, 2024-03-04:
outback has a limited list of batteries it supports.  It has support for Simpliphi, Blue Ion, Discover, Fortress, Active BMS, Energycell PLR and XLC, Kilovault. 

Skybox runs on an industrial beagleboard and has an RJ12 connector to connect to the BMS RJ45 Can port.

carsonfee, 2024-03-04:
I have heard mumblings that Skybox uses Sunspec Modbus but I am not entirely sure. Outback tech support is abysmal

ai-republic, 2024-03-04:
Hi @carsonfee,
so you are trying to connect the Skybox to some other (not supported) BMS using the CANBus right?
I would need some CAN specification from Outback or any supported battery manufacturer to create a binding. Unless its speaks a more common specification like the PylonTech CAN specification.

Another way would be to 'listen in' on the communication of a running system (Skybox and supported BMS) and try to reverse engineer the protocol specification. For that you would need a CAN device hooked up on the CAN bus of the system, monitor the communication and protocol the current values (like SOC, current, voltage, etc.).

carsonfee, 2024-03-04:
I have that 2 Channel Can hat for my RPi. Can I just plug in to the inverter and capture the traffic? 
The supported batteries are in my top post ^^
The batteries I have are sungold power. I think it uses the same bms as EG4 and jaikiper.

ai-republic, 2024-03-04:
Yes, you can connect the 2CH CAN hat to the inverter and use `candump` to capture the traffic.

If you can provide CAN or RS485 or ModBus specifications for your inverter or supported batteries it would make life easier. Maybe inquire at the manufacturer (inverter or any supported battery manufacturer) with your exact inverter model and be persistent that you cannot use it without the CAN/RS485/ModBus communication specification of any supported BMS.

carsonfee, 2024-03-16:
My batteries use a PACE bms. Sungold power

ai-republic, 2024-03-17:
Hi Carson,
can you configure your PACE BMS to use the Pylontech CAN protocol?
I've read somewhere that you can configure it to use it?

carsonfee, 2024-03-17:
Yes I can. It supports a few different protocols, Pylon, Pace, Growatt and
a couple of others. Trying to connect to the battery now, it is supposed to
have a canbus b type of cable using pins 4 & 5 but so far no connection
happening. The SPI says the can hat is initialized, going to try a loopback
test to make sure it's working then see if I can find another pin out for
the cable.

carsonfee, 2024-03-17:
getting somewhere, can bus needed to be set to 500kbps 
[image]
getting this error, so it looks like they are talking just not the same language.
I have the config set to PYLON_CAN on the Pi and BMS Can bus protocol is set to PYLON on the battery
would a candump help ?

carsonfee, 2024-03-18:
candump looks more or less right I think?   35E shows LEOCH as the manufacturer ID. That looks like the frames have correct ids but data is wonky/different from PYLON ?

  can0  351   [8]  00110100 00000010 11010000 00000111 10100000 00001111 10110000 00000001   '4.......'
  can0  355   [8]  00110011 00000000 01100100 00000000 00000000 00000000 00000000 00000000   '3.d.....'
  can0  356   [8]  11110010 00010100 10101000 00000010 11110000 00000000 00000000 00000000   '........'
  can0  35C   [8]  11000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000   '........'
  can0  35E   [8]  01001100 01000101 01001111 01000011 01001000 00100000 00100000 00100000   'LEOCH   '
  can0  359   [8]  00000000 00000000 00000000 00000000 00000100 01010000 01001110 00000000   '.....PN.'

carsonfee, 2024-03-18:
is the endianess
 right? 356 Should be Pack Voltage so 11110010 00010100 is F214 which looks like garbage (-3,564)  but 14F2 = 5362 * .01 which would be 53.62 which seems about right

ai-republic, 2024-03-18:
ok, cool -  that's definitely the Pylon protocol!
Could you do the candump in standard hex format please. Binary is a bit of a hassle reading and converting ;)

So for command:
0x351: 
Max voltage = 56.4V
Charge limit = 200.0A
Discharge limit = 400.0A
Min voltage = 43.2V

0x355: 
SOC: 51%
SOH: 100%

etc.

So that looks all good!

I've added a fix for the SOC and SOH and more logging, so please do a _Clean install_ with the _Configurator_.
Could you also set the log-level set to _debug_ and provide the log file, then I can see what the application is getting.

ai-republic, 2024-03-18:
Ok, so its the short Pylon protocol version. What does the application do? It should read values now from the BMS. Could you please update to the lastest version (_Clean install_ in _Configurator_) and attach the log file (please with _debug_ log-level set).

ai-republic, 2024-03-21:
Hi, 
I wrote some tests with the candump data and it is read and processed correctly. 
I think the problem you might have lies in the communication between your webserver and the application's MQTT broker.

Could you update your application and send me the _BMS-to-Inverter.log_ and the _BMS-to-Inverter_Webserver.log_ from the installations log folder and if possible a screenshot of your browsers debug console output.

Please start the webserver sepeartely after the application is logging some decent data.

First start the application (in terminal 1):
```
java -DconfigFile=config/config.properties -Dlog4j2.configurationFile=file:config/log4j2.xml -jar lib/bms-to-inverter-main-0.0.1-SNAPSHOT.jar
```
Then wait until it reads data from the BMS and after that start the webserver (in terminal 2):
```
java -jar lib/webserver-0.0.1-SNAPSHOT.jar --spring.config.location=file:config/config.properties
```

carsonfee, 2024-03-21:
just running the application it looks like its reading garbage from the can bus.

2024-03-21 11:46:47.406 | INFO  | Thread-3   | verter.BmsToInverter:190   | Sending to inverter NONE on can1...
2024-03-21 11:46:47.407 | DEBUG | Thread-3   | verter.core.Inverter:79    | Inverter received: Buffer (HEX): [null]

2024-03-21 11:46:48.144 | INFO  | Thread-2   | verter.BmsToInverter:164   | Reading BMS #0 PYLON_CAN on can0...
2024-03-21 11:46:48.145 | INFO  | Thread-2   | stoinverter.core.BMS:120   | ---------------------------------> Thread 43
2024-03-21 11:46:48.145 | DEBUG | Thread-2   | ocol.can.JavaCANPort:66    | CAN frame read...
2024-03-21 11:46:48.146 | DEBUG | Thread-2   | ocol.can.JavaCANPort:72    | CAN read frame Buffer (HEX): [0x56, 0x03, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x9E, 0x14, 0xF9, 0xFE, 0xF7, 0x00, 0x00, 0x00]

2024-03-21 11:46:48.147 | DEBUG | Thread-2   | PylonBmsCANProcessor:98    | Pack V       Pack A          Avg Temp
 20.48          0.0             -2506.8
2024-03-21 11:46:48.147 | INFO  | Thread-2   | verter.BmsToInverter:216   |
BMS     SOC       V       A      CellMinV        CellMaxV       CellDiff
#1       204.8  204.8   0.0     0.0(#0)         0.0(#0) 0.0

carsonfee, 2024-03-21:
Pylon protocol is little endian, seems like you're reading big endian

carsonfee, 2024-03-22:
[BMS-to-Inverter.log](https://github.com/ai-republic/bms-to-inverter/files/14715737/BMS-to-Inverter.log)

carsonfee, 2024-03-22:
[BMS-to-Inverter_Webserver.log](https://github.com/ai-republic/bms-to-inverter/files/14715793/BMS-to-Inverter_Webserver.log)

ai-republic, 2024-03-22:
Hi, 
I just committed a fix. Could you please do a _Clean install_ and update your installation.
Let's see what we get :)

carsonfee, 2024-03-22:
[BMS-to-Inverter.log](https://github.com/ai-republic/bms-to-inverter/files/14718560/BMS-to-Inverter.log)

New log. Looks somewhat closer but still not right

carsonfee, 2024-03-22:
your change to little endian for the byte buffer looks good but the actual data that's in the buffer is out of order or maybe padded somehow?  the first 2 bytes look correct e.g. 
CAN read frame Buffer (HEX): [0x55, 0x03, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x1E, 0x00, 0x64, 0x00, 0x00, 0x00, 0x00, 0x00]

 candump reads right, the values in your buffer are the same but there's all those extra 0 bytes.
355   [8]  1E 00 64 00 00 00 00 00

carsonfee, 2024-03-22:
CAN read frame Buffer (HEX): [0x55, 0x03, | **0x00, 0x00, 0x08, 0x00, 0x00, 0x00,** |0x1E, 0x00, 0x64, 0x00, 0x00, 0x00, 0x00, 0x00]

I think the 6 bytes after the ID is the DLC and other parts of the can arbitration field

ai-republic, 2024-03-22:
yes, that is just the extended CAN frame 4 byte frame id, length and flags(3 bytes after length), then comes the 8 byte data.
But good point, forgot to skip the next 4 bytes before reading the data.
I fixed it now - this will give the correct values now! :)

carsonfee, 2024-03-22:
That fixed it!

now to figure out if the skybox can actually give any useful output from a candump

ai-republic, 2024-05-01:
closing since this issue has not been actively continued. Please reopen if issues arise.

closed 2024-05-01 by ai-republic