ai-republic/bms-to-inverter #21: Daly Growatt implementation imminent

Irthinkimlost, 2024-03-05:
Hiya

16 Lishen 272ah @ 1C 
Daly 48v 150A 
Growatt SPF 5000ES
Solar only, generator backup

No comms to BMS

RS485 (Daly) to RJ45 (growatt) fail. Crimped then inverted pins then removed 2 then inverted again. Could not set BMS to any specific (Pylon) protocol. Speedy email response from Daly support asked where I bought it but hasn't replied to me since I told them it was a small south African online shop that has since closed down, with a very bad reputation. 

Daly interface board option but I think I still need to set the battery type on BMS so I don't want to buy it. 

RS485 (Daly) to USB windows success
Daly software could also not change battery protocol. But I had full live Conn to BMS, all voltages etc working fine. 3 wire from RS485 to USB. So the BMS is speaking fine. Just need to relay the message. 

Tomorrow I'm going to get my first ever Pi. Pi 3 and with your suggested IO. Then I'll follow your guide as best that I can. I'm excited. 

I need this. My growatt isn't very good at charging blindly. 😅

ai-republic, 2024-03-06:
Hi,
good to hear that. If you need any help setting up your PI just ask. Hopefully you're getting a PI 3B because only that supports the 64-bit Raspian and you need a 64bit OS to run Java 17+.

About the Daly protocol: This application support the native Daly CAN/RS485 and UART specifications and also the Growatt SPF series. So this should work out of the box with the correct configuration!

If you have questions, just ask :)

Irthinkimlost, 2024-03-06:
Unfortunately they didn't have the waveshare so I got 2 bidirectional USB to RS485 adaptors. Which leads me to a question. Can I use the default DALY RS485 to USB cable to connect the Pi to the BMS? Like how I had it connected to my laptop?  secondly, please confirm the pin mapping for the RS485.

Irthinkimlost, 2024-03-06:
Ok... i'm going to stop tinkering today before I hurt myself

Pi is alive. 
Configurator is successfully installed and running.
I am connected remote desktop to Pi via WiFi. 

When you have a minute would you please help with some guidance:

I'm using the RS485 to USB supplied by DALY. This worked for connecting BMS to Daly PC monitoring software so i'm going to use it for now. 

I crimped an RJ45 onto CAT5 for Pi to Growatt. Used only 3 wires. O/W into 1, Orange into 2 and G/W into 3. Left other wires disconnected. 
Then i screwed:
Orange into 1st port on USB/RS485 adaptor labelled B-. 
O/W into 2nd port labelled A+
G/W into 3rd port labelled GND
Not sure if this goes into the BMS or RS485 port on inverter

Configurator:
I don't know which or the USBs on the Pi are which number or called what. Based on your guide I labelled them:
Bottom left to Daly "/dev/ttyUSB0" 
Top left to Growatt "/dev/ttyUSB1" 
This is if i'm looking into the port. Pushing the USB away from me towards the Pi and into the port. 
There isn't an option for Growatt RS485 in the inverters list. which option should I choose? 

Growatt:
Do i need to change the growatt settings at all? 
Currently it is set to "USE2" and this doesn't give an error for not being connected to BMS. 
I'd like to change it to Lithium, as it should be. Additionally this will give an alarm if it isn't communicating which is a nice confirmation. 

Server:
I don't know if it's something i have(n't) done but I am unable to connect to the view server via my browser. 
Do i need to forward ports somewhere? 
Someone said it may need to be forced to use IPv4

Thank you for making yourself available to help me and others. 

Steve

ai-republic, 2024-03-07:
From the Daly to the PI try the provided Daly cable as I described above.
Or you connect directly to the PI Serial IO pins.
[image]

I'm not sure about the coloring of the RJ45 cable, but going from PI to the inverter it would be easier to use a CAN adapter (PIN4 -> CAN H, PIN5 -> CAN L).

I have not yet implemented the Growatt RS485, but could do that on the weekend or early next week.

sanitariu, 2024-03-07:
Hi, I have similar daly bms and i "hit the wall" with uart2usb adapter. My pi model is - " Raspberry Pi 4 Model B Rev 1.2" and it have problems with USB hardware. Maybe it is from the kernel but i read that it is from the hardware part of PI device. This means that your daly bms  /dev/ttyUSB0 device will be randomly
disconnected and missing. All this is fixable by connecting UART tx/rx/gnd directly to the PI device. No additional hardware. After this you can read data and it is working good. I also have growatt and did not have time to connected it. Hope this weekend waiting for not so sunny days. Also if growatt do no detect the connection with battery it will turn off the output. This is bad because my pi is connected to that output too :) So i bought ups for the pi too. I think you should use LI52 or LI51 but i am not sure yet. I was unable to read any data from the growatt using BMS cable but maybe it needs to be setup first for L51/52 mode ?!

Irthinkimlost, 2024-03-07:
Update for the day!! a long day.. but again, productive.

I went out into the world and got my hand on a Waveshare 2-Channel Isolated CAN FD Expansion HAT! yay

I spent my second ever day working with Pi and this new language figuring out how to put a hat on. I did. Go me.

Pi and CAN hat are alive
CAN tested and I sent and received data
Connected CAN into Daly and into Pi. Used same method to listen but heard nothing
Connected RS485 into Pi USB but didn't know what i was doing. Found this (https://pypi.org/project/dalybms/) to try decipher incoming signal but couldn't make my USB work. I don't know. 
Connected Daly via RS485 to laptop again successfully and again tried to change settings from RS485 to CAN but I can't. So I'm a little worried that even with the CAN hat i may still need to go from RS485 to Pi to CAN to Growatt. 
1 screenshot showing your tool in action. I think i had NONE in the inverter setting on your configurator so this was just me trying to listen. 
the other screenshot showing me trying to listen to CAN signal from Daly.

What say you oh wise one? 

Thanks again

[image]
[image]

sanitariu, 2024-03-07:
Steve you are walking my way ... Do not try CAN connection from your Daly BMS because you are missing the hardware parts for that in your Daly. Also you do not have RS-485 at Daly so it is not going to work too.
You can try UART2USB adapter which came with your Daly but I am pretty sure that your PI is going to loose serial connectivity from time to time. Instead just connect 3 wires tx/rx/gnd direct to your PI and use device /dev/ttyAMA0 for receiving data from Daly BMS. You can setup Inverter for NONE just to see if Daly is working correct and then you can go for the inverter part.
Here is picture where to connect Daly BMS 3 cables to PI.

[image]

Irthinkimlost, 2024-03-07:
Epic. Thank you! I will try this tomorrow

Irthinkimlost, 2024-03-08:
I don't know what I'm doing here guys. 

Does this screenshot tell you anything?

[image]

ai-republic, 2024-03-09:
Hi guys,
sorry I've been away the last couple of days.
Thanks Todor for helping out. 
@Irthinkimlost Yes, it's like I said before, your Daly is most probably lacking the hardware for CAN/RS485 so you need to go via the UART plug of your Daly and connect the RX, TX and GND from the Daly (as in the picture I sent above) to your PI's GPIO pins 6 (GND), 8 (TX) and 10 (RX) like in the picture that @sanitariu posted above. From your 2CH-CAN hat you connect  your cables PIN4 -> CAN H and PIN5 -> CAN L via RJ45 to your Growatt CAN port.

Your PI need to be properly configured:
The serial port of Raspberry Pi is often used to access the shell. However, in some condition you just wanna use it to communicate with UART peripherals. You can disable shell and kernel messages on the serial connection via Raspberry Pi configuration tool:

```
sudo raspi-config
```

Select Advanced Options -> Serial -> <NO> to disable shell and kernel messages on the serial connection. Then edit the /boot/config.txt file and append:

```
enable_uart=1
```

And you will need to set the baud rate to 9600 baud (for now everytime you start your PI):

```
stty -F /dev/ttyAMA0 9600
```

If you ever have the port still open you can close the using process by calling:
```
fuser -k /dev/ttyAMA0
```

Irthinkimlost, 2024-03-09:
Thanks for your reply. 

I don't know what's happening with my BMS now. It was charging this morning and inverter shouted about over voltage +-57v and then shutdown. BMS has been unresponsive since then. This has happened before when charging from solar but the BMS has never done this. 

Battery currently at 53.4v 
I've been using the system for a few months so it's not a setup issue. 

I'll check back in regarding the Pi once I get this sorted. 

Ciao

sanitariu, 2024-03-09:
Somehow you entered your BMS fail safe mode and mosfets are closed. You must use windows softwarre and connect via the ttl2usb to wake it up. I think this is the problem. I do not know your batteries but 57 volts maybe too much ?!

Irthinkimlost, 2024-03-09:
Hi. Thanks for your reply. 

I tried to connect RS485 to USB like I have in the past but it didn't connect like it usually does. I've only ever used pcmaster software to connect on windows. 

Cell charge voltage 3.65 which is 58.4 so I don't think that's the problem. The error and high voltage was reported on the inverter. As soon as this happened I checked battery voltage with voltmeter and it read 53.4v. So I don't think the battery is a problem. It was around 80% SOC when this happened. 

The BMS has disconnected/shutoff die to cell high voltage in the past once but it was easy to restart and continue. This may not be the same situation.

ai-republic, 2024-03-10:
Try to disconnect the Daly shortly from the battery that should restart it.

Irthinkimlost, 2024-03-10:
Good morning 

Thanks for your replies. 

What do you mean by disconnect the battery "shortly"?

I have not yet tried:

 Apply a charge to BMS. I don't have another charger and I can't get the growatt to charge because it doesn't see the battery connected. I've tried changing battery types but no success. I will connect the generator later and see if that helps me. 

I have also not yet tried to connect ttl2usb. I don't know what this means and I looked online but I didn't find anything useful. 

Based on other advice found online I have tried the following unsuccessfully:

Disconnecting and reconnecting BMS
Disconnecting and reconnecting BMS, balance leads and UART. 
Disconnecting overnight and reconnecting BMS the next day
Shorting the pins on UART
Applying 3.3v to UART 
Shorting BMS and battery. I forget which terminals they recommend shorting but I followed instructions. 
Connecting RS485 to laptop

Do you know how I can check if the BMS is dead/broken? 

Thanks guys 👍

ai-republic, 2024-03-10:
Hi,
if you disconnect battery positive (B+) and battery negative (B-) (the two thick wires of your Daly) shortly and then reconnect it should at least restart shortly and you should be able to connect via PC Master and see what errors it reports.

sanitariu, 2024-03-10:
Hi,
I hard similar at the start and as far as i remember i just connected to windows laptop
using ttl2usb adapter which came with Daly BMS and it opened again.
The other ways is somehow to give 50-54 volts like normal solar charing to main cables.
This is what i remember.

Irthinkimlost, 2024-03-11:
I think the BMS is dead. I will look into getting it tested by a professional but I'm looking to replace it. I tried applying current higher than the battery voltage but it didn't work. 

I can easily get another Daly or a JK. What would you recommend?

ai-republic, 2024-03-12:
I would recommend the JK. I have also 8 Daly's, but I've read and watched reviews where the JK is a lot more precise and reliable than the Daly. So if I'd have to choose a new one I guess I'd would go for the JK CAN.

Irthinkimlost, 2024-03-13:
Hi guys

Thank you so so so much for your help. With the relatively low cost to replace the Daly with something better and higher capacity and that will communicate with the growatt out of the box I have purchased a JK 200a BMS. 

I appreciate your efforts to assist the DIY power community and myself and I will be making a contribution to say thank you for everything. 

Wishing you all the best

Steve

closed 2024-03-14 by ai-republic