ai-republic/bms-to-inverter #23: JK BMS doesn't work

danieltorkian, 2024-03-14:
2024-03-14 18:46:34.633 | INFO  | Thread-2   | verter.BmsToInverter:164   | Reading BMS #1 JK_RS485 on /dev/ttyAMA0...
2024-03-14 18:46:34.645 | INFO  | Thread-2   | stoinverter.core.BMS:120   | ---------------------------------> Thread 44
2024-03-14 18:46:34.647 | ERROR | Thread-2   | stoinverter.core.BMS:139   | Error requesting data!
java.lang.UnsupportedOperationException: null
	at java.nio.ByteBuffer.array(ByteBuffer.java:1471) ~[?:?]
	at com.airepublic.bmstoinverter.protocol.rs485.JSerialCommPort.sendFrame(JSerialCommPort.java:115) ~[protocol-rs485-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.bms.jk.rs485.JKBmsRS485Processor.collectData(JKBmsRS485Processor.java:27) ~[bms-jk-rs485-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.core.BMS.process(BMS.java:127) ~[core-api-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.BmsToInverter.lambda$start$2(BmsToInverter.java:165) ~[bms-to-inverter-main-0.0.1-SNAPSHOT.jar:?]
	at java.lang.Thread.run(Thread.java:840) [?:?]

ai-republic, 2024-03-14:
Hi Daniel,
ok, thanks for that. I've not been able to test the RS485 binding for the JK so its good if your willing to help me test it and get it up and running.
I will update the code to fix that exception and then we can go from that. Could you also please set the log level to _debug_ please to get more information?

danieltorkian, 2024-03-15:
sure thing

[logs.txt](https://github.com/ai-republic/bms-to-inverter/files/14609152/logs.txt)

this my config:
bms.pollInterval=1

bms.1.type=JK_RS485
bms.1.id=1
bms.1.portLocator=/dev/ttyAMA0
bms.1.delayAfterNoBytes=2000

###################################################################
###                    Inverter settings                        ###
###################################################################
# The inverter type can be NONE, DEYE_CAN, GROWATT_CAN, SMA_SI_CAN, SOLARK_CAN
inverter.type=SOLARK_CAN
# The port name/device to use to communicate to the  inverter  
inverter.portLocator=can0
# Interval to send data to the inverter (in seconds)
inverter.sendInterval=1

I use SEENGREAT RS485-Dual-CAN-I and also tested two RS485-USB Dongles with same results....

ai-republic, 2024-03-15:
Hi Daniel,
I fixed the exception above.
Could you do a _Clean install_ from the _Configuration_ app please and test again?

danieltorkian, 2024-03-15:
Hi Thank you!
I tried again but still getting an error. See logs.
[BMS-to-Inverter.log](https://github.com/ai-republic/bms-to-inverter/files/14617933/BMS-to-Inverter.log)

danieltorkian, 2024-03-15:
can you please give me the permissions to push a branch and request a PR????

ai-republic, 2024-03-16:
Hi Daniel,
I've added iterating through all commands and error handling in case no data was received.
The program will delay (Delay after no bytes received setting)  a short time and try to read again. If no bytes are received for over 10x then it will try the next command.

Could you please _Clean install_ with the _Configurator_ and lets see if we get any responses.

danieltorkian, 2024-03-17:
Hey Torsten, CRC calculation of spec is wrong.... See according branch. Next step would be to convert the values from response to Battery_Object according to the spec.

ai-republic, 2024-03-18:
Hi Daniel,
thanks for your contribution. 
I merged your code with some minor changes. I skipped the lombok framework (I know its nice) but I'd like to keep the footprint as small as possible. 
I was thinking about requesting all values at once, but kept it consistent with the other bindings. 
But it will be more performant this way so I kept it :)

ai-republic, 2024-03-21:
@danieltorkian could you try a _Clean install_ from the _Configurator_ application and test the JK binding?

ai-republic, 2024-03-28:
I will close this issue, since the JK RS485 binding is working now

closed 2024-03-28 by ai-republic