ai-republic/bms-to-inverter #24: BMS tab is empty

carsonfee, 2024-03-17:
[image]

Can't add a bms in the configuator the dialog is empty

ai-republic, 2024-03-17:
Have you downloaded the latest _Configurator_ application?

carsonfee, 2024-03-17:
Yes I have

ai-republic, 2024-03-18:
Please get the new _Configurator_ app again. I've tested it and it works fine for me

carsonfee, 2024-03-18:
got latest configurator. couldn't do a clean install as the BMS tab was still empty. I copied my config.properties to the location of the new configurator and ran again. it read the configuration and I updated and did a clean install. the install window was also blank but I watched files get added and closed the window when it stopped adding files. ran ./start and the web server fails. I'm attaching the log
[start.log](https://github.com/ai-republic/bms-to-inverter/files/14631704/start.log)

ai-republic, 2024-03-18:
Just tested the _Configurator_ app on my Pi and also there it works.

[image]

The Webserver won't run if the BMS-to-Inverter application is not running. You need to select the Webserver in the _Configurator_ and it will automatically also select the MQTT broker and client.

Have you tried to run the _Configurator_ on a PC or laptop and then transfer the installation folder per e.g. WinSCP or Putty to the Pi?

carsonfee, 2024-03-18:
I have the gui working now. something is messed up with file permissions. if I run sudo configurator.jar all of the dialogs are blank. I tried chmod 777 and ran sh configurator. the gui is correct now but it fails to write the configuration and also fails to install. doing a combination of adding a config folder and a temp folder seemed to let me save and clean install. The web server is still crashing however.  here's a new log : 
[bms2inv_start.log](https://github.com/ai-republic/bms-to-inverter/files/14632964/bms2inv_start.log)

carsonfee, 2024-03-18:
the bms-to-inverter application fails to run:
[bms_cli.log](https://github.com/ai-republic/bms-to-inverter/files/14633233/bms_cli.log)

ai-republic, 2024-03-18:
Could you please replace your _log4j2.xml_ in your _config_ folder with this one (please remove the suffix _.txt_):
[log4j2.xml.txt](https://github.com/ai-republic/bms-to-inverter/files/14633604/log4j2.xml.txt)

and run the program again to get more information. It is very strange that there is not more in the log.

carsonfee, 2024-03-18:
here you go.
[newlog.log](https://github.com/ai-republic/bms-to-inverter/files/14641559/newlog.log)

ai-republic, 2024-03-19:
It looks like your install did not finish downloading all the libraries. Please check your _lib_ directory in the installation folder. I'm pretty sure its missing a file called _netty-common-4.1.94.Final.jar_.

But back to the problem of this issue:
1. Have you tried to run the _Configurator_ application on a PC? How does it look there?
2. Have you tried running the_Configurator_ application directly on the PI without using VNC?
3. Could you tell me what PI model you are using which Java is installed (`java -version` in the terminal)

carsonfee, 2024-03-19:
Its a Pi 4B. 
I'll try running directly on the pi, I need to spin up a linux build on my PC

openjdk version "17.0.10" 2024-01-16
OpenJDK Runtime Environment (build 17.0.10+7-Debian-1deb12u1)
OpenJDK 64-Bit Server VM (build 17.0.10+7-Debian-1deb12u1, mixed mode, sharing)

carsonfee, 2024-03-19:
running directly on the Pi the configurator worked correctly without issues but I still have the crash in bms-inverter

ai-republic, 2024-03-19:
Still the java.lang.ClassNotFoundException: io.netty.util.internal.PlatformDependent exception?
Could you please delete the _.m2_ folder in your home folder and also delete your installation folder and do a fresh _Clean install_ again and be sure to enable the _Webserver_, the _MQTT Broker Server_ and _MQTT Producer_ services.
Maybe the Netty Jar file is corrupt due to a incomplete download.

carsonfee, 2024-03-19:
did a clean install on PC and copied over to the PI. it's running again but the can info is still wrong on the web page. A can dump does look correct but not the values displayed on the page

[image]

carsonfee, 2024-03-19:
can0  35E   [8]  4C 45 4F 43 48 20 20 20   'LEOCH   '
  can0  359   [8]  00 00 00 00 04 50 4E 00   '.....PN.'
  can0  351   [8]  34 02 D0 07 A0 0F B0 01   '4.......'
  can0  355   [8]  42 00 64 00 00 00 00 00   'B.d.....'
  can0  356   [8]  CE 14 F7 FF BE 00 00 00   '........'
  can0  35C   [8]  C0 00 00 00 00 00 00 00   '........'

ai-republic, 2024-03-21:
Hi,
can you confirm that the _Configurator_ is working? Because then I would close this issue and continue the discussion in the issue #20

carsonfee, 2024-03-21:
Yes. Configurator is working correctly. We can close this

closed 2024-03-21 by carsonfee