ai-republic/bms-to-inverter #28: mqtt with authentication?

marekorok, 2024-05-08:
Hello,

is there any way to use mqtt with authentication?

I was trying tcp://login:password@server_ip:1883 but no luck
mosquitto reports in logs:
Client <unknown> disconnected due to protocol error.

cheers

ai-republic, 2024-05-09:
Have you tried tcp://<server-ip>:61616 ? That the locator for the Artemis MQTT broker

marekorok, 2024-05-09:
well, I'm using standard [mosquitto mqtt broker](https://mosquitto.org/) installed on my home assistant environment and I don't want to use anything else.
Then it is using standard ports: 1883 for non-encrypted and 8883 for TLS encrypted traffic.

Every client I have is authenticating to the mqtt for security reasons, which I think is a standard nowadays, example Tasmota:
[image]

That is why I'm asking for such possibility...

ai-republic, 2024-05-12:
Adding the authentication for the Artemis MQ Broker would be no problem. But you would need to a bridge between Artemis MQ to Mosquitto. So you need a client connecting to Artemis MQ to receive the BMS data and a producer to post to your Mosquitto broker on a topic/queue there.
Or you could have home assistant connect as a client directly to the Artemis MQ broker which would make things simpler.

ai-republic, 2024-05-12:
I guess for the integration to HomeAssistant it would be nice to configure (possible multiple) producers to post the BMS data directly to a MQTT broker. You would then need to create a HA configuration first to map the BMS data Json to a HA sensor. At least that's how I understand. I haven't used HA myself so I'm just guessing from the HA overview about connecting MQTT devices.

marekorok, 2024-05-17:
don't bother with HA for the moment, leave it for later, but shortly: the integration is done through specific autodiscovery mqtt topics: https://www.home-assistant.io/integrations/mqtt/

Assuming there is standard MQTT client implementation then there should be no difference which server is used, am I right?
Then I still prefer to stick with lightweight mosquitto mqtt broker and not to add another heavy software package.

ai-republic, 2024-05-19:
Hi, I've added the support to supply a username and passwort now in the mqtt.producer configuration:
```
# The URL of the MQTT broker to send to
mqtt.producer.locator=tcp://127.0.0.1:61616
# The topic name on the MQTT broker to send to
mqtt.producer.topic=energystorage
# The username of the external MQTT broker to send to
mqtt.producer.username=guest
# The password on the external MQTT broker to send to
mqtt.producer.password=guest_pwd
```

This mqtt.producer configuration will now be used to connect to an external broker (like Mosquitto).

ai-republic, 2024-06-01:
I will close this ticket as I haven't heard any problems

closed 2024-06-01 by ai-republic