ai-republic/bms-to-inverter #32: SRNE  inverter support

Dimu1234, 2024-06-12:
Hi & thank you for the project.
Does this support SRNE inverters & if not, do you have any plans to improve this to add SRNE inverter support?

ai-republic, 2024-06-12:
Hi,
thank you for liking this project :)
As far as I researched quickly, your SRNE inverter should support the PYLON CAN protocol if you enabled the CAN support in the inverter (32 - CAN) and CAN mode UZE in option 33.

Which battery system or BMS(s) do you have?

Dimu1234, 2024-06-12:
Thank you for the quick reply.
I'm having JK-BMS with assembled LiFePO4 battery pack. (16 cells of 160Ah, 48 V)

ai-republic, 2024-06-30:
Ah ok, so you can test if you get reading using the JK BMS binding (choose _JK_CAN_ in the _Configurator_ app). For the test please enable _debug_ log-level and as inverter choose _NONE_ for now.
Which CAN hat do you have for your Raspberry PI? If none yet, I would suggest the Waveshare 2CH CAN hat as you'll be needing 2 can ports (one from BMS to RPI and one from RPI to inverter). 

NOTE: We have discovert that the Waveshare 2CH CAN hat does not (currently) work with a RPI 5. If you want to buy a PI for this project I would suggest to go with the RPI 4B for now.

Dimu1234, 2024-07-03:
[image]

This is what I have

ai-republic, 2024-07-06:
Ok. So you want to connect the JK BMS via RS485 and the SRNE inverter via CAN, right?

Dimu1234, 2024-07-09:
I don't think SRNE inverter has CAN port. I could not see any info in the manual. But it has RS485 port. So is this possible RS485 to RS485 ?

ai-republic, 2024-08-09:
Yes, you can also use the CAN port for your _JK_CAN_ BMS binding and RS485 port for your _PYLON_RS485_ inverter binding.

ai-republic, 2024-09-29:
@Dimu1234 any progress on your project? How did it work out? If everything works fine I'd be very thankful for any [donation](https://github.com/sponsors/ai-republic) 😄

ai-republic, 2024-10-04:
I guess I can close this

closed 2024-10-04 by ai-republic