ai-republic/bms-to-inverter #34: SMA CAN Help

ironcowboysolar, 2024-06-24:
Hi! I love what you’ve got going here. I recently built a battery pack with three Daly BMS and I am attempting to establish EXTBMS comms with SMA.

I saw your post on a forum and I have some rPi 4b’s laying around. I’m a novice in programming and raspbian. The most I’ve done on a rPi is emulation and tweak config files.

I’m also new to GitHub and I’m not even sure if this is where I post this kind of stuff.

Anyways, I’m looking for some dumbed down guidance on creating this tool. I’ll gladly support this project and I’ll send more to keep this project going and for your time.

Thanks!

ai-republic, 2024-06-25:
Hi @ironcowboysolar,
first of all, thanks for liking and supporting this project!!!

Before we get you up and running I have a couple of questions on your setup:
1. Did you get your Daly BMSes specifically CAN enabled by the sales support when you bought them?
2. What CAN adapter/hat do you have for your RPI 4B?

Best regards,
Torsten

ironcowboysolar, 2024-06-26:
Thanks for the response! I purchased the BMS from Amazon. They’re Daly Smart BMS with Parallel Module and the Comm interface board. The technical documentation mentioned that they were ready for both CAN and RS485.

Included is a picture of the batteries I build. I’m really hoping to be able to work this out otherwise in future builds I may switch reluctantly to Batrium.

[image]
[image]
[image]

ai-republic, 2024-06-30:
Those boxes look great :)
I have the same setup really, also with 8 Daly's with parallel module and interface board, but nime doesn't look so clean ;)

Did you already receive/install your Waveshare 2CH CAN hat? If you need any help I can assist you with that. 
Once that is installed you can connect all of the interface boards with RJ45 cables like you have now, but in the left ports (which are the CAN ports). Make sure you set the DIP switches for each connected interface board starting from 0 to 2. From the first BMS put a RJ45 cable in the 2nd CAN port of the interface board and from there to your can0 port of you 2CH CAN hat. You need to cut the cable and only use the wires of PIN 4 (CAN L) and PIN 5 (CAN H) to connect to the 2CH CAN hat (see following image) but I also tested PIN 1 (CAN H) and PIN 2 (CAN L) and they also work:

[image]

You count the pins from the left side when the clip of the RJ45 plug is facing _down_. Then you are basically ready to run the _Configurator_ app, choose log-level _debug_ in the general settings and then add your BMSes with IDs 1-3 on port _can0_. Choose inverter as _NONE_ for now. And then start the application. Please read also the [Wiki - How to use](https://github.com/ai-republic/bms-to-inverter/wiki/How-to-use) for guidance.

If you encounter any problems, just contact me :)

ironcowboysolar, 2024-07-17:
Thanks for the response. Most of the questions I have are mostly related to setting up the Rpi. Here's my questions: 

1. For the Raspberry Pi, I'm assuming that I'll need to reformat the SD card and set it up as a fresh install. Do I use Raspberry Pi OS 32 or 64 bit? Should I install the full OS with the desktop environment or just the minimum build? What instructions do you have for installing the CAN hat?

2. Do I install the "Configurator" on my PC or on the Raspberry Pi? Is it possible to set up the Rpi so that it works automatically when its plugged into the power supply, or will I need to bring a monitor and keyboard onsite to set it up? Will it need to be set up and started up manually if it loses power and comes back on?

3. On the Daly BMS - on the "Upper Computer" app, there is an option to change the communication protocol from RS485 (default) to CAN. When I try to do this it doesn't let me make the change. From what I am beginning to understand, plugging the BMS directly from the UART port on the BMS doesn't allow you to change that parameter, and to do this you will need to connect to the BMS somehow via the interface board. The interface board has a RS232/RJ11 port, and it came with an RJ11 to USB cable. Is this how I connect my computer to it?

4. Which models of Rpi will work for this? In the future, if i can achieve the same thing using a Pi Zero or a 2 or 3, I'll probably use one of those instead.

Thanks again!

ai-republic, 2024-07-21:
Alright, cool. So 
1. yes, its probably best to use a clean Raspian OS 64bit. The instructions for you CAN hat can  usually be found on the manufacturer's site and differs from device to device so I can't really give general instructions here ;)
2. You can SSH to your PI if you know its IP, then there's no need for keyboard and monitor. To have it restarted each time it reboots you can add a call to the `start` shell script to you `sudo nano /etc/rc.local` file. As the PI will not automaticall reboot after a power loss you can add a PI battery UPS so that it will not power off when there's an outage
3. The Daly with CAN support needs to ordered specifically via the online support. The Daly's found in online shops usually only support UART and/or RS485. I have mine connected via interface boards which offer both RS485 and CAN. But as I mentioned, if you didn't order these via customer support you probably don't have the CAN version
4. I recommend PI 3B or 4B, but any PI with enough memory and 64bit will do. With the PI 5 I have some issues getting the Waveshare 2CH CAN hat running.
5. Yes, that is what this program is for ;) To translate between different communication specifications and protocols :)

ironcowboysolar, 2024-07-26:
Ok, I have some good news and some not good news.

The good news is that I got the CAN hat installed and was able to validate the function of can1 sending and can0 receiving.

I ended up installing configurator directly on the rPi.

I created a clean install file.

here’s where I ran into an issue:

When I execute the shell script “Start” nothing happens. If I select execute in terminal, the terminal window opens and quickly closes without any indication that it’s doing anything.

so I tried to open terminal to manually execute the start file from root user. When I do that, it says “cannot access the jarfile”

In response to this, I copied the jarfile path from the “lib” folder, as detailed  the “start” file, then from terminal I executed the file directly “Java -jar lib/bms-to-inverterblahblah.jar”
When I did this, a terminal window started reporting CAN data and kept doing it until I closed the window.

In a nutshell, it seems like the start sh file may not be working, but the .jar file it says it doesn’t have access to definitely works.

thoughts?

ai-republic, 2024-07-26:
Did you change the permissions like described in the [Wiki](https://github.com/ai-republic/bms-to-inverter/wiki/How-to-use):

```
chmod 777 start
chmod 777 configurator
```

I think that will solve your issue. 

For easier starting I would suggest you make two shell scripts to start the bms-to-inverter app and webserver separately (I will change that also for the _Configurator_ application to provide 2 scripts):

**_start_**
```
#!/bin/bash
java -DconfigFile=config/config.properties -Dlog4j2.configurationFile=file:config/log4j2.xml -jar lib/bms-to-inverter-main-0.0.1-SNAPSHOT.jar
```

**_startWebserver_**
```
#!/bin/bash
java -jar lib/webserver-0.0.1-SNAPSHOT.jar --spring.config.location=file:config/config.properties
```

ai-republic, 2024-08-02:
@ironcowboysolar did it solve your issue?

ironcowboysolar, 2024-08-08:
Oh man lots of information here to share. First thing is to get the shell script to work I had to put the _specified_file paths for all of the files detailed in the shell script. I also had to set chmod permissions to 755. To make the shell script execute automatically upon startup I had to add a bash script for the start.sh file and I had to add chmod +x to the rc.local file and all the configurator files.

Now when I execute start.sh in terminal I can see all the processes running as it tries to open ports and send and receive data.

Now for the complicated stuff. I have not been able to verify that the CANBUS ports on the interface boards actively transmit any data. Using the DALY app, I have been able to set the communication protocol to SMA CAN, as well as save the setting, but i'm still unable to read anything and I'm not sure if it's actually doing anything. I have a CAN reader for my laptop but the software that comes with it sucks so I'm going to get a better one.

With the BMS to inverter running, when I try to give the command "ifconfig can0" it doesn't show that its receiving any information packets, nor does candump can0.

Maybe you're correct about the BMS being shipped without CAN capabilities, but the amazon page where I bought it from states that it's supposed to have CAN/UART/RS485 support.

I have a few options at this point. Seemingly, the worst case scenario would be that I translate from Daly RS485 to SMA CAN, but from what I can tell, my CAN hat does not support RS485. Seems like I'll need to get a two channel CAN hat that supports both protocols to be able to do this.

I can also get a decent CAN reader from my laptop so that I could verify whether or not the CAN ports transmit data. Due to some of what the log file states, I am suspicious that maybe the CAN hat is improperly installed.

Here is a brief copy and paste from the log file:

**2024-08-07 16:53:04.588 | INFO  | Thread-2   | toinverter.core.Port:113   | Opening port can0 SUCCESSFUL
2024-08-07 16:53:04.589 | DEBUG | Thread-2   | s485.JSerialCommPort:159   | Send: Buffer (HEX): [0xA5, 0x40, 0x50, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x3D]

2024-08-07 16:53:04.589 | ERROR | Thread-2   | ractDalyBmsProcessor:70    | Failed to initialize BMS!
java.lang.NullPointerException: Cannot invoke "com.fazecast.jSerialComm.SerialPort.getOutputStream()" because "this.port" is null
	at com.airepublic.bmstoinverter.protocol.rs485.JSerialCommPort.sendFrame(JSerialCommPort.java:164) ~[protocol-rs485-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.bms.daly.rs485.DalyBmsRS485Processor.sendMessage(DalyBmsRS485Processor.java:66) ~[bms-daly-rs485-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.bms.daly.common.AbstractDalyBmsProcessor.collectData(AbstractDalyBmsProcessor.java:63) ~[bms-daly-common-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.core.BMS.process(BMS.java:137) ~[core-api-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.BmsToInverter.lambda$start$2(BmsToInverter.java:239) ~[bms-to-inverter-main-0.0.1-SNAPSHOT.jar:?]
	at java.lang.Thread.run(Thread.java:840) [?:?]
2024-08-07 16:53:04.590 | INFO  | Thread-2   | toinverter.core.Port:111   | Opening can0 ...
2024-08-07 16:53:04.591 | ERROR | Thread-2   | s485.JSerialCommPort:71    | Could not open port can0!
com.fazecast.jSerialComm.SerialPortInvalidPortException: Unable to create a serial port object from the invalid port descriptor: /dev/can0
	at com.fazecast.jSerialComm.SerialPort.getCommPort(SerialPort.java:502) ~[jSerialComm-2.10.4.jar:2.10.4]
	at com.airepublic.bmstoinverter.protocol.rs485.JSerialCommPort.open(JSerialCommPort.java:58) ~[protocol-rs485-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.core.Port.ensureOpen(Port.java:112) ~[core-api-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.protocol.rs485.JSerialCommPort.sendFrame(JSerialCommPort.java:156) ~[protocol-rs485-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.bms.daly.rs485.DalyBmsRS485Processor.sendMessage(DalyBmsRS485Processor.java:66) ~[bms-daly-rs485-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.bms.daly.common.AbstractDalyBmsProcessor.collectData(AbstractDalyBmsProcessor.java:74) ~[bms-daly-common-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.core.BMS.process(BMS.java:137) ~[core-api-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.BmsToInverter.lambda$start$2(BmsToInverter.java:239) ~[bms-to-inverter-main-0.0.1-SNAPSHOT.jar:?]
	at java.lang.Thread.run(Thread.java:840) [?:?]
Caused by: java.io.IOException
	at com.fazecast.jSerialComm.SerialPort.getCommPort(SerialPort.java:488) ~[jSerialComm-2.10.4.jar:2.10.4]
	... 8 more
2024-08-07 16:53:04.591 | INFO  | Thread-2   | toinverter.core.Port:113   | Opening port can0 SUCCESSFUL
2024-08-07 16:53:04.592 | DEBUG | Thread-2   | s485.JSerialCommPort:159   | Send: Buffer (HEX): [0xA5, 0x40, 0x90, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x7D]

2024-08-07 16:53:04.592 | ERROR | Thread-2   | stoinverter.core.BMS:149   | Error requesting data!
java.lang.NullPointerException: Cannot invoke "com.fazecast.jSerialComm.SerialPort.getOutputStream()" because "this.port" is null
	at com.airepublic.bmstoinverter.protocol.rs485.JSerialCommPort.sendFrame(JSerialCommPort.java:164) ~[protocol-rs485-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.bms.daly.rs485.DalyBmsRS485Processor.sendMessage(DalyBmsRS485Processor.java:66) ~[bms-daly-rs485-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.bms.daly.common.AbstractDalyBmsProcessor.collectData(AbstractDalyBmsProcessor.java:74) ~[bms-daly-common-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.core.BMS.process(BMS.java:137) ~[core-api-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.BmsToInverter.lambda$start$2(BmsToInverter.java:239) ~[bms-to-inverter-main-0.0.1-SNAPSHOT.jar:?]
	at java.lang.Thread.run(Thread.java:840) [?:?]**

I'm interested to hear your ideas on this. Thanks!

ErikssonPer, 2024-08-08:
You could loop both canhat channels together and send some can commands with one and recieve with the other, that way you'll see that the hat works atleast.

ironcowboysolar, 2024-08-08:
> You could loop both canhat channels together and send some can commands with one and recieve with the other, that way you'll see that the hat works atleast.

I have opened two terminal windows and did a candump can0 and then cansend can1 000#11 22 33 44 55 

This works like its supposed to.

ErikssonPer, 2024-08-08:
Check this out, may need to send a message first to get it to start sending data,
https://github.com/Louisvdw/dbus-serialbattery/discussions/561

ai-republic, 2024-08-09:
> First thing is to get the shell script to work I had to put the _specified_file paths for all of the files detailed in the shell script. I also had to set chmod permissions to 755. To make the shell [...]

1. The shell script should work as it is - unless you call it from another directory. In that case you could just add something like `cd ~/bms-to-inverter` at the beginning of the `start.sh` script.
2. Yes, you need to do a `chmod 744 start.sh startWebserver.sh` if you copy the files from Windows to your PI.
3.  Your PI with Raspian OS should have the bash shell installed per default

ai-republic, 2024-08-09:
> Maybe you're correct about the BMS being shipped without CAN capabilities, but the amazon page where I bought it from states that it's supposed to have CAN/UART/RS485 support.

I can recommend chatting to a Daly representative directly. Here's a WhatsApp link from their Website:
[image]

Or just send them an email. You'll probably have to send a picture of the BMS and S/N.
There is also a way to measure the voltage on the pins: CAN Hi usually measures from 2.5V to 3.75V while CAN Lo measures from 2.5V to 1.25V.

ai-republic, 2024-08-09:
> **2024-08-07 16:53:04.588 | INFO | Thread-2 | toinverter.core.Port:113 | Opening port can0 SUCCESSFUL 2024-08-07 16:53:04.589 | DEBUG | Thread-2 | s485.JSerialCommPort:159 | Send: Buffer (HEX): [...]

You have configured the _DALY_RS485_ BMS binding but specified `can0` as port. That cannot work. To use CAN you need to configure the _DALY_CAN_ BMS binding and use `can0` (if connected to `can0` on your hat) or _DALY_RS485_ BMS binding connected to A and B of your RS485 connector.

ai-republic, 2024-08-09:
> Check this out, may need to send a message first to get it to start sending data, [Louisvdw/dbus-serialbattery#561](https://github.com/Louisvdw/dbus-serialbattery/discussions/561)

Once the CAN is connected it usuallly works out of the box. To be sure you can restart the BMS (either via their PC software, or interface board or disconnecting the battery cable shortly).
The Bms-to-Inverter application will send a message to the Daly BMS to first read out some general information like number of cellls in the pack, etc. The Daly BMS will respond immediately. I have the application running with my 8 Daly BMSs myself ;)

ai-republic, 2024-08-09:
> Using the DALY app, I have been able to set the communication protocol to SMA CAN,

If you use the _DALY_CAN_ BMS binding then please set the 'Inverter Manager' to _NONE_ and not to _SMA_! Or you could set the _SMA_CAN_  BMS binding, but I recommend the Daly binding for your Daly ;)

[image]

ironcowboysolar, 2024-08-09:
Sorry, I accidentally posted the part of the log where it was tinkering with the Daly RS485. Here's the same log from when it was set to Daly_CAN

"2024-08-07 17:33:26.249 | INFO  | Thread-2   | verter.BmsToInverter:238   | Reading BMS #1 DALY_CAN on can0...
2024-08-07 17:33:26.250 | INFO  | Thread-2   | stoinverter.core.BMS:130   | ---------------------------------> Thread 19
2024-08-07 17:33:26.252 | DEBUG | Thread-2   | .DalyBmsCANProcessor:46    | SEND: Buffer (HEX): [0x40, 0x01, 0x50, 0x18, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-08-07 17:33:26.253 | DEBUG | Thread-2   | ocol.can.JavaCANPort:113   | CAN frame sending: Buffer (HEX): [0x40, 0x01, 0x50, 0x18, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-08-07 17:33:26.255 | DEBUG | Thread-2   | ocol.can.JavaCANPort:76    | CAN frame read...
2024-08-07 17:33:27.270 | ERROR | Thread-2   | ractDalyBmsProcessor:70    | Failed to initialize BMS!
tel.schich.javacan.platform.linux.LinuxNativeOperationException: Unable to read from the socket - errorNumber=11, errorMessage='Resource temporarily unavailable'
	at tel.schich.javacan.SocketCAN.read(Native Method) ~[javacan-core-3.2.4.jar:?]
	at tel.schich.javacan.AbstractCanChannel.readSocket(AbstractCanChannel.java:213) ~[javacan-core-3.2.4.jar:?]
	at tel.schich.javacan.RawCanChannelImpl.readUnsafe(RawCanChannelImpl.java:85) ~[javacan-core-3.2.4.jar:?]
	at tel.schich.javacan.RawCanChannelImpl.read(RawCanChannelImpl.java:79) ~[javacan-core-3.2.4.jar:?]
	at tel.schich.javacan.RawCanChannelImpl.read(RawCanChannelImpl.java:74) ~[javacan-core-3.2.4.jar:?]
	at com.airepublic.bmstoinverter.protocol.can.JavaCANPort.receiveFrame(JavaCANPort.java:77) ~[protocol-can-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.bms.daly.can.DalyBmsCANProcessor.sendMessage(DalyBmsCANProcessor.java:54) ~[bms-daly-can-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.bms.daly.common.AbstractDalyBmsProcessor.collectData(AbstractDalyBmsProcessor.java:63) ~[bms-daly-common-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.core.BMS.process(BMS.java:137) ~[core-api-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.BmsToInverter.lambda$start$2(BmsToInverter.java:239) ~[bms-to-inverter-main-0.0.1-SNAPSHOT.jar:?]
	at java.lang.Thread.run(Thread.java:840) [?:?]"

ai-republic, 2024-08-09:
Did you configure your CAN hat properly and have you pulled the can0 interface up with:sudo ip link set can0 type can bitrate 500000sudo ifconfig can0 up

ironcowboysolar, 2024-08-11:
This was copy pasted from my rc.local file so that it executes upon startup.

#!/bin/bash
# Initialize CAN BUS
sudo ip link set can0 up type can bitrate 1000000
sudo ip link set can1 up type can bitrate 1000000
sudo ifconfig can0 txqueuelen 65536
sudo ifconfig can1 txqueuelen 65536

ai-republic, 2024-08-11:
Thanks @ironcowboysolar - could you try setting the bitrate to 500000 please? I think the Daly only supports that.

ironcowboysolar, 2024-08-12:
OK, so every time I try this I get further and I think I'm very close to getting this to work.

I was able to get a voltage on the CAN pins on the Daly, but since I'm unsure about it, I installed the RS485 CAN Hat and decided to go with the DALY_RS485 binding.

After I dialed it in some, I was able to confirm that the rPi was definitely receiving RS485 information from the BMS. Upon reboot, my rPi boots to terminal and the start.sh script is executed automatically after it boots. As it started scrolling through all the data, I could briefly see where the rPi was collecting data, such as the voltages of the various cells and the state of charge, etc.

Also, when this was running, I did a candump and could see lots of CAN data. ifconfig also said that there were numerous packets being sent etc.

Now my problem is that I can't seem to get the Sunny Island to recognize the CAN data. When I set up the inverter for Li_Batt_EXT_BMS, upon boot up it works for a minute and then I get a BMS timeout error. During the time that it does run, it does not receive the SOC information from the rPi.

The configurator is set to SMA_CAN  binding on port can0 with a baudrate of 500,000. There is a resistor on the rPi and a resistor on the other SMA. Do you have any suggestions or tips? I feel like I'm close to a breakthrough.

Thanks for your detailed responses.

ai-republic, 2024-08-14:
Hi @ironcowboysolar,
first just a couple of questions:
1. Daly is now connected via RS485 to your RPI right?
2. Your CAN hat is only connected to the SMA inverter (`can0`)? Or also to your Daly? Where did you do the candump? Could you please attach the file here too please?

For the RS485 you do not have to connect A->B, B->A, just connect A->A, B->B. That would only be for serial communication using RX, TX.

Can we test just the Daly binding first please. So only choose the _DALY_RS485_ binding for your BMS with Id #1 at 9600 baud. Then for the inverter please choose _NONE_ as binding. And please also set the log level to _Debug_, Can you please run it and then attach the log file here please.

ironcowboysolar, 2024-08-14:
1. Yes, I have installed the Waveshare RS485/CAN hat. I have the Daly BMS connected to the port serial0 and the SMA is connected to the can0 port. 

I have the BMS to Inv program set to execute on startup after the rPi boots into terminal instead of the desktop environment. When it starts up, it goes through the boot process, and after that I can see it quickly processing data packets on the screen and I can catch bits of the data as it rains down. I can see that from the Daly, it receives the BMS info such as cell voltage, temps, SOC and other things. Lots of RECEIVED  on serial0 and SENDING on can0. Here's the log file details:

I can see that this information is being transmitted via the can0 port.

ironcowboysolar, 2024-08-14:
Here's some more logs that may be useful:

2024-08-11 19:24:18.725 | DEBUG | Thread-11  | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0xEF]

2024-08-11 19:24:18.726 | DEBUG | Thread-11  | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0x0C]

2024-08-11 19:24:18.727 | DEBUG | Thread-11  | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0xF2]

2024-08-11 19:24:18.728 | DEBUG | Thread-11  | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0x0C]

2024-08-11 19:24:18.729 | DEBUG | Thread-11  | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0xF3]

2024-08-11 19:24:18.730 | DEBUG | Thread-11  | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0x4B]

2024-08-11 19:24:18.731 | DEBUG | Thread-11  | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0x88]

2024-08-11 19:24:18.744 | DEBUG | Thread-11  | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0xA5]

2024-08-11 19:24:18.745 | DEBUG | Thread-11  | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0x01]

2024-08-11 19:24:18.746 | DEBUG | Thread-11  | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0x95]

2024-08-11 19:24:18.749 | DEBUG | Thread-11  | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0x08, 0x03]

2024-08-11 19:24:18.752 | DEBUG | Thread-11  | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0x0C, 0xF1]

2024-08-11 19:24:18.752 | DEBUG | Thread-11  | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0x0C]

2024-08-11 19:24:18.753 | DEBUG | Thread-11  | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0xF2]

2024-08-11 19:24:18.754 | DEBUG | Thread-11  | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0x0C]

2024-08-11 19:24:18.755 | DEBUG | Thread-11  | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0xF2]

2024-08-11 19:24:18.756 | DEBUG | Thread-11  | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0x4B]

2024-08-11 19:24:18.757 | DEBUG | Thread-11  | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0x8A]

2024-08-11 19:24:18.769 | DEBUG | Thread-11  | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0xA5]

2024-08-11 19:24:18.770 | DEBUG | Thread-11  | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0x01]

2024-08-11 19:24:18.771 | DEBUG | Thread-11  | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0x95]

2024-08-11 19:24:18.772 | DEBUG | Thread-11  | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0x08]

2024-08-11 19:24:18.773 | DEBUG | Thread-11  | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0x04]

2024-08-11 19:24:18.774 | DEBUG | Thread-11  | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0x0C]

2024-08-11 19:24:18.775 | DEBUG | Thread-11  | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0xF3]

2024-08-11 19:24:18.776 | DEBUG | Thread-11  | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0x0C]

2024-08-11 19:24:18.777 | DEBUG | Thread-11  | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0xF2]

2024-08-11 19:24:18.778 | DEBUG | Thread-11  | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0x0C]

2024-08-11 19:24:18.779 | DEBUG | Thread-11  | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0xF2]

2024-08-11 19:24:18.780 | DEBUG | Thread-11  | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0x4B]

2024-08-11 19:24:18.781 | DEBUG | Thread-11  | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0x8D]

2024-08-11 19:24:18.789 | DEBUG | Thread-2   | alyBmsRS485Processor:67    | SEND: Buffer (HEX): [0xA5, 0x40, 0x95, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x82]

2024-08-11 19:24:18.789 | DEBUG | Thread-11  | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0xA5]

2024-08-11 19:24:18.790 | DEBUG | Thread-11  | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0x01]

[... 266 lines cut ...]
2024-08-11 19:24:19.496 | DEBUG | Thread-6   | verter.core.Inverter:94    | Inverter send: Buffer (HEX): [0x55, 0x03, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x46, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-08-11 19:24:19.497 | DEBUG | Thread-6   | ocol.can.JavaCANPort:113   | CAN frame sending: Buffer (HEX): [0x55, 0x03, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x46, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-08-11 19:24:19.499 | DEBUG | Thread-6   | verter.core.Inverter:94    | Inverter send: Buffer (HEX): [0x56, 0x03, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0xB4, 0x14, 0xB5, 0xFF, 0x00, 0x00, 0x00, 0x00]

2024-08-11 19:24:19.500 | DEBUG | Thread-6   | ocol.can.JavaCANPort:113   | CAN frame sending: Buffer (HEX): [0x56, 0x03, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0xB4, 0x14, 0xB5, 0xFF, 0x00, 0x00, 0x00, 0x00]

2024-08-11 19:24:19.501 | DEBUG | Thread-6   | verter.core.Inverter:94    | Inverter send: Buffer (HEX): [0x59, 0x03, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-08-11 19:24:19.502 | DEBUG | Thread-6   | ocol.can.JavaCANPort:113   | CAN frame sending: Buffer (HEX): [0x59, 0x03, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-08-11 19:24:19.514 | DEBUG | Thread-2   | alyBmsRS485Processor:67    | SEND: Buffer (HEX): [0xA5, 0x40, 0x98, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x85]

2024-08-11 19:24:19.609 | DEBUG | Thread-2   | s485.JSerialCommPort:149   | Next frame: Buffer (HEX): [0xA5, 0x01, 0x98, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x46]

2024-08-11 19:24:19.611 | DEBUG | Thread-2   | alyBmsRS485Processor:85    | RECEIVED: Buffer (HEX): [0xA5, 0x01, 0x98, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x46]

2024-08-11 19:24:19.611 | WARN  | Thread-2   | alyBmsRS485Processor:150   | Command 0x98 to BMS 1 successfully sent and received!
2024-08-11 19:24:19.613 | INFO  | Thread-2   | verter.BmsToInverter:290   | 
BMS	SOC	  V  	  A  	 CellMinV 	 CellMaxV	CellDiff
#1	 70.9	53.0	-7.5	3.306(#16) 	3.315(#15)	0.009 

2024-08-11 19:24:19.613 | INFO  | Thread-2   | verter.BmsToInverter:360   | BMS alarms: 
	NONE

ironcowboysolar, 2024-08-14:
This battery is about 45 minutes from me, so if it's possible, I'd like to go out there with a specific goal or few different things to try, unless you were able to support in realtime. Thank you!

ai-republic, 2024-08-16:
Hi @ironcowboysolar,
thanks for the log files. I have added support for inverter and BMS plugins and for your case I added a plugin to use preset values for batteries if the battery information has not yet been read. I also fixed a few other things concerning aggregation of some values. I also updated and fixed the _Configurator_ app.

For me to access your PI remotely you would need to either install a RDP application (like pi-connect, teamviewer, etc.)  or create port forwarding to your PI's internal IP on the SSH port (22) on your router and send me the external IP of your router and a user login user/pwd on your PI.

ironcowboysolar, 2024-09-02:
I have still been unable to realise communication between the rPi and the SMA. 

When you say this part "I have added support for inverter and BMS plugins and for your case I added a plugin to use preset values for batteries if the battery information has not yet been read. I also fixed a few other things concerning aggregation of some values. I also updated and fixed the Configurator app" 

Does this mean that I just need to re-download configurator, and extract the contents over the old files? When you mention plugins, I'm not sure exactly what you mean and how to apply them. In exactly what way did you add the support and how do I utilise it?

A few weird things: The first is that the BMS-to-Inverter log file, it doesn't show the data that appears every time the app is running. It seems to log sporadic sections of data and I haven't been able to figure out what conditions need to be met for it to record into the log. For this reason I can't even show you what the latest logs of the app were because i can't copy and paste from terminal. The information is gone now.

In my rc.local file, I added code to initialise the CAN driver and to set the bitrate. Here's the code: 

sudo ip link set can0 type can bitrate 500000
sudo ifconfig can0 up
sudo ifconfig can0 txqueuelen 65536

Is it possible that the BMS-to-Inverter app has this already built into the code, which conflicts with this script happening upon start up? Maybe the app sees that can0 is already in use? Should I start the app with can0 down?

From what I can tell the RS485 to the rPi is working. When the app is running in terminal, I can see that it receives all the data including voltage, cell voltages, balance status, temperature, internal resistance and crap like that. The issue seems to be transmitting the data via CAN, the can numbers that it appears to send out don't change. A candump does show lots of data and ifconfig shows packets, but it seems to be the same numbers over and over again, which is maybe a null.

I went back to the Waveshare Wiki for the RS485 CAN hat and realised that i didn't have the python plugin for "can" downloaded. I did that and tested the can via python receive.py and send.py and it worked. I think this is just used for testing but maybe it had soemthing to do with this problem. I can see that when I execute the send and receive test via python, if can0 is up when it starts it will say "busy."

Excited to hear any ideas you may have with this information.

Thanks!

ironcowboysolar, 2024-09-05:
In response to number two: I understand about the logs folder and the logs file. What I was saying is that the file "BMS-to-Inverter.log" does not always log what happens in terminal as the program runs. If I open the file now it only shows logs from earlier that day and doesn't show any of the things I was seeing before I decided to give up.

ironcowboysolar, 2024-09-05:
Do you know what the pinout is for the SMA? They seem to show both CAN and SYNC. Which pins are you using?

ai-republic, 2024-09-06:
[image]

ai-republic, 2024-09-08:
Only connect the CAN_H and CAN_L. Do no connect CAN_GND or anything else. So basically you only use the wires of pin 4 & 5 for the RJ45 cable that goes to your SMA. And only those wires you connect to your RPi's CAN hat CAN H and CAN L respectively.

ironcowboysolar, 2024-09-09:
2024-09-08 20:15:40.708 | INFO  | Thread-5   | verter.BmsToInverter:294   | Sending to inverter SMA_SI_CAN on can0...
2024-09-08 20:15:40.709 | DEBUG | Thread-5   | verter.core.Inverter:152   | Inverter received: Buffer (HEX): [null]

2024-09-08 20:15:40.709 | INFO  | Thread-5   | InverterCANProcessor:66    | Sending SMA frame: Batt(V)=52.4, Batt(A)=-3.0, SOC=51.1
2024-09-08 20:15:40.709 | DEBUG | Thread-4   | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0xA5]

2024-09-08 20:15:40.710 | DEBUG | Thread-5   | verter.core.Inverter:173   | Inverter send: Buffer (HEX): [0x51, 0x03, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x40, 0x02, 0xA0, 0xF6, 0xA0, 0xF6, 0x78, 0x01]

2024-09-08 20:15:40.710 | DEBUG | Thread-5   | ocol.can.JavaCANPort:113   | CAN frame sending: Buffer (HEX): [0x51, 0x03, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x40, 0x02, 0xA0, 0xF6, 0xA0, 0xF6, 0x78, 0x01]

2024-09-08 20:15:40.710 | DEBUG | Thread-5   | verter.core.Inverter:173   | Inverter send: Buffer (HEX): [0x55, 0x03, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x33, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-09-08 20:15:40.710 | DEBUG | Thread-5   | ocol.can.JavaCANPort:113   | CAN frame sending: Buffer (HEX): [0x55, 0x03, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x33, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-09-08 20:15:40.710 | DEBUG | Thread-5   | verter.core.Inverter:173   | Inverter send: Buffer (HEX): [0x56, 0x03, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x78, 0x14, 0xE2, 0xFF, 0x00, 0x00, 0x00, 0x00]

2024-09-08 20:15:40.711 | DEBUG | Thread-4   | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0x01]

2024-09-08 20:15:40.711 | DEBUG | Thread-5   | ocol.can.JavaCANPort:113   | CAN frame sending: Buffer (HEX): [0x56, 0x03, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x78, 0x14, 0xE2, 0xFF, 0x00, 0x00, 0x00, 0x00]

2024-09-08 20:15:40.711 | DEBUG | Thread-5   | verter.core.Inverter:173   | Inverter send: Buffer (HEX): [0x59, 0x03, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-09-08 20:15:40.711 | DEBUG | Thread-5   | ocol.can.JavaCANPort:113   | CAN frame sending: Buffer (HEX): [0x59, 0x03, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-09-08 20:15:40.712 | DEBUG | Thread-4   | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0x95]

2024-09-08 20:15:40.713 | DEBUG | Thread-4   | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0x08]

2024-09-08 20:15:40.714 | DEBUG | Thread-4   | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0x06]

2024-09-08 20:15:40.715 | DEBUG | Thread-4   | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0x0C]

2024-09-08 20:15:40.716 | DEBUG | Thread-4   | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0x71]

2024-09-08 20:15:40.717 | DEBUG | Thread-4   | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0x0C]

2024-09-08 20:15:40.718 | DEBUG | Thread-4   | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0xD7]

2024-09-08 20:15:40.719 | DEBUG | Thread-4   | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0x0C]

2024-09-08 20:15:40.720 | DEBUG | Thread-4   | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0xD9]

2024-09-08 20:15:40.721 | DEBUG | Thread-4   | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0x00]

2024-09-08 20:15:40.722 | DEBUG | Thread-4   | s485.JSerialCommPort:200   | Received: Buffer (HEX): [0x8E]

2024-09-08 20:15:40.801 | DEBUG | Thread-2   | s485.JSerialCommPort:149   | Next frame: Buffer (HEX): [0xA5, 0x01, 0x95, 0x08, 0x01, 0x0C, 0xD7, 0x0C, 0xD5, 0x0C, 0xD7, 0x00, 0xEB]

2024-09-08 20:15:40.802 | DEBUG | Thread-2   | alyBmsRS485Processor:85    | RECEIVED: Buffer (HEX): [0xA5, 0x01, 0x95, 0x08, 0x01, 0x0C, 0xD7, 0x0C, 0xD5, 0x0C, 0xD7, 0x00, 0xEB]

2024-09-08 20:15:40.802 | DEBUG | Thread-2   | n.DalyMessageHandler:289   | BMS #1, Frame No.: 0, Cell No: 1. 3287mV
2024-09-08 20:15:40.803 | DEBUG | Thread-2   | n.DalyMessageHandler:289   | BMS #1, Frame No.: 0, Cell No: 2. 3285mV
2024-09-08 20:15:40.803 | DEBUG | Thread-2   | n.DalyMessageHandler:289   | BMS #1, Frame No.: 0, Cell No: 3. 3287mV
2024-09-08 20:15:40.803 | DEBUG | Thread-2   | s485.JSerialCommPort:149   | Next frame: Buffer (HEX): [0xA5, 0x01, 0x95, 0x08, 0x02, 0x0C, 0xD1, 0x0C, 0xD5, 0x0C, 0xD8, 0x00, 0xE7]

2024-09-08 20:15:40.803 | DEBUG | Thread-2   | alyBmsRS485Processor:85    | RECEIVED: Buffer (HEX): [0xA5, 0x01, 0x95, 0x08, 0x02, 0x0C, 0xD1, 0x0C, 0xD5, 0x0C, 0xD8, 0x00, 0xE7]

2024-09-08 20:15:40.803 | DEBUG | Thread-2   | n.DalyMessageHandler:289   | BMS #1, Frame No.: 1, Cell No: 4. 3281mV
2024-09-08 20:15:40.804 | DEBUG | Thread-2   | n.DalyMessageHandler:289   | BMS #1, Frame No.: 1, Cell No: 5. 3285mV
2024-09-08 20:15:40.804 | DEBUG | Thread-2   | n.DalyMessageHandler:289   | BMS #1, Frame No.: 1, Cell No: 6. 3288mV
2024-09-08 20:15:40.804 | DEBUG | Thread-2   | s485.JSerialCommPort:149   | Next frame: Buffer (HEX): [0xA5, 0x01, 0x95, 0x08, 0x03, 0x0C, 0xD7, 0x0C, 0xD7, 0x0C, 0xD5, 0x00, 0xED]

2024-09-08 20:15:40.805 | DEBUG | Thread-2   | alyBmsRS485Processor:85    | RECEIVED: Buffer (HEX): [0xA5, 0x01, 0x95, 0x08, 0x03, 0x0C, 0xD7, 0x0C, 0xD7, 0x0C, 0xD5, 0x00, 0xED]

2024-09-08 20:15:40.805 | DEBUG | Thread-2   | n.DalyMessageHandler:289   | BMS #1, Frame No.: 2, Cell No: 7. 3287mV
2024-09-08 20:15:40.805 | DEBUG | Thread-2   | n.DalyMessageHandler:289   | BMS #1, Frame No.: 2, Cell No: 8. 3287mV
[... cut ...]
2024-09-08 20:15:46.828 | DEBUG | Thread-2   | n.DalyMessageHandler:289   | BMS #1, Frame No.: 3, Cell No: 12. 3285mV
2024-09-08 20:15:46.828 | INFO  | Thread-2   | n.DalyMessageHandler:307   | Battery #2 voltages:
	#1: 3.285V
	#2: 3.288V
	#3: 3.288V
	#4: 3.283V
	#5: 3.287V
	#6: 3.284V
	#7: 3.283V
	#8: 3.287V
	#9: 3.287V
	#10: 3.288V
	#11: 3.289V
	#12: 3.285V
	#13: 3.287V
	#14: 3.289V
	#15: 3.288V
	#16: 3.187V

2024-09-08 20:15:46.828 | DEBUG | Thread-2   | s485.JSerialCommPort:149   | Next frame: Buffer (HEX): [0xA5, 0x01, 0x95, 0x08, 0x05, 0x0C, 0xD

ironcowboysolar, 2024-09-09:
In an above comment I posted a good chunk of a log. Aside from a HEX [null] in the logs every now and then, I don't see any indicator what's causing my problem. Every time I try to connect the rPi to the ComSyncIn port, nothing happens, and after 60 seconds I get ExtBMSTimeout error.

I've done the following: 

Tested continuity from the CAN H,L wires coming from the rPi to the RJ45 that plugs into the inverter. Tried flipping H and L. Tried instead using the SYNC_H and SYNC_L pins on the RJ45 that connects to the inverter. 

I completely deleted all of the files in my configurator folder, then I redownloaded the jar and set it up as a fresh install.

I also connected the can0 ouput on the rPi to a can tool that I use to read can data for fortress batteries on my laptop. I was able to verify that it is sending can data and was able to send from the rPi and receive on my laptop, although I cannot see what the data it's sending says.

I've also updated to the latest firmware on the Sunny Island inverter.

ai-republic, 2024-09-11:
So I see nothing wrong in the logs and if the data is being sent via `can0` then the inverter should be getting the BMS data. I guess you configured the SI to _Lithium battery_?
And the SI has the CAN terminating resistor RJ45 plugged in the second SYNC port?
[image]
You might need to terminate also the SYNC_H and SYNC_L on the RPi side of the cable. 
So the SMA terminating resistor will terminate both on the SI but SYNC_H and SYNC_L are not terminated on the loose end of the RJ45 going from SI to RPi. The CAN_H and CAN_L should be terminated by the CAN hat on the RPi.

ironcowboysolar, 2024-09-19:
No matter what I do, I cannot get the SMA to do anything with the CAN data. I'm very familiar with SMA inverters, down to the FET's and other components.

I have the inverter set to LI_Ion_ExtBMS. I have a special cable I made with a twisted pair landing on the CAN_H and CAN_L pins, and another twisted pair landing on the SYNC_H and SYNC_L pins - also with a GND on the ComSyncIn port that connects to my rPi.

The CAN_H, CAN_L, and GND wires land on the rPi in the corresponding terminals, and as per your last message, there is a 120 ohm resistor soldered to the SYNC_H and SYNC_L wires on the rPi side of the cable.

I have a fresh install of the latest configurator. Upon initializing start.sh, I can see all of the data from the BMS being received by the Raspberry Pi, I can see it sending SMA frames, but when I turn the SMA on, the SOC says 0%, and there is no temperature information or anything else. After about a minute I receive an ExtBMS_Timeout error and the system shuts down.

I've tried literally everything such as: the latest SMA firmware, unplugging Slave 1 inverter and terminating the port that it was connected to. I've tried flipping CAN_H and CAN_L. I've tried removing the on board terminator jumpers on the CAN hat.

Since then, I've built a separate 16S battery with a Daly BMS I ordered straight from them with CAN enabled, which is connected to a separate inverter (two separate projects). I set up a separate rPi with the DALY_CAN binding for this battery and just like the DALY_RS485 battery, I can see it reading data from the BMS, I can see it sending CAN frames to the SMA, but in both situations, the SMA inverters act the same way. No data showing up from the BMS and an error popping up after a minute or so.

On both projects, I've connected CAN port designated for sending to a CAN reader and confirmed that the rPi is sending CAN data at a baud rate of 500k. After confirming a data stream, I've wired a RJ45 connector on the end of it and plugged it directly into the ComSyncIn port.

Is there something I'm missing? Can you please share exactly what you did to make your Daly BMS communicate with the Sunny Island? Are you using a different firmware?

I've learned so much about communication protocols, raspbian and lots of useful things, but I'm still taking an L on the end result. 

Please help.

kommando828, 2024-09-20:
I am successfully running a Master/Slave setup of two SI5048's set to lithium connected to a JK PB BMS via canbus using the SMA canbus protocol. This is the protocol the JK references as the SMA it uses but I have also set it to Pylontech and that works too.
[SMA CAN protocol.pdf](https://github.com/user-attachments/files/17072093/SMA.CAN.protocol.pdf)

ironcowboysolar, 2024-09-20:
> I am successfully running a Master/Slave setup of two SI5048's set to lithium connected to a JK PB BMS via canbus using the SMA canbus protocol. This is the protocol the JK references as the SMA it [...]

If  you look at my log above vs the SMA protocol in your link, all the necessary data that the SMA requires is represented like this [ 0x123 ] with three digits after the X but the data my rPi is sending has only two digits or characters after the X, for example [ 0x12 ]

What do your debug logs look like when the system is running? Just for what it's worth, I used the SMA can protocol to set up my system, but it's still yet to work.

ai-republic, 2024-09-21:
> I am successfully running a Master/Slave setup of two SI5048's set to lithium connected to a JK PB BMS via canbus using the SMA canbus protocol. This is the protocol the JK references as the SMA it [...]

Yes, that is the protocol specification that is implemented

ai-republic, 2024-09-21:
@ironcowboysolar I will try to implement the non-mandantory sections of the SMA protocol to see if SMA also treats some of them as mandantory (just to prevent people using cheap battery systems 😉 )

kommando828, 2024-09-21:
The protocol in the PDF I posted is for the SI X.0H-11, however my SI5048 running SI6048US firmware 7.304 and the JK version works as does the JK version of Pylontech, Goodwe and Deye. If your SI is a -12 or -13 then maybe they changed the protocol, which would be odd as Pylontech batteries are compatible with all Gens of SI but stranger things have happened.

What I have noticed is that taking out the terminal resistor from ComSyncOut kills canbus comms dead, so maybe the RJ45 plug with resistor you have in your last SI RJ45 ComSyncOut socket is not working.

ai-republic, 2024-09-21:
> The protocol in the PDF I posted is for the SI X.0H-11, however my SI5048 running SI6048US firmware 7.304 and the JK version works as does the JK version of Pylontech, Goodwe and Deye. If your SI is [...]

No, the protocol is still the same. The Pylontech protocol is similar but not quite the same as the SMA. There is no option to select a protocol version in the SI and the documentation states explicitly that it only works with this protocol (that doesn't mean that Pylon won't work if its not to strict)

Yes, please double check all wirings and terminations

ai-republic, 2024-09-29:
Hi @ironcowboysolar,
I've think I found the issue!!! The SMA uses CAN 2.0A 11-bit can-id and my SMA binding was sending it with the 29-bit CAN 2.0B can-id.
I think that will fix your issue, but I will also implement the other CAN commands for completeness 😄

ai-republic, 2024-09-29:
Right, I added the manufacturer code and battery type and -capacity settings also. Let me know how it goes 😄

ironcowboysolar, 2024-10-06:
> Hi @ironcowboysolar, I've think I found the issue!!! The SMA uses CAN 2.0A 11-bit can-id and my SMA binding was sending it with the 29-bit CAN 2.0B can-id. I think that will fix your issue, but I will also implement the other CAN commands for completeness 😄

Freaking sweet! Is it ready to download and install?

Thanks for your continued support.

ai-republic, 2024-10-06:
@ironcowboysolar yes. You'll need to get the latest _Configurator_ application and do a clean install. I you choose an existing installation folder it load your existing configuration file `config.properties`. Just delete the libs folder prior to a _Clean install_  just to be sure you have a fresh installation

ai-republic, 2024-10-31:
@ironcowboysolar did you get a chance to test everything? How's everything working out?

ironcowboysolar, 2024-10-31:
I was able to get the app to work enough so that the inverter stays on. Felt great. There's a few bugs that need to be worked out but for the most part at least its running. Thanks for your help!

A few things I noticed that were strange: It doesn't let me manually change the battery charge values. Does the inverter get this information from the BMS? Because I never set voltages for the Boost, Absorb, etc charges on the BMS, only low battery disconnect and high battery disconnect. The default charging settings on the SMA are based for a (24) cell lead acid battery, so you have to take the desired charge voltage voltage and divide it by 24 to be able to input the parameter on the SMA, the default setting is @2.4v per cell (57.6v) and I need to lower that charging value to about 55.4v @2.3v per cell because 57.6v is the cutoff voltage on the BMS. When I try to change the setting it defaults back to 2.4v immediately after changing it.

What happens is when the inverters charge the battery, the voltage quickly climbs to 56 and then past 57, triggering the cutoff on the BMS.

The second issue may or may not be related to your app. I have an AC coupled inverter connected to the SMA which has been working in open loop mode (VRLA), but since I've had it running on closed loop, the inverter frequency seems to stay shifted up which is keeping the AC coupled inverter throttled down. In a nutshell in closed loop the Sunny Island inverter is not allowing the AC coupled inverters to send it power by keeping the frequency above a certain level. Even at 30-40% SOC the GT inverter never reaches a frequency low enough to be allowed to activate.

I'm going out there today to see if the AC coupled inverter still works if I put it back in closed loop and I'm going to play with the settings on the BMS to see if I can get the charging settings to work better.

Thanks for all your help. Interested to hear any tips or tricks you used to get your system dialed in.

kommando828, 2024-10-31:
With the SI set to Lithium and successfully receiving Canbus from a BMS then the SI only takes the target V to charge up to from the BMS, you may be able to change it as I can on my SI5048 but 5 mins later it is back to the BMS target V. There is one wrinkle to this, when the SI is off grid then as the Frequency Shifting to curtail the PV output lags then the target V is exceeded so SMA reduce the SI Target V by 1V when the SI is off grid, when the SI is on grid then excess can be sent to the grid so the 1V is added back.

So as I run 100% off grid my BMS target V for a 48V nominal Lithium is set to 56,2V and the SI displays its target V as 55.2V.

SI SOH is always 100%, even if the BMS reports 100% SOC at a low V the SI should still charge to the Target V.

There is also a rare bug whereby in Lithium mode the SI frequency goes out of control and shuts the PV inverters down before the Target V. My setup has never exhibited this at all, only been told of one instance. Only fix so far is to revert to SI lead acid and go open loop comms.

ironcowboysolar, 2024-10-31:
> With the SI set to Lithium and successfully receiving Canbus from a BMS then the SI only takes the target V to charge up to from the BMS, you may be able to change it as I can on my SI5048 but 5 mins later it is back to the BMS target V. There is one wrinkle to this, when the SI is off grid then as the Frequency Shifting to curtail the PV output lags then the target V is exceeded so SMA reduce the SI Target V by 1V when the SI is off grid, when the SI is on grid then excess can be sent to the grid so the 1V is added back.
> 
> So as I run 100% off grid my BMS target V for a 48V nominal Lithium is set to 56,2V and the SI displays its target V as 55.2V.
> 
> SI SOH is always 100%, even if the BMS reports 100% SOC at a low V the SI should still charge to the Target V.
> 
> So instead of a 24 cell V on Boost, Full and Float the SI works to this parameter
> 
> Parameter BatChrgVtgMan 55.2V
> 
> There is also a rare bug whereby in Lithium mode the SI frequency goes out of control and shuts the PV inverters down before the Target V. My setup has never exhibited this at all, only been told of one instance. Only fix so far is to revert to SI lead acid and go open loop comms.

I get that there would be some discrepancies with the frequency, but when I was running off of closed loop I looked at the screen on my Sunny Boy inverter and it said the frequency was at 69 hz with the battery at 30%. I’m not sure why the SI would respond in that way but due to being AC coupled, the system is still non-op in closed loop.

I’m very happy to have made progress though.

kommando828, 2024-11-01:
Until there is a fix for this Frequency shifting bug then stay on open loop. If you tell the SI the battery is 46V initially then you go lower than 2.3V per cell so the SI does not go too high on V. 

There is a ticket into SMA on the Freq jump on lithium and also user testing. He has a number of SI's and only 2, which were linked as master slave, are affected, he has others that work and my master/slave works too.

Which SI do you have, the faulty pair are SI6048US's.

kommando828, 2024-11-02:
The high frequency bug fix on an SI6048US is to completely turn it off, remove all power connections using the breakers and wait at least 30mins. This discharges the capacitors so the boards are no longer powered and the temporary memory chips are wiped clean. You can then reboot and the memory is refreshed with good settings from the longer term memory chip. This needs to be done simultaneously on all connected SI's just in case one of the slaves is not updated properly from the master on reboot. As the capacitors will be empty you will need to set up a scheme to cope with the inrush current if the BMS does not recognise it as being different from an accidental short and shutdown.

ai-republic, 2024-11-03:
Hi guys,
great to have this discussion - hope you can help each other out on the configuration side. I have a SI 8.0H-13 (european version) but currently not in use. I upgraded to a 16kW Deye as I didn't want to spent more on a second SI + inverter.

@ironcowboysolar if you want to manipulate the data from the BMS, i.e. max voltage limit, you could also try to use the _ManipulateBatteryPackDataPlugin_ inverter plugin.

Happy to hear the app is working well for you 😄

ironcowboysolar, 2024-11-20:
@ai-republic Did your Deye inverter get locked recently? I've seen some sources say that non Sol-Ark Deye inverters have been remotely deactivated by the company in China because selling them on the US market in competition with Sol-Ark is a breach of contract. If it hasn't been bricked, be careful connecting it to the internet.

On a separate note, I'm having a strange issue and I would like to recruit your help, I'm willing to send a donation that is equivalent to your normal rate. As detailed in the comment a few weeks ago, in Li_Ion_EXT_BMS mode, the frequency walks all the way up to 70hz, which is actually high enough to damage any connected electronic devices. I've done some further troubleshooting, and even when the SMA is in VRLA mode (open loop), if I plug in the Raspbery Pi with the app running the frequency begins the walk up on the inverter. When this happens, it will stay at 70hz, even if I unplug the Raspberry Pi. After this, even with the Pi disconnected, if I put the inverter into stand-by and restart it, it will not fix the frequency issue. The only way to correct the issue is for me to turn the master inverter completely off and then turn it back on without the Pi connected.

This tells me that there's something in the CAN data that causes the software on the inverter to react this way, even if the inverter isn't in closed loop mode.

> Which SI do you have, the faulty pair are SI6048US's.

@kommando828 These are the SI6048's 

> The high frequency bug fix on an SI6048US is to completely turn it off, remove all power connections using the breakers and wait at least 30mins. This discharges the capacitors so the boards are no [...]

@kommando828 Tried this and it didn't correct the issue. The frequency walk begins as soon as the RPi with the CAN data is connected in ANY battery mode.

I will attach a few photos and videos that show the issue.

kommando828, 2024-11-21:
That looks similar to the other faulty SI6048, trouble is the turning off, waiting for total power down and then turning back on was the only fix tried and it worked first time. So any further possible fixes are untried, untested and so potentially could make the situation worse. If you are ok with that risk then this is how I would proceed.

First I would make sure the SI is on the latest firmware and upgrade if necessary.

Latest is FW 1 7.304 FW2 7.300

Second I would redo the turning off of the SI but leave it turned off for a lot longer ie several hours. It worked on one machine so its worth trying again just in case a residual charge kept the setting unchanged.

If its still not working then these are the untried untested items to try. Note 1 relies on you having one good SI, if you don't have access to a working SI then you cannot try this one out. No point in me sending my version as mine are based on 230v 50hz and would mess your SI up, you need a set from a fellow SI6048US user already on the latest firmware but not necessarily on Lithium.

The fault is likely one of these three.

A. Firmware corruption on the SI chip.

B. Parameters gone bad

C. Hardware issue

1. If you have a good supply of SD cards you can try the following to test B. note that for now do not connect any good SI's with bad SI's as a bad master will potentially corrupt currently good slaves.

On your good master, use the menu to copy the parameters to the SD card using 510.02 ParaSto and copy the parameters to set 1 and set 2. 

Take the SD card out and make a copy, put the original back into the good SI..

Now separately without connection to any good SI's fire up the defective SI and set it up as your good SI with the copied SD card in the slot, then as soon as its fired up go into 510.02 and load in from the SD card either set 1 or 2.

If that fixes the issue then it was a parameter issue. If not then it's back to 1 or 3.

To test A then you need to reload the firmware. This is the procedure for going back to a previous firmware, my guess is it will also work to reload the current firmware completely and hopefully overwrite any corruption.

Description
The article describes the firmware update for Sunny Island SI5048.
Note: An update / downgrade always results in the loss of all parameter settings.

Procedure
Save the Update.bin file on the SD card of the Sunny Island.
Switch off (disconnect DC) the Sunny Island and wait approx. 30 seconds (until the display is completely dark).
Insert the SD card into the Sunny Island.
Start the Sunny Island (connect DC).
When starting, press and hold the down arrow key.
As soon as "Update 1/2 Erase" appears on the display, the down arrow key can be released.
When the Sunny Island goes into standby mode, the update has been carried out.
Afterward, check the parameters #312_02 and #312_08 (must correspond to Update.bin).
The Sunny Island can be started again.

Then setup the SI and test.

If that does not work then it makes 3 most likely, what component  it could be is beyond my knowledge, it would have to be a component that works in lead acid but not in Lithium.

ai-republic, 2024-11-23:
@ironcowboysolar my Deye didn't get locked but then again I'm not residing in the US. 
I had/have (not currently in use) a SI 8.0H-13 but never observed this behavior, especially that the frequency goes up above 61Hz. I had mine configured for to use the frequency shifting to regulate the Sunny Boy to lower/cut off its output.
Have you tried configuring the max Hz for the frequency shifting? Maybe it think the batteries are fully charge and trying to tell the PV inverter to lower its DC output. I'm not sure what the default max setting is on your SI6048US inverters but be sure to check those.
Otherwise I would suggest to try @kommando828 solutions.

Calvin983, 2024-12-03:
Interesting video. I see that as soon as you plug it in the frequency starts to ramp up. Mine was doing something completely different. My system had been running off grid trouble free for several months. Then one morning about 7 am the UPS power supplies all started beeping at the same time and all the items are running off of the battery. Eventually, the batteries ran out and everything on those power supplies shut off. The only reason I heard or discovered there was a problem was because the UPS power supplies were all beeping at the same time. There was still 120/240 V power available so I did not know what caused the trouble. Somehow I managed to look at the frequency and saw that it was 70 Hz. I put info on the DIY solar forum. Spent a week trying to troubleshoot it and then the end discovered that removing all power sources and checking 0.0 V on both the AC and the DC solve the problem. Once all the power sources removed and the Sunny Island sat there for half hour or so then turning it back on it ramped up to normal 60 Hz. Before removal of power sources, the problem persisted in all 4 Sunny Island’s. Regardless of the setting and which one was the master and which ones was the slave they always ramped up to 70 Hz. I did try to set it to VRLA settings that also solved the problem. The Sunny Island ran at 60 Hz in open loop without communication. This seems to be different than what you experienced. But the only way to get the communication back up and running close loop was to actually remove the power sources. Still no idea what caused the trouble. SMA says it’s a safety issue so the SI does not charge the battery. Since it happened at 7 AM in the morning, the batteries were nowhere near 100% full. The Sunny Boys were also not producing any power at the time and moments before the frequency was 60 Hz. The error happened just around the time the sunny boys were waking up, but still certainly not producing a lot of power at all.

Note. Turning the JKPBBMS off at the switch does not actually mean that there’s 0 V dc on it. Somehow voltage still leaks through there and only physically disconnecting the DC wires from the Sunny Island let everything reset back to normal.

Like Kommando says, maybe it’s some sort of memory issue. Maybe between your experience in my experience, we can narrow it down a little? I have the SI logs from that time and need to see if anything unusual happened. Since you are working on the programming, maybe you can find somethgin there that causes it and resolve it?

I did try to put a brand new Sunny Island on the same battery with lithium close loop settings and that particular unit ran perfectly fine with the same battery. So it does not seem to be the battery that is the cause of the problem. It was only the other Sunny Islands that had the problem. The problem seem to come out of nowhere.

Calvin983, 2024-12-03:
[Si 70Hz Log 702 am 10-28-24.xlsx](https://github.com/user-attachments/files/17996085/Si.70Hz.Log.702.am.10-28-24.xlsx)
Here is the SI log file in case it helps. Error happened at 7:20 AM, Oct 28. 
Sunny Boys were also just starting to wake up and produce power at that time.

kommando828, 2024-12-03:
Note that like myself Calvin uses JK PB BMS series and has these in closed loop with the SI, no need for an intermediary. The JK PB BMS has a number of Canbus Protocols such as Pylontech etc but also one called FSS-ConnectingBat-TI-en-10 which when you look at the SMA SI protocol PDF I posted is the footer entry. When you select this on the JK then you get a working closed loop with an SMA SI but with an extra decimal on the figures eg SOC is 88.8% not 88% over Pylon which fits in with the PDF. I have never had the 70hz bug but the possibility is that the PDF has an error. IE two coders using the same document of the protocol but on different platforms cause the same issue then the smoking gun is the document.

ironcowboysolar, 2024-12-11:
Hey @ai-republic on new builds with Daly BMS, in DALY_CAN mode, I keep getting this error:

2024-12-11 12:18:49.339 | INFO  | main       | til.SystemProperties:22    | Loading config.properties from: /home/pi/configurator/config/config.properties
2024-12-11 12:18:49.472 | DEBUG | main       | ging.LoggerProviders:152   | Logging Provider: org.jboss.logging.Log4j2LoggerProvider
2024-12-11 12:18:51.785 | INFO  | main       | ore.InverterProducer:76    | Created inverter binding: SMA_SI_CAN
2024-12-11 12:18:51.906 | INFO  | main       | verter.core.Inverter:53    | Using plugins: []
2024-12-11 12:18:51.934 | INFO  | main       | core.BMSListProducer:105   | Intialized BMS #1[DALY_CAN] on port can0
2024-12-11 12:18:51.952 | INFO  | main       | verter.BmsToInverter:261   | Starting BMS receiver...
2024-12-11 12:18:51.958 | INFO  | Thread-2   | verter.BmsToInverter:268   | Reading BMS #1 DALY_CAN on can0...
2024-12-11 12:18:51.960 | INFO  | Thread-2   | toinverter.core.Port:111   | Opening can0 ...
2024-12-11 12:18:52.062 | INFO  | Thread-2   | toinverter.core.Port:113   | Opening port can0 SUCCESSFUL
2024-12-11 12:18:52.066 | DEBUG | Thread-2   | .DalyBmsCANProcessor:46    | SEND: Buffer (HEX): [0x40, 0x01, 0x50, 0x18, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-12-11 12:18:52.069 | DEBUG | Thread-2   | ocol.can.JavaCANPort:115   | CAN frame sending: Buffer (HEX): [0x40, 0x01, 0x50, 0x18, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-12-11 12:18:52.073 | DEBUG | Thread-2   | ocol.can.JavaCANPort:76    | CAN frame read...
2024-12-11 12:18:52.076 | DEBUG | Thread-2   | ocol.can.JavaCANPort:82    | CAN read frame Buffer (HEX): [0x01, 0x80, 0x06, 0x04, 0x08, 0x00, 0x00, 0x00, 0x01, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-12-11 12:18:52.079 | DEBUG | Thread-2   | .DalyBmsCANProcessor:75    | RECEIVED: Buffer (HEX): [0x01, 0x80, 0x06, 0x04, 0x08, 0x00, 0x00, 0x00, 0x01, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-12-11 12:18:52.081 | WARN  | Thread-2   | .DalyBmsCANProcessor:94    | Message has wrong address and command id: Buffer (HEX): [0x01, 0x80, 0x06, 0x04, 0x08, 0x00, 0x00, 0x00, 0x01, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-12-11 12:18:52.083 | DEBUG | Thread-2   | .DalyBmsCANProcessor:46    | SEND: Buffer (HEX): [0x40, 0x01, 0x53, 0x18, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-12-11 12:18:52.086 | DEBUG | Thread-2   | ocol.can.JavaCANPort:115   | CAN frame sending: Buffer (HEX): [0x40, 0x01, 0x53, 0x18, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-12-11 12:18:52.088 | DEBUG | Thread-2   | ocol.can.JavaCANPort:76    | CAN frame read...
2024-12-11 12:18:52.091 | DEBUG | Thread-2   | ocol.can.JavaCANPort:82    | CAN read frame Buffer (HEX): [0x01, 0x80, 0x07, 0x04, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-12-11 12:18:52.094 | DEBUG | Thread-2   | .DalyBmsCANProcessor:75    | RECEIVED: Buffer (HEX): [0x01, 0x80, 0x07, 0x04, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

2024-12-11 12:18:52.096 | WARN  | Thread-2   | .DalyBmsCANProcessor:94    | **Message has wrong address and command id**: Buffer (HEX): [0x01, 0x80, 0x07, 0x04, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

~~~~~~~~~~~~~~~~~~~~~~~~~~~~

2024-12-11 12:18:52.099 | DEBUG | Thread-2   | .DalyBmsCANProcessor:46    | SEND: Buffer (HEX): [0x40, 0x01, 0x5A, 0x18, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-12-11 12:18:52.101 | DEBUG | Thread-2   | ocol.can.JavaCANPort:115   | CAN frame sending: Buffer (HEX): [0x40, 0x01, 0x5A, 0x18, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-12-11 12:18:52.104 | DEBUG | Thread-2   | ocol.can.JavaCANPort:76    | CAN frame read...
2024-12-11 12:18:52.106 | DEBUG | Thread-2   | ocol.can.JavaCANPort:82    | CAN read frame Buffer (HEX): [0x01, 0x80, 0x08, 0x04, 0x08, 0x00, 0x00, 0x00, 0x10, 0x02, 0x00, 0x01, 0xB0, 0x6C, 0x00, 0x00]

Message has wrong address and can id.

[12 more replies]

closed 2025-01-05 by ai-republic