ai-republic/bms-to-inverter #35: Selectronic SP Pro

gus1234sudo, 2024-06-24:
Hi. Total noob here (the extent of my knowledge was putting Victron VRM on pi3B). 

I have a selectronic inverter (I am off grid) and about to replace ABB with Fronius AC couple). I will have upto 3 fronius inverters on different arrays. 

Apparently selectronic supports pylontech but unsure as contacted Batrium and they could not guarantee their bms would work so went with the jk inverter bms. 

Do you know if your programming supports selectronic? 

In the future I would like to get information on home assistant. 

Thanks for your time. 

G

ai-republic, 2024-06-25:
Hi,
if the Selectronic inverter can 'speak' the Pylon protocol then it should be no problem. Otherwise I would need a CAN communication specification for the Selectronic.

And your setup would be the JK BMS -> Selectronic inverter right? Or JK -> Fronius? I got a bit confused reading ;)

gus1234sudo, 2024-06-25:
Sorry, a lot of inverters. JK to selectronic is what I am looking at. When I contacted Batrium about their bms (they advertise compatible with inverters that speak pylontech) they said they had issues with selectronic not working and wouldn’t guarantee it would work. Where would I find a can communication specification? Sorry, I did say I am a noob :)

thanks for your prompt reply

ai-republic, 2024-06-30:
Ah ok, so you can test if you get reading using the JK BMS binding (choose JK_CAN in the Configurator app). For the test please enable debug log-level and as inverter choose NONE for now.
Which CAN hat do you have for your Raspberry PI? If none yet, I would suggest the Waveshare 2CH CAN hat as you'll be needing 2 can ports (one from BMS to RPI and one from RPI to inverter).

NOTE: We have discovert that the Waveshare 2CH CAN hat does not (currently) work with a RPI 5. If you want to buy a PI for this project I would suggest to go with the RPI 4B for now.

ai-republic, 2024-08-06:
@gus1234sudo  Did you test the Pylon on your Selectronic inverter? Or did you give up your project?

ai-republic, 2024-08-09:
Closing this issue since I have not heard anything for a long time. Please create a new issue if you encounter any problems.

closed 2024-08-09 by ai-republic

mattthemadscientist, 2024-11-25:
I'm currently using this with a selectronic 2i, still ironing out the details, as I'm not seeing battery communication on the selectronic side (I suspect it's just a wiring issue). did you ever get it working @gus1234sudo?

It's worth noting that the pylontech protocol isn't very comprehensive and doesn't transmit individual cell details.