ai-republic/bms-to-inverter #36: Support for deye battery

sarzyk, 2024-06-25:
Hi is any chance to make possibility communication between JK BMS (PB version) and Deye RW-M6.1 battery via RS485 or can?

Thank for reply.

ai-republic, 2024-06-30:
Hi, 
I'm not sure what you want to do. The Deye RW-M6.1 is a battery system with its own BMS. What exactly do you want with the do with the JK BMS? Do you want to connect 2 batteries (the Deye and another via JK)? 

If so, yes, it is possible to use different BMSes with this application. You will need a separate CAN port for different BMS types. So you could for example run 5 JK BMSes on one CAN port and the Deye on another.

sarzyk, 2024-07-04:
Hi. Thank for reply.
I wish that :
In deye set charge and discharge maximum for 80A and for the JK for 140A.

ai-republic, 2024-07-06:
I think you can set the max charge/discharges settings with the software that comes for your JK and Deye BMS

ai-republic, 2024-08-09:
Closing this issue since I have not heard anything for a long time. Please create a new issue if you encounter any problems.

closed 2024-08-09 by ai-republic