ai-republic/bms-to-inverter #37: support for Solis Inverter with Sacred Sun lithium Battery

jan-dane, 2024-07-06:
Hi. Nice work helping needy people.

I need help in connecting Sacred Sun Battery (SCIFP48100 model) having only RS232 and RS485 ports with Solis Hybrid inverter 6KW (S6-EH1P6K-L-Pro). But the inverter communicate through CAN port only for BMS.
How to connect battery RS485 port data to Inverter CAN port? .. please help.

Sacred Sun Battery Specs:  https://www.sacredsun.com/uploads/soft/221111/1-221111152351.pdf
                          Manual:   https://www.sacredsun.com/uploads/soft/201211/1-201211100Q0.pdf

Solis Inverter compatible battery data: https://www.solisinverters.com/uploads/file/Solis_Leaflet_Battery_matching_S6-EH1P%283-6%29K-L-PRO_V2%2C3_202406.pdf

Solis Inverter Mannual: (page # 24~27)   https://www.solisinverters.com/uploads/file/Solis_Manual_S6-EH1P%283-8%29K-L-PRO_EUR_V1%2C1%2820240311%29.pdf

Solis Inverter with batteries communication : https://solis-service.solisinverters.com/en/support/solutions/articles/44002484785-communication-ports-for-battery-connection

ai-republic, 2024-07-07:
Hi,
I would need the RS485 communication protocol specification for the Sacred Sun BMS. The Solis understands the PylonTech CAN protocol so that's no problem.

To setup a system with this application you will need a Raspberry PI with a RS485/CAN hat.

jan-dane, 2024-07-17:
Thx for reply...
I don't know exactly about the RS485 communication protocol specification for the Sacred Sun BMS. However, battery can be connected/accessed to PC via RS485 USB converter (as shown in attachment) with following BMS softwares in window10.

https://energytalk.co.za/uploads/short-url/6i0wuNTGFnHFxXLpCFh1D2EIZbl.zip

http://www.crystalradio.cn/forum.php?mod=attachment&aid=MTY0MjYxMDl8ZTJhOTcwZTV8MTcxODQzNTg3OHwwfDIwODk0NjA%3D

PC Connection setup:   https://www.youtube.com/watch?v=IOAFlZLxnPE

[image]

jan-dane, 2024-07-17:
Another software checked working with the same battery.

https://ucca98f13d7d95569659bd8b52f4.dl.dropboxusercontent.com/cd/0/get/CUXcyIOqGSh3JIbVUO9HHQ7iAnENCS63EgUCgwH1-DmMIFAFE6HDO89tw-jRVjGUyEKPRNw5OMuQDmWzaDp0tajW2fEJgi_79oCzZ4EAVc-3JTYuxIzwi78i-ZKAhIbo3s5AnfdUo34vJ8N6KeNIvYfi_fgTd6qQImPQYluFkJlMtg/file?_download_id=6296857176032198077251404144398358272023653671255062195367987459&_notify_domain=www.dropbox.com&dl=1

ai-republic, 2024-07-19:
Ok, after a little (or long) research I found out that your BMS seems to be a rebranded Tian Power BMS that seems to use the same RS485 protocol as the EG4 LiFePower battery packs. What I've found so far is a crude implementation and some reverse engineered documentation on the protocol. I did find a EG4 LiFePower RS485 protocol which some comments believe it to be buggy but that would be a starting point. There is also a user @TianPower here on Github who might help if you give him the serial number of your BMS.

ai-republic, 2024-09-29:
I guess this can be closed. Please open another ticket if you need further assistance.

closed 2024-09-29 by ai-republic