ai-republic/bms-to-inverter #38: Powmr Hvm pro 6.2 kw and jk bms (black one with 2amps balancer)

flyzet-prog, 2024-07-09:
Unfortunately, in this model no settings to activate communication with battery. But it has wifi dongle out(rs232 maybe), which i use to pullout data from inverter to the home assistant over esp32(syssi github project)
In the Powmr HVM 6.2M-48-K it’s already has that feature to connect bms with different protocols. 
My model Pro is pretty similar to this one, the only difference that my model has detachable display.

There is any way to connect bms and communicat over rs232?
Much appreciated for any help.
[image]

[image]
[image]
[image]

ai-republic, 2024-07-14:
Hi,
sorry to answer so late, but I needed to research a little. I did find the specification for the PACE and PYLON RS485/RS232 communication protocol so it should be no problem to create a binding for that.
I will see that I can start the implementation for that maybe today, but I'm a little busy at the moment. I hope to get it finished on the next weekend :)

flyzet-prog, 2024-07-14:
> Hi, sorry to answer so late, but I needed to research a little. I did find the specification for the PACE and PYLON RS485/RS232 communication protocol so it should be no problem to create a binding [...]

So, you believe i can properly make communication to bms even if there is no such option on inverter to chose some protocol? Much appreciated for you reply, hope you can help me to resolve my problem. Thanks

ai-republic, 2024-07-19:
Yes, your JK BMS should be able to communicate via RS485 or CAN depending on your version. This should already be supported by the _JK_485_ or _JK_CAN_ binding respectively for the BMS in the _Configurator_ app. Then for the inverter binding choose _PYLON_RS485_ as binding - once I have implemented it this weekend ;)

flyzet-prog, 2024-07-19:
> Yes, your JK BMS should be able to communicate via RS485 or CAN depending on your version. This should already be supported by the _JK_485_ or _JK_CAN_ binding respectively for the BMS in the [...]

I am planing to use Jk-Pb2a16s20p. But my inverter not uses rs485, it's use rs232 communication. Thank you for you reply, i hope i can use your integration to properly communicate jk bms with my inverter.

ai-republic, 2024-07-26:
It should be ok using the RS485 binding as the RS232 of your inverter also only communicates via serial interface over the 2 wires. Its basically the same.

flyzet-prog, 2024-07-26:
> It should be ok using the RS485 binding as the RS232 of your inverter also only communicates via serial interface over the 2 wires. Its basically the same.

Ok, please let me know when you will create porting to my inverter i can test it out.

ai-republic, 2024-08-09:
@flyzet-prog sorry, just stumbled over your issue again. The _PYLON_RS485_ binding is now ready for BMS and inverter.
Please let me know if you would like to test it!

ai-republic, 2024-09-29:
@flyzet-prog any news on your project? Or can I close this issue?

ai-republic, 2024-10-04:
I guess I can close this

closed 2024-10-04 by ai-republic