ai-republic/bms-to-inverter #40: Jk BMS to Solplanet Inverter

Lino34s, 2024-07-17:
Hey man! Kudos for this amazing project.
Is there any chance on getting my new Nissan leaf battery pack with jk bms to communicate with my asw3680 hybrid inverter via rs485

I guess the inverter doesn't have Can because I had it linked flawlessly with my pylonteck bank via rs485 and I just can't find any reference on the product manual either

ai-republic, 2024-07-19:
Hi,
yes, I'm going to start implementing the RS485 PylonTech protocol this weeking, so if you use your JK BMS and your inverter can communicate via the Pylon RS485 protocol then it should be no problem.

ai-republic, 2024-07-21:
Sorry, didn't get round to it this weekend, but I see what I can do until next :)

Lino34s, 2024-07-21:
Hey Torsten,

I guess I figured out my Solplanet inverter also supports CAN communication.

Do you think I’ll be able to get them to communicate? Because JK BMS also
emulate pylonteck, for example, and I didn’t get

Thank you very much,
Diogo Lino

ai-republic, 2024-07-21:
So your JK does CAN and also the inverter right? And both 'speak' the PylonTech communication specification?

Lino34s, 2024-07-21:
Yes they does, because first, until a couple months ago,  I had my
Solplanet inverter fully integrated with a 5kwh pylonteck bank (by rs485).
I just did a 18kwh powerwall with jk BMS from my old Nissan leaf

Com os melhores cumprimentos,
Diogo Lino

ai-republic, 2024-07-26:
In that case, yes my application can communicate with both your BMS(es) and inverter :)

Lino34s, 2024-07-29:
I would like to take this opportunity to include the document with the communication protocol for my Solplanet ASW3680H-S2 Inverter

[ASW3000~6000H-S2 communication_protocol (CN).pdf](https://github.com/user-attachments/files/16418457/ASW3000.6000H-S2.communication_protocol.CN.pdf)

ai-republic, 2024-08-02:
Thanks for the communication specification. As your Solplanet inverter supports the PylonTech RS485 protocol I would recommend to use this. I'm still implementing the binding but hope to finish it this weekend

Lino34s, 2024-08-02:
Perfect!

I'm also waiting for the rs485 Can Hat for RaspPi to come in the next few days. I guess I might need some little help on the assembly 

Thanks a lot man

ai-republic, 2024-08-04:
@Lino34s  the Pylon RS485 binding is now implemented. Please get the new _Configurator_ , choose _PYLON_RS485_ in the inverter tab and do a _Clean install_.
Please also configure the debug log-level so that I can check if there are any prolems.
Happy testing ;)

Lino34s, 2024-08-17:
Can you please provide me with an image of the CAN Hat/Inverter setup, including the respective cabling and connections?

ai-republic, 2024-08-18:
Hi @Lino34s, 
this is different for each BMS, CAN hat and inverter depending on your model. But in the manuals you should find the corresponding plugs and PIN layout for the connections. Then you just need to connect CAN H to CAN H and CAN L to CAN L with the correct pins. 
In general like this:
[image]

Lino34s, 2024-08-26:
Sorry. Is that the 1CH or 2CH CAN HAT? I just bought the 1xCH one since I thought of using CAN on the BMS side and rs485 on the inverter side. Am I correct?
I am currently preparing the cable connections. If I understand correctly, the BMS connection is made on the pins, and the inverter connection is made on that green SP3485 interface chip from de right. Only those 4x cables from your scheme are used right?

ai-republic, 2024-08-26:
It would be a 1CH CAN/RS485 hat similar to this https://www.waveshare.com/rs485-can-hat.htm

Lino34s, 2024-09-01:
Getting these logs when trying to read BMS data
[image]

ai-republic, 2024-09-02:
Hmm, looks like you specified a CAN port for a RS_485 binding. Could you attach your config.properties file here please?

Lino34s, 2024-09-02:
[config.txt](https://github.com/user-attachments/files/16841342/config.txt)

ai-republic, 2024-09-05:
Ah, you have the JK on `can0` and also the BMS which is a Pylon RS485. The Pylon should be on a port like `/dev/ttyXXX`

ai-republic, 2024-10-04:
I guess I can close this as there has been no news for over a month. Please feel free to reopen this issue if you need further assistance.

closed 2024-10-04 by ai-republic