ai-republic/bms-to-inverter #41: Error: Unable to access jarfile bms-to-inverter-main-0.0.1-SNAPSHOT.jar

wire67, 2024-07-18:
Hello! Thanks for making this promising project :)
Running `./start` from the script subdirectory (?) outputs this error `Error: Unable to access jarfile bms-to-inverter-main-0.0.1-SNAPSHOT.jar`, I do not see this jar to be available in the repo, or am I doing this wrong?

OS: Ubuntu.

ai-republic, 2024-07-19:
Ah, this `start`scipt from the `scripts` folder is outdated. Please use the _Configurator_ application to configure your system (see [Wiki - How to use](https://github.com/ai-republic/bms-to-inverter/wiki/How-to-use)). Once you have done the _Clean install_ you will have a `start` script generated in your installation folder. Please use that to start the :)

wire67, 2024-07-19:
I did followed this wiki and did run the configurator on windows, the generated folder contains:
- config
-- config.properties
-- log4j2.xml
- temp
-- apache-maven-3.9.6
-- bms-to-inverter-main
-- bms-to-inverter.zip

there is no `installation` folder though

ai-republic, 2024-07-21:
Hi,
if you ran the _Clean install_ on Windows then you should have the following folders/files generated in the folder that you set as _Installation path_ in the _Configurator_ **General** tab:
[image]

Which Java JDK is installled? You should have a JDK 21 to run this successfully. If not, then you can download it [here](https://jdk.java.net/java-se-ri/21).

wire67, 2024-07-23:
I run jdk 22 
`"C:\Program Files\Java\jdk-22.0.2\bin\java" -jar configurator.jar` 

and `clean install`
```Installing in: P:\Prog\bms-to-inverter-installer
Configuration in: P:\Prog\bms-to-inverter-installer\config
Temp directory is: P:\Prog\bms-to-inverter-installer\temp
Downloading application...done
Building application...

Application was successfully built!
```

then a message 

[image]

and folder contains 

[image]

ai-republic, 2024-07-26:
Could you please download the latest [_Configurator_](https://github.com/ai-republic/bms-to-inverter/blob/main/configurator/current/configurator.jar) application and try again? Please let me know if it solved your issue :)

ai-republic, 2024-08-03:
@wire67 did it resolve your issue?

wire67, 2024-08-06:
Thanks for the help! The new configurator didn't work better, I also then tried JDK 21 with no success. Next I will run the configurator from another platform.
[image]

ai-republic, 2024-08-07:
Is the P: drive an external USB drive? Could you try to run it on a local drive like C: please? 
I tested it and have it running on Windows and also on my PI so I believe your external drive might cut out (I/O chip on/off) shortly during the build

wire67, 2024-08-07:
P is a virtual disk of C, I have tried on C as well. 
That error message, it's looking for a zip file in the downloaded repo, but that zip doesn't exit in the repo...

wire67, 2024-08-07:
[image]

ai-republic, 2024-08-07:
The zip File is downloaded from GitHub. That contains the complete BMS-to-Inverter repository. Maybe you have the URL blocked by a firewall? It tries this URL: https://github.com/ai-republic/bms-to-inverter/archive/master.zip

ai-republic, 2024-08-09:
Did you solve your problem? Or have you tried running it on your PI?

wire67, 2024-08-09:
URL works fine, no firewall. Apologies I am now busy with other aspects of the system, will be back on this later.

ai-republic, 2024-08-18:
@wire67 I just updated the _Configurator_ application. You might want to try with the latest [_Configurator_](https://github.com/ai-republic/bms-to-inverter/blob/main/configurator/current/configurator.jar) application

ai-republic, 2024-08-30:
I will close this issue as I believe it has been resolved. Feel free to open another issue if you encounter any issues :)

closed 2024-08-30 by ai-republic

wire67, 2024-09-10:
I was still getting an error but the log now had more info now: 
```
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-compiler-plugin:3.10.1:compile (default-compile) on project core-api: Compilation failure
[ERROR] No compiler is provided in this environment. Perhaps you are running on a JRE rather than a JDK?
```
I had multiple instances of java.
After uninstalling oracle's java.
```
The JAVA_HOME environment variable is not defined correctly, 
this environment variable is needed to run this program. 
```
As shown above I was calling 
`"C:\Program Files\Java\jdk-22.0.2\bin\java" -jar configurator.jar` 

but that path call doesn't work, I had to fully switch over to this JDK with the JAVA_HOME variable. 
After updating JAVA_HOME, the configurator worked fine.

ai-republic, 2024-09-11:
ah, so it was your Java environment setup :)
Glad to hear its working now!