ai-republic/bms-to-inverter #42: Support for Samsung battery

jezza61, 2024-07-23:
[715449196-4-84kWh-Standard-Product-Specification-R0-2-20190319.pdf](https://github.com/user-attachments/files/16343200/715449196-4-84kWh-Standard-Product-Specification-R0-2-20190319.pdf)
Hi, 
Is it possible to add support for Samsung batteries?
Included is the file with protocol details

ai-republic, 2024-07-26:
Hi, 
yes, that is definitely possible. Thanks for the specification! I'm finishing off the Pylon BMS RS485 binding. Then I could implement the Samsung BMS binding.

ai-republic, 2024-08-04:
I finished the Pylon RS485 binding and will try to find time to implement the Samsung next weekend :)

ai-republic, 2024-08-06:
@jezza61 I will start the Samsung BMS binding now :)

ai-republic, 2024-08-06:
@jezza61 I have finished implementing the Samsung binding. Please download the latest _Configurator_ application and choose the _SAMSUNG_CAN_ binding for your BMS.

If you have questions - don't hesitate to ask :)

ai-republic, 2024-08-09:
Closing this issue. Please create a new issue if you encounter any problems.

closed 2024-08-09 by ai-republic