ai-republic/bms-to-inverter #43: JK BMS to BYD protocol

ckambiselis, 2024-07-25:
Since you support the BYD BMS, is it possible to add BYD CAN protocol as the output? I'm looking to make a custom JK BMS battery to use with a Fronius inverter, they support BYD batteries CAN protocol.

ai-republic, 2024-07-26:
Sure, that shouldn't be a problem. Have to see if I get any time for it this weekend though as its my sons birthday ;)

ai-republic, 2024-08-04:
I finished the Pylon RS485 bindings and will do the BYD Inverter binding next weekend :)

ai-republic, 2024-08-06:
@ckambiselis I just noticed that the BYD and SMA protocols are very similar and should be compatible. There are only a couple of extra commands in the SMA protocol.
Can you try the selecting BYD protocol for your Fronius using the SMA binding?

ckambiselis, 2024-08-06:
Hi, I'm currently waiting for a lot of parts of the system, so I can't test anything yet.

ai-republic, 2024-10-31:
@ckambiselis any updates on your setup? Is it working for you now?

ckambiselis, 2024-10-31:
Have been really busy at work, so I haven't been able to test it yet.

ai-republic, 2024-12-01:
Hi @mman358, yes I'm sure you can. I will have to create a RS485 binding for the JK-DZ11-B2A24S but that shouldn't be a problem.
Your Deye HV inverter will surely support the Pylontech HV protocol (and probably the BYD as you mentioned) so that is already there.
As hardware I would recommend a Pi 4B and Waveshare RS485-CAN-Hat as your Deye can communicate via CAN.
If you have any questions just ask 😉

mman358, 2024-12-01:
Nice, thanks in advance! 

Any change to make the setup work with Raspberry Pi Pico WH + CAN module? I have couple of extra picos lying around. 
And would use this or similar CAN module
https://www.aliexpress.com/item/1005005191034082.html?spm=a2g0o.cart.0.0.79dc38darMAsHv&mp=1

Also attached approved battery list for my inverter. Pylon is in the list but whichever of these protocols is the easiest to develop should work.
[DeyeApprovedBatterylist(DY-HV(160-800)-021).pdf](https://github.com/user-attachments/files/17968832/DeyeApprovedBatterylist.DY-HV.160-800.-021.pdf)

mman358, 2024-12-03:
@ai-republic, no I haven't. Another way that comes to mind is if you could isolate the piece of code for me I need, I could simply convert that to micropython that runs on pico. That is, if it makes sense to you?

If not, then I may need to do this with Pi 4 as you originally proposed.

ai-republic, 2025-01-05:
I will close this issue now as there hasn't been any further progress. Feel free to open another issue if you encounter problems or need support

closed 2025-01-05 by ai-republic