ai-republic/bms-to-inverter #44: Pace BMS to MyEnergi Libbi inverter

skinsuk2, 2024-07-30:
I'm not sure if the above is possible - please let me know

ai-republic, 2024-08-01:
I would need a little more informatio:
1. which protocols does your PACE BMS and inverter support (CAN, RS485)?
2. which communication specification implementation does your inverter support (PylonTech, SMA, etc.)? You might need to check your inverter's battery configuration options for that

skinsuk2, 2024-08-02:
> I would need a little more information:

Thanks for your reply.
1. supports both CAN and RS485.  Has a load of settings when hooked up to a
pc.
2. I think this is the problem because the Libbi inverter is designed to
work with Libbi batteries.  There is a calibration test on startup which
takes 3 complete cycles.
See the manual here:
https://www.myenergi.com/wp-content/uploads/2023/04/libbi-Install-Manual-Rev-6.2-26.04.23.pdf

Matt

ai-republic, 2024-08-03:
Yes, I think your PACE BMS supports CAN and the MyEnergi inverter RS-485. I would need some more information what batteries the inverter supports (like PylonTech, BYD, etc.) or the RS-485 communication specification for your MyEnergi inverter.
Is there anything in the inverter battery configuration which let's you set different protocols?

skinsuk2, 2024-08-03:
The inverter doesn't work without a battery connected, so it's difficult to
find this info. I've tried searching the web with no luck. The inverter is
only designed to work with their own, premium-priced batteries so I think
this may not work. I knew it was a long shot, but thanks for your help.
Matt

ai-republic, 2024-08-04:
Have you tried contacting the manufacturer directly? Sometimes they will send you the communications specification if you prove that you have bought one of their models. Usually they will ask for the S/N or invoice.

ai-republic, 2024-10-04:
I will close this issue as there has been no correspondance for over a month. Feel free to reopen this ticket if you need further assistance.