ai-republic/bms-to-inverter #46: Installing configurator

jezza61, 2024-08-16:
Hi, I have installed the bms-to-inverter on one rasperypi, no probs, all works OK.
On the second PI, an older model, the application was installed and built OK, however when I clicked Close, it popped out with a message Failed to install the application! java,nio.FileSystemNotFoundException
Can it install into a 32bit version?
The Can interface was OK, as when I run candump can0 , I was getting a stream of valid data.

Second question, the battery change current, and the voltage limit for charge and discharge, could it be modified before being sent to the Inverter?
Essentially, I would like to 'undercharge' BMS to around 85% and discharge to about perhaps 15% (I have plenty of capacity.
I would prefer not to use limits reported by BMS, as charge to ~100% is scary, and discharge too low, wears out the battery.
At present I am testing several Samsung packs, and I am surprised how little data is pulled using CAN, RS485 seems to have about 10 times the amount of data.
However I don't know what data is collected from the BMS and what needs to be sent to the Inverter (is there some tutorial explaining can IDs and data being exchanged?)
And thanks a lot for this program.

ai-republic, 2024-08-18:
Hi @jezza61,
1. could you please tell me what older RPI you are using? You will need a 64-bit OS as the later Java versions only support 64-bit architectures. So the PI 3B and upwards all support the 64-bit Raspian OS
2. yes, I just implemented BMS and inverter plugins which can be used to manipulate the data before sending it to the inverter

Lastly the information what is sent can be identified via the CAN ID and what it matches to in the specification. Usually the data is send using LSB (Least significant bit), so if you have a U_INT_16 or S_INT_16 you need to take the second byte first then the first in your hex calculator to see the value in decimal. Then there is also the unit to check e.g. 0.1A.

jezza61, 2024-08-22:
I have some additional raspberry pi 4b  now, trying again, i think it is the older raspberry that fails.
And I can read and convert can values OK, same with rs485 stuff, just need more understanding what transmit to what, and what data is required by the inverter, I am assuming that BMS is sending the data in can in a 'short' format as compared to rs485 ?
I am assuming that inverters expect sunspec data IDs (are these consistent between inverters? ) and the bms should send specific can IDs ?

jezza61, 2024-08-22:
Commented too soon, a brand new raspberri pi 4b with the latest 64bit os, clean install, jdk17, same error message when running configurator. I did have another raspberry install perfectly fine and run OK. So not sure what the issue is?

jezza61, 2024-08-22:
Redone a clean new raspberry PI install, with a can hat. Installed java
Installing from configurator works OK, and it completes with no error messages, however it never installs a webserver now, even though it was selected.
It did install it previously on another raspberry, and the app runs, pulls down bms data and displays in the webserver.
I did follow the install instructions to the letter, so have I missed something? Also you have mentioned some plugin to modify some data before it is being sent to the inverter?

ai-republic, 2024-08-22:
> nstalling from configurator works OK, and it completes with no error messages, however it never installs a webserver now, even though it was selected.

Yes, I changed it to an embedded webserver as to not go via the Message Broker. This is more efficient. If you've selected the webserver, then it will be automatically started with the application.

jezza61, 2024-08-23:
So, here it is now.
The older raspberry pi, has a waveshare can hat https://www.waveshare.com/wiki/RS485_CAN_HAT_(B), and 64bit bookworm OS, updated to the latest version.
Candump can0 sends a stream of data like that:
 can0  00000500   [8]  A9 15 00 00 59 64 70 8B
  can0  00000501   [8]  00 00 00 00 02 02 00 00
  can0  00000502   [8]  45 02 AC 03 AC 03 C0 01
  can0  00000503   [8]  8C 0F 91 0F 83 0F A9 15
  can0  00000504   [8]  AA 15 A9 15 13 14 13 00
So I am assuming that the can interface is working OK, I have the older bms-to-inverter install, so I start start.sh and startWebserver.sh and I get a web page that displays a web page:
[image]
So I am assuming that is all working OK (No inverter yet, waiting for contractors to install it).
Second raspberry PI new 4B version, with a different can hat, https://www.waveshare.com/wiki/RS485_CAN_HAT
Installed with the latest 64 bit bookworm OS, and drivers (some issues with wiringpi ).
However Can0 interface seems to be OK, as candump can0, also shows a stream of data, same as for the first PI.
The latest bms-to-inverter installed, the one that has start.sh only, this one has ssl server, but the page is blank:
[image]
No data is displayed at all.
I am waiting for another dual can hat to test, however I am assuming that both PI units have a working CAN interface, as both can receive data OK, BMS is sending the data continously.
The BMS also has a data on RS485 interface, and I can pull down modbus input registers, but I am assuming that these would have to be polled by something ?
So I am not sure why the latest software does not show any data, I am just about to make copy of the first PI image and install the latest version of the bms-to-inverter on it, to see if the web page still works OK.
I suppose I could also do the opposite, and try to install the older bms software on the new PI, to see if that works?
I am happy to provide both CAN and RS485 data from live BMSs, I have many of them.
Perhaps I did not install the waveshare drivers correctly?  the instructions seem to be OK for older version of OS

jezza61, 2024-08-23:
One new update, running both RPIs on the older code after adjusting the CAN driver, both display web server with live data.

ai-republic, 2024-08-24:
Thank you for your feedback! I fixed the webserver issue. It was blocking at startup.
Please make a _Clean install_ with the new _Configurator_ application. 
Happy testing and please let me know how it goes :)

jezza61, 2024-08-26:
[Deye CAN BUS.pdf](https://github.com/user-attachments/files/16746679/Deye.CAN.BUS.pdf)
Here is the result of the latest test.
It does work now, the web page works, and the data is being collected and sent.
The BMS tab in the web page does not match cell data from the collected CAN data?
The discharge current limit sent to inverter does not seem right?
I am assuming that my DEYE CAN info is correct (I have included CAN tables I used)?
Is there a way of changeing the data before it is being sent to the inverter?

jezza61, 2024-08-26:
It seems, that I have not done enough research on CAN protocols, most common batteries seem to have the same, or almost the same protocol structure.
So now I have a better understanding. Now I need to read the manual for the inverter I am going to use shortly :)

ai-republic, 2024-08-27:
Hi @jezza61,
great to hear you're making progress :)
Could you please attach the log file here running the application in debug mode. Thanks!

jezza61, 2024-08-27:
Funny this one :)
As if I knew how to start the debug mode :(

jezza61, 2024-08-27:
Hi,
If you can tell me how to start in debug mode it would be great :)
I can also run a CAN trace on the conversation.

Best Regards,

Jerry

ai-republic, 2024-08-27:
In the _Configurator_ app you can set the log level in the general tab. Or your can edit the _config/log4j2.xml_ and set the following to _debug_

[image]

jezza61, 2024-08-27:
Hi,

Here is the log, and I did include candump can0 at the beginning, to show
the BMS data

It is a 8 tray stack of batteries, so more data

Best Regards,

Jerry

ai-republic, 2024-08-27:
Ah, I think you forgot to attach the file ;)

jezza61, 2024-08-27:
Here we go again, a zip file

jezza61, 2024-08-27:
My email shows that I have sent it previously also

jezza61, 2024-08-27:
I can a raw text file but it is big?

ai-republic, 2024-08-27:
You can also Zip it and attach it

jezza61, 2024-08-27:
Yet anothe attempt.
I did get a message that the text file was too large for our side.

Best Regards,

Jerry

ai-republic, 2024-08-27:
Maybe you can split it up and only send the first couple of MB. That would be well enough :)

jezza61, 2024-08-27:
Ok here we go:
CAN0 dum has this repeating pattern:
 can0  500   [8]  A7 15 00 00 55 64 68 15
  can0  501   [8]  00 00 00 00 08 08 00 00
  can0  502   [8]  45 02 B0 0E B0 0E C0 01
  can0  503   [8]  84 0F 8B 0F 80 0F A7 15
  can0  504   [8]  B7 15 99 15 13 16 13 00
  can0  505   [8]  01 03 00 00 00 00 00 00
  can0  510   [8]  9A 15 00 00 55 64 00 00
  can0  511   [8]  00 00 87 0F 7F 0F 16 15
  can0  512   [8]  00 00 00 00 01 02 00 00
  can0  518   [8]  B2 15 00 00 56 64 00 00
  can0  519   [8]  00 00 88 0F 80 0F 14 14
  can0  51A   [8]  00 00 00 00 01 02 00 00
  can0  520   [8]  B7 15 00 00 56 64 00 00
  can0  521   [8]  00 00 8A 0F 82 0F 14 14
  can0  522   [8]  00 00 00 00 01 02 00 00
  can0  528   [8]  B2 15 00 00 55 64 00 00
  can0  529   [8]  00 00 88 0F 81 0F 14 14
  can0  52A   [8]  00 00 00 00 01 02 00 00
  can0  530   [8]  9A 15 00 00 56 64 00 00
  can0  531   [8]  00 00 88 0F 82 0F 14 13
  can0  532   [8]  00 00 00 00 01 02 00 00
  can0  538   [8]  AE 15 00 00 55 64 00 00
  can0  539   [8]  00 00 86 0F 81 0F 13 13
  can0  53A   [8]  00 00 00 00 01 02 00 00
  can0  540   [8]  A3 15 00 00 56 64 00 00
  can0  541   [8]  00 00 8B 0F 81 0F 13 13
  can0  542   [8]  00 00 00 00 01 02 00 00
  can0  548   [8]  99 15 00 00 56 64 00 00
  can0  549   [8]  00 00 8A 0F 80 0F 13 13
  can0  54A   [8]  00 00 00 00 01 02 00 00
  can0  5F0   [8]  01 00 85 0F 85 0F 82 0F
  can0  5F1   [8]  01 00 85 0F 87 0F 83 0F
  can0  5F2   [8]  01 00 85 0F 87 0F 7F 0F
  can0  5F3   [8]  01 00 82 0F 83 0F 82 0F
  can0  5F4   [8]  01 00 85 0F 87 0F 82 0F

Now, the max discharge current, is always the same as max charge current,
according to Samsung Data
Somehow the maxpackDischarge current is all scrambled and it comes out huge.
The log is filled mostly with webserver data.

##########################

 committed
2024-08-27 14:23:21.111 | DEBUG | 1589252-39 | ernal.HttpConnection:555   |
***@***.******@***.***[{l=/192.168.5.36:8443,r=/
192.168.5.223:56580,OPEN,fill=-,flush=-,to=776/30000}] parse
***@***.******@***.***[p=0,l=427,c=20480,r=427]={<<<GET
/alarmMessages HTTP/1...igin\r\nPriority:
u=4\r\n\r\n>>>\x00\x00\x00\x00\x00\x00\x00\x00\x00...\x00\x00\x00\x00\x00\x00\x00}]
2024-08-27 14:23:21.117 | DEBUG | 1589252-39 | etty.http.HttpParser:1614  |
parseNext s=START ***@***.***[p=0,l=427,c=20480,r=427]={<<<GET
/alarmMessages HTTP/1...igin\r\nPriority:
u=4\r\n\r\n>>>\x00\x00\x00\x00\x00\x00\x00\x00\x00...\x00\x00\x00\x00\x00\x00\x00}
2024-08-27 14:23:21.117 | DEBUG | 1589252-31 | tate$ChannelResponse:1291  |
writing last=true
***@***.***[p=0,l=2756,c=2756,r=2756]={<<<{"levelTwoDischargeCurren...:"L2
Cell V Diff High"}>>>} ***@***.******@***.***
https://192.168.5.36:8443/alarmMessages HTTP/1.1}
2024-08-27 14:23:21.116 | DEBUG | 1589252-28 | State$HandlerInvoker:613   |
invoking handler in
***@***.***{handling=Thread[qtp1791589252-28,5,main],
handled=false, send=SENDING, completed=false, ***@***.***
https://192.168.5.36:8443/data HTTP/1.1}
2024-08-27 14:23:21.117 | DEBUG | 1589252-39 | etty.http.HttpParser:2049  |
START --> SPACE1(GET)
2024-08-27 14:23:21.117 | DEBUG | 1589252-39 | etty.http.HttpParser:2047  |
SPACE1 --> URI
[... 1073 lines cut ...]
handled=false, send=LAST_SENDING, completed=false, ***@***.***
https://192.168.5.36:8443/data HTTP/1.1}]
2024-08-27 14:23:21.178 | DEBUG | 1589252-31 | tate$ChannelResponse:1306  |
write succeeded ***@***.******@***.***
https://192.168.5.36:8443/alarmMessages HTTP/1.1}
2024-08-27 14:23:21.185 | DEBUG | 1589252-28 | etty.io.WriteFlusher:427   |
Flushed=true written=11560 remaining=0 ***@***.***{WRITING}->null
2024-08-27 14:23:21.187 | DEBUG | 1589252-28 | etty.io.WriteFlusher:108   |
update ***@***.***{IDLE}->null:WRITING-->IDLE
2024-08-27 14:23:21.187 | DEBUG | 1589252-28 | nection$SendCallback:760   |
generate: DONE for ***@***.***
***@***.***
***@***.*** https://192.168.5.36:8443/data HTTP/1.1}]
***@***.***
2024-08-27 14:23:21.188 | DEBUG | 1589252-28 | .util.ConcurrentPool:191   |
returning reserved entry
***@***.***{terminated=false,multiplex=-1,pooled=null}
for ***@***.***
[strategy=THREAD_ID,inUse=0,size=2,max=256,leaked=0,terminated=false]
2024-08-27 14:23:21.188 | DEBUG | 1589252-28 | Pool$ConcurrentEntry:496   |
enabled ***@***.***
***@***.***
***@***.***[p=0,l=0,c=8192,r=0]={<<<>>>HTTP/1.1
...\x00\x00\x00\x00\x00\x00\x00}]} for ***@***.***
[strategy=THREAD_ID,inUse=0,size=2,max=256,leaked=0,terminated=false]

jezza61, 2024-08-27:
Another trace:

2024-08-27 15:46:35.294 | DEBUG | 0958248-30 | etty.http.HttpParser:2059  |
HEADER:VALUE --> IN_VALUE
2024-08-27 15:46:35.294 | DEBUG | 0958248-42 | etty.http.HttpParser:1202  |
parsedHeader(null) header=null, headerString=[Sec-Fetch-Mode],
valueString=[cors]
2024-08-27 15:46:35.294 | DEBUG | 0958248-30 | etty.http.HttpParser:2061  |
HEADER:IN_VALUE --> FIELD(Referer: https://192.168.5.36:8443/)
2024-08-27 15:46:35.294 | DEBUG | 0958248-42 | etty.http.HttpParser:2059  |
HEADER:FIELD --> IN_NAME
2024-08-27 15:46:35.294 | DEBUG | 0958248-30 | etty.http.HttpParser:1202  |
parsedHeader(null) header=Referer, headerString=[Referer], valueString=[
https://192.168.5.36:8443/]
2024-08-27 15:46:35.294 | DEBUG | 0958248-42 | etty.http.HttpParser:2059  |
HEADER:IN_NAME --> VALUE
2024-08-27 15:46:35.294 | DEBUG | 0958248-42 | etty.http.HttpParser:2059  |
HEADER:VALUE --> IN_VALUE
2024-08-27 15:46:35.294 | DEBUG | 0958248-30 | etty.http.HttpParser:2059  |
HEADER:FIELD --> IN_VALUE
2024-08-27 15:46:35.294 | DEBUG | 0958248-42 | etty.http.HttpParser:2061  |
HEADER:IN_VALUE --> FIELD(Sec-Fetch-Site: same-origin)
2024-08-27 15:46:35.294 | DEBUG | 0958248-30 | etty.http.HttpParser:2061  |
HEADER:IN_VALUE --> FIELD(Authorization: Basic Ym1zOmpmZXJybzE=: Basic
Ym1zOmpmZXJybzE=)
2024-08-27 15:46:35.294 | DEBUG | 0958248-42 | etty.http.HttpParser:1202  |
parsedHeader(null) header=null, headerString=[Sec-Fetch-Site],
valueString=[same-origin]
2024-08-27 15:46:35.294 | DEBUG | 0958248-30 | etty.http.HttpParser:1202  |
parsedHeader(Authorization: Basic Ym1zOmpmZXJybzE=) header=Authorization,
headerString=[Authorization], valueString=[Basic Ym1zOmpmZXJybzE=]
2024-08-27 15:46:35.294 | DEBUG | 0958248-42 | etty.http.HttpParser:2059  |
HEADER:FIELD --> IN_NAME
2024-08-27 15:46:35.294 | DEBUG | 0958248-30 | etty.http.HttpParser:2059  |
HEADER:FIELD --> IN_VALUE
2024-08-27 15:46:35.294 | DEBUG | 0958248-42 | etty.http.HttpParser:2059  |
HEADER:IN_NAME --> VALUE
2024-08-27 15:46:35.294 | DEBUG | 0958248-30 | etty.http.HttpParser:2061  |
HEADER:IN_VALUE --> FIELD(Connection: keep-alive: keep-alive)
2024-08-27 15:46:35.294 | DEBUG | 0958248-42 | etty.http.HttpParser:2059  |
HEADER:VALUE --> IN_VALUE
2024-08-27 15:46:35.294 | DEBUG | 0958248-30 | etty.http.HttpParser:1202  |
parsedHeader(Connection: keep-alive) header=Connection,
headerString=[Connection], valueString=[keep-alive]
2024-08-27 15:46:35.294 | DEBUG | 0958248-42 | etty.http.HttpParser:2061  |
HEADER:IN_VALUE --> FIELD(Priority: u=4)
2024-08-27 15:46:35.294 | DEBUG | 0958248-30 | etty.http.HttpParser:2059  |
HEADER:FIELD --> IN_NAME
2024-08-27 15:46:35.294 | DEBUG | 0958248-42 | etty.http.HttpParser:1202  |
parsedHeader(null) header=null, headerString=[Priority], valueString=[u=4]
2024-08-27 15:46:35.294 | DEBUG | 0958248-30 | etty.http.HttpParser:2059  |
HEADER:IN_NAME --> VALUE
2024-08-27 15:46:35.294 | DEBUG | 0958248-30 | etty.http.HttpParser:2059  |
HEADER:VALUE --> IN_VALUE
2024-08-27 15:46:35.294 | DEBUG | 0958248-30 | etty.http.HttpParser:2061  |
HEADER:IN_VALUE --> FIELD(Sec-Fetch-Dest: empty)
2024-08-27 15:46:35.294 | DEBUG | 0958248-42 | nal.HttpChannelState:262   |
onRequest GET{u=https://192.168.5.36:8443/data,HTTP/1.1,h=12,cl=-1,p=null}
***@***.***{handling=null, handled=false, send=SENDING,
completed=false, request=null}
2024-08-27 15:46:35.294 | DEBUG | 0958248-30 | etty.http.HttpParser:1202  |
parsedHeader(null) header=null, headerString=[Sec-Fetch-Dest],
valueString=[empty]
2024-08-27 15:46:35.294 | DEBUG | 0958248-30 | etty.http.HttpParser:2059  |
HEADER:FIELD --> IN_NAME
2024-08-27 15:46:35.294 | DEBUG | 0958248-30 | etty.http.HttpParser:2059  |
HEADER:IN_NAME --> VALUE
2024-08-27 15:46:35.294 | DEBUG | 0958248-30 | etty.http.HttpParser:2059  |
HEADER:VALUE --> IN_VALUE
2024-08-27 15:46:35.295 | DEBUG | 0958248-30 | etty.http.HttpParser:2061  |
[... 1187 lines cut ...]
tryProduce true
2024-08-27 15:46:37.304 | DEBUG | 0958248-47 | nnection$SslEndPoint:863   |
<fill f=418 uf=false ***@***.******@***.***
[{l=/192.168.5.36:8443,r=/192.168.5.223:55787
***@***.***
{NOT_HANDSHAKING,eio=-1/-1,di=-1,fill=IDLE,flush=IDLE}~>{l=/
192.168.5.36:8443,r=/192.168.5.223:55787
***@***.***[p=HttpParser{s=START,0
of ***@***.******@***.***{handling=null,
handled=false, send=SENDING, completed=false, request=null}]
2024-08-27 15:46:37.304 | DEBUG | 0958248-42 | io.ssl.SslConnection:384   |
>c.onFillable ***@***.******@***.***[{l=/
192.168.5.36:8443,r=/192.168.5.223:55837
***@***.***
{NOT_HANDSHAKING,eio=-1/-1,di=-1,fill=INTERESTED,flush=IDLE}~>{l=/
192.168.5.36:8443,r=/192.168.5.223:55837
***@***.***[p=HttpParser{s=START,0
of ***@***.******@***.***{handling=null,
handled=false, send=SENDING, completed=false, request=null}]
2024-08-27 15:46:37.304 | DEBUG | 0958248-30 | tor$SelectorProducer:561   |
updateable 0
2024-08-27 15:46:37.304 | DEBUG | 0958248-47 | ernal.HttpConnection:523   |
***@***.******@***.***[{l=/192.168.5.36:8443,r=/
192.168.5.223:55787,OPEN,fill=-,flush=-,to=2000/30000}] filled 418
***@***.******@***

jezza61, 2024-08-27:
And a can trace of what is being sent

On Tue, Aug 27, 2024 at 3:49 PM Jerry Sawick ***@***.***> wrote:

> Another trace:
>
>
> 2024-08-27 15:46:35.294 | DEBUG | 0958248-30 | etty.http.HttpParser:2059
>  | HEADER:VALUE --> IN_VALUE
> 2024-08-27 15:46:35.294 | DEBUG | 0958248-42 | etty.http.HttpParser:1202
>  | [...]

jezza61, 2024-08-27:
Hi,

Can ID 351, byte 3 is the one in error, Byte 2 and byte 3 shouldl be the
same, as Samsung always have these 2 values the same.
That is the max charge current and max discharge current.
In my case it would not matter, as my battery current limit being 376A  is
well in excess of the inverter limit.
But for other people with small batteries, the Inverter trying to pull down
over 6000A would be bad :)

Best Regards,

Jerry

On Tue, Aug 27, 2024 at 3:49 PM Jerry Sawick ***@***.***> wrote:

> Another trace:
>
>
> 2024-08-27 15:46:35.294 | DEBUG | 0958248-30 | etty.http.HttpParser:2059
>  | HEADER:VALUE --> IN_VALUE
> 2024-08-27 15:46:35.294 | DEBUG | 0958248-42 | etty.http.HttpParser:1202
>  | [...]

ai-republic, 2024-08-27:
Hi Jerry,
I've deleted the comments with the log dump as otherwise this thread is unreadable ;)
Could you disable the webserver during testing please and just run the BMS please - makes the log entries more readable. But thats a good tip - I'll separate the webserver log from the application log. And please delete the logs in the _logs_ folder first. Then attach the log file as file please.

jezza61, 2024-08-27:
I have switched off the webserver.
Perhaps that helps?

On Tue, Aug 27, 2024 at 4:03 PM Jerry Sawick ***@***.***> wrote:

> Hi,
>
> Can ID 351, byte 3 is the one in error, Byte 2 and byte 3 shouldl be the
> same, as Samsung always have these 2 values the same.
> That is the max charge current and max discharge current.
> In my [...]

ai-republic, 2024-08-27:
Yes please. And only as file please

jezza61, 2024-08-27:
ere is the log

=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2024.08.27 17:01:54 =~=~=~=~=~=~=~=~=~=~=~=
./start.sh
[?2004l2024-08-27 17:02:01.115 | INFO  | main       | erter.core.util.Util:35    | Loading config.properties from: config/config.properties
2024-08-27 17:02:01.145 | DEBUG | main       | ging.LoggerProviders:152   | Logging Provider: org.jboss.logging.Log4j2LoggerProvider
2024-08-27 17:02:01.891 | INFO  | main       | core.BMSListProducer:98    | Intialized BMS #0[SAMSUNG_CAN] on port can0
2024-08-27 17:02:01.898 | INFO  | main       | verter.BmsToInverter:261   | Starting BMS receiver...
2024-08-27 17:02:01.900 | INFO  | Thread-2   | verter.BmsToInverter:268   | Reading BMS #0 SAMSUNG_CAN on can0...
2024-08-27 17:02:01.901 | INFO  | Thread-2   | stoinverter.core.BMS:130   | ---------------------------------> Thread 19
2024-08-27 17:02:01.902 | INFO  | Thread-2   | toinverter.core.Port:111   | Opening can0 ...
2024-08-27 17:02:01.919 | INFO  | Thread-2   | toinverter.core.Port:113   | Opening port can0 SUCCESSFUL
2024-08-27 17:02:01.920 | DEBUG | Thread-2   | ocol.can.JavaCANPort:76    | CAN frame read...
2024-08-27 17:02:01.925 | DEBUG | Thread-2   | ocol.can.JavaCANPort:82    | CAN read frame Buffer (HEX): [0x00, 0x05, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0xA5, 0x15, 0x00, 0x00, 0x55, 0x64, 0x24, 0x5F]

2024-08-27 17:02:01.926 | DEBUG | Thread-2   | msungBmsCANProcessor:96    | 
Pack V 	 Pack A 	 Avg 
SOC 	SOH\n55.4 	0.0  85.0		  100.0
2024-08-27 17:02:01.927 | INFO  | main       | verter.BmsToInverter:290   | Starting inverter sender...
2024-08-27 17:02:01.928 | INFO  | Thread-3   | verter.BmsToInverter:294   | Sending to inverter DEYE_CAN on can0...
2024-08-27 17:02:01.931 | DEBUG | Thread-3   | verter.core.Inverter:152   | Inverter received: Buffer (HEX): [null]

2024-08-27 17:02:01.933 | DEBUG | Thread-3   | verter.core.Inverter:173   | Inverter send: Buffer (HEX): [0x51, 0x03, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-08-27 17:02:01.934 | DEBUG | Thread-3   | ocol.can.JavaCANPort:113   | CAN frame sending: Buffer (HEX): [0x51, 0x03, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-08-27 17:02:01.936 | DEBUG | Thread-3   | verter.core.Inverter:173   | Inverter send: Buffer (HEX): [0x55, 0x03, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x54, 0x00, 0x64, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-08-27 17:02:01.937 | DEBUG | Thread-3   | ocol.can.JavaCANPort:113   | CAN frame sending: Buffer (HEX): [0x55, 0x03, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x54, 0x00, 0x64, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-08-27 17:02:01.938 | DEBUG | Thread-3   | verter.core.Inverter:173   | Inverter send: Buffer (HEX): [0x56, 0x03, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0xA4, 0x15, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-08-27 17:02:01.939 | DEBUG | Thread-3   | ocol.can.JavaCANPort:113   | CAN frame sending: Buffer (HEX): [0x56, 0x03, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0xA4, 0x15, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-08-27 17:02:01.939 | INFO  | Thread-2   | verter.BmsToInverter:320   | 
BMS	SOC	  V  	  A  	 CellMinV 	 CellMaxV	CellDiff
#1	 85.0	55.4	0.0	0.0(#0) 	0.0(#0)	0.0 

2024-08-27 17:02:01.940 | INFO  | Thread-2   | verter.BmsToInverter:390   | BMS alarms: 
	NONE
2024-08-27 17:02:01.940 | DEBUG | Thread-3   | verter.core.Inverter:173   | Inverter send: Buffer (HEX): [0x5E, 0x03, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-08-27 17:02:01.942 | DEBUG | Thread-3   | ocol.can.JavaCANPort:113   | CAN frame sending: Buffer (HEX): [0x5E, 0x03, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-08-27 17:02:01.943 | DEBUG | Thread-3   | verter.core.Inverter:173   | Inverter send: Buffer (HEX): [0x5E, 0x03, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-08-27 17:02:01.944 | DEBUG | Thread-3   | ocol.can.JavaCANPort:113   | CAN frame sending: Buffer (HEX): [0x5E, 0x03, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-08-27 17:02:01.945 | DEBUG | Thread-3   | verter.core.Inverter:173   | Inverter send: Buffer (HEX): [0x59, 0x03, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-08-27 17:02:01.946 | DEBUG | Thread-3   | ocol.can.JavaCANPort:113   | CAN frame sending: Buffer (HEX): [0x59, 0x03, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-08-27 17:02:02.940 | INFO  | Thread-2   | verter.BmsToInverter:268   | Reading BMS #0 SAMSUNG_CAN on can0...
2024-08-27 17:02:02.941 | INFO  | Thread-2   | stoinverter.core.BMS:130   | ---------------------------------> Thread 19
2024-08-27 17:02:02.941 | DEBUG | Thread-2   | ocol.can.JavaCANPort:76    | CAN frame read...
2024-08-27 17:02:02.942 | DEBUG | Thread-2   | ocol.can.JavaCANPort:82    | CAN read frame Buffer (HEX): [0x01, 0x05, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x08, 0x08, 0x00, 0x00]

2024-08-27 17:02:02.942 | INFO  | Thread-2   | verter.BmsToInverter:320   | 
BMS	SOC	  V  	  A  	 CellMinV 	 CellMaxV	CellDiff
#1	 85.0	55.4	0.0	0.0(#0) 	0.0(#0)	0.0 

2024-08-27 17:02:02.942 | INFO  | Thread-2   | verter.BmsToInverter:390   | BMS alarms: 
	NONE
2024-08-27 17:02:03.943 | INFO  | Thread-2   | verter.BmsToInverter:268   | Reading BMS #0 SAMSUNG_CAN on can0...
[... cut ...]
2024-08-27 17:03:19.733 | DEBUG | Thread-3   | verter.core.Inverter:173   | Inverter send: Buffer (HEX): [0x5E, 0x03, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-08-27 17:03:19.734 | DEBUG | Thread-3   | ocol.can.JavaCANPort:113   | CAN frame sending: Buffer (HEX): [0x5E, 0x03, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-08-27 17:03:19.734 | DEBUG | Thread-3   | verter.core.Inverter:173   | Inverter send: Buffer (HEX): [0x5E,

ai-republic, 2024-08-28:
So, the Samsung BMS data looks all good. The Deye 0x351 looks also correct according to the spec (except that the maxDischarge Amps is negative of 376.0A):
[image]

Charge V: 58.1V
Charge A limit: 376.0A
Discharge A limit: 6117.6A
Discharge V limit: 44.8V

On another Deye spec the last 2 bytes are used for the battery discharge voltage limit so that's also ok.

I've changed the Discharge A limit to positive so that it will now show up as  0xB0, 0x0E -> 376.0
Please do a clean install to get the fix :)

ai-republic, 2024-08-30:
I think I can close this issue now a the _Configurator_ issues have been resolved. Feel free to open another issue if you encounter any problems :)

closed 2024-08-30 by ai-republic

jezza61, 2024-08-30:
The only issue I have noticed that in the Web server, once you get an error from the bms it becomes visible and after the error is cleared, the webserver still shows the error.You have to restart the program to clear the error.Is that intentional?Sent from my Galaxy

ai-republic, 2024-08-31:
> The only issue I have noticed that in the Web server, once you get an error from the bms it becomes visible and after the error is cleared, the webserver still shows the error.You have to restart the program to clear the error.Is that intentional?

No, that is definetely not intentional. I will investigate that - thanks!