ai-republic/bms-to-inverter #47: Failed to install the application! java.nio.file.FileSystemNotFoundException

MarkoBursic, 2024-08-21:
Hi,
I get this message at the end, and no files are created. When I start Clean install, everything compiles, builds, when finished the Installing window says: Application was successfully built! But when I close this window, there is another pop-up window with error (like it the tittle of this issue).
I had installed JDK 22, JDK21, running configurator on Windows 10.

ai-republic, 2024-08-21:
Thank you! I ccould reproduce the error when I downloaded the JAR. I'm on it... :)

ai-republic, 2024-08-22:
I fixed the issue of not reading the log file template. Please test. It should be working fine now.

MarkoBursic, 2024-08-22:
I have tested and I can confirm that issue is resolved. Not only it generates the missing output files, but also correctly reads config.properties when restarting the configurator, before it didn't.

closed 2024-08-22 by MarkoBursic