ai-republic/bms-to-inverter #48: Huawei sun2000 4.6ktl supported?

TyngRikku, 2024-08-24:
I’d like to add batteries to this inverter, is that possible?

ai-republic, 2024-08-25:
Hi @TyngRikku,
yes it is possible as the inverter supports RS485 communication with the external battery. I just need the RS485 communication specification for your model or one of its supported batteries

TyngRikku, 2024-08-25:
Hey Torsten, attached the documents describing this for the inverter and the only official battery supported (Luna 2000), although I read comments even that battery isn’t plug & play for some. I’m not interested in this battery but added for completeness

Awesome work here btw, I was amazed when I learned BMS’s are so incompatible, I was naive any battery would do.

[HUAWEI_SUN2000_245KTL28KTL_MODBUS_Interface_Definitions_20150715_JP.pdf](https://github.com/user-attachments/files/16739204/HUAWEI_SUN2000_245KTL28KTL_MODBUS_Interface_Definitions_20150715_JP.pdf)
[LUNA2000B ESS Modbus Port Definitions.pdf](https://github.com/user-attachments/files/16739205/LUNA2000B.ESS.Modbus.Port.Definitions.pdf)

ai-republic, 2024-08-25:
Ok, thanks for the specification. The battery specification is what I need to send the BMS data to the inverter. I'll see when I get around to start with that. Possibly next weekend.

ai-republic, 2024-08-30:
Btw, which BMS are you planning to ues?

TyngRikku, 2024-08-30:
I plan on using the Pylontech CAN protocol but I havn’t found affordable 400V batteries yet to be honest and this inverter does not accept 48V batteries :-(

ai-republic, 2024-08-30:
Yes, the inverter needs HV (high voltage) batteries. You could also put 8-12x 48V packs in series to achieve that.

ai-republic, 2024-09-01:
@TyngRikku I've added a rudimentary support for the Huawei BMS and inverter bindings. Please download the new _Configurator_ application, choose the HUAWEI_MODBUS inverter binding and do a clean install. 
Please let me know if you have issues :)

TyngRikku, 2024-09-01:
Ok will try it this evening!

TyngRikku, 2024-09-01:
> Yes, the inverter needs HV (high voltage) batteries. You could also put 8-12x 48V packs in series to achieve that.

Mm manual explicitly tells to not put them in series, how will the voltage be divided evenly? Do you have experience with this? Wouldn’t you need a BMS @ 400V instead of 9 @ 48V?

ai-republic, 2024-09-02:
Well, the voltage is not really a problem. A BMS could e.g. have 16s4p which mean 4 packs in parallel and 16 cells in series. You just need the right BMS

ai-republic, 2024-10-04:
@TyngRikku any news on your project? Did you find a solution for your battery/BMS?

ai-republic, 2024-10-15:
I will close this issue as there has been no response. Feel free to open another issue if the need arises

closed 2024-10-15 by ai-republic