ai-republic/bms-to-inverter #50: Growatt low voltage (CAN) BMS

TyngRikku, 2024-08-26:
In the application, I can not find this option for the BMS, is it named differently?

Thanks

ai-republic, 2024-08-27:
Could you please tell me what BMS or Growatt battery system you have?

TyngRikku, 2024-08-27:
Its a growatt sph5000 and greenbatt batteries (I have two single phase PV systems, one with growatt and one huawei)

thanks

ai-republic, 2024-08-30:
So the inverter is a Growatt SPH5000 and the batteries are Greenbatt. The inverter is supported by the application. Do you have any information on the Greenbatt battery? Which CAN communication specification does it use? As its chinese I suspect it will support the Pylon specification. Can you log in to the Greenbatt BMS or get a CAN dump?

TyngRikku, 2024-08-30:
Hey Torsten

I can select multiple CAN protocols, of which the Pylontech CAN looks the logical choice. The BMS is from Shanghai Enjie. I have no idea how to do a CAN dump :-)

ai-republic, 2024-09-29:
@TyngRikku I'd need some kind of specification for the BMS to move on

ai-republic, 2024-10-04:
@TyngRikku I will close this as there has been no correspondance for over a month. If you need assistance please feel free to reopen this or create another issue

closed 2024-10-04 by ai-republic