ai-republic/bms-to-inverter #54: JK bms RS485 Modbus V1.0 issue

kovaroman, 2024-09-03:
Hi, I have a JK B2A8S20P with a JK RS485 adapter (GPS UART port to RS485 port), it is working OK on windows with JK monitor app. I have configured on linux your configurator, tried to use ID 0 and 1, 115200 baud rate and 9600, my PC rs485 adapter is based on CH340 chip and starts on port `/dev/ttyUSB0`

On windows my JK protocol is set to `JK BMS RS485 Modbus V1.0` and works well with JK monitor app, when I set this protocol and try to start your script (BMS type: JK_RS485, ID: 1, Port: /dev/ttyUSB0, Baudrate: 115200, Delay: 1000),  it sends request and doesn't receive response.

here is part of log:

```
2024-09-03 10:26:44.916 | INFO  | Thread-10  | verter.BmsToInverter:268   | Reading BMS #1 JK_RS485 on /dev/ttyUSB0...
2024-09-03 10:26:44.917 | INFO  | Thread-10  | stoinverter.core.BMS:130   | ---------------------------------> Thread 28
2024-09-03 10:26:44.917 | DEBUG | Thread-10  | s485.JSerialCommPort:180   | Clearing RX buffers
2024-09-03 10:26:44.918 | DEBUG | Thread-10  | s485.JSerialCommPort:159   | Send: Buffer (HEX): [0x4E, 0x57, 0x00, 0x13, 0x00, 0x00, 0x00, 0x00, 0x06, 0x03, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x68, 0x00, 0x00, 0x01, 0x29]

2024-09-03 10:26:45.221 | DEBUG | Thread-10  | .JKBmsRS485Processor:176   | No bytes received: 1 times!
2024-09-03 10:26:45.222 | DEBUG | Thread-10  | .JKBmsRS485Processor:186   | Waiting for messages to arrive....
2024-09-03 10:26:46.426 | DEBUG | Thread-10  | .JKBmsRS485Processor:176   | No bytes received: 2 times!
2024-09-03 10:26:46.427 | DEBUG | Thread-10  | .JKBmsRS485Processor:186   | Waiting for messages to arrive....
2024-09-03 10:26:47.631 | DEBUG | Thread-10  | .JKBmsRS485Processor:176   | No bytes received: 3 times!
2024-09-03 10:26:47.631 | DEBUG | Thread-10  | .JKBmsRS485Processor:186   | Waiting for messages to arrive....
2024-09-03 10:26:48.835 | DEBUG | Thread-10  | .JKBmsRS485Processor:176   | No bytes received: 4 times!
2024-09-03 10:26:48.835 | DEBUG | Thread-10  | .JKBmsRS485Processor:186   | Waiting for messages to arrive....
2024-09-03 10:26:49.185 | INFO  | Thread-12  | verter.BmsToInverter:294   | Sending to inverter PYLON_RS485 on /dev/ttyUSB0...
2024-09-03 10:26:49.185 | DEBUG | Thread-12  | verter.core.Inverter:187   | No battery data yet received to send to inverter!
2024-09-03 10:26:50.038 | DEBUG | Thread-10  | .JKBmsRS485Processor:176   | No bytes received: 5 times!
2024-09-03 10:26:50.039 | DEBUG | Thread-10  | .JKBmsRS485Processor:186   | Waiting for messages to arrive....
2024-09-03 10:26:51.243 | DEBUG | Thread-10  | .JKBmsRS485Processor:176   | No bytes received: 6 times!
2024-09-03 10:26:51.243 | DEBUG | Thread-10  | .JKBmsRS485Processor:186   | Waiting for messages to arrive....
2024-09-03 10:26:52.448 | DEBUG | Thread-10  | .JKBmsRS485Processor:176   | No bytes received: 7 times!
2024-09-03 10:26:52.449 | DEBUG | Thread-10  | .JKBmsRS485Processor:186   | Waiting for messages to arrive....
2024-09-03 10:26:53.653 | DEBUG | Thread-10  | .JKBmsRS485Processor:176   | No bytes received: 8 times!
2024-09-03 10:26:53.653 | DEBUG | Thread-10  | .JKBmsRS485Processor:186   | Waiting for messages to arrive....
2024-09-03 10:26:54.189 | INFO  | Thread-12  | verter.BmsToInverter:294   | Sending to inverter PYLON_RS485 on /dev/ttyUSB0...
2024-09-03 10:26:54.190 | DEBUG | Thread-12  | verter.core.Inverter:187   | No battery data yet received to send to inverter!
2024-09-03 10:26:54.858 | DEBUG | Thread-10  | .JKBmsRS485Processor:176   | No bytes received: 9 times!
2024-09-03 10:26:54.858 | DEBUG | Thread-10  | .JKBmsRS485Processor:186   | Waiting for messages to arrive....
2024-09-03 10:26:56.062 | DEBUG | Thread-10  | .JKBmsRS485Processor:176   | No bytes received: 10 times!
2024-09-03 10:26:56.063 | ERROR | Thread-10  | .JKBmsRS485Processor:199   | Error sending frame: Buffer (HEX): [0x4E, 0x57, 0x00, 0x13, 0x00, 0x00, 0x00, 0x00, 0x06, 0x03, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x68, 0x00, 0x00, 0x01, 0x29]

2024-09-03 10:26:56.063 | INFO  | Thread-10  | verter.BmsToInverter:320   | 
BMS	SOC	  V  	  A  	 CellMinV 	 CellMaxV	CellDiff

2024-09-03 10:26:56.063 | INFO  | Thread-10  | verter.BmsToInverter:390   | BMS alarms: 
	NONE

```

I tried different config combinations, tried to changed UART protocol on BMS to `(9600) JK BMS RS485 Modbus V1.0` and set Baudrate 9600 in configurator, nothing changed. Does your script support `JK BMS RS485 Modbus V1.0` protocol on RS485?

ai-republic, 2024-09-04:
Have you tried
```
cat /dev/ttyUSB0
```
and see if data is coming?
If your JK is a Modbus version, then the _JK_RS485_ BMS binding will not work. It is for pure RS485 not for Modbus RTU protocol specification.
If you can provide the JK Modbus RTU specification then I can create a binding for that :)

kovaroman, 2024-09-04:
https://github.com/Louisvdw/dbus-serialbattery/discussions/969#discussioncomment-8668592 - here are some info about JK BMS RS485 Modbus V1.0

https://github.com/Louisvdw/dbus-serialbattery/files/14483776/BMS.RS485.Modbus.V1.0.pdf - original
https://github.com/Louisvdw/dbus-serialbattery/files/14483784/Jikong.BMS.RS485.Modbus.universal.protocol.V1.0.pdf - translated

Thanks)

ai-republic, 2024-10-04:
@kovaroman sorry I've been quite busy the last couple of weeks. I will try to implement the JK ModBus BMS binding this or next weekend

ai-republic, 2024-10-06:
@kovaroman I had a look a the specifications you sent but the translated version is all messed up. But I will try to combine the both to make some sense of it 😄

ai-republic, 2024-10-13:
Hi @kovaroman, 
I implemented the JK ModBus BMS binding. Could you download the latest _Configurator_ and test please. Make sure you have log level set to _debug_, Please attach the _bmstoinverter.log_ here so that I can check if everything is reading the data correctly.

ai-republic, 2024-10-31:
I guess this issue can be closed as it has been inactive for over a month

closed 2024-10-31 by ai-republic