ai-republic/bms-to-inverter #55: A problem with Clean install with invertor type NONE

kovaroman, 2024-09-03:
Downloaded your release 1.0.1, unpacked.
Setting Configurator (I want only read BMS settings, no invertor running):

- General tab - selected this folder as Instalation path in Configurator, set OS Platform X86 64bit (UNIX), Log level Debug.
- BMS tab - BMS: JK_RS485, ID: 1, Port: /dev/ttyUSB0, Baudrate: 115200, Delay: 1000, Poll interval: 5
- Inverter tab - Inverter NONE, Port: 0, Baudrate: 0, Send interval: 5
- Services tab - Webserver chacked with default ports

Making Clean install I receive:

```
[INFO] --------------------------------[ jar ]---------------------------------
[WARNING] The POM for com.ai-republic.bms-to-inverter:protocol-can:jar:0.0.1-SNAPSHOT is missing, no dependency information available
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary for bms-to-inverter-parent 0.0.1-SNAPSHOT:
[INFO] 
[INFO] bms-to-inverter-parent ............................. SUCCESS [  0.320 s]
[INFO] core-api-0.0.1-SNAPSHOT ............................ SUCCESS [  4.743 s]
[INFO] protocol-rs485-0.0.1-SNAPSHOT ...................... SUCCESS [  0.714 s]
[INFO] bms-jk-rs485-0.0.1-SNAPSHOT ........................ SUCCESS [  0.873 s]
[INFO] inverter-dummy-0.0.1-SNAPSHOT ...................... FAILURE [  0.046 s]
[INFO] webserver-0.0.1-SNAPSHOT ........................... SKIPPED
[INFO] bms-to-inverter-main-0.0.1-SNAPSHOT ................ SKIPPED
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  6.912 s
[INFO] Finished at: 2024-09-03T11:05:24+03:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal on project inverter-dummy: Could not resolve dependencies for project com.ai-republic.bms-to-inverter:inverter-dummy:jar:0.0.1-SNAPSHOT: The following artifacts could not be resolved: com.ai-republic.bms-to-inverter:protocol-can:jar:0.0.1-SNAPSHOT (absent): Could not find artifact com.ai-republic.bms-to-inverter:protocol-can:jar:0.0.1-SNAPSHOT -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/DependencyResolutionException
[ERROR] 
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <args> -rf :inverter-dummy

Failed to build application!

```

Changing Port and Baudrate in Inverter tab doesn't help.

ai-republic, 2024-09-03:
Ah, please just download the _Configurator_ as described in the [Wik -How-to-usei](https://github.com/ai-republic/bms-to-inverter/wiki/How-to-use#download-the-configurator) and configure your BMS binding and as inverter binding select _NONE_.
Then click _Clean install_ and it will do everything for you :)

kovaroman, 2024-09-04:
I have done everything as is decribed in Howto, all data configured is provided here on the top comment, but Configurator fails install with this error:
`[ERROR] Failed to execute goal on project inverter-dummy: Could not resolve dependencies for project com.ai-republic.bms-to-inverter:inverter-dummy:jar:0.0.1-SNAPSHOT: The following artifacts could not be resolved: com.ai-republic.bms-to-inverter:protocol-can:jar:0.0.1-SNAPSHOT (absent): Could not find artifact com.ai-republic.bms-to-inverter:protocol-can:jar:0.0.1-SNAPSHOT -> [Help 1]`

ai-republic, 2024-09-05:
I just tested the _Configurator_ application and could not reproduce the error. Are you sure you downloaded the latest version [`Configurator application`](https://github.com/ai-republic/bms-to-inverter/blob/main/configurator/current/configurator.jar) and started it with a Java 17 with the `java -jar configurator.jar`?

ai-republic, 2024-09-08:
Did you get it working now?

ai-republic, 2024-09-15:
This issue is solved now. Please try with the latest _Configurator_ application. If you encounter further problems just open another issue 😄

closed 2024-09-15 by ai-republic