ai-republic/bms-to-inverter #56: Support for EG4 6000XP Inverter

danielbeardsley, 2024-09-06:
The [EG4 6000XP](https://signaturesolar.com/eg4-6000xp-off-grid-inverter-split-phase/) has become really popular and I'm interested in connect a DIY battery to it.

The have a ["battery compatibility list"](https://eg4electronics.com/wp-content/uploads/2024/04/EG4-6000XP-Battery-Compatibility-List.pdf) but it's not clear that they have an explicit protocol that the inverter speaks.

ai-republic, 2024-09-08:
You can definetely use the _PYLON_RS485_ binding for the inverter as it only supports RS485. That should work with the existing application :)

ai-republic, 2024-09-29:
I guess this can be closed. Please open another ticket if you need further assistance. If everything works find I'd be very thankful for any [donation](https://github.com/sponsors/ai-republic) 😄

closed 2024-09-29 by ai-republic