ai-republic/bms-to-inverter #60: Daly_RS485 Failed to initialize

jlmontes1697, 2024-09-09:
I have a raspberry py 3B with "WaveShare CAN HAT For Raspberry Pi, 2-Ch RS485 and 1-Ch CAN" (https://www.waveshare.com/rs485-can-hat-b.htm).
A Daly 7s 40A Smart BMS.
They are connected by RS485.

With the python library "python-daly-bms" I can read all the BMS parameters with the port /dev/ttySC0 but with the bms-to-inverter app I get the error: "Failed to initialize BMS"

I have the app configured:
BMS:
DALY_RS485
Id: 1
Port: /dev/ttySC0
Baudrate: 9600
Delay: 200
Inverter: HUAWEI_MODBUS (The inverter is not connected yet)
Services: HTTP

What is wrong with my configuration?

ai-republic, 2024-09-09:
Could you please attach the _config.properties_ file and if possible the log file.
Initially please configure the inverter as _NONE_ to see if you get data from the port. You can also try a `cat /dev/ttySC0` to see if data is coming through. Please also initialize your port before starting the application:
```
stty -F /dev/ttySC0 9600
```

jlmontes1697, 2024-09-09:
When the daemon is running, get this with "journalctl -f"
sc16is7xx spi1.0: ttySC0 (0): invalid RTS setting, using RTS_AFTER_SEND instead

With the app on python, can get the parameters from BMS and dont get this error.
What difference between driver java an python?

ai-republic, 2024-09-11:
Thanks for the config and the log. I noticed that the BMS id is 0. For the Daly the first BMS id needs to be set to 1

jlmontes1697, 2024-09-12:
This is fot other issue. When you set the id 1 on configuration panel, the file config.properties has value to 0. You have to change de value

ai-republic, 2024-09-15:
Btw, I fixed the issue where the BMS id is automatically set in the latest _Configurator_ application.

ai-republic, 2024-09-29:
Hi @jlmontes1697,
I guess this can be closed. Please open another ticket if you need further assistance. If everything works find I'd be very thankful for any [donation](https://github.com/sponsors/ai-republic) 😄

closed 2024-09-29 by ai-republic