ai-republic/bms-to-inverter #61: mpp inverter to lifepo4 can protocol

eugenmat, 2024-09-17:
hello, i have an inverter mpp pip max 11kw ( comunication rs485) and 2 battery H-VAC 9.6kw ( ccomunicaton can), i buy from MPP protocol for LIA CANBUS ( they sent an pdf file), how can i make an executable program to update remote panel ?

ai-republic, 2024-09-21:
Hi @eugenmat,
could you name your inverter and and battery system or BMS and possibly attach user manual and CAN/RS485 specifications?

eugenmat, 2024-09-21:
hello, my inverter is MPP SOLAR PIP 11KW MAX, battery is V-TAC VT48200B, momently i create a CAN cable for comunication but inverter don't see when battery is completly charged. I buy from MPP SOLAR protcol file for CAN comunication between the inverter and battery ( i attach file). Inverter has CAN comunication but only with WECO and SOLTARO battery. I use comunication on USER type.
[(LIA) BMS CAN communication protocol 20220410.pdf](https://github.com/user-attachments/files/17083792/LIA.BMS.CAN.communication.protocol.20220410.pdf)

eugenmat, 2024-09-25:
hello, can i use your application for can comunication to try update can on my inverter?
thanks

ai-republic, 2024-09-29:
Hi @eugenmat,
yes, you can used my application but I need to implement the LIA BMS binding first. For the inverter there's already the Pylon support. I will try and implemnent the LIA BMS binding next weekend. I've been very busy the last week

ai-republic, 2024-11-03:
@eugenmat I finally got time to implement the LIA BMS CAN binding.
Please download the _Configurator_ application and just configure the BMS and loglevel debug for testing. Don't configure an inverter or any optional services yet until we see that everything is reading fine.

ai-republic, 2025-01-05:
I will close this issue now as there hasn't been any further progress. Feel free to open another issue if you encounter problems or need support

closed 2025-01-05 by ai-republic