ai-republic/bms-to-inverter #63: Samsung battery fails test

jezza61, 2024-09-22:
Hi,
I now have the Deye  inverter connected to the Samung battery.
The Raspberry PI sends the following data:
  can1  00000351   [8]  45 02 D0 02 B0 0E C0 01
  can1  00000355   [8]  5F 00 64 00 00 00 00 00
  can1  00000356   [8]  26 16 EF FF 00 00 00 00
  can1  0000035E   [8]  00 00 00 00 00 00 00 00
  can1  0000035E   [8]  00 00 00 00 00 00 00 00
  can1  00000359   [8]  00 00 00 00 00 00 00 00
However Deye will not even look at it.
Also solar assistant fails to see it as valid data.
Is anything missing from it?
Best Regards,

Jerry

jezza61, 2024-09-23:
I have done some more testing.
The Raspberry is sending data to can IDs with id padded with 0 as seen above. This is apparently not acceptable to Deye.
When I send the same data but with canid only 351, 355, 356 and so on, it works and Deye sends a message 305, with all zero data.
Second issue in the web page the voltage is shown with % and not V.
Also the batter current seems to be dived by extra 10, as when the battery reports 31A the web page shows 3.1A.
And the 35C is not present, right?
This is charge enable, discharge enable, force charge values, this I don't believe is reported by Samsung BMS.
Is there any way of dropping the charge voltage of 58.1 down to 56.6  to reduce the risk of overcharge ?

Best Regards,

Jerry

jezza61, 2024-09-23:
Also I used this document:
https://www.studocu.com/row/document/%E5%8D%8E%E5%8D%97%E7%90%86%E5%B7%A5%E5%A4%A7%E5%AD%A6/physical-chemistry/can-bus-protocol-%E5%BE%B7%E4%B8%9Adeye-low-voltage/46210190

And this data appears to match the Deye inverter stuff.

jezza61, 2024-09-23:
And I believe that the system current is FFFF - 2 byte data from ID 500 from Samsung.
First 2 bytes show system voltage in 0.01 units, second 2 bytes show system current, but it appears that you have to deduct these bytes from FFFF and then it gives you the correct value, or at least it appears to do that.

jezza61, 2024-09-24:
Correction, CAN messages do have the correct System current value. RS485 also has the same value.
Somehow the software does not reflect the correct number.

ai-republic, 2024-09-29:
Hi @jezza61,
sorry for the late reply I'm currently very busy - also at the weekends 😒 
What I gather from your findings is that maybe the Deye inverter uses CAN 2.0A with standard length instead of CAN2.0B with extended can-id length. I will see what I can find out.
Thanks for the web-page issues, I will have a look at the values and also the %.

ai-republic, 2024-09-29:
Right, I fixed all your found issues :)
It was correct that Deye uses t he standard CAN 2.0A can-id and not the extended. I also checked the system current and fixed that too as well as the voltage being shown with the percent sign.
Please do a _Clean install_ with the _Configurator_ application and let me know if its all working now 😄

ai-republic, 2024-10-04:
@jezza61 did you get a chance to test the fixes? Please let me know if everything worked out

jezza61, 2024-10-06:
Hi,

I can confirm, that it is working now, on Deye, standard setting Lithium
battery 00, which is default, it does pick up all can messages.
I have not tested alarms, as my system is in production now.
I am going to setup another raspberry and send Samsung alarm messages to
see what is reported.

However I still have the main problem.

The current max charge voltage of 58.1 is too high for my liking, as I can
see that the inverter can potentially overcharge the battery.
Deye, does not seem to support the message to stop the charge.
Low end of battery is 44.8 which is safe, but the recommended setting is to
use 15%  to 85% of battery to maximize the life.
I have noticed a plugin for the inverter to set battery pack limits, but
there is no doco for it, and in my case it does not work, perhaps it is not
meant for it?

So  is there any way of changing at the very least the Max charge voltage
parameter, to something like 56.6V instead of 58.1 ?   56.6 appears to be
85-90% according to the BMS.
Just this one parameter would make my day.
I have also included CAN log and RS485 Modbus log from the BMS.
This time it has a full compliment of 16 battery packs, and messages for
all cells and packs.
The manual I provided previously does not explain all addresses, but the
can messages appear to be OK.

I did look in your code where you send the Charge volt limit message,
however modifying it by myself is a bit much for my skills :)

Best Regards,

Jerry

NO    Direction  Time Scale        Frame Type      Frame Format        Frame ID    Length          Data 
0      Receive      19:52:59:753      Data frame      Standard frame      00000500      8      8f 14 e9 ff 3f 64 aa 11       
1      Receive      19:52:59:758      Data frame      Standard frame      00000501      8      00 00 00 00 10 10 00 00       
2      Receive      19:52:59:761      Data frame      Standard frame      00000502      8      45 02 60 1d 60 1d c0 01       
3      Receive      19:52:59:765      Data frame      Standard frame      00000503      8      bd 0e c2 0e b6 0e 8f 04       
4      Receive      19:52:59:769      Data frame      Standard frame      00000504      8      a0 14 83 14 16 18 15 00       
5      Receive      19:52:59:771      Data frame      Standard frame      00000505      8      01 03 00 00 00 00 00 00       
6      Receive      19:52:59:774      Data frame      Standard frame      00000510      8      84 14 6c ff 3f 64 00 00       
7      Receive      19:52:59:779      Data frame      Standard frame      00000511      8      00 00 bf 0e b9 0e 18 18       
8      Receive      19:52:59:783      Data frame      Standard frame      00000512      8      00 00 00 00 01 02 00 00       
9      Receive      19:52:59:786      Data frame      Standard frame      00000514      8      9b 14 6c ff 40 64 00 00       
10     Receive      19:52:59:788      Data frame      Standard frame      00000515      8      00 00 c0 0e ba 0e 18 17       
11     Receive      19:52:59:791      Data frame      Standard frame      00000516      8      00 00 00 00 01 02 00 00       
12     Receive      19:52:59:796      Data frame      Standard frame      00000518      8      a0 14 6d ff 40 64 00 00       
13     Receive      19:52:59:801      Data frame      Standard frame      00000519      8      00 00 c1 0e bc 0e 18 17       
14     Receive      19:52:59:803      Data frame      Standard frame      0000051a      8      00 00 00 00 01 02 00 00       
15     Receive      19:52:59:806      Data frame      Standard frame      0000051c      8      9a 14 6d ff 40 64 00 00       
16     Receive      19:52:59:810      Data frame      Standard frame      0000051d      8      00 00 bf 0e ba 0e 17 17       
17     Receive      19:52:59:813      Data frame      Standard frame      0000051e      8      00 00 00 00 01 02 00 00       
18     Receive      19:52:59:816      Data frame      Standard frame      00000520      8      84 14 6d ff 40 64 00 00       
19     Receive      19:52:59:819      Data frame      Standard frame      00000521      8      00 00 c0 0e bb 0e 17 16       
20     Receive      19:52:59:825      Data frame      Standard frame      00000522      8      00 00 00 00 01 02 00 00       
21     Receive      19:52:59:830      Data frame      Standard frame      00000524      8      97 14 6c ff 40 64 00 00       
22     Receive      19:52:59:836      Data frame      Standard frame      00000525      8      00 00 bf 0e ba 0e 17 16       
23     Receive      19:52:59:840      Data frame      Standard frame      00000526      8      00 00 00 00 01 02 00 00       
24     Receive      19:52:59:926      Data frame      Standard frame      00000528      8      8c 14 6c ff 40 64 00 00       
25     Receive      19:52:59:932      Data frame      Standard frame      00000529      8      00 00 c2 0e bc 0e 16 16       
26     Receive      19:52:59:938      Data frame      Standard frame      0000052a      8      00 00 00 00 01 02 00 00       
27     Receive      19:52:59:944      Data frame      Standard frame      0000052c      8      83 14 6c ff 40 64 00 00       
28     Receive      19:52:59:949      Data frame      Standard frame      0000052d      8      00 00 c1 0e bb 0e 15 15       
29     Receive      19:52:59:955      Data frame      Standard frame      0000052e      8      00 00 00 00 01 02 00 00       
30     Receive      19:52:59:960      Data frame      Standard frame      00000530      8      89 14 6d ff 3f 64 00 00       
[... cut ...]
494    Receive      19:53:05:403      Data frame      Standard frame      00000500      8      8f 14 e8 ff 3f 64 b2 11       
495    Receive      19:53:05:481      Data frame      Standard frame      00000501      8      00 00 00 00 10 10 00 00       
496    Receive      19:53:05:486      Data frame      Standard frame      00000502      8      45 02 60 1d 60 1d c0 01       
497    Receive      19:53:05:491      Data frame      Standard frame      00000503      8      bc 0e c2 0e b6 0e 8f 04       
498    Receive      19:53:05:497      Data frame      Stan

jezza61, 2024-10-06:
One more note, I have noticed that switching over to lithium battery in
Deye settings now started using the grid power to charge it up, as at
present it is 61%.
It is late at night, and on Volt setting in Deye, it never uses grid to
charge the battery, only uses solar.
Perhaps I need to look at Deye programming for that.
Not sure if it is a function of battery reported voltages or SOC state?
I am going to switch it over to volt mode for now, as I have 2 days of
stored power in the battery, and I do not want to top it up from the grid.

Best Regards,

Jerry

ai-republic, 2024-10-06:
Hi @jezza61,
great to hear its working now 😄 
For your problem with the max charge I can write a small plugin to set some parameters like max and min charge voltages/currents. I'll see that I get that done next weekend.
For your grid charging problem: that you need to set in the Deye inverter settings either via Deye cloud or on the inverter itself.
In Deye cloud it looks like this:
[image]
There you can check the _Grid_ and set the times and voltage levels to charge via grid. Hope that helps 😄

jezza61, 2024-10-06:
Thanks for that.
I have not yet changed any grid stuff config on Deye as yet, as at present
it is OK.
Obviously CAN messages from BMS change this.  more reading is required for
me then.
I am playing with node-red at present to see what CAN messages the Samsung
BMS is sending to individual packs, so I can use some of these in
automation.
That plugin would be great, because it would free me up from worrying about
BMS and Inverter connection.

Best Regards,

Jerry

ai-republic, 2024-10-13:
Hi @jezza61,
I implemented the `ManipulateBatteryPackDataPlugin` plugin that should serve your purpose to manipulate max charge/discharge voltages and currents. You'll need to download the latest _Configurator_ and do a clean install. 
Please let me know if it works for you 😄

jezza61, 2024-10-13:
Hi,I have done a clean install with the plugin.No change.The Max voltage, is still as received from BMS.I did set it to 56.8 in plugin, but it is being sent as 58.1 which is the actual bms limit.Best regardsJerrySent from my Galaxy

ai-republic, 2024-10-13:
@jezza61 found the issue. It was not overwritting the defaults. Fixed it. Please can you do a _Clean install_ and retest?

jezza61, 2024-10-14:
So this time, there is no transmission at all of 351 id or any transmission.
Candump still shows the BMS data, but all output has stopped.
I will try to re-install again?

Best Regards,

Jerry

jezza61, 2024-10-14:
OK, after re-installing again, I had to move stuff from can0 to can1, and
then back to can0 and then it started working.
I am going to reconfigure the inverter and see what result it produces.
However at this time the program does exactly what I wanted. It translates
messages, and allows me to limit max values.

Thanks,

Jerry

jezza61, 2024-10-14:
One more question if possible.The can I'd 35C. It is not used here, right?Bit 7 and 6 are controlling charge and discharge right?I can't see any output for these?Best RegardsJerrySent from my Galaxy

ai-republic, 2024-10-14:
@jezza61 yes, bit 6/7 (discharge/charge) tells the Deye inverver the state of the dis/charging breaker. The Samsung CAN specification actually doesn't reflect this. There is no data for that.
What would you like to do?

jezza61, 2024-10-14:
I have received a full set of rs485 and can specs for my inverter from Deye.I am going to check. I believe one of the registers for Samsung does have these settings.I would need to log the can to see the samsung messages.However at this time the inverter stopped the discharge at 10% exactly as it was configured.We need some sunlight to check the top end of charge. Last 2 weeks we have mostly rain, so I never get 100%.Best regardsJerrySent from my Galaxy

ai-republic, 2024-10-31:
@jezza61 any updates?

jezza61, 2024-10-31:
Not, yet.So far we did not have enough sun to test the top end charge, and the plugin with the limit.So far it only got to 80% charge.However I got It configured to autostart.So it starts automatically, and just works .Very soon we are going to add 6kw of extra panels, to make sure we can fill up the battery even on cloudy days.However so far the program is working perfectly.Best regardsJerrySent from my Galaxy

ai-republic, 2024-11-03:
So I will close this ticket for now. If you encounter any problems please let me know

closed 2024-11-03 by ai-republic

jezza61, 2024-11-03:
Yes please.
I will do more tests once I have my new panels. Then I should be able to
hit the top end of the battery and see how it performs.
Unfortunately it appears that Deye does not have register 35C, where you
can force the inverter to do stuff.
However at present the whole thing works just fine.
Thanks a lot for your help.

Best regards

Jerry