ai-republic/bms-to-inverter #67: I need Narada 48NPFC200 to work with Growatt SPF5000ES, hope you can help.

Tangbunna, 2024-09-27:
Hi, I need narada 48npfc50/100/200 BMS via RS485 working too.
I have no idea about this documents if it is really refer to NPFC series, but it should help some.  

https://www.scribd.com/document/499960699/Narada-BMS-Modbus-V1-6

ai-republic, 2024-09-29:
Hi @Tangbunna,
thanks for the specification. I will see if I can start implementing that next weekend 😄

ai-republic, 2024-10-20:
Hi @Tangbunna,
I've added the _NARADA_MODBUS_ BMS binding now. Please download the latest _Configurator_ application and you will find it in the BMS dropdown list.

ai-republic, 2024-10-31:
@Tangbunna did you have a chance to test the _NARADA_MODBUS_ BMS binding?

Tangbunna, 2024-11-01:
Hi, Thanks for your effort, I have my RP3 broken due too power supply over
voltage. I am waiting for another new RP delivered. Thanks again and will
let you know right after that

ai-republic, 2025-01-05:
I will close this issue now as there hasn't been any further progress. Feel free to open another issue if you encounter problems or need support

closed 2025-01-05 by ai-republic