ai-republic/bms-to-inverter #68: Support for EG4 18kPV

opebakare, 2024-09-28:
I have an **EG4 18kPV** with two **SEPLOS** 280AH batteries, and there has always been some missing data between the two, despite setting the batteries to use PylonTech protocol. I'm happy to be a tester to connect via CAN to inverter, and RS485 to the batteries for full compatibility.

ai-republic, 2024-09-29:
Hi @opebakare,
it would be great if you could test the Seplos BMS binding. Just set the BMS binding to _SEPLOS_ and configure your BMS settings (baudrate, etc.) and set the inverter binding to _NONE_. Also set the log level to _DEBUG_.
Please share the log file here then I can analyze if everything is working 😄

opebakare, 2024-10-06:
Got the CAN hat for RPi-5 today, and will be testing tomorrow. More to come!

ai-republic, 2024-10-06:
@opebakare Great! What CAN hat did you get? I had issues with my Waveshare 2-CH-CAN hat on my RPi 5 and couldn't get it to work. Let me know how it works out for you!

opebakare, 2024-10-06:
Small issue - the Seplos dropdown is a CAN; how would I set it up via RS485 since I have a USB->RS485 device?

opebakare, 2024-10-06:
> @opebakare Great! What CAN hat did you get? I had issues with my Waveshare 2-CH-CAN hat on my RPi 5 and couldn't get it to work. Let me know how it works out for you!

I got the Waveshare RS485 CAN HAT from Amazon, and dmesg shows it's active and working.
[image]

ai-republic, 2024-10-07:
> Small issue - the Seplos dropdown is a CAN; how would I set it up via RS485 since I have a USB->RS485 device?

For the Seplos BMS there is currently only the CAN binding available. If you have a RS485 communication specification for the Seplos BMS I can implement that binding.

closed 2024-10-31 by ai-republic