ai-republic/bms-to-inverter #70: DalyBMS RS485 to Deye RS485

nicodier, 2024-10-07:
Hello, 

I'm trying to use your software to get data from DalyBMS to a Deye inverter using RS485 (Pylon protocol).
Getting data from the Daly BMS'es works, and I can also see this data in the integrated webserver.
However, I get an error when data is sent to the inverter:
`2024-10-06 13:10:06.609 | INFO  | Thread-13  | verter.BmsToInverter:294   | Sending to inverter PYLON_RS485 on /dev/ttyACM0...
2024-10-06 13:10:06.616 | ERROR | Thread-13  | verter.core.Inverter:190   | Failed to send frame to inverter:Buffer (HEX): [null]

java.nio.BufferUnderflowException: null
        at java.nio.HeapByteBuffer.get(HeapByteBuffer.java:190) ~[?:?]
        at com.airepublic.bmstoinverter.inverter.pylon.rs485.PylonInverterRS485Processor.createSystemInfo(PylonInverterRS485Processor.java:62) ~[inverter-pylon-rs485-0.0.1-SNAPSHOT.jar:?]
        at com.airepublic.bmstoinverter.inverter.pylon.rs485.PylonInverterRS485Processor.createSendFrames(PylonInverterRS485Processor.java:40) ~[inverter-pylon-rs485-0.0.1-SNAPSHOT.jar:?]
        at com.airepublic.bmstoinverter.core.Inverter.process(Inverter.java:169) ~[core-api-0.0.1-SNAPSHOT.jar:?]
        at com.airepublic.bmstoinverter.inverter.pylon.rs485.PylonInverterRS485Processor$Proxy$_$$_WeldClientProxy.process(Unknown Source) ~[inverter-pylon-rs485-0.0.1-SNAPSHOT.jar:?]
        at com.airepublic.bmstoinverter.BmsToInverter.lambda$start$4(BmsToInverter.java:295) ~[bms-to-inverter-main-0.0.1-SNAPSHOT.jar:?]
        at java.lang.Thread.run(Thread.java:1583) [?:?]`

I have already tried to comment out this line `frames.add(createSystemInfo(aggregatedPack)); // 0x60`, then it does seem to send data and I no longer get the error, but on the Deye I don't see anything.
The Deye is set to Lithium Mode 12 = Pylon protocol.
When I open a screen to /dev/ttyACM0, I do get some strings from the inverter like this:
~20124661E00212FD30
~20424661E00242FD2A
~20624661E00262FD26
~20824661E00282FD22
~20224662E00222FD2D

I have also purchased a Waveshare USB-CAN-A adapter to use the Deye CAN protocol, but this adapter is presenting itself as a ttyUSB0 device. Also tried to convert this to a network device with slcand, but could't get it to send data.

I prefer to get it working with RS485. Can you help me to get this running?

ai-republic, 2024-10-08:
Hi  @nicodier,
sure, I will have a look at it next weekend as I'm quite busy atm during workdays.

nicodier, 2024-10-09:
Hello @ai-republic, thanks for your reply.
I finally got it to work using another USB-CAN adapter (DSD Tech SH-C30A), but some values are not properly calculated.
For example: charge voltage, average temperature, max charge and discharge current are displayed wrong when all BMS'es get aggregated.
What is working correctly: SoC, Total voltage, Total current

I'm using 4 Daly BMS's.

I will post the JSON data from the webserver here later so we can investigate and find out the best solution.

ai-republic, 2024-10-10:
Ah good to hear!
Yes, please just delete the log-file and configure log-level _info_ and without webserver activated.
That way I can see the values from the BMS easily. Are the values only displayed wrongly in the webserver? If so, please also attach a screenshot of the browser showing the faulty values.

ai-republic, 2024-10-13:
Btw, I also split the logging now to 3 logs: bmstoinverter, webserver and mqtt to make the logs more readable

ai-republic, 2024-10-31:
@nicodier any updates?

ai-republic, 2024-11-10:
I guess I can close this issue as there has been no news since over a month. Feel free to open another issue if you need help 😄

closed 2024-11-10 by ai-republic