ai-republic/bms-to-inverter #72: Huawei inverter connects to third-party battery BMS

930680849, 2024-10-16:
Spend Huawei Inverter Connected to DIY Battery BMS?
[image]
[image]

ai-republic, 2024-10-17:
Yes, that is supported. What is you're issue?

930680849, 2024-10-17:
华为需要DCDC高压电池

ai-republic, 2024-10-18:
Yes, the Huawei Inverter uses HV. So either you have a HV battery or you can put batteries in series to match the required voltage, e.g. put 10x48V in series to achieve 480V.

930680849, 2024-10-18:
> 是的，华为逆变器使用HV。因此，您要么有高压电池，要么您可以将电池串联以匹配所需的电压，例如将10x48V串联以达到480V。

我这里有DCDC模块 380-48双向

930680849, 2024-10-18:
> 是的，华为逆变器使用HV。因此，您要么有高压电池，要么您可以将电池串联以匹配所需的电压，例如将10x48V串联以达到480V。

能否增加pace bms？

ai-republic, 2024-10-18:
I believe you will need a BMS that supports your voltage needs (i.e. 450V or 600V or similar). You cannot put i.e. 48V BMSes in series as the voltage will increase for each battery pack.

930680849, 2024-10-18:
> 我相信您需要一个支持您电压需求的 BMS（即 450V 或 600V 或类似电压）。您不能将 48V BMS 串联起来，因为每个电池组的电压都会增加。

需要你的协议支持 
[image]

930680849, 2024-10-18:
能否增加pace485bms？

ai-republic, 2024-10-19:
What's the image above? It that your Pace BMS? What do you mean with 'increased'? You mean to tweak a Pace BMS from 48V to 480V? I don't know anything about that and I don't think its possible.

930680849, 2024-10-19:
pace 480-48 48-480需要注入华为协议

ai-republic, 2024-10-20:
I'm sorry but I can't make any sense of the translations. You'll need to go into more detail and be more specific what you intend to do and how.

930680849, 2024-10-20:
留下个邮箱

ai-republic, 2024-10-20:
If you need assistance with the application, please post it here. I cannot give you tips on how to tweak your BMS hardware from 48V to 480V.

ai-republic, 2024-10-31:
Closing this issue as it has been inactive for over 2 weeks

closed 2024-10-31 by ai-republic