ai-republic/bms-to-inverter #73: Working with RENAC ESC5000 - no mqtt

windmillBart, 2024-10-20:
Hi, 
Firstly, thanks a lot for sharing this code! it was a trouble to get my DIY battery bank interfacing with my RENAC inverter. Now it works !
I used BMS Daly RS485 and PYLON-CAN protocols.

I just might have made a error with the mqtt service; When the script is executed I can see an error:  "this.mqttProducer" is null

java.lang.NullPointerException: Cannot invoke "com.airepublic.bmstoinverter.core.service.IMQTTProducerService.sendMessage(String)" because "this.mqttProducer" is null
        at com.airepublic.bmstoinverter.BmsToInverter.receivedData(BmsToInverter.java:342) ~[bms-to-inverter-main-0.0.1-SNAPSHOT.jar:?]
        at com.airepublic.bmstoinverter.BmsToInverter.lambda$start$1(BmsToInverter.java:269) ~[bms-to-inverter-main-0.0.1-SNAPSHOT.jar:?]
        at com.airepublic.bmstoinverter.core.BMS.process(BMS.java:199) ~[core-api-0.0.1-SNAPSHOT.jar:?]
        at com.airepublic.bmstoinverter.BmsToInverter.lambda$start$2(BmsToInverter.java:269) ~[bms-to-inverter-main-0.0.1-SNAPSHOT.jar:?]
        at java.lang.Thread.run(Thread.java:840) [?:?]
        
 I'm running a Debian 12 server, do I need to install something extra to make the mqtt producer to work? 
 
 thanks !

windmillBart, 2024-10-20:
this is my mqtt config
# Activate the MQTT producer if you want to send your BMS data to a MQTT broker, e.g. HomeAssistant
mqtt.producer.enabled=true
# The URL of the external MQTT broker to send to
mqtt.producer.locator=http://192.168.0.126:8123

ai-republic, 2024-10-21:
Hi @windmillBart,
great to hear the application is working for your BMS and inverter. I'll have to check your issue with the MQTT producer. Which MQTT broker are you trying to connect to? Have you created / or is auto-creation enabled, in your MQTT broker? Are the credentials set for your MQTT broker (if any)?

windmillBart, 2024-10-25:
hi,
I was trying to connect to Home Assistant mosquitto broker, credentials are set in the broker, and client should be added automatically. Nevertheless, in the mean time I have used a separete wire connection and USB - UART connector, connected with the UART2 on the BMS and using the dreadnought daly-bms-cli library to read the parameters from my bms.
So, that solved my problem. Thanks for looking into it and apologies not providing the feedback earlier.

ai-republic, 2024-10-25:
Great thanks! Will close this issue then.

closed 2024-10-25 by ai-republic