ai-republic/bms-to-inverter #79: Tian power BMS is reading Amperage incorrectly

aren, 2024-11-05:
According to my inverter, my two batteries are producing ~650W of power right now. At 52 volts, that's 12.5 amps, or 6.25 amps each.

But the logs are showing 648 Amps per battery. That can't be right.

```
2024-11-05 08:31:07.719 | DEBUG | Thread-10  | o.ModbusRTUTransport:266   | Sent: 01 04 10 00 00 17 B4 C4
2024-11-05 08:31:07.789 | DEBUG | Thread-10  | o.ModbusRTUTransport:443   | Response: 01 04 2E 14 57 FD 44 1A CD 00 C5 00 D2 00 00 00 00 1E 00 02 8C 03 E8 29 16 00 0C 27 10 0C B8 0C B5 00 00 00 C8 00 BE 00 DC 00 00 15 E0 27 10 00 00 22 E9
2024-11-05 08:31:07.789 | INFO  | Thread-10  | verter.BmsToInverter:320   |
BMS     SOC       V       A      CellMinV        CellMaxV       CellDiff
#1       65.2   52.0    648.3   3.253(#0)       3.256(#0)       0.003
#2       64.2   52.0    648.4   3.253(#0)       3.256(#0)       0.003

2024-11-05 08:31:07.789 | INFO  | Thread-10  | verter.BmsToInverter:390   | BMS alarms:
        NONE
2024-11-05 08:31:07.789 | INFO  | Thread-10  | verter.BmsToInverter:268   | Reading BMS #2 TIAN_MODBUS on /dev/ttyAMA0...
2024-11-05 08:31:07.839 | DEBUG | Thread-10  | o.ModbusRTUTransport:266   | Sent: 02 04 10 00 00 17 B4 F7
2024-11-05 08:31:07.911 | DEBUG | Thread-10  | o.ModbusRTUTransport:443   | Response: 02 04 2E 14 57 FD 4D 1A 11 00 C8 00 D2 00 00 00 00 1E 00 02 82 03 E8 28 98 00 24 27 10 0C B9 0C B3 00 00 00 C8 00 C8 00 DC 00 00 15 E0 27 10 00 00 4F CA
2024-11-05 08:31:07.911 | INFO  | Thread-10  | verter.BmsToInverter:320   |
BMS     SOC       V       A      CellMinV        CellMaxV       CellDiff
#1       65.2   52.0    648.3   3.253(#0)       3.256(#0)       0.003
#2       64.2   52.0    648.4   3.251(#0)       3.257(#0)       0.006
```

aren, 2024-11-05:
Ah here's the issue. The bytes are FD 4D. The code is interpreting them as unsigned short, which is 64845. However, the TianPower BMS docs show this is a signed short, so it should be -691 (or 6.91A).

ai-republic, 2024-11-10:
Hi @aren,
could you attach the Tian ModBus spec again. I can't seem to find it - thanks!
I'm wondering if the registers are actaully all 2 bytes long (short or char) and not 4 byte (int). I've got a lot of specs that are basically the same as the Tian and all use 2 byte registers which are also really the ModBus standard.

aren, 2024-11-10:
See https://github.com/ai-republic/bms-to-inverter/discussions/71

ai-republic, 2024-11-10:
Thanks, I found that in my specs collection but wasn't sure which manufacturer that belongs to 😉 
Here it actually states the registers are holding 2 bytes - hmm, where did I see that with the 4 bytes 🤔

ai-republic, 2024-11-10:
I changed it according to the spec. Could you do a _Clean install_ and check what the Tian BMS binding comes up with?

aren, 2024-11-14:
It's way off now.

```
2024-11-13 19:44:07.899 | DEBUG | Thread-10  | o.ModbusRTUTransport:443   | Response: 02 04 2E 14 82 00 00 15 A4 00 AF 00 C8 00 00 00 00 1C 00 03 D3 02 30 16 19 00 1A 27 10 0C D3 0C CF 00 00 00 B4 00 AA 00 C8 00 00 15 E0 27 10 00 00 07 AA
2024-11-13 19:44:07.900 | INFO  | Thread-10  | verter.BmsToInverter:320   |
BMS	SOC	  V  	  A  	 CellMinV 	 CellMaxV	CellDiff
#1	 0.0	0.0	52.5	0.0(#0) 	0.0(#0)	0.0
#2	 0.0	0.0	52.5	0.0(#0) 	0.0(#0)	0.0

2024-11-13 19:44:07.900 | INFO  | Thread-10  | verter.BmsToInverter:388   | BMS alarms:
	BMS #1:	WARNING -> PACK_TEMPERATURE_HIGH
	BMS #1:	WARNING -> PACK_TEMPERATURE_LOW
	BMS #1:	WARNING -> DISCHARGE_CURRENT_HIGH
	BMS #1:	WARNING -> CHARGE_MODULE_TEMPERATURE_HIGH
	BMS #2:	ALARM -> FAILURE_SENSOR_PACK_TEMPERATURE
	BMS #2:	WARNING -> ENCASING_TEMPERATURE_HIGH
	BMS #2:	WARNING -> PACK_TEMPERATURE_LOW
	BMS #2:	WARNING -> DISCHARGE_CURRENT_HIGH
	BMS #2:	WARNING -> CHARGE_MODULE_TEMPERATURE_HIGH
	BMS #2:	ALARM -> FAILURE_COMMUNICATION_INTERNAL
```

ai-republic, 2024-11-14:
oh 😮 - let me change it back and try the s_int_16

ai-republic, 2024-11-14:
@aren please try now with a _Clean install_

aren, 2024-12-19:
It still isn't reading signed shorts correctly.

```
2024-12-19 07:00:38.411 | INFO  | Thread-10  | verter.BmsToInverter:268   | Reading BMS #1 TIAN_MODBUS on /dev/ttyAMA0...
2024-12-19 07:00:38.535 | INFO  | Thread-10  | verter.BmsToInverter:320   |
BMS     SOC       V       A      CellMinV        CellMaxV       CellDiff
#1       46.1   50.2    653.0   3.0(#0)         3.178(#0)       0.178
#2       48.6   50.2    653.3   3.091(#0)       3.178(#0)       0.087
```

ai-republic, 2024-12-21:
ok, this looks better again though. So the off value is the current  (A) I guess. What would the expected values be? Maybe its in the wrong unit? Could it be 6.5A or 65.0A?

aren, 2024-12-21:
Yeah. Take this response for example:

`02 04 2E 14 57 FD 4D 1A 11 00 C8 00 D2 00 00 00 00 1E 00 02 82 03 E8 28 98 00 24 27 10 0C B9 0C B3 00 00 00 C8 00 C8 00 DC 00 00 15 E0 27 10 00 00 4F CA`

`02 04 ` -> Response and type
`2E` -> length of response (46 bytes)
`14 57` -> field #1, which is battery pack voltage. value is 5207 or 52.07v
 `FD 4D` -> field #2, which is battery pack current. value is 64845 (648.45A) in unsigned int, but -691 (-6.91A) as a signed int. -6.91A is correct, because the battery pack is discharging.

Field #2, `FD4D` needs to be interpreted as a signed in (using 2's complement), not an unsigned int.

ai-republic, 2024-12-22:
Great, I will check these later and provide a patch

ai-republic, 2024-12-26:
Could you update (_Clean install_ in the latest  _Configurator_ app) and retest please?

aren, 2024-12-27:
Yes it's good now. Thanks!

```
2024-12-27 12:42:24.687 | INFO  | Thread-10  | verter.BmsToInverter:320   |
BMS	SOC	  V  	  A  	 CellMinV 	 CellMaxV	CellDiff
#1	 99.9	52.7	-6.0	3.298(#0) 	3.301(#0)	0.003
#2	 99.9	52.7	-5.9	3.297(#0) 	3.301(#0)	0.004
```

ai-republic, 2025-01-05:
I guess this issue can be closed then :)

closed 2025-01-05 by ai-republic