ai-republic/bms-to-inverter #81: Daly BMS to Solis inverter.

DMIINC, 2024-11-10:
As a title say. What can be done and what would you need?
Nice project you got going. Sorry if I did not read everything and possibly asking this question the wrong way.

Thanks,

ai-republic, 2024-11-11:
Which communication (CAN/RS485/ModBus) does your Solis inverter support?

DMIINC, 2024-11-11:
CAN/RS485 
Solis S6-EH1P11.4K-H-US
I do have what appears to be  communication manual
Thank you

ai-republic, 2024-11-14:
@DMIINC could you please attach the communications protocol so that I can check?

DMIINC, 2024-11-14:
[RS485_MODBUS Solis PV Grid-Connected Inverter.pdf](https://github.com/user-attachments/files/17747358/RS485_MODBUS.Solis.PV.Grid-Connected.Inverter.pdf)

Thank you

ai-republic, 2024-11-16:
Hi @DMIINC,
I had a look at the communication specification but it does not specify any BMS data that is passed.
If had a look at the inverter data sheet and the battery compatibility sheet:
[Compatibility_List_Solis_S6_Hybrid_v2,4_08,23,23.pdf](https://github.com/user-attachments/files/17782942/Compatibility_List_Solis_S6_Hybrid_v2.4_08.23.23.pdf)

This states that it supports the Pylontech. Have you tried to configure the _PYLON_HV_CAN_ binding for your inverter? I think that should work. You might have to do a corresponding configuration in your Solis inverter.

DMIINC, 2024-11-19:
Ok. Will try here in day or two. I will report my findings here. Thank you .

DMIINC, 2024-11-22:
2024-11-21 20:33:27.904 | INFO  | Thread-2   | toinverter.core.Port:111   | Opening can_0 ...
2024-11-21 20:33:27.905 | ERROR | Thread-2   | toinverter.core.Port:116   | Opening port can_0 FAILED!
tel.schich.javacan.platform.linux.LinuxNativeOperationException: Failed to resolve the interface: can_0 - errorNumber=19, errorMessage='No such device'
        at tel.schich.javacan.platform.linux.LinuxNetworkDevice.resolveInterfaceName(Native Method) ~[javacan-core-3.2.4.jar:?]
        at tel.schich.javacan.platform.linux.LinuxNetworkDevice.lookup(LinuxNetworkDevice.java:70) ~[javacan-core-3.2.4.jar:?]
        at tel.schich.javacan.NetworkDevice.lookup(NetworkDevice.java:53) ~[javacan-core-3.2.4.jar:?]
        at tel.schich.javacan.CanChannels.newRawChannel(CanChannels.java:73) ~[javacan-core-3.2.4.jar:?]
        at com.airepublic.bmstoinverter.protocol.can.JavaCANPort.open(JavaCANPort.java:60) ~[protocol-can-0.0.1-SNAPSHOT.jar:?]
        at com.airepublic.bmstoinverter.core.Port.ensureOpen(Port.java:112) ~[core-api-0.0.1-SNAPSHOT.jar:?]
        at com.airepublic.bmstoinverter.protocol.can.JavaCANPort.sendExtendedFrame(JavaCANPort.java:100) ~[protocol-can-0.0.1-SNAPSHOT.jar:?]
        at com.airepublic.bmstoinverter.bms.daly.can.DalyBmsCANProcessor.sendMessage(DalyBmsCANProcessor.java:47) ~[bms-daly-can-0.0.1-SNAPSHOT.jar:?]
        at com.airepublic.bmstoinverter.bms.daly.common.AbstractDalyBmsProcessor.collectData(AbstractDalyBmsProcessor.java:74) ~[bms-daly-common-0.0.1-SNAPSHOT.jar:?]
        at com.airepublic.bmstoinverter.core.BMS.process(BMS.java:173) ~[core-api-0.0.1-SNAPSHOT.jar:?]
        at com.airepublic.bmstoinverter.BmsToInverter.lambda$start$2(BmsToInverter.java:269) ~[bms-to-inverter-main-0.0.1-SNAPSHOT.jar:?]
        at java.lang.Thread.run(Thread.java:840) [?:?]
2024-11-21 20:33:27.906 | DEBUG | Thread-2   | ocol.can.JavaCANPort:115   | CAN frame sending: Buffer (HEX): [0x40, 0x02, 0x90, 0x18, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-11-21 20:33:27.907 | ERROR | Thread-2   | stoinverter.core.BMS:192   | Error requesting data!
java.lang.NullPointerException: Cannot invoke "tel.schich.javacan.RawCanChannel.write(tel.schich.javacan.CanFrame)" because "this.canChannel" is null
        at com.airepublic.bmstoinverter.protocol.can.JavaCANPort.sendExtendedFrame(JavaCANPort.java:117) ~[protocol-can-0.0.1-SNAPSHOT.jar:?]
        at com.airepublic.bmstoinverter.bms.daly.can.DalyBmsCANProcessor.sendMessage(DalyBmsCANProcessor.java:47) ~[bms-daly-can-0.0.1-SNAPSHOT.jar:?]
        at com.airepublic.bmstoinverter.bms.daly.common.AbstractDalyBmsProcessor.collectData(AbstractDalyBmsProcessor.java:74) ~[bms-daly-common-0.0.1-SNAPSHOT.jar:?]
        at com.airepublic.bmstoinverter.core.BMS.process(BMS.java:173) ~[core-api-0.0.1-SNAPSHOT.jar:?]
        at com.airepublic.bmstoinverter.BmsToInverter.lambda$start$2(BmsToInverter.java:269) ~[bms-to-inverter-main-0.0.1-SNAPSHOT.jar:?]
        at java.lang.Thread.run(Thread.java:840) [?:?]

Tried can0 also but no dice. What you think. Thanks, Using board from your recommend list Waveshare 2CH-CAN hats. Anything special for it?

DMIINC, 2024-11-22:
2024-11-21 21:44:59.909 | INFO  | Thread-2   | toinverter.core.Port:111   | Opening can0 ...
2024-11-21 21:44:59.910 | ERROR | Thread-2   | toinverter.core.Port:116   | Opening port can0 FAILED!
tel.schich.javacan.platform.linux.LinuxNativeOperationException: Failed to resolve the interface: can0 - errorNumber=19, errorMessage='No such device'
	at tel.schich.javacan.platform.linux.LinuxNetworkDevice.resolveInterfaceName(Native Method) ~[javacan-core-3.2.4.jar:?]
	at tel.schich.javacan.platform.linux.LinuxNetworkDevice.lookup(LinuxNetworkDevice.java:70) ~[javacan-core-3.2.4.jar:?]
	at tel.schich.javacan.NetworkDevice.lookup(NetworkDevice.java:53) ~[javacan-core-3.2.4.jar:?]
	at tel.schich.javacan.CanChannels.newRawChannel(CanChannels.java:73) ~[javacan-core-3.2.4.jar:?]
	at com.airepublic.bmstoinverter.protocol.can.JavaCANPort.open(JavaCANPort.java:60) ~[protocol-can-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.core.Port.ensureOpen(Port.java:112) ~[core-api-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.protocol.can.JavaCANPort.sendExtendedFrame(JavaCANPort.java:100) ~[protocol-can-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.bms.daly.can.DalyBmsCANProcessor.sendMessage(DalyBmsCANProcessor.java:47) ~[bms-daly-can-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.bms.daly.common.AbstractDalyBmsProcessor.collectData(AbstractDalyBmsProcessor.java:74) ~[bms-daly-common-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.core.BMS.process(BMS.java:173) ~[core-api-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.BmsToInverter.lambda$start$2(BmsToInverter.java:269) ~[bms-to-inverter-main-0.0.1-SNAPSHOT.jar:?]
	at java.lang.Thread.run(Thread.java:840) [?:?]
2024-11-21 21:44:59.911 | DEBUG | Thread-2   | ocol.can.JavaCANPort:115   | CAN frame sending: Buffer (HEX): [0x40, 0x02, 0x90, 0x18, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-11-21 21:44:59.912 | ERROR | Thread-2   | stoinverter.core.BMS:192   | Error requesting data!
java.lang.NullPointerException: Cannot invoke "tel.schich.javacan.RawCanChannel.write(tel.schich.javacan.CanFrame)" because "this.canChannel" is null
	at com.airepublic.bmstoinverter.protocol.can.JavaCANPort.sendExtendedFrame(JavaCANPort.java:117) ~[protocol-can-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.bms.daly.can.DalyBmsCANProcessor.sendMessage(DalyBmsCANProcessor.java:47) ~[bms-daly-can-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.bms.daly.common.AbstractDalyBmsProcessor.collectData(AbstractDalyBmsProcessor.java:74) ~[bms-daly-common-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.core.BMS.process(BMS.java:173) ~[core-api-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.BmsToInverter.lambda$start$2(BmsToInverter.java:269) ~[bms-to-inverter-main-0.0.1-SNAPSHOT.jar:?]
	at java.lang.Thread.run(Thread.java:840) [?:?]
^C2024-11-21 21:45:00.175 | INFO  | Thread-1   | verter.BmsToInverter:426   | Shutting down BMS and inverter threads...FAILED
2024-11-21 21:45:00.179 | INFO  | Thread-1   | verter.BmsToInverter:458   | Shutting down ports...OK
Weld SE container 162ddb10-1104-431e-a516-81639b049a92 shut down by shutdown hook
dalibor@powerwall2:~ $ ^C
dalibor@powerwall2:~ $

DMIINC, 2024-11-22:
Or how would you setup this board. Please do tell. Thanks

ai-republic, 2024-11-23:
Hi @DMIINC,
it definitely looks like your 2CH-CAN hat is not properly set up. Did you follow the instructions on the Waveshare Wiki?
You do not need to install the Python or the example/wiring PI code. 
Which model exactly did you get? The [2-CH-CAN-FD-HAT](https://www.waveshare.com/wiki/2-CH_CAN_FD_HAT) or the [2-CH-CAN-HAT](https://www.waveshare.com/wiki/2-CH_CAN_HAT)?

DMIINC, 2024-11-23:
I think I solved that issue. I have another one. What do you think? I got this one https://www.waveshare.com/wiki/2-CH_CAN_HAT%2B 

024-11-23 06:12:48.119 | ERROR | Thread-2   | stoinverter.core.BMS:192   | Error requesting data!
tel.schich.javacan.platform.linux.LinuxNativeOperationException: Unable to write to the socket - errorNumber=100, errorMessage='Network is down'
	at tel.schich.javacan.SocketCAN.write(Native Method) ~[javacan-core-3.2.4.jar:?]
	at tel.schich.javacan.AbstractCanChannel.writeSocket(AbstractCanChannel.java:237) ~[javacan-core-3.2.4.jar:?]
	at tel.schich.javacan.RawCanChannelImpl.writeUnsafe(RawCanChannelImpl.java:102) ~[javacan-core-3.2.4.jar:?]
	at tel.schich.javacan.RawCanChannelImpl.write(RawCanChannelImpl.java:92) ~[javacan-core-3.2.4.jar:?]
	at com.airepublic.bmstoinverter.protocol.can.JavaCANPort.sendExtendedFrame(JavaCANPort.java:117) ~[protocol-can-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.bms.daly.can.DalyBmsCANProcessor.sendMessage(DalyBmsCANProcessor.java:47) ~[bms-daly-can-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.bms.daly.common.AbstractDalyBmsProcessor.collectData(AbstractDalyBmsProcessor.java:74) ~[bms-daly-common-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.core.BMS.process(BMS.java:173) ~[core-api-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.BmsToInverter.lambda$start$2(BmsToInverter.java:269) ~[bms-to-inverter-main-0.0.1-SNAPSHOT.jar:?]
	at java.lang.Thread.run(Thread.java:840) [?:?]

DMIINC, 2024-11-23:
We are getting somewhere.

dalibor@powerwall2:~ $ sudo ./start.sh
2024-11-23 07:08:01.145 | INFO  | main       | til.SystemProperties:22    | Loading config.properties from: config/config.properties
2024-11-23 07:08:02.224 | INFO  | main       | ore.InverterProducer:76    | Created inverter binding: PYLON_HV_CAN
2024-11-23 07:08:02.280 | INFO  | main       | verter.core.Inverter:53    | Using plugins: []
2024-11-23 07:08:02.304 | INFO  | main       | core.BMSListProducer:105   | Intialized BMS #0[DALY_CAN] on port can0
2024-11-23 07:08:02.315 | INFO  | main       | verter.BmsToInverter:261   | Starting BMS receiver...
2024-11-23 07:08:02.327 | INFO  | Thread-2   | verter.BmsToInverter:268   | Reading BMS #0 DALY_CAN on can0...
2024-11-23 07:08:02.334 | INFO  | Thread-2   | toinverter.core.Port:111   | Opening can0 ...
2024-11-23 07:08:02.405 | INFO  | Thread-2   | toinverter.core.Port:113   | Opening port can0 SUCCESSFUL
2024-11-23 07:08:02.461 | WARN  | Thread-2   | .DalyBmsCANProcessor:94    | Message has wrong address and command id: Buffer (HEX): [0x00, 0x42, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-11-23 07:08:03.459 | WARN  | Thread-2   | .DalyBmsCANProcessor:94    | Message has wrong address and command id: Buffer (HEX): [0x00, 0x42, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-11-23 07:08:04.458 | WARN  | Thread-2   | .DalyBmsCANProcessor:94    | Message has wrong address and command id: Buffer (HEX): [0x00, 0x42, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-11-23 07:08:05.458 | WARN  | Thread-2   | .DalyBmsCANProcessor:94    | Message has wrong address and command id: Buffer (HEX): [0x00, 0x42, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-11-23 07:08:06.459 | WARN  | Thread-2   | .DalyBmsCANProcessor:94    | Message has wrong address and command id: Buffer (HEX): [0x00, 0x42, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-11-23 07:08:07.459 | WARN  | Thread-2   | .DalyBmsCANProcessor:94    | Message has wrong address and command id: Buffer (HEX): [0x00, 0x42, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-11-23 07:08:08.459 | WARN  | Thread-2   | .DalyBmsCANProcessor:94    | Message has wrong address and command id: Buffer (HEX): [0x00, 0x42, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-11-23 07:08:09.459 | WARN  | Thread-2   | .DalyBmsCANProcessor:94    | Message has wrong address and command id: Buffer (HEX): [0x00, 0x42, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-11-23 07:08:10.459 | WARN  | Thread-2   | .DalyBmsCANProcessor:94    | Message has wrong address and command id: Buffer (HEX): [0x00, 0x42, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-11-23 07:08:11.459 | WARN  | Thread-2   | .DalyBmsCANProcessor:94    | Message has wrong address and command id: Buffer (HEX): [0x00, 0x42, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-11-23 07:08:12.459 | WARN  | Thread-2   | .DalyBmsCANProcessor:94    | Message has wrong address and command id: Buffer (HEX): [0x00, 0x42, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-11-23 07:08:13.459 | WARN  | Thread-2   | .DalyBmsCANProcessor:94    | Message has wrong address and command id: Buffer (HEX): [0x00, 0x42, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-11-23 07:08:14.459 | WARN  | Thread-2   | .DalyBmsCANProcessor:94    | Message has wrong address and command id: Buffer (HEX): [0x00, 0x42, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-11-23 07:08:14.460 | INFO  | main       | verter.BmsToInverter:290   | Starting inverter sender...
2024-11-23 07:08:14.462 | INFO  | Thread-3   | verter.BmsToInverter:294   | Sending to inverter PYLON_HV_CAN on can1...
2024-11-23 07:08:14.463 | INFO  | Thread-3   | toinverter.core.Port:111   | Opening can1 ...
2024-11-23 07:08:14.464 | INFO  | Thread-3   | toinverter.core.Port:113   | Opening port can1 SUCCESSFUL
2024-11-23 07:08:14.477 | INFO  | Thread-2   | verter.BmsToInverter:320   | 
BMS	SOC	  V  	  A  	 CellMinV 	 CellMaxV	CellDiff
#1	 -0.1	0.0	0.0	0.0(#0) 	0.0(#0)	0.0 

2024-11-23 07:08:14.479 | INFO  | Thread-2   | verter.BmsToInverter:390   | BMS alarms: 
	NONE
2024-11-23 07:08:15.480 | INFO  | Thread-2   | verter.BmsToInverter:268   | Reading BMS #0 DALY_CAN on can0...
2024-11-23 07:08:15.481 | WARN  | Thread-2   | .DalyBmsCANProcessor:94    | Message has wrong address and command id: Buffer (HEX): [0x00, 0x42, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-11-23 07:08:15.495 | ERROR | Thread-3   | verter.core.Inverter:198   | Failed to send frame to inverter PYLON_HV_CAN :Buffer (HEX): [null]

tel.schich.javacan.platform.linux.LinuxNativeOperationException: Unable to read from the socket - errorNumber=11, errorMessage='Resource temporarily unavailable'
	at tel.schich.javacan.SocketCAN.read(Native Method) ~[javacan-core-3.2.4.jar:?]
	at tel.schich.javacan.AbstractCanChannel.readSocket(AbstractCanChannel.java:213) ~[javacan-core-3.2.4.jar:?]
[... cut ...]
	at com.airepublic.bmstoinverter.inverter.pylonhv.can.PylonHVInverterCANProcessor.readRequest(PylonHVInverterCANProcessor.java:41) ~[inverter-pylon-hv-can-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.core.Inverter.process(Inverter.java:147) ~[core-api-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.inverter.pylonhv.can.PylonHVInverterCANProcessor$Proxy$_$$_WeldClientProxy.process(Unknown Source) ~[inverter-pylon-hv-can-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.BmsToInverter.lambda$start$4(BmsToInverter.java:295) ~[bms-to-inverter-main-0.0.1-SNAPSHOT.jar:?]
	at java.lang.Thread.run(Thread.java:840) [?:?]

ai-republic, 2024-11-24:
Hi @DMIINC,
it looks better now. It has initialized the CAN interface now. This is the configuration that I use to start my 2-CH-CAN-FD hat:
```
#!/usr/bin/env bash
sudo ip link set down can0
sudo ip link set down can1
sudo ip link set can0 type can bitrate 500000 restart-ms 100 fd off
sudo ip link set can1 type can bitrate 500000 restart-ms 100 fd off
sudo ifconfig can0 txqueuelen 65536
sudo ifconfig can1 txqueuelen 65536
sudo ip link set up can0
sudo ip link set up can1
```
But sometimes I get the same error that 
```
tel.schich.javacan.platform.linux.LinuxNativeOperationException: Unable to read from the socket - errorNumber=11, errorMessage='Resource temporarily unavailable'
```
if the device is not connected properly.

For now - to test the BMS - I would recommend to set the inverter to the _NONE_ binding.
My guess is that your BMS Id is not configured correctly. The Daly BMS usually starts with ID 1. You'll need to set that in the configuration. Do you just have one Daly BMS? Do you use the interface board or are you connected straight to the Daly CAN port? And does your Daly really support CAN (you often need to contact a Daly salesperson to get a CAN activated BMS)?

DMIINC, 2024-11-24:
Thanks for the reply. I can use either interface board or can lines from BMS itself. I know I get canbus data out from BMS since in some  combinations on inverter and BMS i will get at least correct SOC. I will try above script and see if it helps. Thank you.

ai-republic, 2024-11-24:
ok, great. Yes, then I think its either the BMS id or a wrong CAN configuration (e.g. baudrate). You can set the BMS id via the DIP switches on the interface board.

DMIINC, 2024-11-24:
Dip switches on interface board are all in off position. Do you suggest something different. What baud rate would you try?

DMIINC, 2024-11-24:
no previous error as of now. I do get this

G | Thread-2   | .DalyBmsCANProcessor:75    | RECEIVED: Buffer (HEX): [0x00, 0x42, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-11-24 08:07:20.281 | WARN  | Thread-2   | .DalyBmsCANProcessor:94    | Message has wrong address and command id: Buffer (HEX): [0x00, 0x42, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-11-24 08:07:20.281 | DEBUG | Thread-2   | .DalyBmsCANProcessor:46    | SEND: Buffer (HEX): [0x40, 0x01, 0x98, 0x18, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-11-24 08:07:20.282 | DEBUG | Thread-2   | ocol.can.JavaCANPort:115   | CAN frame sending: Buffer (HEX): [0x40, 0x01, 0x98, 0x18, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-11-24 08:07:20.282 | DEBUG | Thread-2   | ocol.can.JavaCANPort:76    | CAN frame read...
2024-11-24 08:07:21.280 | DEBUG | Thread-2   | ocol.can.JavaCANPort:82    | CAN read frame Buffer (HEX): [0x00, 0x42, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-11-24 08:07:21.281 | DEBUG | Thread-2   | .DalyBmsCANProcessor:75    | RECEIVED: Buffer (HEX): [0x00, 0x42, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-11-24 08:07:21.281 | WARN  | Thread-2   | .DalyBmsCANProcessor:94    | Message has wrong address and command id: Buffer (HEX): [0x00, 0x42, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-11-24 08:07:21.281 | INFO  | Thread-2   | verter.BmsToInverter:320   | 
BMS	SOC	  V  	  A  	 CellMinV 	 CellMaxV	CellDiff
#1	 -0.1	0.0	0.0	0.0(#0) 	0.0(#0)	0.0 

2024-11-24 08:07:21.282 | INFO  | Thread-2   | verter.BmsToInverter:390   | BMS alarms: 
	NONE
2024-11-24 08:07:21.513 | INFO  | Thread-3   | verter.BmsToInverter:294   | Sending to inverter PYLON_HV_CAN on can1...
2024-11-24 08:07:21.513 | DEBUG | Thread-3   | ocol.can.JavaCANPort:76    | CAN frame read...
2024-11-24 08:07:21.514 | DEBUG | Thread-3   | ocol.can.JavaCANPort:82    | CAN read frame Buffer (HEX): [0x59, 0x03, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x50, 0x4E, 0x00]

2024-11-24 08:07:21.514 | DEBUG | Thread-3   | verter.core.Inverter:150   | Inverter PYLON_HV_CAN received: Buffer (HEX): [0x59, 0x03, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x50, 0x4E, 0x00]

2024-11-24 08:07:21.514 | DEBUG | Thread-3   | verter.core.Inverter:160   | Inverter PYLON_HV_CAN received (after running plugins): Buffer (HEX): [0x59, 0x03, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x50, 0x4E, 0x00]

2024-11-24 08:07:22.282 | INFO  | Thread-2   | verter.BmsToInverter:268   | Reading BMS #1 DALY_CAN on can0...
2024-11-24 08:07:22.283 | DEBUG | Thread-2   | .DalyBmsCANProcessor:46    | SEND: Buffer (HEX): [0x40, 0x01, 0x90, 0x18, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-11-24 08:07:22.283 | DEBUG | Thread-2   | ocol.can.JavaCANPort:115   | CAN frame sending: Buffer (HEX): [0x40, 0x01, 0x90, 0x18, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-11-24 08:07:22.283 | DEBUG | Thread-2   | ocol.can.JavaCANPort:76    | CAN frame read...
2024-11-24 08:07:22.284 | DEBUG | Thread-2   | ocol.can.JavaCANPort:82    | CAN read frame Buffer (HEX): [0x00, 0x42, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-11-24 08:07:22.284 | DEBUG | Thread-2   | .DalyBmsCANProcessor:75    | RECEIVED: Buffer (HEX): [0x00, 0x42, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-11-24 08:07:22.284 | WARN  | Thread-2   | .DalyBmsCANProcessor:94    | Message has wrong address and command id: Buffer (HEX): [0x00, 0x42, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-11-24 08:07:22.284 | DEBUG | Thread-2   | .DalyBmsCANProcessor:46    | SEND: Buffer (HEX): [0x40, 0x01, 0x91, 0x18, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-11-24 08:07:22.284 | DEBUG | Thread-2   | ocol.can.JavaCANPort:115   | CAN frame sending: Buffer (HEX): [0x40, 0x01, 0x91, 0x18, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-11-24 08:07:22.285 | DEBUG | Thread-2   | ocol.can.JavaCANPort:76    | CAN frame read...
2024-11-24 08:07:23.281 | DEBUG | Thread-2   | ocol.can.JavaCANPort:82    | CAN read frame Buffer (HEX): [0x00, 0x42, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-11-24 08:07:23.281 | DEBUG | Thread-2   | .DalyBmsCANProcessor:75    | RECEIVED: Buffer (HEX): [0x00, 0x42, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-11-24 08:07:23.281 | WARN  | Thread-2   | .DalyBmsCANProcessor:94    | Message has wrong address and command id: Buffer (HEX): [0x00, 0x42, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-11-24 08:07:23.282 | DEBUG | Thread-2   | .DalyBmsCANProcessor:46    | SEND: Buffer (HEX): [0x40, 0x01, 0x92, 0x18, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2024-11-24 08:07:23.282 | DEBUG | Thread-2   | ocol.can.JavaCANPort:115   | CAN frame sending: Buffer (HEX): [0x40, 0x01, 0x92, 0x18, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

[... cut ...]
^C2024-11-24 08:07:25.209 | INFO  | Thread-1   | verter.BmsToInverter:426   | Shutting down BMS and inverter threads...FAILED
2024-11-24 08:07:25.210 | INFO  | Thread-1   | ocol.can.JavaCANPort:127   | Shutting down port 'can0'...OK
2024-11-24 08:07:25.211 | INFO  | Thread-1   | ocol.can.JavaCANPort:127   | Shutting down port 'can1'...OK
2024-11-24 08:07:25.211 | INFO  | Thread-1   | verter.BmsToInverter:458   | Shutting down ports...OK
Weld SE container 04ba2258-9fcd-499d-8a28-497a0d58e605 shut down by shutdown hook

DMIINC, 2024-11-24:
It do not feel like it is transmitting to inverter at all

dalibor@powerwall2:~ $ ifconfig
can0: flags=193<UP,RUNNING,NOARP>  mtu 16
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 65536  (UNSPEC)
        RX packets 73  bytes 584 (584.0 B)
        RX errors 0  dropped 39  overruns 0  frame 0
        TX packets 35  bytes 280 (280.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

can1: flags=193<UP,RUNNING,NOARP>  mtu 16
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 65536  (UNSPEC)
        RX packets 442  bytes 3528 (3.4 KiB)
        RX errors 0  dropped 238  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.182  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::280d:a310:ae22:2ce8  prefixlen 64  scopeid 0x20<link>
        ether 2c:cf:67:3d:fc:bb  txqueuelen 1000  (Ethernet)
        RX packets 16783  bytes 1901627 (1.8 MiB)
        RX errors 0  dropped 368  overruns 0  frame 0
        TX packets 24838  bytes 29186064 (27.8 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 106

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 20  bytes 2311 (2.2 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 20  bytes 2311 (2.2 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.185  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::ed58:290a:5acc:18f  prefixlen 64  scopeid 0x20<link>
        ether 2c:cf:67:3d:fc:bc  txqueuelen 1000  (Ethernet)
        RX packets 848  bytes 86229 (84.2 KiB)
        RX errors 0  dropped 255  overruns 0  frame 0
        TX packets 73  bytes 8556 (8.3 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

DMIINC, 2024-11-24:
but I think inverter is listening

DMIINC, 2024-11-24:
[image]

ai-republic, 2024-11-25:
I think you need to set your DIP with the no. 1 switch to ON. Then you should received data from your BMS. Also select the BMS binding _DALY_CAN_ and the ID 1 and baudrate 500000.

DMIINC, 2024-11-25:
Tried reversing call from can0 to can and it worked. Either can outputs on the boar are misprinted or software is coded reverse

DMIINC, 2024-11-25:
[image]

DMIINC, 2024-11-25:
However what is read by your software and what inverter is different. Do you suggest different settings. Amps are high and it throws errors

[image]

ai-republic, 2024-11-26:
Great! You're getting data. That's a start 😄 
The current looks like its x10. Is that on the latest version? Can you confirm this? Then I could have a look at it and fix it.

DMIINC, 2024-11-26:
Yes. The latest version. Thank you.

DMIINC, 2024-11-26:
[image]

Additional information that might help.

ai-republic, 2024-11-27:
ok great, I'll have a look at it on the weekend!

DMIINC, 2024-11-27:
Here is the one that works best for me. My inverter specific. I got my hands on it today.

[CAN Communication Protocol - Solis S6 Hybrid (13).pdf](https://github.com/user-attachments/files/17940860/CAN.Communication.Protocol.-.Solis.S6.Hybrid.13.pdf)

ai-republic, 2024-12-01:
Thanks for the protocol. It is the same as the Pylon HV but the datatype are not s_int_16 but u_int_16 so I changed those.
Could you please do a _Clean install_ on the _Configurator_ app and send me some feedback. Thanks!

ai-republic, 2024-12-03:
Hmm, the code is correct. And the Solis specification you supplied and Pylon HV are identically. The Daly reports 0A current in the above example right? And you have not configured any plugins either right?

DMIINC, 2024-12-03:
Correct. I have also noticed that if I change bitrate or delay information displayed in the CMD window will change. Maybe timing is off somewhere. How would you go about setting it up correctly? Thanks

DMIINC, 2024-12-04:
No matter what i do this is as best the inverter sees. I have changed every setting I can change multiple times. 184A always appears and inverter goes into overload fault. Other info looks to be ok for most part. Any suggestions. Thanks.

[image]

ai-republic, 2024-12-07:
What would be the correct values for the faulty voltage/current? Maybe its a type format error.

DMIINC, 2024-12-07:
Exactly what it says in command window.

ai-republic, 2024-12-14:
This is what is says in your spec:
[image]
And this is how its set in the application:
[image]
So 2nd u_int_16 (unsigned short / char) is setting the current with an offset of -3000A so 0A would be the value 3000 and as unit is 0.1A it would be 30000 = 0.0A

So I guess the Solis CAN protocol sends its total current apparently in a different way.  Can you try to use the ManipulateBatteryPackDataPlugin to set the current to 0 and see what the inverter shows?

DMIINC, 2024-12-14:
I will for sure. I did some more experimenting with pylon protocol straight from bms to inverter. I have found that BYD protocol on inverter shows correct AMPs but wrong voltage and no other info. See pic.[image]

[image]

DMIINC, 2024-12-14:
> This is what is says in your spec: [...]

So is this the only difference you see or are there any others? So it is different then Pylon? Thank you for your help so far. You are getting somewhere .

DMIINC, 2024-12-14:
No matter what number I set for current I get 184.5 See pics

[image]

DMIINC, 2024-12-14:
Also discharge current is off. 

[image]

DMIINC, 2024-12-14:
[image]

ai-republic, 2024-12-15:
ok, so the Max Discharge Current Limit value implies that the previous implementation with the signed shorts was correct.
The value 6503.6 is basically 65536 (maximum value an u_int_16 can have) -500 (50A unit 0.1A with negative sign for discharge).
I will see that I change that back as it has been known to work - not that others update and get errors.
For the battery current I think this must come from some other value. I will try to check the specs again and see if it might be transferred with some other CAN Id. But it'll be probably not before next weekend as I'm quite busy towards the end of the year

ai-republic, 2024-12-15:
So, I reverted the _PYLON_HV_CAN_ bindings back to use the original datatypes and created a _SOLIS_HV_CAN_ inverter binding which we can test/fix on without impacting users which use the _PYLON_HV_CAN_

DMIINC, 2024-12-15:
> So, I reverted the _PYLON_HV_CAN_ bindings back to use the original datatypes and created a _SOLIS_HV_CAN_ inverter binding which we can test/fix on without impacting users which use the _PYLON_HV_CAN_

Trying to play with it but it does not come up in inverter menu selection.

ai-republic, 2024-12-21:
Ah, sorry! Forgot to upload the latest configurator.jar. Please try again

DMIINC, 2024-12-21:
Some updates:
This is the first time I was able  to get any data under Lithium battery HV

[image]

[image]

[image]

[image]

We are moving in right direction I would say.
Same errors as you pointed out in your previous post before.

ai-republic, 2024-12-22:
Ok, good. I will can fix the charge current limit later and provide a fix

ai-republic, 2024-12-25:
Hi @DMIINC,
I merged your changes. Please let me know how it goes...

DMIINC, 2024-12-25:
I just do not understand on what kind of math does this operate on. LOL. Majority of numbers look wrong. What if we remove offset?

[image]

ai-republic, 2024-12-25:
Could you delete the log files and send me the log file plus the screenshots of your inverter values please?

DMIINC, 2024-12-25:
how do I get to log files? Screen shoot above is inverter values or what inverter sees coming in.

ai-republic, 2024-12-25:
The bms-to-inverter log file is in your application log folder on your PI. Please delete them before restarting the application. Then please take fresh screenshots of your inverter values. I can then compare what comes in from the BMS (Daly) and what gets sent out by the application and how the inverter interprets the values (inverter screenshots)

DMIINC, 2024-12-25:
[image]

DMIINC, 2024-12-25:
From Solis Cloud

[image]

ai-republic, 2024-12-26:
Oh, were you connected with another software to the Daly? 
It has lots of message replies that are not valid replies for the requests. This often happens when/after you connect with another application, e.g. Daly software or bluetooth app. 
Can you try to restart your BMS? And please also set the log level to _debug_ and add the new log file.

DMIINC, 2024-12-26:
[BMS-to-Inverter.log](https://github.com/user-attachments/files/18248914/BMS-to-Inverter.log)

DMIINC, 2024-12-26:
yes bms id is 1 and baud rate is 500000

DMIINC, 2024-12-26:
####  Simple single port configuration ####
# bms.pollIntervall - is the interval to request BMS data (in seconds)
# bms.x.type - can be (DALY_CAN, DALY_RS485, JK_CAN, PYLON_CAN or SEPLOS_CAN 
# bms.x.portLocator - is the locator/device to use to communicate to the BMS, eg. can0, /dev/ttyUSB0, com3, etc.  
# bms.x.baudRate - is the locator/device baudrate to use to communicate to the BMS, eg. 9600, 500000, etc.  
# bms.x.delayAfterNoBytes - is the delay after receiving no data (in ms)
bms.pollInterval=1

bms.1.type=DALY_CAN
bms.1.id=1
bms.1.portLocator=can1
bms.1.baudRate=500000
bms.1.delayAfterNoBytes=200

###################################################################
###                    Inverter settings                        ###
###################################################################
# The inverter type can be NONE, DEYE_CAN, GROWATT_CAN, SMA_SI_CAN, SOLARK_CAN
inverter.type=SOLIS_HV_CAN
# The port name/device to use to communicate to the  inverter  
inverter.portLocator=can0
# The port baud rate to use to communicate to the  inverter  
inverter.baudRate=500000
# Interval to send data to the inverter (in seconds)
inverter.sendInterval=1

###################################################################
###                 Optional services settings                  ###
###################################################################

ai-republic, 2024-12-26:
> [BMS-to-Inverter.log](https://github.com/user-attachments/files/18248914/BMS-to-Inverter.log)

This looks better 😄

DMIINC, 2024-12-26:
yes

[34 more replies]

closed 2026-01-30 by ai-republic