ai-republic/bms-to-inverter #82: Selectronic SP Pro 2i

mattthemadscientist, 2024-11-15:
G'day mate, I've finally got my pi up and running and reading the 4 Samsung batteries nicely. I'm using a 4b and 2-ch can hat. unfortunately, the individual cell data isn't coming across to the selectronic inverter. 

Does the Pylontech protocol support individual battery cell data within a stack? I'll get onto selectronic too, as this may just be a setting on their end.

Also, there are only 2 pins, CAN_H and CAN_L on the selectronic, and no ground, does this need to be connected for it to equalise and communicate properly? and if so, should I be earthing the inverter chassis to the batteries?

Thanks mate!

ai-republic, 2024-11-16:
Hi @mattthemadscientist,
unfortunately the Pylontech protocol does not support single cell voltage information.
CAN ground usually does not have to be connected from my experience. Some people even state that you definitely should not connect it. 
But, what you must have is CAN termination with a 120ohm resistor. This might already be done automatically in your inverter/BMS, but you should check that if you encounter communication problems

mattthemadscientist, 2024-11-26:
I've got it connected to the selectronic, but am getting some odd data and its throwing the selectronic into protection mode. 

odd results in the selectronic are as follows. 

charge voltage limit seems to be reading as 14.5v
discharge current limit reading as -6365.6 A
BMS serial is reading as 0

See below log snippet

2024-11-26 11:10:45.009 | INFO  | Thread-2   | verter.BmsToInverter:390   | BMS alarms: 
	NONE
2024-11-26 11:10:45.010 | INFO  | Thread-2   | verter.BmsToInverter:268   | Reading BMS #4 SAMSUNG_CAN on can0...
2024-11-26 11:10:45.010 | DEBUG | Thread-2   | ocol.can.JavaCANPort:76    | CAN frame read...
2024-11-26 11:10:45.010 | DEBUG | Thread-2   | ocol.can.JavaCANPort:82    | CAN read frame Buffer (HEX): [0x01, 0x05, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x04, 0x04, 0x00, 0x00]

2024-11-26 11:10:45.010 | INFO  | Thread-2   | verter.BmsToInverter:320   | 
BMS	SOC	  V  	  A  	 CellMinV 	 CellMaxV	CellDiff
#1	 77.0	54.0	0.0	3.875(#0) 	3.887(#0)	0.012 
#2	 77.0	54.0	0.0	3.875(#0) 	3.886(#0)	0.011 
#3	 77.0	54.0	0.0	3.875(#0) 	3.887(#0)	0.012 
#4	 77.0	54.0	0.0	3.875(#0) 	3.887(#0)	0.012 

2024-11-26 11:10:45.010 | INFO  | Thread-2   | verter.BmsToInverter:390   | BMS alarms: 
	NONE
2024-11-26 11:10:46.010 | INFO  | Thread-2   | verter.BmsToInverter:268   | Reading BMS #1 SAMSUNG_CAN on can0...
2024-11-26 11:10:46.010 | DEBUG | Thread-2   | ocol.can.JavaCANPort:76    | CAN frame read...
2024-11-26 11:10:46.011 | DEBUG | Thread-2   | ocol.can.JavaCANPort:82    | CAN read frame Buffer (HEX): [0x03, 0x05, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x29, 0x0F, 0x2E, 0x0F, 0x23, 0x0F, 0x20, 0x15]

2024-11-26 11:10:46.011 | DEBUG | Thread-2   | msungBmsCANProcessor:168   | 
Max Cell mV 	Min Cell mV
3886000.0	3875000.0
2024-11-26 11:10:46.011 | INFO  | Thread-2   | verter.BmsToInverter:320   | 
BMS	SOC	  V  	  A  	 CellMinV 	 CellMaxV	CellDiff
#1	 77.0	54.0	0.0	3.875(#0) 	3.886(#0)	0.011 
#2	 77.0	54.0	0.0	3.875(#0) 	3.886(#0)	0.011 
#3	 77.0	54.0	0.0	3.875(#0) 	3.887(#0)	0.012 
#4	 77.0	54.0	0.0	3.875(#0) 	3.887(#0)	0.012 

2024-11-26 11:10:46.011 | INFO  | Thread-2   | verter.BmsToInverter:390   | BMS alarms: 
	NONE
2024-11-26 11:10:46.011 | INFO  | Thread-2   | verter.BmsToInverter:268   | Reading BMS #2 SAMSUNG_CAN on can0...
2024-11-26 11:10:46.011 | DEBUG | Thread-2   | ocol.can.JavaCANPort:76    | CAN frame read...
2024-11-26 11:10:46.011 | DEBUG | Thread-2   | ocol.can.JavaCANPort:82    | CAN read frame Buffer (HEX): [0x10, 0x05, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x27, 0x15, 0xFC, 0xFF, 0x4D, 0x64, 0x00, 0x00]

2024-11-26 11:10:46.011 | INFO  | Thread-2   | verter.BmsToInverter:320   | 
BMS	SOC	  V  	  A  	 CellMinV 	 CellMaxV	CellDiff
#1	 77.0	54.0	0.0	3.875(#0) 	3.886(#0)	0.011 
#2	 77.0	54.0	0.0	3.875(#0) 	3.886(#0)	0.011 
#3	 77.0	54.0	0.0	3.875(#0) 	3.887(#0)	0.012 
#4	 77.0	54.0	0.0	3.875(#0) 	3.887(#0)	0.012 

2024-11-26 11:10:46.012 | INFO  | Thread-2   | verter.BmsToInverter:390   | BMS alarms: 
	NONE
2024-11-26 11:10:46.012 | INFO  | Thread-2   | verter.BmsToInverter:268   | Reading BMS #3 SAMSUNG_CAN on can0...
2024-11-26 11:10:46.012 | DEBUG | Thread-2   | ocol.can.JavaCANPort:76    | CAN frame read...
2024-11-26 11:10:46.012 | DEBUG | Thread-2   | ocol.can.JavaCANPort:82    | CAN read frame Buffer (HEX): [0x10, 0x05, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x27, 0x15, 0xFC, 0xFF, 0x4D, 0x64, 0x00, 0x00]

2024-11-26 11:10:46.012 | INFO  | Thread-2   | verter.BmsToInverter:320   | 
BMS	SOC	  V  	  A  	 CellMinV 	 CellMaxV	CellDiff
#1	 77.0	54.0	0.0	3.875(#0) 	3.886(#0)	0.011 
#2	 77.0	54.0	0.0	3.875(#0) 	3.886(#0)	0.011 
#3	 77.0	54.0	0.0	3.875(#0) 	3.887(#0)	0.012 
#4	 77.0	54.0	0.0	3.875(#0) 	3.887(#0)	0.012 

2024-11-26 11:10:46.012 | INFO  | Thread-2   | verter.BmsToInverter:390   | BMS alarms: 
	NONE
2024-11-26 11:10:46.012 | INFO  | Thread-2   | verter.BmsToInverter:268   | Reading BMS #4 SAMSUNG_CAN on can0...
2024-11-26 11:10:46.012 | DEBUG | Thread-2   | ocol.can.JavaCANPort:76    | CAN frame read...
2024-11-26 11:10:46.012 | DEBUG | Thread-2   | ocol.can.JavaCANPort:82    | CAN read frame Buffer (HEX): [0x14, 0x05, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x27, 0x15, 0xFC, 0xFF, 0x4E, 0x64, 0x00, 0x00]
[... 86 lines cut ...]
2024-11-26 11:10:48.030 | INFO  | Thread-2   | verter.BmsToInverter:390   | BMS alarms: 
	NONE
2024-11-26 11:10:48.030 | INFO  | Thread-2   | verter.BmsToInverter:268   | Reading BMS #3 SAMSUNG_CAN on can0...
2024-11-26 11:10:48.030 | DEBUG | Thread-2   | ocol.can.JavaCANPort:76    | CAN frame read...
2024-11-26 11:10:48.031 | DEBUG | Thread-2   | ocol.can.JavaCANPort:82    | CAN read frame Buffer (HEX): [0xF0, 0x05, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x03, 0x00, 0x2C, 0x0F, 0x2E, 0x0F, 0x2C, 0x0F]

2024-11-26 11:10:48.031 | INFO  | Thread-2   | verter.BmsToInverter:320   | 
BMS	SOC	  V  	  A  	 CellMinV 	 CellMaxV	CellDiff
#1	 77.0	54.0	0.0	3.875(#0) 	3.886(#0)	0.011 
#2	 77.0	54.0	0.0	3.875(#0) 	3.886(#0)	0.011 
#3	 77.0	54.0	0.0	3.875(#0) 	3.887(#0)	0.012 
#4	 77.0	54.0	0.0	3.875(#0) 	3.887(#0)	0.012 

2024-11-26 11:10:48.032 | INFO  | Thread-2   | verter.BmsToInverter:390   | BMS alarms: 
	NONE
2024-11-26 11:10:48.032 | INFO  | Thread-2   | verter.BmsToInverter:268   | Reading BMS #4 SAMSUNG_CAN on can0...
2024-11-26 11:10:48.032 | DEBUG | Thread-2   | ocol.can.JavaCANPort:76    | CAN frame read...
2024-11-26 11:10:48.032 | DEBUG | Thread-2   | ocol.can.JavaCANPort:82    | CAN read frame Buffer (HEX): [0xF1, 0x05, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x03, 0x00, 0x2A, 0x0F, 0x29, 0x0F, 0x2C, 0x0F]

2024-11-26 11:10:48.032 | INFO  | Thread-2   | verter.BmsToInverter:320   | 
BMS	SOC	  V  	  A  	 CellMinV 	 CellMaxV	CellDiff
#1	 77.0	54.0	0.0	3.875(#0) 	3.886(#0)	0.011 
#2	 77.0	54.0	0.0	3.875(#0) 	3.886(#0)	0.011 
#3	 77.0	54.0	0.0	3.875(#0) 	3.887(#0)	0.012 
#4	 77.0	54.0	0.0	3.875(#0) 	3.887(#0)	0.012

mattthemadscientist, 2024-11-26:
haha I just noticed this, 

2024-11-26 11:10:46.011 | DEBUG | Thread-2 | msungBmsCANProcessor:168 |
Max Cell mV Min Cell mV
3886000.0 3875000.0

that would be 3,886 volts... something funny is going on

ai-republic, 2024-11-27:
@mattthemadscientist I guess the Max/MinCellmV needs to be divided by 1000 not multiplied 😉 
I will check the other issues/chats and see if other users had the same problem. Possibly your Samsung BMS is a different version and needs another BMS binding. What Samsung batteries do you have?

mattthemadscientist, 2024-11-27:
mine are the elpm482-00005, I linked the architecture in an earlier post. the communication protocol can be found in the attached pdf... if it attaches for me. 
[4.84kWh_Standard_Product_Specification(1)[92].pdf](https://github.com/user-attachments/files/17932207/4.84kWh_Standard_Product_Specification.1.92.pdf)

ai-republic, 2024-12-01:
Hi @mattthemadscientist I updated some datatypes. For the cell voltage it was just outputting it wrong, but the reading was correct. Please do a _Clean install_ with the _Configurator_ app and let me now how it behaves 😄

ai-republic, 2025-01-05:
@mattthemadscientist did you get a chance to test it?

mattthemadscientist, 2025-01-05:
yup, still having an issue with the calculated voltage though. ill get back there in the next two days and get a log file. thanks again mate, you're an absolute legend.