ai-republic/bms-to-inverter #83: Support for Goodwe SBP Retrofit

corty8, 2024-11-21:
Hi there

Is it possible to add the Goodwe SBP (GW5000S-BP AC Coupled Retrofit Charger?

Thanks

ai-republic, 2024-11-23:
Sure, do you have the specification? I see it supports CAN and RS485. Better is CAN.

corty8, 2024-11-23:
does this help
[GoodweCAN_BUS_Protocol.pdf](https://github.com/user-attachments/files/17878071/GoodweCAN_BUS_Protocol.pdf)

ai-republic, 2024-11-23:
Yes, this is actually the same as the Pylontec CAN protocol. Could you please test it with the _PYLON_CAN_ BMS binding?

corty8, 2024-11-23:
ok great, I am waiting on a couple of components to be delivered and I will test it and let you know

ai-republic, 2025-01-05:
@corty8 did you get a chance to test it?

corty8, 2025-01-07:
> @corty8 did you get a chance to test it?

no, still haven't been able to
The wifi module on the inverter got damaged when it was being moved and I have not been able to source the correct one
have ordered 2 now and they have sent the wrong one on both occasions

closed 2026-01-30 by ai-republic