ai-republic/bms-to-inverter #85: Binding for Growatt SPH10000-TL3-BH-UP CAN protocol

ErikssonPer, 2024-12-20:
Hi again,
I have a Growatt SPH10000-TL3-BH-UP and ARK bms with 4 modules.

I did a quick test setting the app up for PYLON_HV_CAN but that didn't work, inverter reported _bms error_.
So I've  pulled some logs to try and figure out the protocol.

It seems growatt has went away from pylon protocol and made it's own, this is what I found so far.

Tomorrow i'll disconnect 1 module and run logs again with only 3 and see if I can find what changes.

Inverter sends this until the BMS responds
3010 8 0E DF 14 00 00 00 00 00
4200 8 02 00 00 00 00 00 00 E3
4200 8 00 00 00 00 00 00 00 E3
8210 8 AA 00 00 00 00 00 00 00
3010 8 0E E0 14 00 00 00 00 00
And so on.......
3010 byte 0 and 1 seem to be a counter beacuse it's always incrementing with 1, the rest never changes.

When BMS is powered up and responds for the first time inverter only sends this with 1 second interval, byte 0 and 1 still increments with 1.
3010 8 0E EF 14 00 00 00 00 00

BMS response look like this, ID's not matching anything I've seen and the data is arranged differently also.

_(213,6V,  no charge/discharge  24C  SOC 100%  SOH100%)_

3020   [8]  00 00 00 00 00 00 00 00      Never changes
3030   [8]  18 0C 13 17 1C 17 00 01    Some type of counter, maybe Time and date?
3110   [8]  08 E0 00 00 00 FA 30 41      08E0   Max voltage 227.2     00FA 25A    00FA 25A Max discharge and charge     
3120   [8]  00 00 00 00 00 00 00 00      Never changes
3130   [8]  08 58 00 00 00 F0 64 64     0848 =213,6V      0000  Charge/discharge current (Decimal from signed 2's complement )       
                                                                                          00F0  24C         64 SOC 100%              64  SOH 100%
3140   [8]  13 4A 13 4C 47 54 01 28                                                                                              0128 =  296 Number of cycles
3150   [8]  07 60 01 27 00 40 00 04     0760 never changes   0127 29,5  highest cell temp?    40 64 cells?    04  4 Modules?
3160   [8]  00 00 04 2D 01 18 00 DD                                                                                                     00DD   22.1 Lowest cell temp?
3170   [8]  01 03 04 0E 64 05 01 70
3180   [8]  00 00 00 01 00 40 00 41
3190   [8]  00 0D 12 0D 09 00 00 00
3200   [8]  00 00 02 00 00 00 0E 10
3220   [8]  00 00 00 00 00 00 00 02
3230   [8]  01 30 30 32 32 33 34 30
3240   [8]  01 04 DD B9 01 05 51 0B
3250   [8]  11 FF 11 FF 11 FF 11 FF     Never changes
3260   [8]  00 00 00 00 00 00 00 00     Never changes
3270   [8]  00 00 00 00 00 00 00 00
3F00   [8]  00 00 00 00 00 00 00 00     Never changes

3110 BYTE 7  
01 =  Idle?
02 =Charging
03 =Discharge
41 = Balancing?
42 = fully charged?  
61 = Initiating?
At startup shows 61 then 41  then 01 until charge or discharge begins
When fully charged shows 42 for a brief moment then 41

[Growatt SPH inverter with ARK bms startup.txt](https://github.com/user-attachments/files/18215562/Growatt.SPH.inverter.with.ARK.bms.startup.txt)

ai-republic, 2024-12-21:
Yeah, that's not a protocol I've seen before. Have you tried to contact Growatt support and as for the CAN specification?

ErikssonPer, 2024-12-21:
Many have tried but they are extremly tight lipped.

romanbetwo, 2024-12-21:
Let me fix that for you.

romanbetwo, 2024-12-21:
[Communication protocol of Growatt high voltage battery CAN V1.10 (3).pdf](https://github.com/user-attachments/files/18217694/Communication.protocol.of.Growatt.high.voltage.battery.CAN.V1.10.3.pdf)

romanbetwo, 2024-12-21:
and bonus  track > all possible errors 
[All error list EN.xlsx](https://github.com/user-attachments/files/18217696/All.error.list.EN.xlsx)

ErikssonPer, 2024-12-21:
What kind of magic superpowers do you possess?
That's golden.

romanbetwo, 2024-12-21:
Probably all kind of magic, but I still need to "kick" my self to start learning how CAN communication and translation works inclusing python and all that fancy stuff. I worked in R&D and testing equipment with all kind of brands. Equipment like IFR, Agilent, Yello, Keyence and many more. Whirpool, Asko, Gorenje, Arçelik, Candy, Bosch.. 

I work very closely with growatt engineering team since I start working for company that fix issues :)

ErikssonPer, 2024-12-21:
Do you have the low voltage protocol also, the version I have is half chinese half english?

romanbetwo, 2024-12-21:
abra-ka-dabra 
[Communication protocol of Growatt low voltage battery CAN V1.04 (1).pdf](https://github.com/user-attachments/files/18219728/Communication.protocol.of.Growatt.low.voltage.battery.CAN.V1.04.1.pdf)

ai-republic, 2024-12-22:
@romanbetwo sweet! I will check them out later and provide a new binding for the Growatt HV over the holidays

ErikssonPer, 2024-12-22:
It looks like both HV and LV protocols are more or less identical excapt they have changed the ID's, 311 to 3110, 312 to 3120 and so on.
It's a 2 fer 1 kinda deal.

In my case that might be beneficial since the emus G1 bms I have now have a setting to emulate growatt LV protocol.

I have to send another heartbeat command with ID 301 to start it but then it looks like it matches.
A short snippet from that looks like this, 
[image]

The different heartbeats I think is due to emus following another version of the protocol.

ai-republic, 2025-01-05:
@ErikssonPer, @romanbetwo FYI: I've implemented the Growatt HV and LV CAN bindings for inverter and BMS and are ready for further testing 😉

ErikssonPer, 2025-01-07:
@ai-republic 
Downloaded the Configurator and ran the setup, no issues.

But when I started it the following happens,
jevlar@rpi:~/bms $ ./start.sh
2025-01-07 20:02:40.388 | INFO  | main       | til.SystemProperties:22    | Loading config.properties from: config/config.properties
2025-01-07 20:02:40.509 | DEBUG | main       | ging.LoggerProviders:152   | Logging Provider: org.jboss.logging.Log4j2LoggerProvider
2025-01-07 20:02:42.981 | INFO  | main       | ore.InverterProducer:76    | Created inverter binding: GROWATT_HV_CAN
2025-01-07 20:02:43.102 | INFO  | main       | verter.core.Inverter:53    | Using plugins: []
Exception in thread "main" org.jboss.weld.exceptions.UnproxyableResolutionException: WELD-001480: Bean type class com.airepublic.bmstoinverter.bms.growatthv.can.GrowattHVBmsCANProcessor is not proxyable because it contains a final method public final void com.airepublic.bmstoinverter.core.BMS.process(java.lang.Runnable) - <unknownjakarta.enterprise.inject.spi.Bean instance>.
	at org.jboss.weld.util.Proxies.getUnproxyableClassException(Proxies.java:244)
	at org.jboss.weld.util.Proxies.getUnproxyableTypeException(Proxies.java:199)
	at org.jboss.weld.util.Proxies.getUnproxyableTypeException(Proxies.java:162)
	at org.jboss.weld.bean.proxy.ClientProxyProvider.getClientProxy(ClientProxyProvider.java:238)
	at org.jboss.weld.manager.BeanManagerImpl.getReference(BeanManagerImpl.java:678)
	at org.jboss.weld.manager.BeanManagerImpl.getReference(BeanManagerImpl.java:666)
	at org.jboss.weld.bean.builtin.InstanceImpl.getBeanInstance(InstanceImpl.java:262)
	at org.jboss.weld.bean.builtin.InstanceImpl.get(InstanceImpl.java:114)
	at com.airepublic.bmstoinverter.core.BMSListProducer.createBMS(BMSListProducer.java:94)
	at com.airepublic.bmstoinverter.core.BMSListProducer.produceBMSList(BMSListProducer.java:80)
	at java.base/jdk.internal.reflect.DirectMethodHandleAccessor.invoke(DirectMethodHandleAccessor.java:103)
	at java.base/java.lang.reflect.Method.invoke(Method.java:580)
	at org.jboss.weld.injection.StaticMethodInjectionPoint.invoke(StaticMethodInjectionPoint.java:95)
	at org.jboss.weld.injection.StaticMethodInjectionPoint.invoke(StaticMethodInjectionPoint.java:85)
	at org.jboss.weld.injection.producer.ProducerMethodProducer.produce(ProducerMethodProducer.java:103)
	at org.jboss.weld.injection.producer.AbstractMemberProducer.produce(AbstractMemberProducer.java:161)
	at org.jboss.weld.bean.AbstractProducerBean.create(AbstractProducerBean.java:180)
	at org.jboss.weld.contexts.unbound.DependentContextImpl.get(DependentContextImpl.java:64)
	at org.jboss.weld.bean.ContextualInstanceStrategy$DefaultContextualInstanceStrategy.get(ContextualInstanceStrategy.java:100)
	at org.jboss.weld.bean.ContextualInstance.get(ContextualInstance.java:50)
	at org.jboss.weld.manager.BeanManagerImpl.getReference(BeanManagerImpl.java:684)
	at org.jboss.weld.manager.BeanManagerImpl.getReference(BeanManagerImpl.java:666)
	at org.jboss.weld.manager.BeanManagerImpl.getInjectableReference(BeanManagerImpl.java:784)
	at org.jboss.weld.injection.FieldInjectionPoint.inject(FieldInjectionPoint.java:92)
	at org.jboss.weld.util.Beans.injectBoundFields(Beans.java:345)
	at org.jboss.weld.util.Beans.injectFieldsAndInitializers(Beans.java:356)
	at org.jboss.weld.injection.producer.ResourceInjector$1.proceed(ResourceInjector.java:69)
	at org.jboss.weld.injection.InjectionContextImpl.run(InjectionContextImpl.java:48)
	at org.jboss.weld.injection.producer.ResourceInjector.inject(ResourceInjector.java:71)
	at org.jboss.weld.injection.producer.BasicInjectionTarget.inject(BasicInjectionTarget.java:117)
	at org.jboss.weld.bean.ManagedBean.create(ManagedBean.java:161)
	at org.jboss.weld.contexts.AbstractContext.get(AbstractContext.java:96)
	at org.jboss.weld.bean.ContextualInstanceStrategy$DefaultContextualInstanceStrategy.get(ContextualInstanceStrategy.java:100)
	at org.jboss.weld.bean.ContextualInstanceStrategy$ApplicationScopedContextualInstanceStrategy.get(ContextualInstanceStrategy.java:140)
	at org.jboss.weld.bean.ContextualInstance.get(ContextualInstance.java:50)
	at org.jboss.weld.bean.proxy.ContextBeanInstance.getInstance(ContextBeanInstance.java:102)
	at org.jboss.weld.bean.proxy.ProxyMethodHandler.getInstance(ProxyMethodHandler.java:131)
	at com.airepublic.bmstoinverter.BmsToInverter$Proxy$_$$_WeldClientProxy.start(Unknown Source)
	at com.airepublic.bmstoinverter.BmsToInverter.main(BmsToInverter.java:88)
2025-01-07 20:02:43.125 | INFO  | Thread-1   | verter.BmsToInverter:426   | Shutting down BMS and inverter threads...FAILED
2025-01-07 20:02:43.130 | INFO  | Thread-1   | verter.BmsToInverter:458   | Shutting down ports...OK
Weld SE container 57ea3c97-126f-46f5-9ad7-0abadb3173a3 shut down by shutdown hook

romanbetwo, 2025-01-07:
I will do the test tomorrow, but it looks like an issue

ai-republic, 2025-01-08:
@ErikssonPer oh, thanks for finding that. It should be fixed now. Please _Clean install_ again. Thanks!

romanbetwo, 2025-01-08:
okay so , I've been playin arround and finaly figure out that .. i have bad HAT for this application.. 2xRS 1xCAN .. however I need 2xCAN... 
anyway
I try to sim battery for Growatt HV CAN
Inverter does nothing.
settings:  
bms none
inv growatt hv can > can0 > 500k 1

the log >>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
2025-01-08 17:21:47.118 | ERROR | Thread-3   | InverterCANProcessor:78    | Error creating send frames:
java.lang.StringIndexOutOfBoundsException: String index out of range: 0
        at java.lang.StringLatin1.charAt(StringLatin1.java:48) ~[?:?]
        at java.lang.String.charAt(String.java:1517) ~[?:?]
        at com.airepublic.bmstoinverter.inverter.growatthv.can.GrowattHVInverterCANProcessor.sendBatteryCapacity(GrowattHVInverterCANProcessor.java:239) ~[inverter-growatt-hv-can-0.0.1-SNAPSHOT.jar:?]
        at com.airepublic.bmstoinverter.inverter.growatthv.can.GrowattHVInverterCANProcessor.createSendFrames(GrowattHVInverterCANProcessor.java:64) ~[inverter-growatt-hv-can-0.0.1-SNAPSHOT.jar:?]
        at com.airepublic.bmstoinverter.core.Inverter.process(Inverter.java:177) ~[core-api-0.0.1-SNAPSHOT.jar:?]
        at com.airepublic.bmstoinverter.inverter.growatthv.can.GrowattHVInverterCANProcessor$Proxy$_$$_WeldClientProxy.process(Unknown Source) ~[inverter-growatt-hv-can-0.0.1-SNAPSHOT.jar:?]
        at com.airepublic.bmstoinverter.BmsToInverter.lambda$start$4(BmsToInverter.java:295) ~[bms-to-inverter-main-0.0.1-SNAPSHOT.jar:?]
        at java.lang.Thread.run(Thread.java:840) [?:?]
2025-01-08 17:21:47.122 | DEBUG | Thread-3   | verter.core.Inverter:188   | Calling inverter plugin (onSend): SimulatedBatteryPackPlugin
2025-01-08 17:21:47.124 | DEBUG | Thread-3   | verter.core.Inverter:193   | Inverter GROWATT_HV_CAN send: Buffer (HEX): [0x00, 0x00, 0x31, 0x10, 0x08, 0x00, 0x00, 0x00, 0x09, 0x24, 0x00, 0xC8, 0xFF, 0x38, 0x00, 0x00]

2025-01-08 17:21:47.127 | DEBUG | Thread-3   | ocol.can.JavaCANPort:115   | CAN frame sending: Buffer (HEX): [0x00, 0x00, 0x31, 0x10, 0x08, 0x00, 0x00, 0x00, 0x09, 0x24, 0x00, 0xC8, 0xFF, 0x38, 0x00, 0x00]

2025-01-08 17:21:47.129 | DEBUG | Thread-3   | verter.core.Inverter:188   | Calling inverter plugin (onSend): SimulatedBatteryPackPlugin
2025-01-08 17:21:47.132 | DEBUG | Thread-3   | verter.core.Inverter:193   | Inverter GROWATT_HV_CAN send: Buffer (HEX): [0x00, 0x00, 0x31, 0x20, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2025-01-08 17:21:47.135 | DEBUG | Thread-3   | ocol.can.JavaCANPort:115   | CAN frame sending: Buffer (HEX): [0x00, 0x00, 0x31, 0x20, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2025-01-08 17:21:47.138 | DEBUG | Thread-3   | verter.core.Inverter:188   | Calling inverter plugin (onSend): SimulatedBatteryPackPlugin
2025-01-08 17:21:47.139 | DEBUG | Thread-3   | verter.core.Inverter:193   | Inverter GROWATT_HV_CAN send: Buffer (HEX): [0x00, 0x00, 0x31, 0x30, 0x08, 0x00, 0x00, 0x00, 0x07, 0xD0, 0x00, 0x00, 0x00, 0xFA, 0x18, 0x64]

2025-01-08 17:21:47.141 | DEBUG | Thread-3   | ocol.can.JavaCANPort:115   | CAN frame sending: Buffer (HEX): [0x00, 0x00, 0x31, 0x30, 0x08, 0x00, 0x00, 0x00, 0x07, 0xD0, 0x00, 0x00, 0x00, 0xFA, 0x18, 0x64]

2025-01-08 17:21:47.496 | INFO  | Thread-2   | verter.BmsToInverter:268   | Reading BMS #1 NONE on none...
2025-01-08 17:21:47.498 | INFO  | Thread-2   | verter.BmsToInverter:320   |
BMS     SOC       V       A      CellMinV        CellMaxV       CellDiff
#1       -0.1   0.0     0.0     0.0(#0)         0.0(#0) 0.0

2025-01-08 17:21:47.501 | INFO  | Thread-2   | verter.BmsToInverter:390   | BMS alarms:
        NONE
2025-01-08 17:21:48.143 | INFO  | Thread-3   | verter.BmsToInverter:294   | Sending to inverter GROWATT_HV_CAN on can0...
2025-01-08 17:21:48.145 | DEBUG | Thread-3   | ocol.can.JavaCANPort:76    | CAN frame read...
2025-01-08 17:21:48.147 | DEBUG | Thread-3   | ocol.can.JavaCANPort:82    | CAN read frame Buffer (HEX): [0x00, 0x42, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x02, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x06]

2025-01-08 17:21:48.150 | DEBUG | Thread-3   | verter.core.Inverter:150   | Inverter GROWATT_HV_CAN received: Buffer (HEX): [0x00, 0x42, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x02, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x06]

2025-01-08 17:21:48.151 | DEBUG | Thread-3   | verter.core.Inverter:156   | Calling inverter plugin (onReceive): SimulatedBatteryPackPlugin
2025-01-08 17:21:48.154 | DEBUG | Thread-3   | verter.core.Inverter:160   | Inverter GROWATT_HV_CAN received (after running plugins): Buffer (HEX): [0x00, 0x42, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x02, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x06]

2025-01-08 17:21:48.156 | DEBUG | Thread-3   | verter.core.Inverter:171   | Calling inverter plugin (onBatteryAggregation): SimulatedBatteryPackPlugin
2025-01-08 17:21:48.160 | DEBUG | Thread-3   | InverterCANProcessor:142   | Sending max/min charge and discharge voltage and current limits: Buffer (HEX): [0x00, 0x00, 0x31, 0x10, 0x08, 0x00, 0x00, 0x00, 0x09, 0x24, 0x00, 0xC8, 0xFF, 0x38, 0x00, 0x00]

2025-01-08 17:21:48.163 | DEBUG | Thread-3   | InverterCANProcessor:205   | Sending alarms: Buffer (HEX): [0x00, 0x00, 0x31, 0x20, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2025-01-08 17:21:48.166 | DEBUG | Thread-3   | InverterCANProcessor:225   | Sending battery status: Buffer (HEX): [0x00, 0x00, 0x31, 0x30, 0x08, 0x00, 0x00, 0x00, 0x07, 0xD0, 0x00, 0x00, 0x00, 0xFA, 0x18, 0x64]

[... cut ...]
Battery shows correct SOC but inverter is pretty much confused,.. like realy confused .. start charging and sh+t with no agreements on set parameters or settings ...so pretty heavy to game play&fxxk arround to find out :D.

todo: get a 2x can hat  /current is usable for MOD or huawei
We are missing something in growatt HV plugin
I will create new topic for "my battery" 
If I missed something let me know 
[bms_ems_v01_dbc.xlsx](https://github.com/user-attachments/files/18350381/bms_ems_v01_dbc.xlsx)

ErikssonPer, 2025-01-09:
Ran a quick test with it connected inbetween the inverter on can0 and battery on can1.

The heartbeat message that's sent out on can1 to the bms looks like this,
00003111    8  00  00  00  00  00  00  00  00

So the battery won't respond with any data.

It looks like an error is reported in the log.
[BMS-to-Inverter.log](https://github.com/user-attachments/files/18366435/BMS-to-Inverter.log)

ai-republic, 2025-01-11:
So I'll try to answer one by one 😉 
1. I fixed the IndexOutOfBoundsException so you can try the simulated battery plugin if you like
2. The CAN data you captured definitely matches the _PYLON_HV_CAN_ or _SOLIS_HV_CAN_
3. In the log file I see that the port was not initialized properly (sometimes happens on the PI). If you see this - and you're sure it configured properly - just try to reboot it (a few times).

PS: When doing tests please disable the webserver as it makes the log file hard to read 😉

romanbetwo, 2025-01-11:
Will do sir 🫡

ErikssonPer, 2025-01-18:
I'm not having any luck,
Candump from Can0 only have traffic coming from the BMS
Candump from Can1 only have data from the inverter

If I start the App it wont send anything out on either.

ai-republic, 2025-01-19:
hmm, it looks like its something with your CAN configuration for your CAN adapter is not correctly set up. Or its not (yet) started properly. The error message comes straight from the SocketCAN unix module.

ai-republic, 2025-01-21:
This part is the crucial part:
2025-01-20 22:31:05.790 | ERROR | Thread-2   | stoinverter.core.BMS:192   | Error requesting data!tel.schich.javacan.platform.linux.LinuxNativeOperationException: **_Unable to read from the socket - errorNumber=11, errorMessage='Resource temporarily unavailable'_**        at tel.schich.javacan.SocketCAN.read(Native Method) ~[javacan-core-3.2.4.jar:?] 

This implies that the socket it not available. So either in use, not initialized yet, or not configured correctly.
Is your adapter using the SocketCAN library or something else?

ErikssonPer, 2025-01-28:
It's a waveshare 2 channel fd can hat 

Can't figure it out, everything seems to check out and I can send and recieve on both channels using command line or a simple python script. 

And the app is able to send

ErikssonPer, 2025-01-28:
Gonna try to use it in legacy mode and see what that does.

ai-republic, 2025-01-29:
Maybe also try to re-install the Waveshare hat on a fresh Raspian OS installation and reconfigure it without adding any other libraries like the python scripts etc.
Another thing that comes to mind, did you set the baudrate properly for the BMS and CAN port?

ErikssonPer, 2025-01-29:
Both channels are set to 500kbit and I can log data being sent from both the inverter and bms with candump so it's correct.

I noticed something today, for fun I ran the configurator again and set it to pylon hv bms and growatt hv inverter.

With those settings it sends out a different heartbeat to the bms, (ID 4200) which is correct for pylon protocol.
[image]

Remember it sends this when setup as growatt_hv:
**0 00003111 X 8 00 00 00 00 00 00 00 0**
There's no ID like that in the growatt protocol.

But it doesn't show any errors in the log now, 

jevlar@rpi:~ $ ./start.sh
2025-01-29 20:05:56.961 | INFO  | main       | til.SystemProperties:22    | Loading config.properties from: config/config.properties
2025-01-29 20:05:57.057 | DEBUG | main       | ging.LoggerProviders:152   | Logging Provider: org.jboss.logging.Log4j2LoggerProvider
2025-01-29 20:05:58.992 | INFO  | main       | ore.InverterProducer:76    | Created inverter binding: GROWATT_HV_CAN
2025-01-29 20:05:59.113 | INFO  | main       | verter.core.Inverter:53    | Using plugins: []
2025-01-29 20:05:59.141 | INFO  | main       | core.BMSListProducer:105   | Intialized BMS #1[PYLON_HV_CAN] on port can0
2025-01-29 20:05:59.170 | INFO  | main       | verter.BmsToInverter:261   | Starting BMS receiver...
2025-01-29 20:05:59.177 | INFO  | Thread-2   | verter.BmsToInverter:268   | Reading BMS #1 PYLON_HV_CAN on can0...
2025-01-29 20:05:59.180 | INFO  | Thread-2   | toinverter.core.Port:111   | Opening can0 ...
2025-01-29 20:05:59.256 | INFO  | Thread-2   | toinverter.core.Port:113   | Opening port can0 SUCCESSFUL
2025-01-29 20:05:59.269 | DEBUG | Thread-2   | lonHVBmsCANProcessor:58    | SEND: Buffer (HEX): [0x00, 0x42, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2025-01-29 20:05:59.272 | DEBUG | Thread-2   | ocol.can.JavaCANPort:115   | CAN frame sending: Buffer (HEX): [0x00, 0x42, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2025-01-29 20:05:59.277 | DEBUG | Thread-2   | ocol.can.JavaCANPort:76    | CAN frame read...
2025-01-29 20:06:00.282 | INFO  | Thread-2   | verter.BmsToInverter:320   | 
BMS	SOC	  V  	  A  	 CellMinV 	 CellMaxV	CellDiff

2025-01-29 20:06:00.282 | INFO  | main       | verter.BmsToInverter:290   | Starting inverter sender...
2025-01-29 20:06:00.283 | INFO  | Thread-2   | verter.BmsToInverter:390   | BMS alarms: 
	NONE
2025-01-29 20:06:00.286 | INFO  | Thread-3   | verter.BmsToInverter:294   | Sending to inverter GROWATT_HV_CAN on can1...
2025-01-29 20:06:00.288 | DEBUG | Thread-3   | verter.core.Inverter:207   | No battery data yet received to send to inverter GROWATT_HV_CAN!
2025-01-29 20:06:01.285 | INFO  | Thread-2   | verter.BmsToInverter:268   | Reading BMS #1 PYLON_HV_CAN on can0...

Could this be a clue?

ai-republic, 2025-02-02:
I don't get who sends the 0x3111 ID?  Neither the _PYLON_HV_CAN_ nor the _GROWATT_HV_CAN_ bindings send these IDs

ErikssonPer, 2025-02-02:
Trust me,  it's sent with a 1 second interval as soon as I start the app with it setup for Growatt hv BMS

22:31:04.668 | INFO  | main       | core.BMSListProducer:105   | Intialized BMS https://github.com/ai-republic/bms-to-inverter/issues/1[GROWATT_HV_CAN] on port can1
2025-01-20 22:31:04.757 | DEBUG | Thread-2   | ocol.can.JavaCANPort:115   | CAN frame sending: Buffer (HEX): [0x00, 0x00, 0x31, 0x11, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

cmax1-gmx, 2025-02-07:
Good evening,
My System is a Growatt SPH10000-TL3-BH-UP and Battery ARK HV with 3 modules.
I'd like to get a better insight in the battery system, so no need to intercept the Traffic from and to the inverter - I just want to read the info flying by.

So config is Growatt_HV_CAN for BMS, NONE for Inverter. Using that with Log-Level Info as is default it looked good, no error in the log - but no data visible via Web-Server.

Setting Log-Level to Debug revealed the following Problem:

------------
2025-02-07 17:16:10.852 | INFO  | Thread-10  | verter.BmsToInverter:268   | Reading BMS #0 GROWATT_HV_CAN on can0...
2025-02-07 17:16:10.853 | DEBUG | Thread-10  | attHVBmsCANProcessor:115   | SEND: Buffer (HEX): [0x00, 0x00, 0x31, 0x10, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2025-02-07 17:16:10.854 | DEBUG | Thread-10  | ocol.can.JavaCANPort:115   | CAN frame sending: Buffer (HEX): [0x00, 0x00, 0x31, 0x10, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2025-02-07 17:16:10.855 | DEBUG | Thread-10  | ocol.can.JavaCANPort:76    | CAN frame read...
2025-02-07 17:16:10.856 | DEBUG | Thread-10  | ocol.can.JavaCANPort:82    | CAN read frame Buffer (HEX): [0x10, 0x31, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x06, 0xA8, 0x00, 0xFA, 0x00, 0xFA, 0x30, 0x01]

2025-02-07 17:16:10.857 | DEBUG | Thread-10  | attHVBmsCANProcessor:121   | RECEIVED: Buffer (HEX): [0x10, 0x31, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x06, 0xA8, 0x00, 0xFA, 0x00, 0xFA, 0x30, 0x01]

2025-02-07 17:16:10.857 | ERROR | Thread-10  | stoinverter.core.BMS:192   | Error requesting data!
java.lang.NullPointerException: Cannot invoke "com.airepublic.bmstoinverter.bms.growatthv.can.GrowattHVBmsCANProcessor$Command.ordinal()" because "command" is null
	at com.airepublic.bmstoinverter.bms.growatthv.can.GrowattHVBmsCANProcessor.sendMessage(GrowattHVBmsCANProcessor.java:160) ~[bms-growatt-hv-can-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.bms.growatthv.can.GrowattHVBmsCANProcessor.collectData(GrowattHVBmsCANProcessor.java:104) ~[bms-growatt-hv-can-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.core.BMS.process(BMS.java:173) ~[core-api-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.bms.growatthv.can.GrowattHVBmsCANProcessor$Proxy$_$$_WeldClientProxy.process(Unknown Source) ~[bms-growatt-hv-can-0.0.1-SNAPSHOT.jar:?]
	at com.airepublic.bmstoinverter.BmsToInverter.lambda$start$2(BmsToInverter.java:269) ~[bms-to-inverter-main-0.0.1-SNAPSHOT.jar:?]
	at java.lang.Thread.run(Thread.java:840) [?:?]
-----------

Do you have an idea what could be wrong there ?

BTW, I've not seen other errors so far, just the lines above repeating for every received frame, regardless what ID the frames have (if you need full log I can provide it)

Thank you for you great work!

ai-republic, 2025-02-08:
@cmax1-gmx I fixed the NPE. Please let me know if it works for you and also please attach the log file again. I'm curious what command you receive that is not mapped

ai-republic, 2025-02-08:
> Trust me, it's sent with a 1 second interval as soon as I start the app with it setup for Growatt hv BMS I tested without anything else connected to the can so it's for sure the app.
> 
> 22:31:04.668 | INFO | main | core.BMSListProducer:105 | Intialized BMS [#1](https://github.com/ai-republic/bms-to-inverter/issues/1)[GROWATT_HV_CAN] on port can1 2025-01-20 22:31:04.757 | DEBUG | Thread-2 | ocol.can.JavaCANPort:115 | CAN frame sending: Buffer (HEX): [0x00, 0x00, 0x31, 0x11, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

@ErikssonPer I had a look again at the code. Yes, it will send out the 0x3111 to communicate with the second BMS. Could you try to configure your BMS ID to 0 please?

cmax1-gmx, 2025-02-08:
Looks better, but still a problem:

```
2025-02-08 10:14:57.091 | INFO  | Thread-10  | verter.BmsToInverter:268   | Reading BMS #0 GROWATT_HV_CAN on can0...
2025-02-08 10:14:57.094 | DEBUG | Thread-10  | attHVBmsCANProcessor:115   | SEND: Buffer (HEX): [0x00, 0x00, 0x31, 0x10, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2025-02-08 10:14:57.096 | DEBUG | Thread-10  | ocol.can.JavaCANPort:115   | CAN frame sending: Buffer (HEX): [0x00, 0x00, 0x31, 0x10, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2025-02-08 10:14:57.098 | DEBUG | Thread-10  | ocol.can.JavaCANPort:76    | CAN frame read...
2025-02-08 10:14:57.101 | DEBUG | Thread-10  | ocol.can.JavaCANPort:82    | CAN read frame Buffer (HEX): [0x20, 0x30, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2025-02-08 10:14:57.103 | DEBUG | Thread-10  | attHVBmsCANProcessor:121   | RECEIVED: Buffer (HEX): [0x20, 0x30, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2025-02-08 10:14:57.107 | WARN  | Thread-10  | attHVBmsCANProcessor:219   | Message could not be interpreted Buffer (HEX): [0x20, 0x30, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2025-02-08 10:14:57.110 | DEBUG | Thread-10  | attHVBmsCANProcessor:115   | SEND: Buffer (HEX): [0x00, 0x00, 0x31, 0x20, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2025-02-08 10:14:57.112 | DEBUG | Thread-10  | ocol.can.JavaCANPort:115   | CAN frame sending: Buffer (HEX): [0x00, 0x00, 0x31, 0x20, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]

2025-02-08 10:14:57.115 | DEBUG | Thread-10  | ocol.can.JavaCANPort:76    | CAN frame read...
2025-02-08 10:14:57.117 | DEBUG | Thread-10  | ocol.can.JavaCANPort:82    | CAN read frame Buffer (HEX): [0x10, 0x30, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0xFF, 0x8B, 0x38, 0x00, 0x00, 0x00, 0x00, 0x00]

2025-02-08 10:14:57.120 | DEBUG | Thread-10  | attHVBmsCANProcessor:121   | RECEIVED: Buffer (HEX): [0x10, 0x30, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0xFF, 0x8B, 0x38, 0x00, 0x00, 0x00, 0x00, 0x00]

2025-02-08 10:14:57.122 | WARN  | Thread-10  | attHVBmsCANProcessor:219   | Message could not be interpreted Buffer (HEX): [0x10, 0x30, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0xFF, 0x8B, 0x38, 0x00, 0x00, 0x00, 0x00, 0x00]

```
as before the result for all IDs is identical.

This is what is logged at the end of one cycle:

```
2025-02-08 10:14:56.087 | INFO  | Thread-10  | verter.BmsToInverter:320   |
BMS     SOC       V       A      CellMinV        CellMaxV       CellDiff
#1       -0.1   0.0     0.0     0.0(#0)         0.0(#0) 0.0

2025-02-08 10:14:56.089 | INFO  | Thread-10  | verter.BmsToInverter:390   | BMS alarms:
        NONE
2025-02-08 10:14:56.875 | INFO  | Thread-11  | verter.BmsToInverter:294   | Sending to inverter NONE on can1...
```

IDs seen: 0x3010 0x3020 0x3030 0x3110 0x3120 0x3130 0x3140 0x3150 0x3160 0x3170 0x3180 0x3190 0x3200 0x3220 0x3230 0x3240 0x3250 0x3260 0x3270 0x3280 0x3F00

Full Log:
[BMS-to-Inverter.log](https://github.com/user-attachments/files/18717843/BMS-to-Inverter.log)

cmax1-gmx, 2025-02-08:
an update: it seems that the problem is in line 151 in File GrowattHVBmsCANProcessor.java

```                   final Command command = Command.forCommand(canId - 0x00001000); ```

After changing that to

```                   final Command command = Command.forCommand(canId); ```

I get correct results (at least these look good to me) for most of the defined CAN IDs with exception of 0x3110, 0x3160 and 0x3170:

0x3110 - BufferUnderflowException
0x3160 - IndexOutOfBoundsException
0x3170 - IndexOutOfBoundsException

For details see attached partial Log (I've added Debug-Log-Enttries for the processed Command too): 

[BMS-to-Inverter-part.log](https://github.com/user-attachments/files/18720591/BMS-to-Inverter-part.log)

The other IDs that I've seen as listed in my post above are not processed at the moment as they are not defined.

I hope my infos can help - I've only very basic  java-skills so I'm not able to do more then trying to help identifying the problem.

ai-republic, 2025-02-09:
@cmax1-gmx , @ErikssonPer,
Thanks a lot for the logs. That really helped to fix the issues quickly!
I've updated the code so please try a _Clean install_ and test again.

The issue with the initial readings of 0x3010-0x3030 I also covered 😄

cmax1-gmx, 2025-02-10:
@ai-republic - have you overlooked my comment above related to the problem in File GrowattHVBmsCANProcessor.java or was it intentionally not to change that ? 

```
153c153
<                     final Command command = Command.forCommand(canId - 0x00001000);
---
>                     final Command command = Command.forCommand(canId);
```

Without that change I did not get any message parsed.

Otherwise good news - no more exceptions thrown, the IDs defined are parsed. Some more comments about the values received:

Looking at the Line printed in the log

```
BMS     SOC       V       A         CellMinV          CellMaxV         CellDiff
#1       97.0   1459.8  -460.9  59.148(#47)     63.244(#3)      4.096
```
it is obvious that the values are not correct (LiFePo Cells have a Voltage of ~3.2V, my Battery with three blocks should have ~150 to 160V. This is a Byte-Order-Problem, converting the values above to Big-Endian gives the following:

```
BMS     SOC       V       A      CellMinV        CellMaxV       CellDiff
#1        97	159,3  -1,9.  3,321 (#47)    3,343 (#3)     0,022
```
Some of the other parsed IDs have additional small errors respectively few missing values. If needed I can provide details on that. Sometimes it ist difficult to get the right info out of the Growatt-Protocol-PDF (naming is misleading, wrong explications etc. - so for example for ID 0x3030 they mention that Time is epoch - that is completely wrong, each part of the date is just converted to hex, e.g. Year = 19, Month = 02, Day = 0A etc.)

I noticed that each time when the programm is started my inverter restarted too (Growatt's 5 Mins Startup Check). I suppose the app is sending out each received frame, not taking into account that I have configured "Inverter: NONE". After commenting out the send command no more restart, so perhaps you could take that into consideration (not sending anything when inverter is NONE - the dummy inverter does not send frames, but the BMS-Code does when NONE-Inverter is configured).

romanbetwo, 2025-02-12:
Finaly getting back to life ,.. just joking still no time at all .. 
SPH with pylon HV battery (like) 

[BMS-to-Inverter.log](https://github.com/user-attachments/files/18769781/BMS-to-Inverter.log)

2025-02-12 15:37:58.095 | ERROR | Thread-4   | InverterCANProcessor:78    | Error creating send frames:
java.lang.StringIndexOutOfBoundsException: String index out of range: 1
        at java.lang.StringLatin1.charAt(StringLatin1.java:48) ~[?:?]
        at java.lang.String.charAt(String.java:1517) ~[?:?]
        at com.airepublic.bmstoinverter.inverter.growatthv.can.GrowattHVInverterCANProcessor.sendManufacturerAndMaxCellVoltage(GrowattHVInverterCANProcessor.java:394) ~[inverter-growatt-hv-can-0.0.1-SNAPSHOT.jar:?]
        at com.airepublic.bmstoinverter.inverter.growatthv.can.GrowattHVInverterCANProcessor.createSendFrames(GrowattHVInverterCANProcessor.java:76) ~[inverter-growatt-hv-can-0.0.1-SNAPSHOT.jar:?]
        at com.airepublic.bmstoinverter.core.Inverter.process(Inverter.java:177) ~[core-api-0.0.1-SNAPSHOT.jar:?]
        at com.airepublic.bmstoinverter.inverter.growatthv.can.GrowattHVInverterCANProcessor$Proxy$_$$_WeldClientProxy.process(Unknown Source) ~[inverter-growatt-hv-can-0.0.1-SNAPSHOT.jar:?]
        at com.airepublic.bmstoinverter.BmsToInverter.lambda$start$5(BmsToInverter.java:317) ~[bms-to-inverter-main-0.0.1-SNAPSHOT.jar:?]
        at java.lang.Thread.run(Thread.java:840) [?:?]

this is probably last issue preventing to "run" 

log>>

cheers
Roman

cmax1-gmx, 2025-02-13:
@romanbetwo 
do you happen to have a recent version of the "Growatt Inverter Modbus RTU Protocol", the version I've found via Google is 1.24 also called "latest", which dates 2020-04-28. There are a number of values which are clearly incorrect, so I suppose there have been changes since then ?

I'd like to compare what I can get via canbus vs modbus.

romanbetwo, 2025-02-13:
Hi I'll request for latest and let you know.

cmax1-gmx, 2025-02-13:
Thanks a lot !

ai-republic, 2025-02-15:
@cmax1-gmx , I removed the response mask and changed the frame to big endian. Let's see if you can get some correct values now! 😄

romanbetwo, 2025-02-17:
@cmax1-gmx still no answer from growatt about latest and "greatest" modbus document .. they are so lazy after new year :D 

@ai-republic I am still playing arround with SPH and pylon HV battery, Last issue seems to be gone but I do believe there is anotheone seems that pylon HV see 2#BMS 

I want to continue in this threat, because its SPH related even if other part of build can be a problem. 

2 logs >  
solo BMS 
and with the inverter 

SPH shows that battery is OPEN, which means there is 0 voltage on terminals which is probably just a data from 2nd BMS.

[bms_only_pylon HV.log](https://github.com/user-attachments/files/18827418/bms_only_pylon.HV.log)

[BMS-to-Inverter 17.2.log](https://github.com/user-attachments/files/18827421/BMS-to-Inverter.17.2.log)

cmax1-gmx, 2025-02-19:
@ai-republic changing the whole frame to big endian is problematic - that changes the CAN ID too and decoding daoes not work any more.

To test I just moved that command further down

```
153                } else {
154                    frameReceived = true;
155                    // request has 4th byte 0x10 and response 0x20
156                    final Command command = Command.forCommand(canId);
-->                    receiveFrame.order(ByteOrder.BIG_ENDIAN);
157                    // move position to the data part
158                    receiveFrame.getInt();

```
to see if it would work - and it does. But I think a cleaner way would be to define a buffer for just the data part of the can-frame (the 8 Bytes at the end).

To check the results I compiled a table to compare my manual decoding of the frames for IDs 3110-3200 coming out of my system using the Protocol-PDF from above with the values spit out from bms-to-inverter - see attached file.

[Growatt-HV-CAN-Mappings.ods](https://github.com/user-attachments/files/18869233/Growatt-HV-CAN-Mappings.ods)

ai-republic, 2025-02-24:
Thanks @cmax1-gmx I will have a look at it. Very busy at the moment and currently suffering from the flu

ai-republic, 2025-03-03:
@cmax1-gmx I moved the setting to big endian like you suggested. I hope it works 😉 
Its strange to use two encodings (little and big endian) in one frame though..... 🤔

cmax1-gmx, 2025-03-04:
> [@cmax1-gmx](https://github.com/cmax1-gmx) I moved the setting to big endian like you suggested. I hope it works 😉 I

Yes, it does

> Its strange to use two encodings (little and big endian) in one frame though..... 🤔

Digging a bit deeper I found the following:

- The whole frame as it is received from the net is big endian (as is default for network data). Byte-Order for CAN-Bus Data can be either big or little endian, depends on the implementation, so the CAN-BUS Controller (receiver) needs to take care of this. 
- without specifying big or little endian when fetching the data in a buffer by default system byte order is used - which is little endian for most systems
- Growatt uses big endian (see Protocol-PDF) for high and low volt systems - as opposed to many others for example pylon, byd and other vendors

I think a possible way to adress that could be by defining the CAN-Bus Byte Order as a Config-Value and taking care of the byte order for the receiving buffer once directly in JavaCANPort.java when reading the frame, that could make it easier to add new Devices.

romanbetwo, 2025-07-23:
Hi, I had few issues with getting things to work, my bms sends some other CAN messages not related to pylon, just for maintenance program access. After filtering those out, this app works fine.

Leodien, 2025-07-23:
I'm using the design by Stuart Pittaway.
https://github.com/stuartpittaway/diyBMSv4ESP32

I scanned canbus via mqtt; see the attached files.

[image]

[image]

Leodien, 2025-07-23:
I'm using the design by Stuart Pittaway.
https://github.com/stuartpittaway/diyBMSv4ESP32

I scanned canbus via mqtt; see the attached files.
Image

Leodien, 2025-07-24:
I've found the cause.
As soon as the DIY BMS parameter 4220 reaches Charge Cutoff Voltage, the inverter displays the "high battery voltage" message.
I'd appreciate it if anyone has an example of the minimum registers required in the CAN communication.
Any further experiences would be welcome.