ai-republic/bms-to-inverter #88: JKBMS via RS485 readonly?

humpataa, 2024-12-29:
Sending commands to JK BMS systems via RS485 does not work, only using Bluetooth.
Can anyone confirm this?

ai-republic, 2024-12-30:
Could you share some more details please? What JK BMS model? Baudrate? Log file? etc...

humpataa, 2024-12-30:
Hi Torsten, I am using a JK-BD6A24S15P and similar models.
I couldn't find anything in your code (RS485!) that actually sends commands like turning on/off charging and discharging.
By searching the net and quite a few other projects working with JKBMS nobody seems to have done this in practice. Only via bluetooth! So I asked because it says "read and monitor data from BMSes and write data to an inverter using any protocol" ... maybe you have found a way to do it?

ai-republic, 2024-12-31:
There are some commands defined in the specification to accomplish this:

[image]

What would be your usecase for this?

humpataa, 2024-12-31:
Yes I know, but wanted to make sure that they only work via bluetooth, not RS485. Thank you!

humpataa, 2024-12-31:
> What would be your usecase for this?

A rather special use case: a closed battery pack including a JKBMS will be put into a waterproof compartment. The hatch of the compartment has a magnet touching the battery pack when closing. This magnet closes a reed contact inside the pack bound to an ESP32 which then connects to the BMS & turns on discharging. Whenever the hatch is opened, the ESP32 turns off discharging again.

ai-republic, 2025-01-01:
I would assume that these commands also work via RS485 as they are defined in that specification.
My application does not use those commands though to dis-/enable dis-/charging as of now. Hence my question about the usecase 😉

closed 2025-01-05 by ai-republic