ai-republic/bms-to-inverter #89: Help with Batery Unipower/UCBPower and Inverter Sumry MPS-V Max 10.2 KW

FlavioCostaeSilva, 2025-01-06:
Hi,
First, congratulations for this nice project!

I have a Unipower/UCBPower LiFePo4 (model UPLFP48-1003U) battery, and a chinese Inverter ([Sumry MPS-V Max](https://www.sumryenergy.com/Clean-100kw-Solar-Energy-Storage-System-For-Micro-grid-pd512659698.html)).

The chinese inverter: I believe it will communicate using the PYLON RS485, when setted into configuration \o/

The battery, it uses RS484 and Modbus RTU for communication.
I was able to access some data, using a chinese software and a RS485 to USB adapter. The BMS seems to be a model ( LL15S100P10G1-V2-Anti-Tf ), after search, the part LL15S100P10G1, brough me to a chinese company, [Hunan GCE](https://www.hngce.com/sale-32590526-15s-48v-100a-master-bms-for-telecom-base-station-battery-energy-storage-system.html), I believe them made this BMS.

My question about the battery: The RS485 communication and Modbus RTU (if it uses it), for this BMS, have an specific implementation, or can i set some modbus from configurator, like Tian Modbus, Narada Modubus or JK Modbus, to test it? 
Case not, and a new binding must be made, i will add a technical paper that i found **(See page 28)** and a print from the software to help you

Count on me to test it, this battery sells very well in Brazil
Thanks for advance, 

[Manual_T_cnico_Bateria_de_Litio_Unipower.pdf](https://github.com/user-attachments/files/18313356/Manual_T_cnico_Bateria_de_Litio_Unipower.pdf)

[image]

ai-republic, 2025-01-08:
Hi @FlavioCostaeSilva,
have you tried the _PYLON_RS485_ binding with the battery? If they are chinese it could be using that. 
You can of course also try the other RS485/ModBus bindings, set the inverter to _NONE_ and see if you get any sensible readings from the BMS.
For which BMS is the BMS_TOOLS software? Maybe its possible to see which batteries it supports and which protocols those batteries support.

FlavioCostaeSilva, 2025-01-09:
> Hi @FlavioCostaeSilva, have you tried the _PYLON_RS485_ binding with the battery? If they are chinese it could be using that. You can of course also try the other RS485/ModBus bindings, set the inverter to _NONE_ and see if you get any sensible readings from the BMS. For which BMS is the BMS_TOOLS software? Maybe its possible to see which batteries it supports and which protocols those batteries support. Otherwise I would need to implement a new binding for the specification you provided (need to translate first).

Hi @ai-republic,
I haven't tested any binding yet, I bought an old raspberry pi 2B and it was delivered today, I will test some bindings on this weekend, and return the results :)

This software, BMS_TOOLS, I saw on my battery technical paper, so i found it being used on a [battery (EG4 LL 48V 100AH)](https://eg4electronics.com/categories/batteries/eg4-ll-48v-100ah-lithium-iron-phosphate-battery/), from this brand, EG4Eletronics, in a [youtube video](https://www.youtube.com/watch?v=Axhc8_22Go0), so, I was able download this software. 
Now looking on site, i found a paper, with this information about protocols and communications (Page 16) 
[EG4-LL-48V-100AH-Manual.pdf](https://github.com/user-attachments/files/18356070/EG4-LL-48V-100AH-Manual.pdf)
_______________
[image]

ai-republic, 2025-01-11:
I will have a look at it this weekend

FlavioCostaeSilva, 2025-01-13:
Hi again @ai-republic, I made some tests, but, seems, I'm geting some connection problems, like **Could not open modbus master port ttyUSB0!**, **Could not open modbus master port ttyUSB0!** and **This port appears to have been shutdown or disconnected.**

I'm using this [RS485 usb adapter](https://www.mercadolivre.com.br/adaptador-conversor-usb-serial-rs485-clp-automaco-pic-plc/p/MLB32492073?pdp_filters=item_id:MLB4749884024), it have CH340 microcontroller.
And, I used Ubuntu 22.04 on X86 64bit UNIX

See some tests and results:

> Intialized BMS #1[PYLON_RS485] on port ttyUSB0
2025-01-10 23:44:33.690 | INFO  | main       | verter.BmsToInverter:261   | Starting BMS receiver...
**2025-01-10 23:44:33.691 | INFO  | Thread-2   | verter.BmsToInverter:268   | Reading BMS #1 PYLON_RS485 on ttyUSB0...**
2025-01-10 23:44:33.692 | INFO  | Thread-2   | toinverter.core.Port:111   | Opening ttyUSB0 ...
2025-01-10 23:44:33.702 | INFO  | Thread-2   | toinverter.core.Port:113   | Opening port ttyUSB0 SUCCESSFUL
2025-01-10 23:44:33.703 | INFO  | Thread-2   | toinverter.core.Port:111   | Opening ttyUSB0 ...
2025-01-10 23:44:33.703 | INFO  | Thread-2   | toinverter.core.Port:113   | Opening port ttyUSB0 SUCCESSFUL
**2025-01-10 23:44:33.704 | ERROR | Thread-2   | stoinverter.core.BMS:192   | Error requesting data!
com.fazecast.jSerialComm.SerialPortIOException: This port appears to have been shutdown or disconnected.**

> **2025-01-11 00:16:42.245 | INFO  | Thread-2   | verter.BmsToInverter:268   | Reading BMS #1 HUAWEI_MODBUS on ttyUSB0...**
> 2025-01-11 00:16:42.246 | INFO  | Thread-2   | toinverter.core.Port:111   | Opening ttyUSB0 ...
**> 2025-01-11 00:16:42.246 | ERROR | Thread-2   | dbus.J2ModMasterPort:75    | Could not open modbus master port ttyUSB0!**
> com.fazecast.jSerialComm.SerialPortInvalidPortException: Unable to create a serial port object from the invalid port descriptor: /dev/ttyUSB0
> 	at com.fazecast.jSerialComm.SerialPort.getCommPort(SerialPort.java:502) ~[jSerialComm-2.10.4.jar:2.10.4]
> 	at com.ghgande.j2mod.modbus.net.SerialConnection.open(SerialConnection.java:96) ~[j2mod-3.2.1.jar:3.2.1]

> **2025-01-11 00:08:05.196 | INFO  | main       | core.BMSListProducer:105   | Intialized BMS #1[SACREDSUN_RS485] on port ttyUSB0**
> 2025-01-11 00:08:05.203 | INFO  | main       | verter.BmsToInverter:261   | Starting BMS receiver...
> 2025-01-11 00:08:05.206 | INFO  | Thread-2   | verter.BmsToInverter:268   | Reading BMS #1 SACREDSUN_RS485 on ttyUSB0...
> 2025-01-11 00:08:05.207 | INFO  | Thread-2   | toinverter.core.Port:111   | Opening ttyUSB0 ...
> 2025-01-11 00:08:05.225 | INFO  | Thread-2   | toinverter.core.Port:113   | Opening port ttyUSB0 SUCCESSFUL
> 2025-01-11 00:08:05.226 | INFO  | Thread-2   | toinverter.core.Port:111   | Opening ttyUSB0 ...
> 2025-01-11 00:08:05.227 | INFO  | Thread-2   | toinverter.core.Port:113   | Opening port ttyUSB0 SUCCESSFUL
> 2025-01-11 00:08:05.228 | ERROR | Thread-2   | stoinverter.core.BMS:192   | Error requesting data!
> **com.fazecast.jSerialComm.SerialPortIOException: This port appears to have been shutdown or disconnected.**
> 	at com.fazecast.jSerialComm.SerialPort$SerialPortOutputStream.write(SerialPort.java:2090) ~[jSerialComm-2.10.4.jar:2.10.4]
> 	at com.fazecast.jSerialComm.SerialPort$SerialPortOutputStream.write(SerialPort.java:2072) ~[jSerialComm-2.10.4.jar:2.10.4]
> 	at com.airepublic.bmstoinverter.protocol.rs485.JSerialCommPort.sendFrame(JSerialCommPort.java:164) ~[protocol-rs485-0.0.1-SNAPSHOT.jar:?]

> **2025-01-10 23:55:28.769 | INFO  | Thread-2   | verter.BmsToInverter:268   | Reading BMS #1 NARADA_MODBUS on ttyUSB0...**
> 2025-01-10 23:55:28.770 | INFO  | Thread-2   | toinverter.core.Port:111   | Opening ttyUSB0 ...
> **2025-01-10 23:55:29.077 | ERROR | Thread-2   | dbus.J2ModMasterPort:75    | Could not open modbus master port ttyUSB0!**
> java.io.IOException: Port [ttyUSB0] cannot be opened after [3] attempts - valid ports are: [ttyUSB0]
> 	at com.ghgande.j2mod.modbus.net.SerialConnection.open(SerialConnection.java:140) ~[j2mod-3.2.1.jar:3.2.1]
> 	at com.ghgande.j2mod.modbus.facade.ModbusSerialMaster.connect(ModbusSerialMaster.java:94) ~[j2mod-3.2.1.jar:3.2.1]
> 	at 

> **2025-01-10 23:59:20.996 | INFO  | main       | core.BMSListProducer:105   | Intialized BMS #1[TIAN_MODBUS] on port ttyUSB0**
> 2025-01-10 23:59:20.999 | INFO  | main       | verter.BmsToInverter:261   | Starting BMS receiver...
> 2025-01-10 23:59:21.000 | INFO  | Thread-2   | verter.BmsToInverter:268   | Reading BMS #1 TIAN_MODBUS on ttyUSB0...
> 2025-01-10 23:59:21.000 | INFO  | Thread-2   | toinverter.core.Port:111   | Opening ttyUSB0 ...
> **2025-01-10 23:59:21.324 | ERROR | Thread-2   | dbus.J2ModMasterPort:75    | Could not open modbus master port ttyUSB0!**

> **2025-01-10 23:41:00.545 | INFO  | Thread-2   | verter.BmsToInverter:268   | Reading BMS #2 DALY_RS485 on ttyUSB0...**
> 2025-01-10 23:41:00.545 | INFO  | Thread-2   | toinverter.core.Port:111   | Opening ttyUSB0 ...
> 2025-01-10 23:41:00.546 | INFO  | Thread-2   | toinverter.core.Port:113   | Opening port ttyUSB0 SUCCESSFUL
> 2025-01-10 23:41:00.546 | INFO  | Thread-2   | toinverter.core.Port:111   | Opening ttyUSB0 ...
> 2025-01-10 23:41:00.547 | INFO  | Thread-2   | toinverter.core.Port:113   | Opening port ttyUSB0 SUCCESSFUL
> **2025-01-10 23:41:00.548 | ERROR | Thread-2   | ractDalyBmsProcessor:70    | Failed to initialize BMS!
> com.fazecast.jSerialComm.SerialPortIOException: This port appears to have been shutdown or disconnected.**
> 	at com.fazecast.jSerialComm.SerialPort$SerialPortOutputStream.write(SerialPort.java:2090) ~[jSerialComm-2.10.4.jar:2.10.4]
> 	at

ai-republic, 2025-01-17:
Hmm, it looks like your adapter is not properly set up

FlavioCostaeSilva, 2025-01-17:
> Hmm, it looks like your adapter is not properly set up in your Ubuntu

Nice, I'll try with windows so. Thanks

FlavioCostaeSilva, 2025-01-18:
Hi @ai-republic, 
Today I tried it on Windows, but no success, I attached some results
I'm sending too a .txt file with BMS_TOOLS software communication data extracted, i believe it can help to identify something

[work_Communication.txt](https://github.com/user-attachments/files/18464968/work_Communication.txt)

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

ai-republic, 2025-01-19:
@FlavioCostaeSilva how many BMSes have you connected? Is it only 1? In your logs I always see BMS #3? Maybe try with ID 1 and Pylon. The attached _communications.txt_ from the BMSTOOL doesn't show anything I familiar with and it does not show a pure RS485 frame communication.

FlavioCostaeSilva, 2025-01-20:
> [@FlavioCostaeSilva](https://github.com/FlavioCostaeSilva) how many BMSes have you connected? Is it only 1? In your logs I always see BMS [...]

Hi @ai-republic, thanks for the analysis of BMSTOOL text file. 
So, I have two batteries, I use ID 3 for the first and ID 4 for the second, because, on Sumry Inversor manual, seems it is setted to ID 2, then, I have ID 2 (Inverter), ID 3 (Battery 1) and ID 4 (Battery 2). 
In my test, I used only battery 1, as ID 3... but you is correct, I should try setting ID 1 to my battery 1 (BMS) and choose 1 on software. I will try it, thanks again for advance and patience

[image]

FlavioCostaeSilva, 2025-01-20:
Hi again @ai-republic, now I made new tests with BMS using ID 1. 
The results with Tian Modbus and Narada Modbus seems interesting, maybe my BMS uses a custom Modbus RTU :/

[image]
[image]
[image]
[image]
[image]

ai-republic, 2025-01-21:
Hmm, the Pylon RS485 seems to connect but gets no data. Tian and Narada also (it tries to read some registers but doesn't get anything). The others COM3 seem to be still in use.
So I wonder if there is some kind of start command that needs to be sent to activate the BMS sending data. Or its just a the wrong requests and it doesn't know what to do with it and sends nothing.
In any case I would need the communication specification to know what's expected.

FlavioCostaeSilva, 2025-01-21:
> Hmm, the Pylon RS485 seems to connect but gets no data. Tian and Narada also (it tries to read some registers but doesn't get anything). The others COM3 seem to be still in use. So I wonder if there [...]

Hi, I understand, I will try to ask Mr. Jeffrey from Hunan GCE Technology to give me a RS485 communication sheet for  LL15S100P10G1-V2-Anti-Tf BMS

FlavioCostaeSilva, 2025-01-22:
Mr. Jeffrey sent me only this paper about the bms, but i think, it's not about the communication. I asked again, about some paper of RS485 communication, and he said "no, only EsS version bms has", I think he doesn't understood or is not interested in share this information

[GCE 15S100A-BMS-P10G3+Burglar lock(V13R06）specification.pdf](https://github.com/user-attachments/files/18499355/GCE.15S100A-BMS-P10G3%2BBurglar.lock.V13R06.specification.pdf)

FlavioCostaeSilva, 2025-01-22:
> Hmm, the Pylon RS485 seems to connect but gets no data. Tian and Narada also (it tries to read some registers but doesn't get anything). The others COM3 seem to be still in use. So I wonder if there [...]

One thing more, I was thinking, is possible to intercept the communication between BMS_TOOLS software and the battery BMS? To see the commands that are being passing

ai-republic, 2025-01-25:
Could be possible. I'm sure there are tools out there which accomplish that. But the easiest way would be to contact 'Mr. Jeffrey' again and request the RS486/ModBus communication specification.

closed 2025-02-10 by FlavioCostaeSilva